crankyoldgit/IRremoteESP8266 #1318: Suggestion for enhanced Midea Library

fraschizzato, 2020-11-04:
I'm trying to add some commands to Midea library, but at this time I can't really understand how. 
Tried to pass the single command modifying the Swing command but without success.

The commands are related to the follow_me function. The goal is to use an esp to send room temperature to the ac, acting as remote thermostat.
I've grabbed a lot of data and get it work with arduino (IRremote library) and attiny85.

The tipical messages is: 0xBA45 (address) and 0xD22D0EF1 (message)
The address is always the same, message is for half Room Temperature and the other half Temperature Setted.

In that case:
Byte1 (D2) is room temperature 18°C - Hex conversion of the Decimal RoomTemp+192
Byte2 (2D) is complementar value of Byte1 --> So FF-D2 =2D --> FF-Byte1 = Byte 2
Byte3 (0E) is Temperature setting and Mode --> first nibble is temperature accordingly to the midea table, second nibble is the mode accordingly the same table
Byte4 (F1) the same as byte2 but for byte 3
 
The setting in the message does not affect the ac operational mode, so if the messages says 30°C Cool but the AC is at 17°C Heat it will remain 17°C Heat.

The Remote used is a RG57A7/BGEF - The AC is a Kaysun (Midea). Other working remotes with the device are RG57A6/BGEFU1 and RG57B/BGE (the one supplied with the AC but without FollowMe function - however the AC supports It)

All the grabbed data is in this google sheet: https://docs.google.com/spreadsheets/d/1f4dTW9s5U7kJhvi8UtJVX93kKzXYAMLAMR5qrxaTaxc/edit?usp=sharing

The messages are used also if FollowMe Function is not enabled in the AC (If I enable the function in another room and then put the remote back the ac starts using remote temperature as ambient temp).

Remote RG57A7/BGEF was used because the messages logic seems to be more simple to understand.

The remote sends a message every 2 minutes, after 7 minutes without messages the AC unit starts using it's own room temperature, a new message will restart the follow me mode.

In the link, in "MyTable" sheet there's a list of all tested commands 0 to 40°C
The other sheet are all the grabbed data with mode/temp/special function for each remote, also with some decoding.

As the variety of commands (and remotes) it could be usefull to add a generic send to the Midea Library that accepts byte data. Something like the ir_AC library from that fork of IRremote (https://github.com/marceloburegio/Arduino-IRremote) 

I've made a simple sender with esp-01 but the data sended is not working - checked with arduino receiver and messages are not the ones I send. I don't think is an hardware fault but it would be great If someone can post a working circuit schematic.
I was Using GPIO2 and TSAL6400, tried with/without 2n2222 transistor and/or different resistor.
Also the signal is too low and I can't reach distance over 1meter (neither the "bad" message can be received). Original remote does flawlessly about 7-8 meters (not tested over - end of the room XD)
The room temperature is achieved with DS18B20 using RX-Pin that I didn't use.
Any help is apreciated, thanks in advance.

crankyoldgit, 2020-11-04:
> As the variety of commands (and remotes) it could be usefull to add a generic send to the Midea Library that accepts byte data.

That already exists. See: https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a37d91b3a77b36509abdc53e2fec20a67

I'll look at the rest of the issue later when I get some spare time.

crankyoldgit, 2020-11-04:
> I've made a simple sender with esp-01 but the data sended is not working - checked with arduino receiver and messages are not the ones I send. I don't think is an hardware fault but it would be great If someone can post a working circuit schematic.

If you checked the Wiki page(s) for this project, you would have found: https://github.com/crankyoldgit/IRremoteESP8266/wiki/ESP01-Send-&-Receive-Circuit#sender

fraschizzato, 2020-11-04:
Ok, many thanks, I will give a try in the next days to give a feedback.
Sorry, I've seen the wiki but I've missed the circuit part, thanks for linking that.
Probably the problem in transmission is related to the "instability" of GPIO0-2, I will try with GPIO-3.
Same for sendMidea, I was thinking that some conversion were made as the sended message was received differently, so the problem could be the hardware.
Only a question about sendMidea: using previous example, the command would be `sendMidea( 0xBA45D22D0EF1, 48, 1)` or I have to add the zero's like the hardcoded swing mode, so it will be sendMidea( 0x0000BA45D22D0EF1, 48, 1)? Obviously passing data as uint64_t.
Thanks

crankyoldgit, 2020-11-04:
> Only a question about sendMidea: using previous example, the command would be `sendMidea( 0xBA45D22D0EF1, 48, 1)`

That should work, assuming you want to send it with a _single_  repeat. i.e. two messages.
Otherwise just `sendMidea(0xBA45D22D0EF1);` will work. The default is 48 bits, and no repeats.
e.g.
```
const uint16_t kMideaBits = 48;
const uint16_t kNoRepeat = 0;
const uint16_t kMideaMinRepeat = kNoRepeat;
```
If the hexadecimal code (`0xBA45D22D0EF1`) is what IRrecvDumpV2-3 reported, then yes, that value should work.

fraschizzato, 2020-11-04:
Yes, the data is from IRrecvDump but from a forked version on IRremote that supports 48bit. That shows 0xBA45 as address and the rest as data. I suppose that's a trick to send 48bit message without full modification of the library. Sending that data (with IRremote) works. And probably all my problems with IRremoteESP8266 were from GPIO0-2 Issue.
I will give a feedback soon.

crankyoldgit, 2020-11-04:
> Yes, the data is from IRrecvDump but from a forked version on IRremote that supports 48bit.

I can't speak for someone else's library. They may have a different bit order for how they construct the message. So, only if you capture it with this library can you be sure. But hey, try it. Good luck.

fraschizzato, 2020-11-04:
Ok, thanks, I've made a mistake considering both library as the same.
Here the first catch with ESP board and IRremoteESP8266 library:
```
16:06:29.303 -> Protocol  : COOLIX
16:06:29.303 -> Code      : 0xBA576E (24 Bits)
16:06:29.303 -> Mesg Desc.: Power: On, Mode: 3 (Heat), Fan: 2 (Med), Temp: 21C, Zone Follow: Off, Sensor Temp: 23C
16:06:29.350 -> uint16_t rawData[199] = {4418, 4344,  550, 1622,  524, 520,  550, 1622,  540, 1606,  530, 1616,  524, 520,  550, 1620,  524, 520,  550, 520,  550, 1622,  524, 520,  574, 498,  550, 522,  550, 1622,  522, 522,  550, 1622,  524, 520,  552, 1620,  524, 520,  550, 1622,  524, 520,  550, 1622,  524, 1622,  522, 1622,  524, 1620,  522, 520,  552, 1622,  524, 522,  550, 1622,  522, 520,  550, 520,  552, 520,  550, 520,  550, 1622,  522, 1622,  524, 520,  550, 1622,  524, 1620,  522, 1622,  524, 520,  552, 1622,  522, 522,  550, 520,  550, 1622,  524, 520,  552, 520,  552, 520,  550, 1622,  522, 5182,  4392, 4370,  550, 1594,  550, 522,  548, 1594,  552, 1592,  552, 1592,  552, 518,  578, 1568,  576, 494,  578, 496,  576, 1568,  578, 494,  552, 518,  578, 492,  554, 1590,  552, 520,  578, 1566,  578, 492,  578, 1566,  578, 494,  578, 1566,  580, 494,  578, 1564,  578, 1566,  580, 1564,  580, 1564,  580, 490,  580, 1566,  580, 492,  578, 1566,  580, 492,  580, 492,  580, 492,  580, 490,  580, 1566,  578, 1566,  578, 492,  580, 1564,  580, 1564,  580, 1564,  580, 492,  578, 1566,  580, 492,  580, 494,  578, 1564,  578, 494,  580, 492,  578, 492,  578, 1566,  578};  // COOLIX BA576E
16:06:29.443 -> uint64_t data = 0xBA576E;
```
The unit is recognized as Coolix with only 24 bit data. Follow me temperature is already reported! Unfortunatly Zone Follow remains off also if enabled. 
Now I'm waiting first FollowMe messages and then make a try on transmission.

I'm using a NodeMCU v1 because with ESP-01 (tried gpio 0-2-3) I always get strange data, didn't know why. It's like something other happen on the gpio (Ir receiver led keeps blinking "randomly"). In my experience that the first time with a similiar issue with esp01.

Next follow_up (probably for closing the issues) will be about transmission and confirmation of support for that remote and AC unit (Kaysun) midea Based (recognized as Coolix)

fraschizzato, 2020-11-04:
Ok, all command works with Coolix integration, except for the one I need... The setSensorTemp is not accepted/used by the AC. It seems does nothing, probably i'm doing something wrong. Here the code I use and tested ok with on/off, mode, setTemp, etc.

```
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Coolix.h>

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRCoolixAC ac(kIrLed);  // Set the GPIO to be used to sending the message.

void setup() {
  ac.begin();
  Serial.begin(115200);
}

void loop() {
  Serial.println("Sending...");

  // Set up what we want to send. See ir_Coolix.cpp for all the options.
  ac.setFan(kCoolixFanZoneFollow);
  ac.setSensorTemp(17);
#if SEND_COOLIX
  // Now send the IR signal.
  ac.send();
  Serial.println("Sended");
#else  // SEND_COOLIX
  Serial.println("Can't send because SEND_ARGO has been disabled.");
#endif  // SEND_COOLIX

  delay(60000);
}
```

crankyoldgit, 2020-11-04:
> not grabbed with IRremoteESP8266 library but with IR remote

Can you please make it clear in your reports/data when you are NOT using our library?

The raw data you collected (`UNKNOWN 2800A054`) is only half of a `MIDEA` message.

I've verified that _our_ library is correctly sending a full length `MIDEA` message via `sendMidea()` and that it also can decode that sent message too. See: https://github.com/crankyoldgit/IRremoteESP8266/commit/1ba5a68cbc66d3b6170318c95fd5ff207fb0fbf2

fraschizzato, 2020-11-04:
Yes, sorry. Last two posts are with Your library only. For sending and receiving i'm using two esp8266 board (wemos clone), both with Your library, one with the sending sketch above (modified version of the argo example) and the other with IRreceiveDumpV3 or V2. 
The message from the second remote is correctly reported as midea (0xA4A3477F1979), if I try to send the same data with esp, the receiver (same used for remote) receives only the UNKNOWN data. 
thanks for your time.

crankyoldgit, 2020-11-05:
> if I try to send the same data with esp, the receiver (same used for remote) receives only the UNKNOWN data.

Please post your code for that. You haven't provided that. As far as I can tell via my own testing, `irsend.sendMidea(0xA4A3477F1979);` is doing exactly the right thing. I suggest you re-check your code.

fraschizzato, 2020-11-05:
Here the code, I've seen now the error... 

```
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Midea.h>

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRsend ac(kIrLed);  // Set the GPIO to be used to sending the message.

void setup() {
  ac.begin();
  Serial.begin(115200);
}

void loop() {
  Serial.println("Sending...");

  // Set up what we want to send. See ir_Coolix.cpp for all the options.
#if SEND_COOLIX
  // Now send the IR signal.
  ac.sendMidea(0xA4A3477F1979);
  Serial.println("Sended");
#else  // SEND_
  Serial.println("Can't send because SEND_ARGO has been disabled.");
#endif  // SEND_COOLIX

  delay(60000);
}
```
 The fault I suppose is in` if SEND_COOLIX` , now I think It has to be `SEND_MIDEA`
I will give a try and probably the sendMidea issues is my fault, have to check if the data does the followMe trick.

However I can't understand why with Coolix Integration works flawlessly with all commands but not with the `setSensorTemp`
data sended from the remote is the same of the esp (checked as mentioned before) but the AC doesn't get that.
Thanks

fraschizzato, 2020-11-05:
Corrected code now works, device responds correctly. Final report is:

My Device KaysunAC works with Coolix Integration for all commands except FollowMe mode (`setSensorTemp()`), data is sended correctly but AC does not respond about that. Remote used for tests is RG57A7/BGEF

My Device also works correctly with Midea Integration for all commands, excpept FollowMe that is not implemented (more data below). The FollowMe code from the remote, if replied to the device, works and sets AC room temperature to sended value). Remote used for tests is RG57A6/BGEFU1. 

The FM mode is listed in the google-sheet from cpp (https://docs.google.com/spreadsheets/d/1TZh4jWrx4h9zzpYUI9aYXMl1fYOiqu-xVuOOMqagxrs/edit?usp=sharing)
I think temperature data is in °F, in my case it's in °Celsius, So that 
`0xA482407F174D` is setTemp 17°C and roomTemp 22°C (Auto mode-fan Auto)
`0xA482487F1644` is setTemp 22°C and roomTemp 21°C (Auto mode-fan Auto)
Where Byte4 is Room Temperature in the form HEX2DEC(Byte4)-1

Thanks for the support, hope that can be helpfull for full integration

crankyoldgit, 2020-11-05:
> My Device KaysunAC works with ...

Any chance you can provide us with the Model number for the actual Kaysun A/C so we can add it to the list of supported devices?

crankyoldgit, 2020-11-05:
> Where Byte4 is Room Temperature in the form HEX2DEC(Byte4)-1

Can you elaborate/expand on this please?
Your spreadsheet data for Byte4 (https://docs.google.com/spreadsheets/d/1TZh4jWrx4h9zzpYUI9aYXMl1fYOiqu-xVuOOMqagxrs/edit#gid=1191265844&range=P:Q)
Indicates that the Follow Me is in the Fahrenheit scale.
e.g.
> `0xA482407F174D` is setTemp 17°C and roomTemp 22°C (Auto mode-fan Auto)

So `Byte[4]` would be `0x17`, and according to your [sheet](https://docs.google.com/spreadsheets/d/1TZh4jWrx4h9zzpYUI9aYXMl1fYOiqu-xVuOOMqagxrs/edit#gid=1191265844&range=P26:Q26) that would be `56` (decimal) 56°F in Celsius is 13.33°C, which is a long way off your suggested `22°C` in the text above.

fraschizzato, 2020-11-05:
AC Unit is "Kaysun Casual CF".

Addititional remote working with this is RG57B/BGE (the one supplied with the device), correctly reported from Your library as Coolix

Byte[4], converting 17 from hex to dec gives me 23, that's 1 more of display °C (22)
same for other grabs: 1A from hex to dec gives me 26, that's 1 more of display °C (25)

That's not my sheet, is the one listed in the ir_Midea.cpp from your library, I've supposed that it's for °F and so I've listed some data messages in °C to explain how I suppose it works.

crankyoldgit, 2020-11-05:
> AC Unit is "Kaysun Casual CF".

Thanks. I'll update that it supports it under Coolix in a moment.

> Byte[4], converting 17 from hex to dec gives me 23, that's 1 more of display °C (22)
same for other grabs: 1A from hex to dec gives me 26, that's 1 more of display °C (25)

Ah. Okay. I get it now. I was confused, obviously.
So, what are the minimum and maximum values for the "room" (sensor) temp based on your analysis?
I'm guessing a value of `0x00` in `Byte[4]` would be 0°C. Correct? If so, what's the maximum room/sensor temp you've been able to generate?

This is the range we currently allowed for/assumed:
```
const uint8_t kCoolixSensorTempMin = 16;  // Celsius
const uint8_t kCoolixSensorTempMax = 30;  // Celsius
```

fraschizzato, 2020-11-05:
Have to re-test with your library but as per assumption 0°C would be `0x01` So decimal would be 1 and minus one it will be 0.

I have to take some dump with your library to confirm that, but (testing with the other lib) I've seen that the device goes down and still trasmit at 0°C. For max temp I have to test it (max reached was 27°C but "limited by the environment"), I haven't found a secure way to heat the remote without damaging it. I suppose that 30 to 50°C can be reached
Any idea about? For 0 i've put the remote in the fridge XD.

The actual grabbed data is the following:
```
FollowMe Temp | FollowMe message
21 | A4A3477F1673
25 | A4A3477F1A7B
24 | A4A3477F1979
22 | A4A3477F1772
22 | A482407F174D
```

17°C packet was created manually and it works. 

Before tomorrow  I can add other values (down to 0°C), and, if a good way is found, also values over 25.

crankyoldgit, 2020-11-05:
Na, that's fine. If you've recorded an approx 1-2°C or similar, I'll take your word for it. It probably bottoms out at 0°C. That seems supportable.
Max temp. yeah, don't go melting your remote. I'll set it to 63°C (i.e. `0x3F` / `0b111111`).

**_Hmmm_**. Actually. Looking at the structure of the message again. We've got some overlap.
Currently for Coolix, we've got the Sensor Temp using these bits (a bit mask of `0xF00`) i.e. 8th to the 11th (inclusive) _least significant_ bits of the message code, 4 bits long. And the Fan speed control has a bit mask of `0x7000`. i.e. 12th to the 14th (inclusive) _least significant_ bits of the message code, 3 bits long. There is no way to fit more "Sensor" temp values than the current 16 (4 bits) in that space, because otherwise we start over-writing the fan speed control location(s).

So we should collect _real_ readings from the remote while it's sensor temp is (well) below 16°C, and above 32°C to confirm what it's really doing. i.e. How it's packing in more that 16 values (4bits) into a 4 bit space, or for us to work out if there is something special we don't understand about these "sensor" messages.

crankyoldgit, 2020-11-05:
> Any idea about? For 0 i've put the remote in the fridge XD.

To get values over 25C, maybe put the remote in a ziplock bag (to keep it water proof-ish) and hold it safely above a boiling kettle?!?! i.e. Steam is 100C, so be careful etc. Don't need a value that high of course. 34C or higher should give us the data we need. Or near a heater/fireplace .. or over a candle etc. Or an oven that set very low? Often they can be set to ~40C.

Just be careful, and don't damage your remote. ;-)

fraschizzato, 2020-11-05:
wait wait, I think we are confused, unfortunatly we are talking about two different remotes that uses two different protocols:

This data:
```
FollowMe Temp | FollowMe message
21 | A4A3477F1673
25 | A4A3477F1A7B
24 | A4A3477F1979
22 | A4A3477F1772
22 | A482407F174D
```
and the considerations about HEX to DEC is related only to the Midea protocol.
Probably the overlapping is avoided by the fact that FollowMe Messages have starting byte at` 0xA4` instead of "normal" messages that are starting with `0xA1`. In fact, the overlapping would be constant if we watch at the °F data (that starts as listed, from 62°F to 86°F)

-------------------------------------------------------------------------------------------------------------------------
The coolix remote sends the same data that I can send via your Library, the fact is not the data itself but that the AC is not using that (tell me if I had to clarify, with my limited english I'm probably not always comprensible)
I've seen a little difference in raw data but I don't think it can be the issue (The other commands works, so can't be a general timing issue)

-------------------------------------------------------------------------------------------------------------------------

Now both remotes are in fridge (XD) next post will cointain data from both remotes at different sub16°C values. 
Next step are temperature over 32°C, I've found the simpliest and easiest way to heat the remote (i'm feeling dumb about this, XD), place the remote near the AC that now I'm using in heat mode...

fraschizzato, 2020-11-05:
I will made two posts to avoid confusion:

This is FollowMe data about Midea protocol

|Midea Temp From Remote Screen | Midea Message | Byte 5 | TEST Midea HEX2DEC(Byte5)-1|
|-- | -- | -- | --|
|0 - MIN | 0xA482467F015D | 01 | 0|
|4 | 0xA482467F0559 | 05 | 4|
|7 | 0xA482467F0854 | 08 | 7|
|9 | 0xA482467F0A57 | 0A | 9|
|11 | 0xA482467F0C50 | 0C | 11|
|13 | 0xA482467F0E53 | 0E | 13|
|14 | 0xA482467F0F52 | 0F | 14|
|15 | 0xA482467F104C | 10 | 15|
|16 | 0xA482467F114D | 11 | 16|
|17 | 0xA482467F124F | 12 | 17|
|18 | 0xA482467F134E | 13 | 18|
|25 | 0xA482467F1A47 | 1A | 25|
|28 | 0xA482467F1D41 | 1D | 28|
|30 | 0xA482467F1F42 | 1F | 30|
|31 | 0xA482467F206C | 20 | 31|
|33 | 0xA482467F226F | 22 | 33|
|33 | 0xA482467F226F | 22 | 33|
|35 | 0xA482467F2468 | 24 | 35|
|37 | 0xA482467F266B | 26 | 37|
|37-MAX | 0xA482467F266B | 26 | 37|

Test was always with Mode Auto - Fan Auto - SetTemp 23°C

fraschizzato, 2020-11-05:
That's the Coolix protocol grab.

|Coolix Temp From Remote Screen | Coolix Message | Byte 2 | TEST Coolix HEX2DEC(Byte2)-192|
|-- | -- | -- | --|
|0 - MIN | 0xBAC04A | C0 | 0|
|5 | 0xBAC54A | C5 | 5|
|8 | 0xBAC84A | C8 | 8|
|10 | 0xBACA4A | CA | 10|
|12 | 0xBACC4A | CC | 12|
|14 | 0xBACE4A | CE | 14|
|15 | 0xBACF4A | CF | 15|
|16 | 0xBAD04A | D0 | 16|
|17 | 0xBAD14A | D1 | 17|
|18 | 0xBAD24A | D2 | 18|
|19 | 0xBAD34A | D3 | 19|
|26 | 0xBADA4A | DA | 26|
|27 | 0xBADBDE | DB | 27|
|29 | 0xBADDDE | DD | 29|
|30 | 0xBADEDE | DE | 30|
|32 | 0xBAE0DE | E0 | 32|
|34 | 0xBAE2DE | E2 | 34|
|36 | 0xBAE4DE | E4 | 36|
|37 | 0xBAE5DE | E5 | 37|
|40 | 0xBAE8DE | E8 | 40|
|44 | 0xBAECDE | EC | 44|
|46 | 0xBAEEDE | EE | 46|
|50 - MAX | 0xBAF2DE | F2 | 50|

From 0 to 26 - Set Mode was Auto , fan Auto, setTemp 24°C
From 27 to 50 - Set Mode was Heat, fan Auto, setTemp 24°C

The most of the grabbed data from both remote was saved in that txt file https://drive.google.com/file/d/1xuPcXIQRbwZgwuclgjMhuv_OMf-1vTpb/view?usp=sharing

Max temp for Midea  is 37°C
Max temp for Coolix is 50°C

crankyoldgit, 2020-11-06:
@fraschizzato Ahh. Thanks ever so much for pointing out where I was going wrong, and thanks for the data and clarifications. P.S. Your English is fine. It's just I can be thick some times. ;-) Plus, text isn't the greatest way to communicate some times.

I'm going to try to focus on fixing/improving `MIDEA` first. When that's working to what you need, then move on to the `COOLIX` protocol. To that end, I've created a new branch (https://github.com/crankyoldgit/IRremoteESP8266/tree/kaysun) that adds _some_ support for your "FollowMe" feature thing.
Can I please get you to download and test it out? I also need you to collect some sensor data when the remote is in Fahrenheit mode to make sure we know what it does there as the A/C class object supports both C & F operations.

As it stands, the latest commit to that branch should decode the Sensor/FollowMe temps in C. It does NOT yet handle the `0xA4` vs `0xA1` bit yet. I want some more data from you before I try decoding/guessing what that byte actually does.
I suspect it's a normal message, but the bits might also control if the A/C beeps etc. i.e. It needs some more experimentation/data.

fraschizzato, 2020-11-06:
Here first °F data (tell me if more is needed)

|Fahrenheit | Message | Byte 5|
|-- | -- | --|
|82 | 0xA482607F3353 | 33|
|81 | 0xA482607F3252 | 32|
|78 | 0xA482607F2F45 | 2F|
|76 | 0xA482607F2D47 | 2D|
|75 | 0xA482607F2C46 | 2C|
|74 | 0xA482607F2B43 | 2B|

There will be an obvious overlapping with Celsius data, probably the distinction between two units is based on setTemp value.
In Fahrenheit mode 62°C is the lowest temperature can be set and it's Byte3 `0x60`  (at setTemp 79°F Byte3 is `0x71`, so is linear)
In Celsiud mode 30°C is the maximun temperature can be set and it's Byte3 `0x4D` (at setTemp 23°C is `0x46`, so it's linear)

I suppose that any setTemp over a certain limit says to the AC if it's °F or °C - in that case every byte3 over `0x4D` can't be Celsius (but didn't know if different remotes have same limit). Ideally, as setTemp must be in a known limited range it can be "caged" between fixed values.

PS. Byte1 `0xA4` seems to be always fixed in FollowMe messages, Byte2 changes with Mode/Fan change (`0x82` Auto/Auto , `0xA0` Cool/Auto, `0x90` Cool/Speed2), Byte3 is setTemp, Byte5 is roomTemp, Byte6 is CRC, Byte4 (`0x7F`) seems to be FollowMe "status". So Byte1 `0xA4` means "FollowMe message", then Byte4 is "kind of message" where `0x7F` is Update,`0xFF` is On and `0x3F` is Off. Note: AC uses Update message also if function was not activated (tested by entering FollowMe mode in the remote without let the AC unit know - other room)

Next Post with test results of Kaysun branch

crankyoldgit, 2020-11-06:
Thanks for the new data. I think I've corrected it to handle Sensor Temps in Fahrenheit now. Feel free to download that branch again and test.

> then Byte4 is "kind of message" where 0x7F is Update,0xFF is On and 0x3F is Off.

So, to confirm I (don't :) understand it correctly before I write code to support it.

- `7F` is _"this is a message from the remote telling you that the temp at the remote is Blah. Probably don't make a beep or anything because that would be annoying every few mins"_.
- `3F` is _"Go back to normal operation mode before I set you into FollowMe mode, oh and probably make a beep so the user knows I got the message"_.
- `FF` is _"Use the SensorTemp to adjust your operation mode (if set), oh, and probably make a beep to let the user know I got the message"_.

**Note:** `FF` is also the value used in typical _"user mashing buttons on the remote to control the A/C"_ mode messages. i.e. Non-FollowMe mode. e.g. When `0xA1` is used at the start.

Is that correct? If not, what am I getting wrong/missing with respect to that 4th byte?

fraschizzato, 2020-11-06:
Exactly, also about the "beep". Now I'm testing the branch supplied bevore. Actually I can confirm working: on/off, all modes (auto/heat/cool/dry/fan). Follow me seems not working and it beeps (when the attended is not). For other function I have to check if I missed something in my test code, here the one I'm watching now (commented code is working):

```
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Midea.h>

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRMideaAC ac(kIrLed);  // Set the GPIO to be used to sending the message.

void setup() {
  ac.begin();
  Serial.begin(115200);
  //ac.on(); //OK default °F
  //ac.off(); //OK
  ac.setUseCelsius(true);
  //ac.setMode(kMideaACHeat);
  //ac.setTemp(17);
  //ac.send();
  //Serial.println("Heat 17");
  //delay(15000);
  ac.setTemp(25);
  ac.send();
  Serial.println("Heat 25");
  delay(15000);
  ac.setTemp(30);
  ac.send();
  Serial.println("Heat 30");
  delay(15000);
  ac.setFan(kMideaACFanLow);
  ac.send();
  Serial.println("Fan Low");
  delay(15000);
  ac.setFan(kMideaACFanMed);
  ac.send();
  Serial.println("Fan Med");
  delay(15000);
  ac.setFan(kMideaACFanHigh);
  ac.send();
  Serial.println("Fan High");
  delay(15000);
  ac.setFan(kMideaACFanAuto);
  ac.send();
  Serial.println("Fan Auto");
  delay(15000);
  ac.setSleep(true);
  ac.send();
  Serial.println("Sleep ON");
  delay(15000);
  ac.setSleep(false);
  ac.send();
  Serial.println("Sleep OFF");
  delay(15000);
  ac.setSwingVToggle(true);
  ac.send();
  Serial.println("Swing ON");
  delay(15000);
  ac.setSwingVToggle(false);
  ac.send();
  Serial.println("Swing OFF");
  delay(15000);
  ac.setEconoToggle(true);
  ac.send();
  Serial.println("Eco ON");
  delay(15000);
  ac.setEconoToggle(false);
  ac.send();
  Serial.println("Eco OFF");
  delay(15000);
  //ac.setMode(kMideaACFan);
  //ac.send();
  //Serial.println("Fan Mode");
  //delay(15000);
  //ac.setMode(kMideaACCool);
  //ac.send();
  //Serial.println("Cool Mode");
  //delay(15000);
  //ac.setMode(kMideaACDry);
  //ac.send();
  //Serial.println("Dry Mode");
  //delay(15000);
  //ac.setMode(kMideaACAuto);
  //ac.send();
  //Serial.println("Auto Mode");
  //delay(15000);
  ac.setSensorTemp(37,true);
  ac.send();
  Serial.println("FollowMe 37");
  delay(60000);
  ac.setSensorTemp(0,true);
  ac.send();
  Serial.println("FollowMe 0");
  delay(60000);
  ac.setSensorTemp(17,true);
  Serial.println("FollowMe 17");
  ac.send();
  //delay(60000);
  //ac.off();
  //ac.send();
  //Serial.println("Off - End");
  
}

void loop() {
} 
```

Note that first test was a single run of that code (so nothing was commented)

fraschizzato, 2020-11-06:
Mid-report about working feature:

- ON/OFF
- All Modes (Heat, Cool, Dry, Fan, Auto)
- All Fan Speed (Min,Med,Max,Auto)
- All Temp Settings (Tested 17 - 25 - 30)
- Sleep Mode (on/off)

Not Working:

- Swing (I will catch packet from remote)
- Follow Me (Device starts Timer Mode! XD)
- Eco Mode (Can't test - Not supported by the AC and Remotes)

Not implemented (I will post data about that):
-Turbo Mode (used to reach desidered temp in less time)
-Led (used to switch on/off AC display)
-Self-Clean (not supported by my AC but provided by the remote)
-Timer ON (delay on start device)
-Timer OFF (delay on stop device)

For the last two it makes some time because there's a "time" variable (selectable by multiple press)

Sorry for multiple posts but doing multiple task in  the same time is the only way to provide all informations without confusing and not reminding things

fraschizzato, 2020-11-06:
`0xA18248FFFF54` - Swing toggle ON/OFF from library
`0xA202FFFFFF7E` - Swing mode ON/OFF on remote (note the starting byte at `0xA2`
`0xA201FFFFFF7C` - "Direct" command - allow to move manually the "swing vertical things". Each press will move the (sorry) "things that is moving when swing" by a little (to direct the air flow) 

`0xA209FFFFFF74` - Turbo Mode ON/OFF

`0xA208FFFFFF75` - LED Mode ON/OFF (switch on/off AC "screen" and beeps)

`` - Self-Clean - Not supported by my device, no ir signal when pressed (checked also with camera) - Only printed on button :-)

Follow Me:
`ac.setSensorTemp(37,true);` produces `0xA18248FF26A3`, so `0xA1` instead of `0xA4`, `0x3F` `0xFF` `0x7F` not as stated above. I think that `0x7F` must be the default (as the device uses that temp - without beeping - also if function is enabled or disabled)

Timer ON - Step 0,5 hour until 9,5 hours and then step by 1. Moving byte is 5, step between 0,5 value is 2 and betwen 1 value is 4, latest value disables timer.

|Message | Hour Delay | Descr | Byte 5|
|-- | -- | -- | --|
|0xA1A34DFF8121 | 0,5 | Half Hour | 81|
|0xA1A34DFF8322 | 1 |   | 83|
|0xA1A34DFF8526 | 1,5 |   | 85|
|0xA1A34DFF8724 | 2 |   | 87|
|0xA1A34DFF892E | 2,5 |   | 89|
|0xA1A34DFF8B2C | 3 |   | 8B|
|0xA1A34DFF8D2A | 3,5 |   | 8D|
|0xA1A34DFF8F28 | 4 |   | 8F|
|0xA1A34DFF913E | 4,5 |   | 91|
|0xA1A34DFF933C | 5 |   | 93|
|0xA1A34DFF953A | 5,5 |   | 95|
|0xA1A34DFF9738 | 6 |   | 97|
|0xA1A34DFF9936 | 6,5 |   | 99|
|0xA1A34DFF9B34 | 7 |   | 9B|
 | | .... | Until 9,5 step is 0,5 |  |
|0xA1A34DFFA704 | 10 | From 10 step is 1 | A7|
|0xA1A34DFFAB0C | 11 |   | AB|
|  | ... |   |  |
|0xA1A34DFFDB54 | 23 |   | DB|
|0xA1A34DFFDF50 | 24 |   | DF|
|0xA1234DFFFFE0 | 0 | OFF - Disable Timer | FF|

Timer OFF - same logic as above but moving byte is Byte 4 (the rest of the message is the same). Also changes off command.

|Message | Hours | Descr | Byte 4|
|-- | -- | -- | --|
|0xA1A34D81FF21 | 0,5 |   | 81|
|0xA1A34D83FF22 | 1 |   | 83|
 | | ... |   |  |
|0xA1A34D8BFF2C | 3 |   | 8B|
|  | ... |   |  |
|0xA1A34DA7FF04 | 10 |   | A7|
|0xA1A34DABFF0C | 11 |   | AB|
|  | ... |   |  |
|0xA1A34DDFFF50 | 24 |   | DF|
|0xA1A34DFFFF60 | 0.0 | OFF | FF|

`0xA212FFFFFF6E` "Silence FP" key on remote - AC does nothing - didn't know who-where-why-when and how :-)

I think (hope XD) that's all.
Thank for your time and your support

crankyoldgit, 2020-11-09:
Thanks for the data. Just getting back to this issue now.

> 0xA202FFFFFF7E - Swing mode ON/OFF on remote (note the starting byte at 0xA2
0xA201FFFFFF7C - "Direct" command - allow to move manually the "swing vertical things". Each press will move the (sorry) "things that is moving when swing" by a little (to direct the air flow)
0xA209FFFFFF74 - Turbo Mode ON/OFF
0xA208FFFFFF75 - LED Mode ON/OFF (switch on/off AC "screen" and beeps)

I have a theory, `0xA2` at the start means "Swing" in some context. Is it possible that you tested/captured the Turbo and LED messages with "Swing" being something other that the initial default state?
Alternative theory is `0xA2` is a "This is a special (non-basic operation) mode setting message". e.g. Swing or Direct or Turbo or LED

Looking at the options for the first byte:
`0xA1` = 0b1010**0001** = Normal operation (i.e. if the last bit is `1`)
`0xA2` = 0b1010**0010** = Special (or it could be "swing") operation (i.e. if the second last bit is `1`)
`0xA4` = 0b1010**0100** = Follow Me operation (i.e. if the third last bit is `1`)

Does the data you've collected match that?

crankyoldgit, 2020-11-09:
P.S. Lets not worry about timers for now, unless they change the first byte some how. :)

fraschizzato, 2020-11-09:
for `0xA1` the assumption is correct. A1 commands covers (ON/OFF,Mode,Temp,Timer and Sleep)

for `0xA2` i think is any other command, the only "command" that not match with this is "Sleep". 

About that i've found that sleep command also "rewrites" ac operation mode. (example: AC in heat mode, remote switched to cool without the AC known, remote used to set "sleep mode" --> the AC goes to Cool Mode). So, I think it must be considered like an operational mode instead of a command. 

All commands starting with `0xA2` were verified and confirmed (nothing to do with other modes, tested starting from fresh unit).

Till now `0xA4` is only for "FollowMe" function that has unique features.

------------------------------------------------------------------------------------------------------------------------------------

Another approach can be about command content: `0xA1` and `0xA4` had vars (mode-temp), `0xA2` no. So only examining first byte the software can know if it's a "one shot" command (and had to search it's feature in a "functions array") or the command has to be "dismounted" to get all vars.

crankyoldgit, 2020-11-10:
Okay, thanks for the info/analysis. I work on implementing setting the first byte for the `0xA1` & `0xA4` cases initially, and lets see how we go from there.

crankyoldgit, 2020-11-10:
I've updated the branch (https://github.com/crankyoldgit/IRremoteESP8266/tree/kaysun) to support setting & reporting the message type. (i.e. first byte) and setting `0x7F` for Follow Me etc.

Use the function `IRMideaAC::setType()` to do so.
This should allow you to construct your required messages (I think). Let me know etc.

fraschizzato, 2020-11-10:
Follow Me Confirmed to Work. Following sketch used for test (commented out code is already tested as Working):

```
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Midea.h>

const uint16_t kIrLed = 3;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRMideaAC ac(kIrLed);  // Set the GPIO to be used to sending the message.

void setup() {
  ac.begin();
  ac.setUseCelsius(true);
  //ac.setMode(kMideaACHeat);
  //ac.setTemp(24,true);
  //ac.send();
  delay(60000);
  ac.setType(kMideaACTypeFollow);
  ac.setSensorTemp(16,true);
  ac.send();
  delay(60000);
  ac.setSensorTemp(37,true);
  ac.send();
  delay(60000);
  ac.setSensorTemp(0,true);
  ac.send();
}

void loop() {
}
```

16°C - 37°C - 0°C Tested and working on my AC Unit

Tell me if You need more tests (also for other functions).

Only a suggestion: for `setSensorTemp` I think that  `setType(kMideaACTypeFollow)` must be hard-coded in the function instead of the need of specification in end-user code. It's the only way/only need about this. So I think that "setType(followMe)" and "setType(Normal)" must be (at top and at bottom) of setSensorTemp Function.

crankyoldgit, 2020-11-10:
> Only a suggestion: for `setSensorTemp` I think that `setType(kMideaACTypeFollow)` must be hard-coded in the function instead of the need of specification in end-user code.

Done. My bad. I thought for some reason I had a captured (i.e. real remote message) that started with `0xA1` **and** had a sensor temp value in it. Thus it needed to be separate. I checked the data I had, and that didn't exist. So, I've done as you've suggested. It's now built in to the `setSensorTemp()` & `setEnableSensorTemp()` functions in the latest update to that branch.

crankyoldgit, 2020-11-11:
I've added support for the On & Off timer(s) to that branch. Please download and test that they work as expected.

I think that leaves Turbo and Led toggling to go for Midea. Correct?

fraschizzato, 2020-11-11:
I will try all tomorrow.
Didn't know (didn't checked) if also swing/direct were already implemented/corrected.
If both are working (i will check) the last steps are Turbo and Led (note that all four commands are in the same category `0xA2`).

crankyoldgit, 2020-11-11:
Yep. Swing toggle and Econo toggle is already.

crankyoldgit, 2020-11-11:
Turbo & Light toggling should now also be supported in that branch too.

fraschizzato, 2020-11-11:
Both Timers Confirmed to Work:

```
  ac.setOnTimer(30);
  ac.send();
  delay(15000);
  ac.setOnTimer(60);
  ac.send();
  delay(15000);
  ac.setOnTimer(12*60);
  ac.send();
  delay(15000);
  ac.setOnTimer(24*60);
  ac.send();
  delay(15000);
  ac.setOnTimer(0);
  ac.send();
  delay(15000);
  ac.setOffTimer(30);
  ac.send();
  delay(15000);
  ac.setOffTimer(60);
  ac.send();
  delay(15000);
  ac.setOffTimer(12*60);
  ac.send();
  delay(15000);
  ac.setOffTimer(24*60);
  ac.send();
  delay(15000);
  ac.setOffTimer(0);
  ac.send();
  delay(15000);
```
"0" minutes used to switch OFF
-------------------------------------------------------------------------------------------------------------------------------------------

Sleep Confirmed to Work:

```
  ac.setSleep(true);
  ac.send();
  delay(15000);
  ac.setSleep(false);
  ac.send();
  delay(15000);
```
Led, Turbo and Swing doesn't produce desidered output: (all commands are followed by `ac.send();`)

`ac.setLightToggle(false);`  produces `0xA18248FFFF54` instead of `0xA208FFFFFF75` (same for True)
`ac.setSwingVToggle(false);` produces '0xA18248FFFF54' instead of `0xA202FFFFFF7E` (same for true)
`ac.setTurboToggle(false);`   produces `0xA18248FFFF54` instead of `0xA209FFFFFF74` (same for true - in arduinoIDE command is not "orange" colored)

I think that there's something wrong in `0xA2` implementation (or I'm wrong using that)

Direct command is not implemented `0xA201FFFFFF7C` - It allow to move the wing "manually" setting a specified position by multiple press.

crankyoldgit, 2020-11-11:
Those toggle commands should produce more than 1 message. A "normal" command message, then the "special" message.

fraschizzato, 2020-11-11:
Original Remote produces only one message (here the Swing as example)

```
23:56:33.411 -> Timestamp : 000027.561
23:56:33.411 -> Library   : v2.7.11
23:56:33.411 -> 
23:56:33.411 -> Protocol  : MIDEA
23:56:33.411 -> Code      : 0xA202FFFFFF7E (48 Bits)
23:56:33.411 -> Mesg Desc.: Type: 2 (Special), Swing(V) Toggle: Off, Econo Toggle: On, Turbo Toggle: Off, Light Toggle: Off
23:56:33.411 -> uint16_t rawData[199] = {4384, 4362,  548, 1618,  522, 520,  550, 1620,  522, 520,  548, 520,  550, 520,  550, 1618,  522, 520,  548, 522,  548, 520,  550, 520,  548, 520,  550, 520,  550, 520,  550, 1620,  522, 520,  550, 1618,  522, 1618,  522, 1620,  522, 1620,  522, 1618,  522, 1618,  522, 1618,  522, 1618,  522, 1618,  522, 1620,  524, 1618,  522, 1620,  522, 1618,  522, 1618,  522, 1618,  522, 1620,  522, 1620,  522, 1620,  520, 1620,  522, 1618,  522, 1620,  524, 1618,  522, 1620,  522, 1618,  522, 520,  550, 1620,  522, 1620,  522, 1620,  522, 1618,  522, 1620,  522, 1618,  522, 520,  550, 5174,  4384, 4362,  550, 522,  548, 1592,  550, 520,  550, 1592,  550, 1590,  552, 1588,  552, 520,  550, 1590,  552, 1590,  550, 1590,  552, 1590,  550, 1590,  550, 1590,  552, 1590,  550, 518,  552, 1588,  552, 518,  552, 520,  552, 518,  552, 518,  552, 518,  550, 520,  578, 492,  576, 492,  552, 518,  552, 518,  552, 518,  552, 518,  550, 518,  552, 518,  550, 518,  552, 518,  552, 520,  576, 492,  552, 518,  552, 518,  576, 494,  576, 494,  552, 518,  550, 518,  580, 1562,  552, 518,  552, 518,  550, 520,  556, 514,  550, 518,  552, 518,  552, 1590,  578};  // MIDEA A202FFFFFF7E
23:56:33.504 -> uint64_t data = 0xA202FFFFFF7E;
```

Here the only one produced by the library:

```
00:00:27.217 -> Timestamp : 000261.398
00:00:27.217 -> Library   : v2.7.11
00:00:27.217 -> 
00:00:27.217 -> Protocol  : MIDEA
00:00:27.217 -> Code      : 0xA18248FFFF54 (48 Bits)
00:00:27.217 -> Mesg Desc.: Type: 1 (Command), Power: On, Mode: 2 (Auto), Celsius: On, Temp: 25C/77F, On Timer: Off, Off Timer: Off, Fan: 0 (Auto), Sleep: Off, Swing(V) Toggle: Off, Econo Toggle: Off, Turbo Toggle: Off, Light Toggle: Off
00:00:27.265 -> uint16_t rawData[255] = {4480, 4508,  578, 1680,  574, 578,  550, 1696,  550, 576,  548, 578,  550, 576,  550, 578,  548, 1674,  574, 1696,  550, 576,  550, 576,  550, 576,  550, 576,  550, 576,  550, 1672,  574, 578,  548, 576,  552, 1672,  574, 576,  552, 574,  550, 1668,  578, 574,  576, 552,  576, 550,  576, 1644,  578, 1670,  576, 1668,  604, 1642,  606, 1642,  604, 1638,  606, 1642,  606, 1642,  608, 1638,  608, 1638,  606, 1640,  606, 1640,  604, 1642,  606, 1640,  574, 1672,  576, 1672,  576, 576,  576, 1644,  602, 552,  574, 1646,  574, 580,  548, 1670,  576, 554,  570, 556,  572, 5590,  4518, 4450,  574, 552,  574, 1672,  574, 552,  574, 1672,  574, 1674,  570, 1674,  574, 1672,  574, 556,  572, 554,  574, 1672,  572, 1674,  572, 1674,  572, 1676,  570, 1674,  572, 556,  570, 1678,  572, 1674,  570, 556,  570, 1674,  570, 1676,  570, 556,  570, 1676,  568, 1678,  568, 1680,  542, 582,  568, 558,  570, 556,  568, 558,  544, 582,  568, 556,  570, 556,  570, 558,  568, 558,  568, 556,  558, 568,  568, 558,  568, 558,  568, 558,  568, 558,  568, 560,  542, 1702,  568, 558,  568, 1678,  568, 580,  548, 1700,  546, 580,  546, 1702,  520, 1726,  546, 5632,  4458, 4512,  518, 1728,  518, 608,  518, 1730,  518, 608,  516, 610,  540, 588,  520, 1726,  540, 588,  522, 604,  540, 586,  522, 606,  540, 586,  540, 586,  542, 584,  542, 586,  516, 1730,  540, 1708,  540, 1706,  540, 1708,  540, 1708,  538, 1732,  514, 1734,  514, 1734,  488, 1784,  464, 1784,  436, 1838,  408};  // MIDEA A18248FFFF54
00:00:27.405 -> uint64_t data = 0xA18248FFFF54;
```

code used is:

```
  ac.setSwingVToggle(true);
  ac.send();
```

crankyoldgit, 2020-11-11:
I've added a bigger gap between `MIDEA` messages. i.e. added a decent trailing gap of 100ms+. That should hopefully make the device and IRrecvDumpVx see the two messages as separate.

The library won't send a _single_ message for those, because a user could do something like:
```
ac.setMode(kMideaACCool);
ac.send();
delay(10000);
ac.setMode(kMideaACHeat);
ac.setSwingVToggle(true);
ac.send();  // Here it needs to send two messages. One to change the mode, the other to change the swing etc.
```
The library can't tell the intention of the user, and it tries not to assume knowledge of the existing A/C state where possible.
It can't be smart and say _"oh, don't send the command state again because we've already sent it"_ because we (library) doesn't know why the user has requested it. Maybe they want a re-transmission due to interference etc.

I've also added `setSwingVStep()` (aka. Direct) but you have to `#define KAYSUN_AC true` etc.
Turns out there are multiple brands using the `MIDEA` protocol, and they use different "special" commands for each brand/model.
e.g. the "default" SwingVToggle, is actually your Kaysun's "Direct"/Step.

You can view the latest commit for the chaos it's introduced :-) https://github.com/crankyoldgit/IRremoteESP8266/commit/7875b7a2a8bc20ea773465fa29e1d252c97740f1

Please download and check it out.

fraschizzato, 2020-11-11:
Ok, it makes sense. But as the "special" function are independent from device state I'd rather prefer that they can be sended without changing/confirmig device state (personal opinion). In a case where also uart/cloud is used, operational states were issued by that, and only not supported function are issued via IR (like FollowMe and Led). But I understand that It will be different from "standard" library use and can generate confusion in user that didn't know what happen in the "back".
  
Tomorrow I will give the entire test (waiting daylight if something goes wrong). I hope the gap let the device and the "dump" receiver get both messages (because also the AC doesn't see them).

Correct consideration about `#define KAYSUN_AC`, I can say that both remotes (Mr.Cool and Kaysun) had the "direct" button
I suppose that there are also hidden features.

Really great job! Thanks

crankyoldgit, 2020-11-12:
One option is to add a `sendSwingVToggle()` function, that does just that, for those that really want to send a single message.
But you can do that with a `sendMidea()` call anyway. /shrug.

fraschizzato, 2020-11-12:
Exactly, I've already done that via `sendMidea()`

Led is working
Turbo is working

But if defined `#define KAYSUN_AC true` id ir_Midea.h I get:

```
ir_Midea.cpp: In member function 'bool IRMideaAC::isSwingVStep() const':

ir_Midea.cpp:349:28: error: 'kMideaACStepVStep' was not declared in this scope
   return _.remote_state == kMideaACStepVStep;
```

crankyoldgit, 2020-11-12:
D'oh! Thanks. I'll fix it tomorrow. Replacing it with `kMideaACSwingVStep` should fix it.

crankyoldgit, 2020-11-13:
Fixed in the latest commit to the branch. Also fixed a minor bug with the decoding of the "Step" setting.

fraschizzato, 2020-11-13:
All confirmed to work. For other users: Uncommenting `#define KAYSUN_AC true` in ir_Midea.h it will allow to use the swing/direct commands mentioned above.

Great job!

crankyoldgit, 2020-11-15:
Now that v2.7.12 has been put to bed, I'll start on the Coolix side of this Issue shortly.

fraschizzato, 2020-11-15:
Ok, let me know what you need. As I remember the first problem was in FollowMe Mode: data packet seems to be sended correctly (checked via Dump sketch and compared to the remote one) but (and didn't know why) not used by the AC unit. Then I think that, probably, some commands must be implemented (Direct,Turbo,Sleep,Led,etc).

crankyoldgit, 2020-11-15:
Okay. I tried implementing a "sensor temp" on Coolix that ranges from 0-50C .. but that takes up 6 bits of space. Which overlaps with the Fan bit range (by 1 bit)
(Data from your earlier comment: https://github.com/crankyoldgit/IRremoteESP8266/issues/1318#issuecomment-722658664)

So something in that "area" needs to be worked out if your data is correct.
e.g.
Your data suggests there is 6 bits of data for the sensor temp. i.e. `50`C = `0x32` = `0b110010`
The sensor temp bits start at 9th bit (from the Least significant side) this 9 + 6 = 15. So the 15th Least Significant Bit (LSB) will be the start of the "Fan" bits, which takes up 3 bits. Currently we have the "Fan" starting on the **14th** LSB.
i.e. the masks are:
`0b000000000000111100000000` for the _existing_ Sensor Temp setting &
`0b000000000011111100000000` for the _proposed_ Sensor Temp setting &
`0b000000001110000000000000` for the existing Fan setting.
You can see the overlap.

I'm fairly sure the fan bits are in the correct space, but there is some existing "oddness" with how it's done.

How sure are you that it goes up to 50C?
If it only went up to 30C i.e. `0x1E` it would be only 5 bits.  We have plenty of data that indicates a value of `0x1F` is the "ignore" or Sensor Temp disabled value for other devices.

fraschizzato, 2020-11-15:
https://github.com/crankyoldgit/IRremoteESP8266/issues/1318#issuecomment-722658664 This data is directly from remote, maybe there's a "flag" telling the unit that it's only a FollowMe Message (like the `0xA4` in Midea Protocol). Basically the AC unit didn't need to know Temp/OpMode/FanMode in that kind of messages. As it seems the "FM temperature report" doesn't neither follow the "strange" temperature schema, it seems linear. Note that last byte in the linked comment changes from `4A` to `DE` when operating mode was changed (had to switch to Heat mode from Auto Mode - and as I remember that switch also causes different unit approach about fan mode). Also moving from this https://github.com/crankyoldgit/IRremoteESP8266/issues/1318#issuecomment-721829757 seems that FMmode (mentioned as ZoneFollow) the "flag" is the FanMode (6). Probably that (assumed) FanMode hides something different from the stated till now.

crankyoldgit, 2020-11-15:
There is a special ZoneFollow/Sensor bit:
https://github.com/crankyoldgit/IRremoteESP8266/blob/a56c88b741dbe79ff85a488e7341dbb305120946/src/ir_Coolix.h#L45-L46

Earlier reports of the Sensor had an offset of +16C to a `00` value i.e. 0 == 16C, 1 == 17C

This is the original user-derived analysis of the protocol.
If you can provide a better analysis that covers/matches their observations I'm happy to use it/re-work the code.
See issue https://github.com/crankyoldgit/IRremoteESP8266/issues/484 

I can see their `x` bit being an extra bit of the sensor/zf temp. i.e. 5 bits instead of 4.

fraschizzato, 2020-11-15:
I will take a look in next days (from direct grabs), as first see I can suppose that the "x" bit is part of the temperature and the difference between messages is made by "op"

fraschizzato, 2020-11-15:
Here the grabbed data https://docs.google.com/spreadsheets/d/1kPc2pyFwBD93j4EM-LOSMrTx3g0QxciL2MsnUekY5w8/edit?usp=sharing

Decoding is at really early stage, but as I can see the "overlapping" is avoided from the first byte that is different

Firstly all FollowMe Messages starts with `0xBA` and modes ones with `0xB2` now I'm searching a path for all messages.
At this stage "Direct" command isn't recognized as Coolix, it's the only one not recognized. Here the grab:

```
18:51:39.085 -> Timestamp : 002124.406
18:51:39.085 -> Library   : v2.7.11
18:51:39.085 -> 
18:51:39.085 -> Protocol  : UNKNOWN
18:51:39.085 -> Code      : 0xB0473CC8 (50 Bits)
18:51:39.085 -> uint16_t rawData[99] = {4386, 4366,  546, 1622,  520, 522,  548, 1622,  496, 1646,  520, 524,  548, 522,  548, 1622,  522, 522,  548, 520,  550, 1620,  520, 522,  548, 522,  546, 1622,  496, 1646,  522, 520,  526, 1646,  520, 522,  548, 522,  524, 546,  550, 520,  550, 1620,  498, 1644,  520, 1622,  522, 1622,  520, 1620,  522, 1620,  524, 1618,  524, 1618,  524, 520,  550, 520,  550, 522,  550, 518,  552, 1618,  524, 1618,  524, 1618,  524, 520,  550, 520,  550, 520,  550, 520,  550, 520,  552, 516,  552, 518,  550, 522,  550, 1618,  526, 1616,  524, 1618,  524, 1618,  524, 1618,  550};  // UNKNOWN B0473CC8
```

crankyoldgit, 2020-11-16:
You're `UNKNOWN` "Direct" message appears to have underlying data that matches the `COOLIX` patterns. e.g. Byte followed by inverted Byte. three times. For a total of 24bits.

So without looking into it deeply (finding the errant time entry), I'm guessing it's a transient timing issue/fault. i.e. Try again.

fraschizzato, 2020-11-16:
Probably a misanderstanding, all data grabbed from that button is not recognized, here some grabs from different distance, angle, etc.:
```

17:07:08.119 -> IRrecvDump is now running and waiting for IR input on Pin 4
17:07:12.972 -> Timestamp : 000004.941
17:07:12.972 -> Library   : v2.7.11
17:07:12.972 -> 
17:07:12.972 -> Protocol  : UNKNOWN
17:07:12.972 -> Code      : 0xB0473CC8 (50 Bits)
17:07:12.972 -> uint16_t rawData[99] = {4386, 4362,  548, 1620,  522, 520,  550, 1620,  522, 1620,  522, 520,  548, 522,  548, 1620,  522, 522,  548, 522,  548, 1620,  522, 522,  548, 520,  550, 1620,  522, 1620,  520, 520,  550, 1620,  522, 520,  550, 520,  548, 522,  548, 518,  550, 1620,  520, 1620,  522, 1620,  522, 1620,  522, 1620,  520, 1620,  520, 1620,  522, 1618,  522, 520,  548, 522,  548, 522,  548, 522,  548, 1620,  520, 1620,  520, 1620,  522, 520,  548, 522,  546, 522,  546, 522,  548, 522,  548, 522,  548, 522,  548, 520,  548, 1622,  520, 1622,  520, 1620,  520, 1622,  520, 1620,  520};  // UNKNOWN B0473CC8
17:07:13.019 -> 
17:07:13.019 -> 
17:07:15.091 -> Timestamp : 000007.041
17:07:15.091 -> Library   : v2.7.11
17:07:15.091 -> 
17:07:15.091 -> Protocol  : UNKNOWN
17:07:15.091 -> Code      : 0xB0473CC8 (50 Bits)
17:07:15.091 -> uint16_t rawData[99] = {4384, 4364,  548, 1620,  522, 520,  548, 1620,  522, 1620,  520, 522,  550, 520,  548, 1620,  522, 520,  550, 520,  550, 1620,  522, 520,  548, 520,  548, 1620,  522, 1620,  520, 520,  548, 1620,  520, 522,  548, 520,  550, 520,  548, 520,  548, 1620,  522, 1620,  522, 1620,  522, 1620,  522, 1618,  522, 1618,  522, 1618,  522, 1620,  520, 522,  546, 522,  548, 522,  548, 522,  548, 1620,  522, 1620,  520, 1620,  522, 520,  550, 522,  548, 520,  548, 522,  548, 522,  548, 520,  548, 520,  574, 496,  548, 1620,  520, 1620,  522, 1620,  522, 1620,  520, 1622,  520};  // UNKNOWN B0473CC8
17:07:15.138 -> 
17:07:15.138 -> 
17:07:17.730 -> Timestamp : 000009.679
17:07:17.730 -> Library   : v2.7.11
17:07:17.730 -> 
17:07:17.730 -> Protocol  : UNKNOWN
17:07:17.730 -> Code      : 0xB0473CC8 (50 Bits)
17:07:17.730 -> uint16_t rawData[99] = {4392, 4356,  550, 1616,  552, 494,  550, 1614,  526, 1614,  526, 520,  550, 544,  526, 1614,  528, 518,  548, 546,  526, 1614,  524, 522,  548, 538,  532, 1616,  524, 1616,  550, 496,  548, 1616,  552, 496,  548, 520,  550, 544,  524, 544,  526, 1614,  526, 1616,  524, 1616,  552, 1588,  552, 1588,  550, 1590,  550, 1590,  552, 1566,  574, 520,  552, 516,  526, 544,  524, 520,  550, 1616,  552, 1590,  552, 1590,  550, 520,  552, 518,  526, 518,  576, 494,  550, 520,  572, 520,  550, 520,  528, 542,  552, 1588,  526, 1614,  526, 1590,  550, 1616,  526, 1614,  528};  // UNKNOWN B0473CC8
17:07:17.776 -> 
17:07:17.776 -> 
17:07:20.040 -> Timestamp : 000011.980
17:07:20.040 -> Library   : v2.7.11
17:07:20.040 -> 
17:07:20.040 -> Protocol  : UNKNOWN
17:07:20.040 -> Code      : 0xB0473CC8 (50 Bits)
17:07:20.040 -> uint16_t rawData[99] = {4384, 4362,  548, 1620,  522, 520,  548, 1622,  520, 1620,  520, 522,  546, 522,  548, 1620,  520, 522,  546, 522,  548, 1620,  522, 522,  522, 546,  548, 1622,  520, 1620,  520, 522,  548, 1622,  520, 524,  546, 524,  546, 522,  546, 524,  522, 1644,  520, 1620,  520, 1622,  522, 1620,  520, 1620,  520, 1622,  520, 1620,  522, 1620,  522, 522,  548, 522,  548, 520,  550, 520,  548, 1620,  520, 1620,  522, 1618,  522, 520,  548, 524,  548, 520,  550, 520,  548, 522,  546, 522,  548, 522,  548, 520,  548, 1620,  522, 1620,  520, 1620,  520, 1620,  520, 1620,  522};  // UNKNOWN B0473CC8
17:07:20.087 -> 
17:07:20.087 -> 
17:07:22.727 -> Timestamp : 000014.707
17:07:22.727 -> Library   : v2.7.11
17:07:22.727 -> 
17:07:22.727 -> Protocol  : UNKNOWN
17:07:22.727 -> Code      : 0xB0473CC8 (50 Bits)
17:07:22.774 -> uint16_t rawData[99] = {4384, 4362,  548, 1620,  498, 546,  522, 1644,  496, 1644,  520, 522,  548, 522,  524, 1644,  522, 520,  548, 520,  526, 1642,  522, 522,  550, 520,  550, 1618,  522, 1620,  520, 522,  550, 1620,  522, 518,  524, 548,  522, 546,  548, 522,  548, 1620,  522, 1618,  522, 1620,  522, 1618,  520, 1620,  520, 1620,  520, 1620,  522, 1620,  522, 522,  546, 522,  546, 522,  548, 522,  548, 1620,  522, 1618,  522, 1620,  522, 520,  548, 520,  548, 520,  550, 522,  548, 520,  548, 522,  548, 520,  550, 520,  548, 1618,  524, 1616,  524, 1616,  522, 1620,  524, 1616,  522};  // UNKNOWN B0473CC8
17:07:22.819 -> 
17:07:22.819 -> 
17:07:24.992 -> Timestamp : 000016.968
17:07:24.992 -> Library   : v2.7.11
17:07:24.992 -> 
17:07:24.992 -> Protocol  : UNKNOWN
17:07:24.992 -> Code      : 0xB0473CC8 (50 Bits)
17:07:24.992 -> uint16_t rawData[99] = {4382, 4366,  546, 1622,  518, 526,  546, 1620,  496, 1644,  520, 524,  546, 524,  546, 1620,  520, 520,  548, 524,  524, 1644,  520, 524,  546, 522,  524, 1646,  520, 1620,  498, 546,  524, 1644,  522, 520,  548, 522,  546, 524,  548, 522,  548, 1620,  520, 1620,  522, 1620,  520, 1620,  522, 1620,  522, 1620,  524, 1618,  522, 1618,  522, 522,  548, 522,  548, 522,  548, 520,  550, 1618,  522, 1618,  524, 1618,  522, 520,  550, 520,  550, 520,  550, 520,  524, 544,  548, 522,  550, 520,  550, 520,  548, 1620,  522, 1618,  522, 1620,  522, 1618,  522, 1618,  524};  // UNKNOWN B0473CC8
17:07:25.085 -> 
17:07:25.085 -> 
```

Watching the actual Coolix integration it seems fine about the rest. Can't find an issue about "overlapping", like the "special function" buttons the FollowMe has a specific pattern. 

The fact about the timing can be the reason why FollowMe messages generated by the library dind't work on the AC. 
Here the two grabs https://github.com/crankyoldgit/IRremoteESP8266/issues/1318#issuecomment-721829757 tell me if you need more

crankyoldgit, 2020-11-16:
Okay, I found the problem decoding the "Direct" messages. Not a timing error at all. See https://github.com/crankyoldgit/IRremoteESP8266/commit/85c5b4dac37df5fd5ac3d9a5cf9ac4865ad30ced for details.
It seems "Direct" is an odd-one-out message w.r.t. the Coolix protocol. i.e. No repeat.
Btw. What does this "Direct" button actually do?

Please download & test branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Kaysun_Coolix
I'll be using this branch going forward for the Coolix changes for this Issue, as we work them out.

[21 more replies]

PR #1340 (2020-11-24): Coolix: Improve Sensor(ZoneFollow) and add Vane Step support.
* Increase sensor temp range.
* using `setSensorTemp()` now correctly sets up ZoneFollow mode.
* ZoneFollow mode detection.
* Misc code cleanup.
* Refactor code to use bit fields.
* Additional & updated Unit tests.
* Add support for fan vane step commands.

Fixes #1318

* Confirmed working by user in https://github.com/crankyoldgit/IRremoteESP8266/issues/1318#issuecomment-733085483
commit 34c283d2 (2020-11-24): Coolix: Improve Sensor(ZoneFollow) and add Vane Step support. (#1340)

* Increase sensor temp range.
* using `setSensorTemp()` now correctly sets up ZoneFollow mode.
* ZoneFollow mode detection.
* Misc code cleanup.
* Refactor code to use bit fields.
* Additional & updated Unit tests.
* Add support for fan vane step commands.

Fixes #1318

* Confirmed working by user in https://github.com/crankyoldgit/IRremoteESP8266/issues/1318#issuecomment-733085483
closed 2020-11-24 by crankyoldgit