crankyoldgit/IRremoteESP8266 #1319: Weird LG behavior when sending a RAW code

GiampaMan, 2020-11-04:
Hi @crankyoldgit ! 

Sorry for the delays on creating the new issue, I was a bit busy these days. I uploaded the video to youtube so you can see which is my problem. I will put the link here:

https://youtu.be/CJ6YPOvcs0s

Thanks a lot!
Giampaolo

crankyoldgit, 2020-11-05:
Okay, I've watched your video.

Lets start with some of the basics. (sorry if they are repeated questions I've asked before, but I get so many reported issues, they all tend to blur together.)

- What "software" are you running on the ESP32? i.e. Can you please include your code? I need to be sure what signals are being sent and how the LG ac class is being handled.
- Can you capture the signal from the `ESP32` using `IRrecvDumpV2` (or V3)` on another ESP module for when it (The A/C device) ISN'T working?
- Can you please capture what you think is the "same" message you are sending from the ESP32, but from the Original Remote? This way I can compare what it really takes to turn on your LG device. (please also document what button presses you needed to perform etc to produce the message on the original remote too!)

I agree it (hardware) appears to be able generate IR messages correctly, so we can probably rule that out.
The library is generating valid codes and the LG A/C can understand (most) of them. So it's unlikely a protocol timing issue.
If it can send one and have it understood, they pretty much all should. The problem is probably how the parts of the class (or the code values you're passing) are generating what "codes" to send, and maybe there is something in the order or the sequence we don't understand properly, or there is some other part of a "special" message we haven't seen before that is required to wake up _your_ A/C. It seems once it is awake, it responds correctly to everything. Thus maybe there is some special "wake up" message we need.

As I said earlier, I really need to see the code you're using to work out the **exact** same sequence and compare it to the **real** remote messages.

GiampaMan, 2020-11-05:
Hi @crankyoldgit ! Thanks for the reply

- I have a really large code so I will try to simplify it so you can test it. It's running on Arduino Framework.
- Great I will do it, I will capture when the ESP is failing with another one.
- Sure, I can capture the original remote code for the same command I think the ESP is sending.

As I told you before, sometimes this takes long to produce the failure, but ASAP as it produces the failure I will record the command and send it to you.

I want also to explain you how I think this A/C model work. I can turn on the air directly with the temperature command (pressing on button don't send anything "extra"). For example in the video I send you, I didn't worked on the turn on command, but that behavior can also happen when the A/C is turned on and I just want to change the temperature.

The weird part is that is not a consistent failure (sometimes it fails on the first minute and sometimes takes days to see this failing) seems as you said before, something like a timing issue.

In the meanwhile while I try to generate again the fail leaving the device connected and monitoring it trough another one, I will try to simplify my actual code so you can test it.

Thanks a lot for the help and have a great day!
Giampaolo

crankyoldgit, 2020-11-06:
> * I have a really large code so I will try to simplify it so you can test it. It's running on Arduino Framework.
> * Great I will do it, I will capture when the ESP is failing with another one.
> * Sure, I can capture the original remote code for the same command I think the ESP is sending.

Thanks. Looking forward to it.

I suspect that it's a coding problem in your code. Has your existing code got logging at all? i.e. Are you logging debug info to the serial port or something for what hex codes you are sending, and in what sequence etc etc?
e.g. How certain are you that when the A/C isn't responding to the message that you are actually sending the message code you _think_ you are sending?

GiampaMan, 2020-11-09:
Hi @crankyoldgit !

Well finally I get the bug again and I did what you requested to me. I captured the two IR signals. The first one was the "bad" signal from the ESP32 and the second one was the "good" signal from the original LG A/C remote.

First of all, responding to some of your questions:

- I know my code is not the problem because I tested this with other A/C brands and it worked perfectly without having this issues.
- I know that the command I'm sending to the air conditioner is the right one because I captured the command from the original A/C remote and it works fine, but at some point it stops to work well (only on this brand).

I will give you the data that I captured:

From my ESP32 (the "bad" signal):

```
Timestamp : 000000.000
Protocol  : UNKNOWN
Code      : 0xD5C3694D (30 Bits)
Library   : v2.7.11

Raw Timing[59]:
   +  3454, -  9758,    +   594, -  1490,    +   578, -   514,    +   536, -   504, 
   +   540, -   496,    +   592, -  1480,    +   566, -   474,    +   574, -   462, 
   +   608, -   430,    +   578, -   458,    +   556, -   474,    +   572, -   468, 
   +   546, -   494,    +   572, -  1494,    +   574, -   446,    +   616, -   450, 
   +   564, -   466,    +   590, -  1478,    +   542, -   476,    +   590, -   472, 
   +   530, -   504,    +   594, -   442,    +   594, -   432,    +   594, -   470, 
   +   504, -   530,    +   558, -   480,    +   598, -   426,    +   594, -   470, 
   +   630, -   406,    +   604

uint16_t rawData[59] = {3454, 9758,  594, 1490,  578, 514,  536, 504,  540, 496,  592, 1480,  566, 474,  574, 462,  608, 430,  578, 458,  556, 474,  572, 468,  546, 494,  572, 1494,  574, 446,  616, 450,  564, 466,  590, 1478,  542, 476,  590, 472,  530, 504,  594, 442,  594, 432,  594, 470,  504, 530,  558, 480,  598, 426,  594, 470,  630, 406,  604};  // UNKNOWN D5C3694D
```

From my LG original remote (the "good" signal):

```
Timestamp : 000000.000
Protocol  : LG2
Code      : 0x8800808 (28 Bits)
Library   : v2.7.11

Raw Timing[59]:
   +  3382, -  9864,    +   532, -  1588,    +   490, -   524,    +   510, -   544, 
   +   488, -   542,    +   488, -  1570,    +   512, -   524,    +   514, -   542, 
   +   488, -   552,    +   488, -   536,    +   536, -   520,    +   484, -   546, 
   +   510, -   506,    +   510, -   542,    +   488, -   546,    +   484, -   546, 
   +   490, -   532,    +   516, -  1572,    +   490, -   550,    +   486, -   544, 
   +   490, -   542,    +   488, -   542,    +   490, -   542,    +   490, -   524, 
   +   516, -   542,    +   490, -  1574,    +   488, -   536,    +   510, -   542, 
   +   490, -   546,    +   486

uint16_t rawData[59] = {3382, 9864,  532, 1588,  490, 524,  510, 544,  488, 542,  488, 1570,  512, 524,  514, 542,  488, 552,  488, 536,  536, 520,  484, 546,  510, 506,  510, 542,  488, 546,  484, 546,  490, 532,  516, 1572,  490, 550,  486, 544,  490, 542,  488, 542,  490, 542,  490, 524,  516, 542,  490, 1574,  488, 536,  510, 542,  490, 546,  486};  // LG2 8800808
uint32_t address = 0x88;
uint32_t command = 0x80;
uint64_t data = 0x8800808;
```

That data is for the 23 degrees celsius lowest fan command.

Thanks a lot for all the help and have a great day!
Giampaolo

crankyoldgit, 2020-11-09:
Looking at the "Bad" signal, it clearly is a `LG2 0x8800808` message. Just the timings are slightly off.

The data indicates that the bit mark values are the "old" incorrect bit mark values. e.g. They are higher than what they should be by about 50. They are close enough to probably work, but perhaps them being closer to the expected value might give you a more reliable signal. They were fixed recently.

I think your "transmitter" code is not using a version of the library that includes this patch/commit: https://github.com/crankyoldgit/IRremoteESP8266/commit/3787433e4d209ab3cf6d1187e264aace0a21e251

This lowers the bit mark time (uses the correct value) of the `LG2` protocol.

So, download the current/latest `master` branch and rebuild/re-upload both your own code, and the dump program.
You want to make sure it is using the `master` branch, not a cached copy of the `2.7.11` release version.

GiampaMan, 2020-11-09:
Hi @crankyoldgit !

Thanks for the reply! Ok. I think I have that patch already implemented. To be sure I will download the library again, just in case.
Cool! So now we know is a timing issue. But one question, that patch also helps with sending the RAWs? Because I'm not sending the HEX, I send RAW.

Thanks for all!
Giampaolo

crankyoldgit, 2020-11-10:
> But one question, that patch also helps with sending the RAWs?

No, it won't.

> Because I'm not sending the HEX, I send RAW.

Why on earth are you sending via RAW?!?! :-O Hex (`sendLG2()`) is better and more efficient. ;-)

If you are going to continue to transmit using `sendRaw()` I suggest you use the capture from the "good" signal above instead of the what ever raw array you are using now for that message.

If you are sending a potentially borderline raw message each time, of course it is going to have a similar borderline response.

I really do recommend sending via the purpose built `sendLG2()` method rather than RAW, as this means you/we can systematically fix or improve the timings for the protocol to suit your actual device.

Assuming your code is correct etc and has no bugs etc etc ... my guess is the A/C is having trouble with the "poor" raw message. It may be that while the A/C is sleeping/in low power mode it has less tolerance for incoming IR messages, so it doesn't wake up accidentally. e.g. The message you're currently replaying may have subtle noise/errors in it. Think of it as an Analogue-to-Analogue recording, verses a digital-to-digital one. AtoA is inherently lossey. DtoD is not, it should be a perfect copy.
See: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Why_is_the_raw_data_for_a_button_or_AC_state_always_different_for_each_capture

GiampaMan, 2020-11-12:
Hi @crankyoldgit !
Thanks for the reply

> No, it won't.

Ok, so that explains why the new modifications didn't worked for my specific case.

> Why on earth are you sending via RAW?!?! :-O Hex (sendLG2()) is better and more efficient. ;-)

haha, I will try using HEX to see if works. The reason why I'm using RAW is because my program works in this way:

- You can capture a signal from a remote (example: the LG remote), then the program saves the RAW on a file inside ESP32 SPIFFS memory.
- When I want to send the command to the A/C, the program read the RAW from the SPIFFS and send it.

> I suggest you use the capture from the "good" signal above instead of the what ever raw array you are using now for that message.

That's the problem, that I captured the "good" one from the original remote (and that's the signal that the ESP is supposed to send, the good one). The problem is that sometimes it sends ok, but sometimes it sends as the bad signal I sent you. But I'm capturing it from the "good" one. I hope I can explain myself.

> I really do recommend sending via the purpose built sendLG2() method rather than RAW, as this means you/we can systematically fix or improve the timings for the protocol to suit your actual device. [...]

Yes, the code is correct. I think like you, that might be a noisy signal the one that the ESP32 sends. But I wonder why? Because we are capturing it from the "good" signal, and it sends the "bad" one. This only happens with my LG A/C.
Also thanks for the link, was interesting. I liked the way it was explained 👍🏻 

The really weird part is that I don't remember that this was failing when I was using an ESP8266. I changed to ESP32 because I needed more space for variables and the ESP8266 memory didn't support it.

I will try in the meantime sending the HEX with LG2 protocol. I will let you know the results.

Let me know what do you think.
Have a great day and thanks a lot for helping me!
Giampaolo 😃

crankyoldgit, 2020-11-13:
Depending what else the ESP32 is doing (e.g. bluetooth & wifi etc) it could be the case that there are other processes on the device that are slowing it down when it's sending the IR message.

Something to try is to use the second processor to send the IR message on the ESP32.
See: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Some_other_library_eg_Bluetooth_is_interfering_with_IRsend_What_can_I_do_ESP32

Can you retrieve the "raw" message from the ESP's flash for the message you are having issues with? That way we can compare what the ESP32 was trying to send to what it actually sent (timing-wise).

crankyoldgit, 2020-11-18:
Friendly chase up.

GiampaMan, 2020-11-18:
Hi @crankyoldgit !
How is everything going?

Sorry for the delays, I was doing a lot of tests and wanted to tell you my results

> Depending what else the ESP32 is doing (e.g. bluetooth & wifi etc) it could be the case that there are other processes on the device that are slowing it down when it's sending the IR message. [...]

I thought that this has a lot of sense so I decided to implement the second core just to only send the IR Raw. I'm impressed because now works worse than before. The second core ("1" for coding purposes) only has one task, which is the sending of the Raw code. Then, after sendRaw function is called, I delete the task until it is necessary to send another Raw code again.

What do you think about this? I think like you too, the second core should work better because is free. Do you think is any part of the chip that is common to both cores and is interfering? I'm using WIFI, SPIFFS and Irrecv on core 1 ("0" for coding purposes), and this is set by default by Arduino.

I tried everything, for example:

- Using semaphores to stop all the process on the core 1 and only use core 2 for the moment when the signal is sent, then stop core 2 and continuing with core one. (In a synchronous way)
- Without using semaphores, so in the moment when the IR code is sent, the other core is continuing with the code and the rest of the processes. (In an asynchronous way)

I did all kind of variants on the points I mentioned above but still working worse than before.

> Can you retrieve the "raw" message from the ESP's flash for the message you are having issues with? That way we can compare what the ESP32 was trying to send to what it actually sent (timing-wise).

Sure! Do you mean the message on the SPIFFS memory? So we can compare with the correct one? Let me know and I will send it to you.

I'm very grateful because of the patience on this case and all your help! Let me know your opinion.
Giampaolo

crankyoldgit, 2020-11-18:
> I did all kind of variants on the points I mentioned above but still working worse than before.

Well, that's a bummer. :-(

> Sure! Do you mean the message on the SPIFFS memory? So we can compare with the correct one? Let me know and I will send it to you.

Yes. Exactly.

Another thing to try/eliminate is to use the "good" signal (i.e. `uint16_t rawData[59] = {3382, 9864,  532, 1588,  490, 524,  510, 544,  488, 542,  488, 1570,  512, 524,  514, 542,  488, 552,  488, 536,  536, 520,  484, 546,  510, 506,  510, 542,  488, 546,  484, 546,  490, 532,  516, 1572,  490, 550,  486, 544,  490, 542,  488, 542,  490, 542,  490, 524,  516, 542,  490, 1574,  488, 536,  510, 542,  490, 546,  486};  // LG2 8800808`)
or `irsend.sendLG2(0x8800808);` in the [IRsendDemo.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino) example code. That is, the simplest possible running code to eliminate any possible interference from any other activity on the ESP32.
If it still behaves badly then, then we know it has to be something fundamental with the hardware, or with the A/C unit.

GiampaMan, 2020-11-21:
Hi @crankyoldgit 

> Well, that's a bummer. :-(

That's right :(

I'm doing a test right now. It's what you mentioned. I will try to send the signal with an empty ESP32 so we can check.

Before that, I took an ESP32 with nothing. Just a receiver and the dumpv2 code. Now I'm waiting the device to fail so I can send you the following:

- The "fail" signal captured directly from the ESP32 when it fails to turn on the A/C.
- Also I will capture the "good" signal that the ESP32 sends when the A/C works with it.

- The good signal that the remote uses

- The data read from the SPIFFS so you can check the timing as you mentioned in your latest comment.

Have a great day and thanks a lot for your help! I will tell you the results as soon as possible.

Giampaolo

crankyoldgit, 2020-11-21:
> Now I'm waiting the device to fail so I can send you the following:

Well personally, I'm hoping the stupidly simple "send" demo code won't fail you. ;-)
If it's working 95% of the time, then there probably really isn't anything we can do to improve it. It could just be your A/C being quirky. O_o

GiampaMan, 2020-11-23:
Hi @crankyoldgit !

The device had failed again. Now I have the information I told you:

From SPIFFS I'm getting this information:
`3396, 9810, 580, 1572, 526, 464, 500, 580, 500, 532, 502, 1530, 548, 532, 452, 580, 498, 494, 518, 526, 554, 556, 454, 526, 548, 534, 448, 1614, 454, 538, 548, 502, 500, 580, 506, 1560, 454, 582, 504, 526, 452, 1616, 528, 470, 548, 532, 502, 532, 452, 580, 450, 580, 454, 584, 448, 580, 500, 1564, 452` And the RAW number 59 (all this was read from SPIFFS)

This is the signal that the ESP sends reading the information above and the A/C works perfectly:
```
Timestamp : 000414.006
Library   : v2.7.11

Protocol  : LG2
Code      : 0x8808901 (28 Bits)
Mesg Desc.: Model: 2 (AKB75215403), Power: On, Mode: 0 (Cool), Temp: 24C, Fan: 0 (Quiet)
uint16_t rawData[59] = {3400, 9828,  588, 1568,  554, 438,  504, 578,  506, 528,  502, 1534,  552, 530,  450, 586,  504, 492,  510, 538,  558, 556,  448, 536,  552, 532,  476, 1592,  450, 546,  552, 502,  528, 554,  504, 1568,  448, 592,  504, 530,  450, 1624,  530, 472,  562, 518,  506, 530,  474, 562,  450, 584,  456, 586,  452, 580,  504, 1564,  458};  // LG2 8808901
uint32_t address = 0x88;
uint32_t command = 0x890;
uint64_t data = 0x8808901;
```

Now, this is the bad signal, when the ESP don't work (also sending using the same SPIFFS read):
```
Timestamp : 000115.909
Library   : v2.7.11

Protocol  : LG2
Code      : 0x8808901 (28 Bits)
Mesg Desc.: Model: 2 (AKB75215403), Power: On, Mode: 0 (Cool), Temp: 24C, Fan: 0 (Quiet)
uint16_t rawData[59] = {3402, 9828,  582, 1574,  530, 462,  502, 582,  504, 534,  502, 1534,  524, 560,  448, 588,  508, 486,  472, 574,  532, 580,  448, 536,  526, 558,  446, 1620,  448, 548,  526, 526,  504, 580,  502, 1566,  448, 594,  500, 534,  446, 1626,  530, 474,  526, 556,  502, 536,  448, 588,  450, 586,  448, 594,  448, 584,  502, 1568,  448};  // LG2 8808901
uint32_t address = 0x88;
uint32_t command = 0x890;
uint64_t data = 0x8808901;

```

The "bad" and the "good" signals that I provided were taken from another ESP32 running dumpv3. Let me know what do you think.

> Well personally, I'm hoping the stupidly simple "send" demo code won't fail you. ;-)
> If it's working 95% of the time, then there probably really isn't anything we can do to improve it. It could just be your A/C being quirky. O_o

Haha, I will check that too.

Thanks a lot for your help. Have a great day!
Giampaolo

crankyoldgit, 2020-11-23:
> From SPIFFS I'm getting this information:
> `3396, 9810, 580, 1572, 526, 464, 500, 580, 500, 532, 502, 1530, 548, 532, 452, 580, 498, 494, 518, 526, 554, 556, 454, 526, 548, 534, 448, 1614, 454, 538, [...]

I'm pretty sure this is where your problem is likely to be stemming from.

Your original "recording" (i.e. stored in SPIFFS) is a poor one.
According to our `LG2` timings (used in a `sendLG2()`), every _odd_ number (past the header) should be fairly close to:
https://github.com/crankyoldgit/IRremoteESP8266/blob/102facd1bf966d656df4e040d7c3889327a526e0/src/ir_LG.cpp#L44
e.g. Your SPIFFS data has the 3rd entry being `580`, instead of an _ideal_ `480`. It's off by 100us (or approx 20%) so its probably border-line for detection by the A/C unit.

Every _even_ number (except the header_ should be ideally, either:
https://github.com/crankyoldgit/IRremoteESP8266/blob/102facd1bf966d656df4e040d7c3889327a526e0/src/ir_LG.cpp#L29-L30
Those values are closer to your SPIFFS data, but still a fair bit off in some cases.

Btw, this is one of the reasons we recommend using a supported protocol send routine vs, using Raw data. e.g. `sendLG2()`.
Think of it as you're using a bootleg recording in a live concert from someone with a tape recorder in the audience, vs. a studio digital mastered CD etc.

In short, your source (SPIFFS) raw data is a very border-line signal at best, so replaying it means that other random fluctuations in the environment, on the ESP, in your transmitting circuit, or in the receiving A/C unit will make that magnified.
No matter how often you repeat it, it's going to be a "poor" signal.
If you want to still keep using raw data stored in SPIFFS, I suggest you clean up the data manually, or replace it with the data in https://github.com/crankyoldgit/IRremoteESP8266/issues/1319#issuecomment-729963403

However, the "best" solution is to use `sendLG2(0x8808901);` instead. ;-)

GiampaMan, 2020-11-30:
Hi @crankyoldgit !

Sorry for the delays, was doing a lot of tests and have a lot of information. First of all, I did what you mentioned (just use the simple send example). This presented the same Issue than before. For this reason I started to do the following things:

- I put a simple code to read the LG A/C remote from an ESP32, and an ESP8266. The 8266 throws me less unknown's if we compare it to the ESP32. When I saw that, I tought that maybe the ESP8266 is capturing the signal in a good way and more accurate, so what I did was to put that RAW code also on the simple send example and again the A/C didn't do anything. I started to do all kind of combinations with the ESP32 and ESP8266 for sending/reading, also tested with different IR receptor components and different boards, put new batteries on the A/C control but the result was always the same. Apart all of this, used three different controls (I have 3 same A/C of that same model on my house).
- What I can say is that I captured a long time ago one button on the LG remote and I have a RAW that always works, but the new ones that I read and send don't do anything. This is the 100% good signal: 
`uint16_t rawData[59] = {3232, 9892, 492, 1604, 464, 560, 460, 588, 484, 556, 488, 1616, 464, 556, 488, 548, 484, 548, 484, 552, 484, 548, 484, 548, 488, 556, 460, 588, 484, 556, 464, 572, 484, 556, 484, 1604, 464, 548, 484, 552, 484, 548, 484, 548, 488, 548, 484, 556, 484, 568, 484, 1612, 464, 556, 484, 548, 484, 548, 488};` This incredibly always works without problems when I send (in this case is 23°C).

- Apart all of this I tried to send the same code (23°C) but using the HEX on sendLG2, again, a simple send code without any good results although I used the HEX :( NOTE: Something really weird about this is that sendLG2 never worked, so I decided to test using sendLG and only worked one time (I don't know why). The HEX code is this: `irsend.sendLG(0x8808800);`

I think that maybe the problem could be on the capturing and not on the sending as I thought before. But don't know what to do. I was all this days testing a lot but I don't know what could be the next step to test.

Thanks a lot for all your help! Let me know If you have any idea.
Giampaolo

crankyoldgit, 2020-11-30:
> * What I can say is that I captured a long time ago one button on the LG remote and I have a RAW that always works, but the new ones that I read and send don't do anything. This is the 100% good [...]

A quick skim of that raw code looks pretty much what I'd expect `sendLG2()` to produce. The non-header values are within 20us of the values we consider "ideal" for `LG2`.  i.e. Almost as perfect a signal as we could expect. So, I'm going to scratch my head and give up.
Without an oscilloscope on your system to verify what really is going on, and using alternative IR leds etc, it's hard to say where to go from here.
I'm **really** surprised `sendLG2()` isn't working for you.

One last thing to try, which really is a last grasp, is to try adding a [`irsend.calibrate()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#ad1776aa6c699f9eeca1eef9bb4fe355b) call at boot. i.e. well before you send a message. This will optimise any internal _sending_ timing adjustments internally, just in case one of your ESP devices is running at some strange speed.

If that doesn't work, I'm out of ideas except for replacing the IR transmitting hardware & circuit with something simpler (which is a really low probability that that's the issue). Try different transmission distances maybe?!?
I'm really out of ideas.

GiampaMan, 2020-12-11:
Hi @crankyoldgit !

Sorry for the delays, had busy days. But also was testing a couple of things:

- Tested different distances without any change on the behavior.
- Tested to calibrate as you said but still is not working good.
- Used a Raspberry Pi with Lirc to capture the signal and try to send it through the ESP.

Nothing of the above worked but I have an idea. What do you think about capturing the IR code at night in complete dark? Maybe is some kind of interference. Also if this is related I found an IC that maybe could help (want to know your opinion).

Vishay has an IC with the following reference: VSOP98260 and as far as I understood is to remove interference and correct signals of IR when receiving. Maybe with a filtering like that, removing all the noise from the signal, the ESP could capture it in a good way. What do you think?

Thanks a lot and have a good day!
Giampaolo

crankyoldgit, 2020-12-11:
> Nothing of the above worked but I have an idea. What do you think about capturing the IR code at night in complete dark? Maybe is some kind of interference.

Yes, that's a good thing to do/try. Both sending and receiving are better in a dark and cool room.

>  Also if this is related I found an IC that maybe could help (want to know your opinion).
Vishay has an IC with the following reference: VSOP98260 and as far as I understood is to remove interference and correct signals of IR when receiving. Maybe with a filtering like that, removing all the noise from the signal, the ESP could capture it in a good way. What do you think?

I'm pretty sure capture (via a computer) is not the problem at this stage. It's the _sending_ part you seem to have a problem with.

i.e. There is a subtle problem between the ESP telling the hardware to produce the IR signal and the A/C decoding it.
e.g. It could be any of:
* A bad ESP (unlikely based on tests so far)
* A poor transmitting circuit. (unlikely)
* Poor quality IR LEDs (Possible)
* Underpowered IR LEDs (Possible)
* Too powerful IR LEDs. (Possible)
* Burnt-out IR LEDs. (unlikely)
* IR interference in the environment. (Possible)
* A poor quality IR decoder in the A/C. (Possible)
* Finicky software in the A/C. (Possible)

Any of those (and combinations) are possibilities. For example, it could be your IR LEDs are to powerful, **and** because where your A/Cs IR sensor is located, it might be getting lots of IR reflections off the wall etc from the transmitter.

GiampaMan, 2021-01-21:
Hi @crankyoldgit 
Thanks a lot for all the help provided and sorry for all the delays on finally writing again related to this issue. First of all I hope you had a great holidays and a happy New Year! How is everything going?

I finally discovered what was happening with this issue and I want to share this with you, is a combination of various things that I'm going to tell now. I was investigating and doing some tests.

- **I realized that the code that I was sending to the air is not correct:** I recorded the temperature code only using the temperature up and down buttons which seemed to work. But there is a combination of various things that made this behavior occurs. Apparently the IR buttons that I was capturing (which is not the correct for "turning on") works on my room LG A/C for some magic reasons all the time (the same model as the other one), but this doesn't work on the other. I realized this because I tried to turn on both A/Cs using temperature up and down buttons and didn't worked.
- **My solution:** I tried to turn on the A/C with "power on" code on a specific temperature and this seems to work. So in the first place I set the temperature that I wanted to record on the remote, turn it off and then captured the "turn on" signal for the desired temperature. Also, this is a good solution because if you send this command when the A/C is turned on it also works at the same way as when it was not turned on.
- **Here is a combination of various factors:** First of all, a poor quality IR decoder in the A/C as you said. I consider this because this codes that I was recording with my ESP, that are supposed to be wrong, works on my room LG A/C but not on the others, so seems that this means that even if the code recorded was wrong, the A/C has some weird tolerance that still can't differentiate wether the IR code I'm sending is the "turn on" signal or just the "set temperature to" signal. Also, we add here another tolerance when sending from the ESP device, because the changing temperature signal sent was misunderstood by the A/C and this only happens if this was sent from the ESP, but if you use the original remote, it also don't turns on. So, until this moment, we can think that this "turn on with x temperature" command is pretty similar to the "set A/C with x temperature" command.

I also tested what you suggested like recording at night with lights off but before the solution I comment above it still not worked.

I want to send you a big thanks because I found your help very useful. I'm very grateful with you for helping me along the issue discovering process.

Thanks a lot!
Giampaolo 😄

crankyoldgit, 2021-01-21:
Glad you finally got to the bottom of it. Using "stored" codes is not the greatest or most flexible, as you've found out.
This is one of the advantages of using the `IRLgAc` or `IRac` class interface via this library (to avoid saving stored codes) ;-)

Thanks for the update & getting back to us on this issue, and yep, that certainly probably is what the problem was.

I'll mark this resolved/closed now. Phew! What a long strange trip it's been!

closed 2021-01-21 by crankyoldgit