crankyoldgit/IRremoteESP8266 #1320: I'm getting a core panic

jvanlangen, 2020-11-04:
Very unclear template.

Using the IRremoteESP8266 v2.7.11 on a esp32 d1 mini
irrecv.enableIRIn(); is called in the setup. It is connected on PIN 35, (i've tried several pins)

When I call "irrecv.decode(&results)" i'm getting this:
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.

It doesn't say which exception

NiKiZe, 2020-11-04:
Do you have anything else in your code, or is it a simple example?

Have you run anything else on the same hardware before?

Quick search gives some examples of storage or power issues for the `Guru Meditation Error: Core 0 panic'ed (IllegalInstruction). Exception was unhandled`.

Could you be more precise in what you find unclear with the template?

jvanlangen, 2020-11-04:
Hi Christian,

I'm using the neopixelbus and the websocket library. I haven't tried it on a simple example. Without the irremote code, it runs steady.

Regards,
Jeroen.

NiKiZe, 2020-11-04:
Could you try without other libs as a test and see if it works?

If it is not an issue with your module itself (mentioned earlier in terms of memory or power as examples) then it could be that some of the other things that you are using does conflict with this, for example timers or similar.

Without being able to reproduce the issue (for example by you sharing your full code) I think you will have to test your way to find why/where this happens.

I would strongly suggest that you try a minimal example with only IRremoteESP8266 example code and make sure if that works first.

jvanlangen, 2020-11-04:
I'll try that first, thank you.

crankyoldgit, 2020-11-04:
After you've tried the example code etc, here are some things to play with which _might_ be causing an issue.
Use of "delay()" calls:
https://github.com/crankyoldgit/IRremoteESP8266/blob/9f19d55f390c6620ece54435ca1312025d556733/src/IRremoteESP8266.h#L742-L748

Incompatible use of a system timer with some other library (i.e. `timer_num`):
https://github.com/crankyoldgit/IRremoteESP8266/blob/9f19d55f390c6620ece54435ca1312025d556733/src/IRrecv.cpp#L138-L156

You're not using a `yield()` call frequently enough?

You've run out of `IRAM` in your program/ESP32, and thus the interrupt handler(s) are no longer in IRAM and causes a CPU exception. (maybe)

Are you using an up-to-date copy of the core library for the ESP32? (Check your `board manager` etc)?

crankyoldgit, 2020-11-04:
Some other suggestions can be found in https://github.com/crankyoldgit/IRremoteESP8266/issues/1073

crankyoldgit, 2020-11-04:
> Without being able to reproduce the issue (for example by you sharing your full code) I think you will have to test your way to find why/where this happens.

^^^^^^^
This. 100 times this!
See points 4 & 5 of https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#reporting-an-issue

crankyoldgit, 2020-11-07:
@jvanlangen Any updates on your testing?

jvanlangen, 2020-11-07:
Nope, sorry, not at home. Thanks for asking

Verzonden vanaf mijn Galaxy

crankyoldgit, 2020-11-24:
Another chase-up ping.

jvanlangen, 2020-11-25:
I appreciate your helpfulness, but I see little opportunity to do anything with this soon. I don't have the time to dive deep into this. I now use another way without IR.

closed 2020-11-25 by jvanlangen

crankyoldgit, 2020-12-03:
FYI This may have been fixed with #1351