crankyoldgit/IRremoteESP8266 #1324: 'IRsend' does not name a type

beaglesparks, 2020-11-07:
When compiling, I keep getting " 'IRsend' does not name a type" and  "'irsend' was not declared in this scope"

I have checked the library folder and besides irremoteESP8266.h I have no other IRsend reference. I did a text search if the folder

Arduino IDE 1.8.13
IRremoteESP8266.h  2.7.11
[error.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/5505017/error.txt)

Code --------------------------------

//#include <IRremote.h>
  #include <IRremoteESP8266.h>
//#include <IRremote.h>
//#include <ESP8266WiFi.h>
//#include <ESP8266WebServer.h>
//#include <WiFiUdp.h>
//#include <functional>

IRsend irsend;//(16);    // IR LED out pin  D0 which is GPIO16     <===== ERROR HERE
unsigned long SamsungPower;

const int LED_SCK=14;
const int LED_WIFI=2; // this pin is inverted, Low turns it on

const int SamsungBits = 32;
//const unsigned long SamsungPower=3772793023;   //Power button

void setup() {
  // put your setup code here, to run once:
pinMode(LED_SCK, OUTPUT);
pinMode(LED_WIFI, OUTPUT);
digitalWrite(LED_SCK, LOW);
SamsungPower=3772793023;
//irsend.begin();
}
void loop() {
  //for (int i = 0; i < 2; i++) {
    digitalWrite(LED_SCK, HIGH);
    irsend.sendSAMSUNG(SamsungPower, 32);
    delay(40);
    //irsend.sendSAMSUNG(SamsungPower, SamsungBits);
    delay(60);
    digitalWrite(LED_SCK, LOW);
  //}
  delay(900); //delay between each signal burst
}

NiKiZe, 2020-11-07:
To me it looks like you have taken an example from some other IR lib and then try to shoe horn IRremoteESP8266 into their example? Have you considers using One of our [examples](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/ControlSamsungAC/ControlSamsungAC.ino) as a base instead?

Also relevant: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me

And also relevant, Why did you delete the issue template, instead of reading it and filling it in?

beaglesparks, 2020-11-07:
Sorry about not following protocol. Didn't think it mattered. Now I know better.

Thanks for your response. I looked at your example and realized i left out   #include <IRsend.h>

that fixed it.

beaglesparks, 2020-11-07:
thanks NiKiZe

closed 2020-11-07 by beaglesparks