crankyoldgit/IRremoteESP8266 #1329: RecvDumpv2  example code not woring in ESP-IDF

MarkEvens, 2020-11-11:
I am using esp-idf v 4.0.1 and liberary v 2.7.11
below is esp-idf code that is just a example code

```
#include <Arduino.h>
#include <assert.h>
#include <IRrecv.h>
#include <IRremoteESP8266.h>
#include <IRac.h>
#include <IRtext.h>
#include <IRutils.h>

const uint16_t kRecvPin = 19;

const uint32_t kBaudRate = 115200;

const uint16_t kCaptureBufferSize = 1024;

#if DECODE_AC
const uint8_t kTimeout = 50;
#else   // DECODE_AC
const uint8_t kTimeout = 15;
#endif  // DECODE_AC

const uint16_t kMinUnknownSize = 12;

#define LEGACY_TIMING_INFO false

IRrecv irrecv(kRecvPin, kCaptureBufferSize, kTimeout, true);
decode_results results;  // Somewhere to store the results

extern "C" {
void app_main();
}

void app_main(void) {

#if defined(ESP8266)
  Serial.begin(kBaudRate, SERIAL_8N1, SERIAL_TX_ONLY);
#else  // ESP8266
  Serial.begin(kBaudRate, SERIAL_8N1);
#endif  // ESP8266
  while (!Serial)  // Wait for the serial connection to be establised.
    delay(50);
  // Perform a low level sanity checks that the compiler performs bit field
  // packing as we expect and Endianness is as we expect.
  assert(irutils::lowLevelSanityCheck() == 0);

  Serial.printf("\n" D_STR_IRRECVDUMP_STARTUP "\n", kRecvPin);
#if DECODE_HASH
  // Ignore messages with less than minimum on or off pulses.
  irrecv.setUnknownThreshold(kMinUnknownSize);
#endif  // DECODE_HASH
  irrecv.enableIRIn();  // Start the receiver

	while (1) {
            // Check if the IR code has been received.
      if (irrecv.decode(&results)) {
         // Display a crude timestamp.
         uint32_t now = millis();
         Serial.printf(D_STR_TIMESTAMP " : %06u.%03u\n", now / 1000, now % 1000);
         // Check if we got an IR message that was to big for our capture buffer.
         if (results.overflow)
            Serial.printf(D_WARN_BUFFERFULL "\n", kCaptureBufferSize);
         // Display the library version the message was captured with.
         Serial.println(D_STR_LIBRARY "   : v" _IRREMOTEESP8266_VERSION_ "\n");
         // Display the basic output of what we found.
         Serial.print(resultToHumanReadableBasic(&results));
         // Display any extra A/C info if we have it.
         String description = IRAcUtils::resultAcToString(&results);
         if (description.length()) Serial.println(D_STR_MESGDESC ": " + description);
         yield();  // Feed the WDT as the text output can take a while to print.
      #if LEGACY_TIMING_INFO
         // Output legacy RAW timing info of the result.
         Serial.println(resultToTimingInfo(&results));
         yield();  // Feed the WDT (again)
      #endif  // LEGACY_TIMING_INFO
         // Output the results as source code
         Serial.println(resultToSourceCode(&results));
         Serial.println();    // Blank line between entries
         yield();             // Feed the WDT (again)
      }
      vTaskDelay(1);
	}
}
```

code compile and upload perfectly but nothing happened when i press my ac remote button to the receiver and give watch dog reset

serial monitor log is below
```
[0;31mE (10346) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:[0m
[0;31mE (10346) task_wdt:  - IDLE0 (CPU 0)[0m
[0;31mE (10346) task_wdt: Tasks currently running:[0m
[0;31mE (10346) task_wdt: CPU 0: main[0m
[0;31mE (10346) task_wdt: CPU 1: IDLE1[0m
[0;31mE (15346) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:[0m
[0;31mE (15346) task_wdt:  - IDLE0 (CPU 0)[0m
[0;31mE (15346) task_wdt: Tasks currently running:[0m
[0;31mE (15346) task_wdt: CPU 0: main[0m
[0;31mE (15346) task_wdt: CPU 1: IDLE1[0m

```

crankyoldgit, 2020-11-11:
Have you tried the un-modified example code at all?! I'm fairly sure it's your modifications to it that are causing the problem.
As I said, you need a `delay()` or a `yield()` call regularly.

The problem is not with the library or example code, it's with the rest of your code.

Again, unless you're modifying the example code in a trivial way (i.e. Changing a GPIO etc), we are not going to support your modifications.
**Again**, see https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me

We are NOT here to teach you about the Arduino, or ESP framework. If you want to make modifications, you need to understand what you're doing first. If you have a genuine issue with the library's unmodified (except for obvious minor settings as suggested in the examples) example code, we will look into it, but I'm not going to support non-typical usage or your custom code changes.

Duplicate of #1328 1328

MarkEvens, 2020-11-11:
@crankyoldgit  sorry for my trouble but i am really shocked because in past i did successfully  received data by this code but in my new try i am failed, Thank you because of your comment i deep dig in my code and found problem in sdkconfig

closed 2020-11-11 by crankyoldgit