crankyoldgit/IRremoteESP8266 #1334: Support for Fujitsu AR-REF1E

rbswift, 2020-11-16:
_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revision of the library used

v2.7.11
### Expected behavior

Support for AR-REF1E

### Actual behavior
Some controls do not seem to be properly supported

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
Thing is I'm using Tasmota 9.1.0 so not sure if that is a showstopper or whether it's outputting useful raw data to the log:

```
06:09:27 MQT: tele/bedroomirblaster/RESULT = {"IrReceived":{"Protocol":"FUJITSU_AC","Bits":56,"Data":"0x0x146300101002FD","Repeat":0,"IRHVAC":{"Vendor":"FUJITSU_AC","Model":1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":16,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":"+3225-1680+445-395Cd+440-1210CdEfCdEdEdEfCfEdEdEdEfEfEdEdE-400EdEgEgEgEgEgEgEgEgEgEfEgEgEgEgEgEgEgE-1215EgEgEgEgEhEgEgEgEg+435gIgIhEgIhEhEhEhEhEhE","RawDataInfo":[115,115,0]}}
06:09:31 MQT: tele/bedroomirblaster/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":130,"Hash":"0xC682B7A1","Repeat":0,"RawData":"+3225-1680+445-395+440dE-1210EdEfEdEdEdEfEfCdEdE-400EfEfEdEdEdEgEgEgEgEgEgEgEgEgEgEfEgEgEgEgEgEgEgE-1215EgEgEgEg+435hEhEhEhEhEhEhEhEgEgIhEgIgI-405+430j+410kLkLkLkL-1240LmLkLkLm+415-425LkLkLmLkLkL-1245LmLkLkLkLkLkLkJkJiJkJpLkJkJiJiGi+360-475Q-1290J-1250JkJtJiQrQrQs+365rQrQrQrQrUrQrQrQsQrQrQsUrQrQrQrQrQrQrQrD-1255UrQrD-1260QrDeGeDvDvQsQsD","RawDataInfo":[259,259,0]}}
06:09:33 MQT: tele/bedroomirblaster/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":58,"Hash":"0xD376FA61","Repeat":0,"RawData":"+3250-1655+440-400CdC-1215CdCeCdCdCdCeCeCdCd+435dFeCeCdCdFdFdF-405Fg+410-430HiHiHiHiHiHiH-1240FgHiHiHiHiHiHiHjHiHiHiHiHiH-1245HjHiHjHjHiHkHjHiHiHkHiHiHkH","RawDataInfo":[115,115,0]}}
06:09:34 MQT: tele/bedroomirblaster/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":58,"Hash":"0xD376FA61","Repeat":0,"RawData":"+3250-1655+440-400C-405+435-1215CdC-1210CdCdCdCgCgCdCdFdFgCgCdCdFdFdFe+410-430IjIjIjIjIjIjIjI-1240IjIjIjIjIjIjEjIkIjIjIjIjIjI-1245IkIjIkIkIjIkIkIjIjIlIjIjIlI","RawDataInfo":[115,115,0]}}
06:09:35 MQT: tele/bedroomirblaster/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":58,"Hash":"0xD376FA61","Repeat":0,"RawData":"+3250-1655+440-400CdC-1210CdCeCdCdC-405+435-1215ChCdCdCdChChCdCdCdGdGfGf+410-430IjIjIjIjIjIjI-1240IjIjIjIjIjIjIjI-1245IjIjIjIjIjIlIlIjIkIkIjIlIkIjIjFlIjFjFlI","RawDataInfo":[115,115,0]}}
06:09:37 MQT: tele/bedroomirblaster/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":130,"Hash":"0xD742D415","Repeat":0,"RawData":"+3250-1655+440-400CdC-1210CdCeCdCdCdCeCeCdCdCdCeCeCdCdCdCdCdCd+435-405FdFdFdFdFgFdF-1215CdFdFdFgFgFg+410-430I-1240FgFgIjIjIkFhFhFhFhFhFhFhFdIjIkIjIjIjIjIjIjIjIjI-1245IkIjIjIjGjGjGjGlIjGjGlIlIjGjGfGf+360-475Mn+365nMnMnM-1290MnMpMnMnMnMnMpMpMnMpMnMnMnMpOnOnMnMnMnMnMnMn+395-1255OnMnMpOnQ-445QcQcQcQcQcQcDrDcQcQcQrQcQcQcQrQrDrQ","RawDataInfo":[259,259,0]}}
```

### What brand/model IR demodulator are you using?
Mirabella Genio Wi-Fi Smart IR Universal Remote Controller running Tasmota
https://www.kmart.com.au/product/mirabella-genio-wi-fi-smart-ir-universal-remote-controller/2622812

### Steps to reproduce the behavior
For example, adjusting the swing or set (vertical position) does not seem to be recognised.

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes but unsure how to follow when using Tasmota or if that's even possible.

### Has this library/code previously worked as expected for you?
No

### Other useful information
I am very happy to attempt to reverse engineer the protocol myself and could flash a non-Tasmota firmware to the device if needed if that's the only way so very grateful for a pointer in the best direction and I'll do some work on getting this remote supported. I think it's quite a common Fujitsu remote in Australia. Cheers

NiKiZe, 2020-11-16:
For the lines where you have unknown we need real raw data, we might be able to use the output you have given, but DumpV2+ would give us the data without deciphering, and also with less probability of timing issues.

The first line you have that is detected as Fujitsu, is it anything in there that was incorrect?

rbswift, 2020-11-16:
Thanks for your reply.

The first command was just me turning off the AC. So my guess is that is a simpler command and common with the existing supported protocol. The subsequent commands adjusted swing/set positions and so were more customised and presumably that is why they were not interpreted.

Is DumpV2+ a firmware that I can upload OTA to replace a tasmota firmware? As long as I can flash back OTA to Tasmota then I can give something new a go and get the most useful raw data. Curious if there is a guide to this?

crankyoldgit, 2020-11-16:
We typically follow this when adding a new protocol: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol

IRdumpV3 can have OTA enabled.
You'll need access to the serial pins for the output.

If you're unsure, I suggest using a ESP dev board with a quality IR demodulator. I.e. no risk of bricking.

rbswift, 2020-11-16:
Thanks. I certainly don't have access to the serial pins on my device unfortunately. I'll see if I can get a dev board rigged up

crankyoldgit, 2020-11-16:
Have you tried sending with all of the supported model types yet?
https://github.com/crankyoldgit/IRremoteESP8266/blob/a56c88b741dbe79ff85a488e7341dbb305120946/src/IRsend.h#L119-L126

crankyoldgit, 2020-11-16:
Another OTA Friendly Serial-less option is [IRMQTTServer](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRMQTTServer) but you will have to enable raw output:
https://github.com/crankyoldgit/IRremoteESP8266/blob/a56c88b741dbe79ff85a488e7341dbb305120946/examples/IRMQTTServer/IRMQTTServer.h#L192-L194

rbswift, 2020-11-16:
Thanks, yes, I found `ARREB1E` is the best option. It seems to support about 75% of the functionality but it's hard to know with some commands because there is sometimes no easy way to verify the AC actually did what I asked ... so I thought it would be best to get all the commands received correctly first then I'd know the sending was dependable.

crankyoldgit, 2020-11-16:
Not all functions/settings are available via the interface Tasmota uses. It's limited to:
https://github.com/crankyoldgit/IRremoteESP8266/blob/a56c88b741dbe79ff85a488e7341dbb305120946/src/IRac.cpp#L842-L856

Not all options for each of the supported settings are available either.
e.g. some swing stuff is either "swing" or "no swing", some protocols support positions. Some have a "step" the vanes one position etc. This latter one isn't available via the `IRac`/Tasmota interface

rbswift, 2020-11-16:
That's interesting thanks. I will have a look and try to harmonise what is available with Tasmota with what is available in HomeKit (I'm using HomeBridge and mqttthing so I can control the AC from my iPhone/Siri) and check if maybe it's just a not supported/overlapping feature. That list looks good enough for my needs but things like fan speed didn't seem to work well. Perhaps it's a mapping 0-100% values wrongly. A few things to check. Cheers

crankyoldgit, 2020-11-16:
Fan speed is "converted" here:
https://github.com/crankyoldgit/IRremoteESP8266/blob/a56c88b741dbe79ff85a488e7341dbb305120946/src/ir_Fujitsu.cpp#L691-L703
&
https://github.com/crankyoldgit/IRremoteESP8266/blob/a56c88b741dbe79ff85a488e7341dbb305120946/src/ir_Fujitsu.cpp#L718-L729

In short there are 5 supported Fujitsu fan speeds:
https://github.com/crankyoldgit/IRremoteESP8266/blob/a56c88b741dbe79ff85a488e7341dbb305120946/src/ir_Fujitsu.h#L60-L64

rbswift, 2020-11-25:
Still working on it. The documentation page here seems to have been corrupted. https://tasmota.github.io/docs/Tasmota-IR/

I'm not super technical but am making some progress today I think... Will update again soon. thanks

rbswift, 2020-11-25:
Does anyone know what the difference between a 2 beep response and a 3 beep response is from a Fujitsu AC?

I'm having some problem with `"Econo":"Off"` which seems to toggle (rather than discrete "off") the economy mode with each successive message and produce a 3 beep response from the AC.

I will keep working on it and update&close this if I get a conclusion but if this issue ticket is no longer thought appropriate then please feel free to close while I keep working. Thanks

crankyoldgit, 2020-11-26:
Yes, it's entirely possible that the code can produce three (3) successive messages using the interface you're using.
See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/d02bc1c5fa79ab229808c1bdba933fece00b4aab/src/IRac.cpp#L857-L907
e.g. A message with `Turbo` & `Econo` set will produce three messages, and thus three beeps for some Models. If only one of those is set, you'll probably get two beeps.
i.e. `Turbo` & `Econo` Fujitsu are special commands that must be sent independently. Yes, they are probably toggles.

If you download Branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1334 / PR #1345 it _should_ handle those toggles for you better.

rbswift, 2020-11-30:
Thanks for all your help with this. I seem to have it working pretty well now and will finesse some more parameters when I get a bit more time. Cheers

closed 2020-11-30 by rbswift

crankyoldgit, 2021-01-04:
FYI, the changes mentioned above have now been included in the new [v2.7.14 release](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.7.14) of the library.

rbswift, 2021-01-05:
awesome, thank you!