crankyoldgit/IRremoteESP8266 #1335: I can't find the YB1FA remote sharp type

syahrir1, 2020-11-18:
I can't find the YB1FA remote sharp type, can you help me?

crankyoldgit, 2020-11-18:
Try following these steps: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#how-do-i-work-out-what-protocol-my-remote-uses 

What you've provided isn't enough for us to go on, other than to say _"Try the `SHARP` or `SHARP_AC` protocol" because you say it's a **Sharp** remote."_.

crankyoldgit, 2020-11-21:
Have you tried any/all of those steps? What was the outcome?

crankyoldgit, 2020-11-24:
Friendly ping

syahrir1, 2020-11-25:
hello mr. david I've found an alternative to my Sharp YB1FA remote using
the "TurnOnKelvinatorAC" type and yes it works completely. thank you before

>

crankyoldgit, 2020-11-25:
That's great. Can you please tell us the Model number of the A/C unit **and** the remote? So we can add it to https://github.com/crankyoldgit/IRremoteESP8266/blob/master/SupportedProtocols.md

syahrir1, 2020-11-26:
Yes, of course. I use SHARP AC-A5VEY with YB1FA Remote

crankyoldgit, 2020-11-26:
Thanks for that.

PR #1346 (2020-11-26): Kelvinator: Update supported models.
Closes #1335
commit 022a54e5 (2020-11-26): Kelvinator: Update supported models. (#1346)

CLoses #1335
closed 2020-11-26 by crankyoldgit