crankyoldgit/IRremoteESP8266 #1338: Request example for Midea generic minisplit

scudrunner, 2020-11-23:
I just downloaded your library this weekend, and I was able to properly decode my Pioneer (Midea) remote, but I am having difficulty understanding the class structure, and how to send the midea commands, and what portions to #include.  When I look at some of the other examples, I can not easily see what needs to change.  

for example, in your Mitsubishi example, you have ac.setFan;
ac.setMode(kMitsubishiACCool);

My AC does not work properly - it gets way warmer then the settings, so I want to program a wemos D1 mini to read the temp from a DHT11 sensor, and then send out the IR signal turn the system on or off to better control the temperature. 

I will settle for having set on / off commands I capture from the remote, but I would like to be able to fully use the capability of your code, and change the settings from the wemos as my project grows. 

Thank you

NiKiZe, 2020-11-23:
So with the example you have been working with, what is not working?

Take any example, https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/TurnOnToshibaAC/TurnOnToshibaAC.ino
Remove the print and change IRToshibaAC to IRMideaAC 
the include from toshiba to ir_Midea.h

You can read the source, or the library documentation as mentioned in the readme, see https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRMideaAC.html

scudrunner, 2020-11-23:
I am working on it some more today, and figuring out the documentation, and learning what I need to change, and what needs to be included...  I will work with this example and see if I can get it compiled correctly.  Thanks for the quick response.

crankyoldgit, 2020-11-23:
You can see someone else's example code here: https://github.com/crankyoldgit/IRremoteESP8266/issues/1318#issuecomment-723118427

It should give you an idea of how to use the `IRMideaAC` class.

scudrunner, 2020-11-23:
I have it compiling right now, and I am watching the com port for output.  
I am still having trouble finding modes - I think I have the following working:
  ac.setMode(kMideaACHeat);
  ac.setFan(kMideaACFanAuto);
  ac.setTemp(70);  // 70F
  ac.setPower(false); // on 

but the Vtoggle does not seem to be setting 
  ac.setSwingVToggle(true); // not working

I also can not get the printstate to work properly - it does not want to print the raw.  I got an error on the ac.getRaw()

I am sure it is a simple command to print the raw, but I not sure what it is.  I did the following 
 uint64_t ir_code = ac.getRaw();
  Serial.print("IR Code: 0x");
  //for (uint64_t i = 0; i < 48; i++)
  //  Serial.printf("%02X", ir_code[i]);
  Serial.printf("%02X", ir_code);
  Serial.println();

  and I print out part of the raw - but it looks like it is missing the header and some other data.  
 

I will get my IR LED set up in a but to check the IR portion, but right now I am going to work on some of the rest of the code, and assume this works.  

I can submit my midea example once it is working.

crankyoldgit, 2020-11-23:
> I can submit my midea example once it is working.

Please don't. We have far to many code examples for A/Cs as it is. We deliberately don't want one for every type of AC, as we'd have to maintain them, and check & compile them with **every** change to the library. This would be wasted time and resources. The examples at present are just a _guide_ to how to do it.

> I am still having trouble finding modes

I _strongly_ suggest you skim-over/read https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Midea.h

> I also can not get the printstate to work properly - it does not want to print the raw. I got an error on the ac.getRaw()
etc
etc
etc

Please see the FAQ on this topic: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me

You're really asking programming questions that are beyond your current programming skill level. We don't have the resources to help you skill-up. You need to invest in that yourself.

scudrunner, 2020-11-23:
Understood, and thanks for all you put into this!

closed 2020-11-23 by scudrunner

fraschizzato, 2020-11-24:
> ac.setSwingVToggle(true); // not working
Some AC uses different message to do this. Use dump example to get the correct command from original remote and then use that via `ac.sendMidea()` that needs `IRsend ac(kIrLed);` instead of `IRMideaAC ac(kIrLed);` . You can find almost every Midea and Coolix integration use in https://github.com/crankyoldgit/IRremoteESP8266/issues/1318 . Tip: if you need to use both `IRsend` and `IRMideaAC` use something different from `ac`. I.e: if `IRsend ac(kIrLed);` for swing, then use `IRMideaAC acmidea(kIrLed);` for the other commands.