crankyoldgit/IRremoteESP8266 #1339: Crash when calling IRac::sendAc() with SAMSUNG_AC and prev=NULL

s-hadinger, 2020-11-24:
### Version/revision of the library used
2.7.12

### Expected behavior
Fix crash sending SAMSUNG_AC

### Actual behavior
When calling `IRac::sendAc(const stdAc::state_t desired, const stdAc::state_t *prev)` with vendor `SAMSUNG_AC`, there is a crash when `prev = nullptr`.

The problem happens here: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRac.cpp#L2535-L2537
calling `prev->power` without checking first that `prev` is not null.

One easy fix would be to duplicate the following: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRac.cpp#L2554-L2555

I let you decide whether you want to add these two lines to SAMSUNG_AC or if you want to add it globally upfront in the code.

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
NA

### What brand/model IR demodulator are you using?
SAMSUNG_AC

### Steps to reproduce the behavior
The crash is systematic with Tasmota

#### Example code used

#### Circuit diagram and hardware used (if applicable)

### I have followed the steps in the [Troubleshooting Guide]
No not needed

### Has this library/code previously worked as expected for you?
Yes

### Other useful information

crankyoldgit, 2020-11-24:
Acknowledged. I'll add something to make it _safe_ soon.

crankyoldgit, 2020-11-24:
@s-hadinger Should be fixed when PR #1341 is merged.
I'll look at creating a new release when it is.
In the meantime, can you please confirm this fixes you're issue?
My tests indicate it does, but extra confirmation from the reporter is always best. ;-)

X-ref: https://github.com/arendst/Tasmota/pull/9969

closed 2020-11-25 by crankyoldgit

s-hadinger, 2020-11-25:
Thanks. I did a quick fix in Tasmota and will wait for the next release of your lib. No urgency.

crankyoldgit, 2020-11-25:
> Thanks. I did a quick fix in Tasmota and will wait for the next release of your lib. No urgency.

I saw, but I want to make sure my fix fixes your problem so when it replaces your fix I know everything is still fine. ... before I publish the new release (probably in less than 24h)

PR #1341 (2020-11-25): [Bug] Fix crash when IRac::sendAc(state_t, *state_t) called with SAMSUNG_AC & NULL
* Ensure protocols that use the `prev` pointer's power state are okay when it's NULL.
* Add unit test to ensure this doesn't happen again.

Fixes #1339
commit 71f42c63 (2020-11-25): [Bug] Fix crash when IRac::sendAc(state_t, *state_t) called with SAMSUNG_AC & NULL (#1341)

* Ensure protocols that use the`prev` pointer's power state are okay when it's NULL.

Fixes #1339