crankyoldgit/IRremoteESP8266 #1347: Supported protocol shows as unknown

Fettkeewl, 2020-11-29:
### Version/revision of the library used
IRremoteESP8266 v2.7.12 in tasmota-ir.bin (9.1.0)

### Expected behavior
I expected this library to decode my AC remote control
because it is already listed as a supported protocol for my model.
[CS-E9CKP series A/C (PANASONIC_AC32)](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/SupportedProtocols.md)

[image]

### Actual behavior
Sending IR Commands the supported protocol is not decoded.
Registers as unknown.
I've even cranked up protocol decryption sensitivity in tasmota to max
using 

```
SetOption38 | 6..255 = set IRReceive protocol detection sensitivity minimizing UNKNOWN protocols
```

```
07:58:26 MQT: tele/TasmotaIR/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":136,"Hash":"0xBD0233BF","Repeat":0}}
07:58:27 MQT: tele/TasmotaIR/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":136,"Hash":"0xBD0233BF","Repeat":0}}
07:58:29 MQT: tele/TasmotaIR/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":136,"Hash":"0x48AAC23B","Repeat":0}}
07:58:31 MQT: tele/TasmotaIR/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":136,"Hash":"0xDBCF37CF","Repeat":0}}
```

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
I used the raw capture mode in tasmota.
The order follows as

1. Power On/Off
2. Temp up
3. Temp down

```
08:07:58 MQT: tele/TasmotaIR/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":136,"Hash":"0xBD0233BF","Repeat":0,"RawData":"+3495-3445+915-830Cd+910dE-2575CfCdCfEdCdEdEdC-2570CfCdEfEdCdEfCfEdEdCdEdCdCdEfCfEdEdEdEdEd+3530bCdCdCdEfCfEdCfEdCdEdEdCfCfCdEfCdEdCfCfCdCdCdEdE-835EdEfCfCdEdCdEdCdHbC-13920HbCdEdCdEdEdE-2610+880-870+875kL-865NdEmNoNmNkLmLkLoLfE-2580EiEkLfCoNiEiEfEfEiEfEfEiEdH-3450EoLdEiEdEoNpEoLfEoLoLdEiEiE-2600+890mNrS-855SfEpEiEkNfCoLiEiEfEfEtSfCfEdEt+3505qC","RawDataInfo":[271,271,0]}}
08:08:04 MQT: tele/TasmotaIR/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":136,"Hash":"0x48AAC23B","Repeat":0,"RawData":"+3535-3445+915-2575+910-835E-870+875dEdE-855+890-2580EfEdC-830EfEkE-2610+880fEdClE-860+885kEdE-2600J-865NiPgHgHlCdCmNqJgHlEfEi+3505bCdElEfEdEdCrH-2585+905rHdE-840UfEdEdElEdEgHfEdEkEdEfEvUfEgHiJkEqJdEgHfEgHiS-3450E-13920A-3480NrNrHiPgHfEmNiJdEfEgNrHgHrNmNrNdEiPkEdEgHkEdClEfEgHmNdEfEqJdEiJr+3495wCfElEiJrHgHdCgHmHfElCrNrNiJkEfEdEgHdEdEgHdEmNfEfElEdCdElEqJdCgHgZwE","RawDataInfo":[271,271,0]}}
08:08:07 MQT: tele/TasmotaIR/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":136,"Hash":"0xDBCF37CF","Repeat":0,"RawData":"+3530-3450+915-835+910-830E-870+875-2575E-2600+890fEiE-855KdEdE-865+880-2610NjKgHoNdEgH-2580EoNpEdEdEgHfE-860KiEiEiEq+885dEmNg+3495-3445CdEdEdEoNiCfEpEmHdEgNmNiEiEqRoNmNdEiEpEiEgNfEdEmNmHjKiEiElKlRmNmSbE-13950+3500bEqKfEgHdEfEpEmHpElKgHdEdEmHiEfEiEdEiEiEmNoNoNgHmNdEiCjKdEiEiEdEl+3505bEfCmHmHgHdEpEdEiCqRlRdEfEdEoNmNoNdEpEiEdEiEiEmHdEdEoNiEdEiEpEmNm+3490bE","RawDataInfo":[271,271,0]}}
```

### What brand/model IR demodulator are you using?
Looks like TSOP4838 

### Steps to reproduce the behavior
Try the library with a similar remote...
Wether it be standalone or in tasmota
[image]

#### Example code used
Tasmota 9.0.1 which includes your library

### Has this library/code previously worked as expected for you?
No, first try

NiKiZe, 2020-11-29:
We can't really use the data you have provided.
It is not in a format we use, you might be able to convert it to a normal raw array.
Also the timings when other code is running might affect it
As described, please provide _Output of raw data from IRrecvDumpV2+_

Does your receiver work with any other remote?

Fettkeewl, 2020-11-29:
I imagined id get a response similar to that :P
So I just finished flashing a Wemos D1 Mini with IRrecvDumpV2

Following is 
1. On/Off
2. Temp up
3. Temp down

```
`Timestamp : 000044.926
Library   : v2.7.13

Protocol  : UNKNOWN
Code      : 0xC5999209 (118 Bits)
uint16_t rawData[236] = {4692, 12,  2388, 6900,  12, 1888,  2490, 14,  12, 12,  12, 12,  12, 682,  3798, 12,  1698, 8668,  12, 1786,  1614, 12,  1634, 3990,  12, 1428,  1362, 12,  150, 5118,  12, 12,  58, 2058,  12, 1700,  1372, 12,  92, 10676,  12, 12,  1566, 2894,  12, 12,  560, 13384,  12, 2410,  6862, 1942,  2606, 12,  12, 12,  12, 590,  3746, 12,  1772, 1420,  12, 12,  7146, 12,  1866, 1566,  12, 12,  1648, 3904,  12, 1548,  14, 1234,  12, 252,  4994, 12,  12, 166,  1956, 12,  1842, 1224,  12, 12,  206, 12,  8606, 12,  68, 1858,  12, 1682,  2766, 12,  12, 704,  1618, 12,  28, 1622,  12, 142,  9794, 12,  2556, 2918,  18, 12,  12, 12,  11878, 4766,  14, 12,  2026, 5126,  12, 68,  6574, 12,  420, 3334,  12, 2072,  8412, 12,  86, 2128,  12, 12,  1350, 3856,  12, 1632,  1218, 12,  268, 2564,  12, 12,  12, 12,  828, 2482,  12, 12,  1160, 1380,  12, 222,  1716, 12,  1802, 2888,  12, 12,  532, 3374,  12, 54,  5200, 2234,  12, 1536,  1292, 12,  224, 2212,  12, 12,  1228, 2652,  12, 12,  916, 1544,  12, 70,  7536, 12,  1348, 1432,  12, 140,  8960, 12,  1736, 16510,  12, 12,  746, 1616,  12, 1870,  12, 1722,  4282, 12,  12, 962,  1548, 12,  1996, 4680,  12, 12,  348, 1878,  1836, 7368,  12, 1298,  1490, 12,  1912, 4424,  12, 12,  752, 1668,  12, 1922,  8180, 12,  2266, 3168,  12, 12,  12, 12,  12, 12 };  // UNKNOWN C5999209

Timestamp : 000061.407
Library   : v2.7.13

Protocol  : UNKNOWN
Code      : 0x115FE612 (77 Bits)
uint16_t rawData[154] = {5820, 12,  1154, 2748,  12, 648,  6172, 12,  864, 3086,  12, 12,  12, 12,  270, 4650,  12, 628,  11152, 12,  1076, 7752,  12, 974,  9250, 12,  1214, 2732,  12, 12,  12, 608,  14756, 1050,  2242, 12,  12, 1158,  13624, 12,  12, 12,  260, 2318,  12, 1268,  6720, 12,  12, 118,  2282, 12,  1304, 9252,  12, 1184,  6442, 12,  476, 2176,  12, 1340,  12070, 16,  12, 12,  3240, 12,  262, 2246,  12, 1324,  11246, 12,  12, 856,  5850, 12,  1206, 12218,  12, 134,  18, 12,  12, 110,  164, 12,  12, 1144,  3140, 12,  11804, 18658,  12, 410,  4268, 12,  12, 934,  16318, 12,  1202, 7670,  12, 1034,  8096, 12,  554, 11014,  12, 1260,  7782, 12,  888, 9194,  12, 1298,  10936, 12,  1232, 16634,  12, 786,  9522, 12,  954, 2146,  12, 1276,  4702, 12,  12, 506,  2090, 1456,  7572, 12,  1150, 2556,  12, 12,  794, 4462,  12, 840,  2088, 12,  1372, 8736,  12, 1706,  3472, 12 };  // UNKNOWN 115FE612

Timestamp : 000067.730
Library   : v2.7.13

Protocol  : UNKNOWN
Code      : 0xBD1EB4F2 (71 Bits)
uint16_t rawData[142] = {5324, 12,  1698, 7546,  12, 1188,  7676, 12,  1040, 9344,  12, 1114,  2578, 12,  12, 12,  764, 4472,  12, 830,  9380, 12,  1100, 2128,  12, 1282,  14880, 12,  842, 2024,  12, 1456,  13804, 12,  12, 30,  11570, 722,  4286, 12,  12, 12,  858, 10054,  14, 12,  12, 12,  12, 326,  3142, 12,  364, 4034,  12, 1222,  5976, 12,  1054, 2104,  12, 1296,  3390, 12,  12, 12,  11508, 12,  796, 1984,  12, 1490,  15986, 12,  1516, 2028,  12, 12672,  5188, 12,  1878, 11274,  12, 12,  786, 4410,  12, 902,  10854, 12,  1372, 9016,  12, 1430,  7442, 12,  1286, 2294,  12, 1060,  7448, 12,  1320, 7362,  12, 1402,  13698, 12,  12, 12,  12, 66,  11988, 12,  238, 3948,  12, 1332,  16274, 12,  1224, 7738,  12, 936,  4458, 12,  12, 648,  3108, 12,  392, 11062,  12, 1254,  7850, 12,  796, 9282,  12, 1220,  1852, 12 };  // UNKNOWN BD1EB4F2
```

crankyoldgit, 2020-11-29:
See https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#im-getting-some-random-odd-values--50-usecs-in-my-capture-of-an-ir-message-what-is-up-with-that

Panasonic uses a freq of 36.7kHz.

You've likely got a 38kHz decoder.

Fettkeewl, 2020-11-29:
> See https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#im-getting-some-random-odd-values--50-usecs-in-my-capture-of-an-ir-message-what-is-up-with-that
> 
> Panasonic uses a freq of 36.7kHz.
> 
> You've likely got a 38kHz decoder.

Thank you for pointing the issue, now I understand the Why.
Is there a "how" to fix this? Is there an software workaround?

Fettkeewl, 2020-11-29:
Might have solved it myself, I was testing with a module that had a trimpot on it. Figuring it might control the modules frequency i tried diling it down and got some proper readings from my remote !

Huzzah! :) I hope this is sendable though.

```
Timestamp : 000136.656
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0x65836A0 (32 Bits)
uint16_t rawData[271] = {3624, 3340,  1018, 676,  1060, 688,  1056, 686,  1064, 2466,  1028, 2472,  1002, 692,  1054, 2484,  1018, 670,  1068, 680,  1056, 690,  1050, 696,  1052, 2488,  1010, 2478,  996, 694,  1048, 2496,  1006, 678,  1066, 678,  1058, 2492,  990, 2494,  1008, 678,  1064, 680,  1056, 692,  1048, 696,  1048, 696,  1056, 686,  1064, 2480,  994, 2498,  990, 694,  1058, 682,  1064, 682,  1056, 688,  1048, 698,  3670, 3364,  992, 692,  1046, 700,  1044, 698,  1052, 2490,  1002, 2490,  986, 700,  1046, 2496,  1004, 682,  1058, 686,  1048, 698,  1044, 700,  1050, 2492,  1004, 2486,  990, 698,  1044, 2500,  1000, 682,  1060, 684,  1052, 2502,  982, 2500,  1002, 682,  1058, 686,  1048, 698,  1042, 702,  1044, 700,  1054, 688,  1060, 2490,  986, 2512,  978, 696,  1054, 688,  1058, 686,  1050, 696,  1042, 704,  3668, 3384,  970, 13878,  3586, 3386,  958, 708,  1038, 704,  1048, 694,  1056, 686,  1052, 694,  1042, 2524,  964, 704,  1048, 2502,  982, 696,  1040, 706,  1036, 708,  1042, 700,  1050, 690,  1056, 2508,  964, 708,  1034, 2522,  980, 688,  1052, 2512,  964, 2524,  974, 694,  1056, 2502,  972, 2522,  968, 700,  1052, 690,  1056, 688,  1050, 2516,  964, 2522,  982, 688,  1056, 2508,  968, 2522,  972, 696,  1056, 686,  3660, 3400,  970, 692,  1058, 686,  1052, 692,  1044, 702,  1038, 706,  1044, 2502,  996, 686,  1050, 2512,  968, 704,  1046, 696,  1056, 686,  1058, 686,  1050, 696,  1040, 2518,  978, 692,  1060, 2486,  990, 698,  1042, 2506,  994, 2490,  998, 688,  1048, 2504,  988, 2494,  1006, 680,  1056, 688,  1046, 700,  1046, 2498,  1006, 2480,  998, 690,  1048, 2500,  998, 2484,  1006, 682,  1052, 694,  3672, 3364,  1000};  // PANASONIC_AC32 65836A0
uint64_t data = 0x65836A0;

Timestamp : 000143.285
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0x65836A0 (32 Bits)
uint16_t rawData[271] = {3670, 3308,  1066, 650,  1092, 654,  1082, 662,  1078, 2436,  1062, 2424,  1064, 658,  1080, 2436,  1054, 662,  1088, 654,  1090, 656,  1082, 662,  1074, 2442,  1052, 2432,  1058, 660,  1078, 2442,  1044, 670,  1078, 664,  1084, 2432,  1046, 2444,  1042, 672,  1078, 664,  1082, 662,  1078, 668,  1070, 676,  1066, 676,  1072, 2444,  1048, 2442,  1036, 678,  1066, 678,  1072, 672,  1078, 664,  1076, 668,  3682, 3332,  1042, 666,  1074, 670,  1066, 678,  1062, 2460,  1038, 2446,  1038, 676,  1064, 2460,  1030, 678,  1072, 670,  1072, 672,  1066, 680,  1060, 2470,  1026, 2456,  1036, 674,  1062, 2470,  1016, 684,  1066, 678,  1070, 2462,  1016, 2476,  1014, 684,  1066, 676,  1068, 676,  1062, 684,  1054, 690,  1054, 690,  1058, 2474,  1020, 2472,  1006, 694,  1050, 692,  1056, 686,  1064, 678,  1062, 684,  3668, 3362,  1010, 13834,  3606, 3360,  1002, 694,  1042, 704,  1040, 704,  1044, 698,  1052, 688,  1058, 2480,  996, 706,  1038, 2492,  1008, 684,  1058, 692,  1046, 700,  1038, 706,  1040, 704,  1046, 2482,  1008, 692,  1044, 2494,  996, 698,  1050, 2480,  1006, 2486,  996, 704,  1044, 2482,  1012, 2478,  998, 704,  1040, 702,  1048, 692,  1058, 2476,  1004, 2488,  998, 700,  1050, 2478,  1010, 2482,  996, 704,  1044, 698,  3668, 3368,  984, 704,  1046, 694,  1056, 682,  1060, 684,  1052, 698,  1042, 2490,  1004, 688,  1060, 2476,  1004, 698,  1042, 704,  1044, 698,  1052, 686,  1060, 682,  1058, 2482,  998, 702,  1044, 2484,  1012, 682,  1056, 2486,  998, 2488,  1010, 682,  1060, 2480,  1000, 2490,  1006, 688,  1060, 682,  1058, 690,  1048, 2490,  1000, 2484,  1012, 682,  1058, 2482,  998, 2488,  1012, 682,  1062, 682,  3668, 3364,  1008};  // PANASONIC_AC32 65836A0
uint64_t data = 0x65836A0;

Timestamp : 000145.734
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0x65836A0 (32 Bits)
uint16_t rawData[271] = {3660, 3318,  1048, 660,  1090, 654,  1088, 656,  1080, 2438,  1046, 2438,  1060, 654,  1086, 2434,  1046, 670,  1078, 666,  1086, 658,  1088, 654,  1084, 2436,  1044, 2442,  1060, 654,  1088, 2430,  1046, 670,  1074, 668,  1082, 2432,  1058, 2432,  1046, 670,  1076, 666,  1084, 660,  1088, 654,  1086, 660,  1076, 668,  1074, 2442,  1058, 2428,  1056, 662,  1076, 668,  1074, 668,  1082, 662,  1088, 654,  3694, 3316,  1050, 660,  1088, 654,  1086, 658,  1078, 2440,  1048, 2436,  1062, 654,  1084, 2434,  1046, 670,  1080, 662,  1088, 654,  1090, 654,  1084, 2434,  1048, 2438,  1062, 652,  1088, 2428,  1050, 668,  1078, 666,  1086, 2426,  1062, 2428,  1050, 668,  1080, 662,  1088, 654,  1092, 652,  1086, 660,  1078, 666,  1076, 2438,  1062, 2424,  1058, 660,  1078, 666,  1076, 666,  1082, 660,  1092, 652,  3696, 3314,  1054, 13782,  3664, 3310,  1062, 650,  1088, 656,  1080, 664,  1078, 666,  1082, 662,  1090, 2422,  1062, 658,  1080, 2434,  1058, 658,  1092, 650,  1092, 652,  1086, 660,  1080, 664,  1080, 2430,  1068, 648,  1092, 2422,  1056, 664,  1082, 2428,  1072, 2416,  1060, 662,  1080, 2430,  1070, 2416,  1066, 654,  1084, 662,  1082, 660,  1088, 2420,  1072, 2418,  1060, 662,  1082, 2426,  1074, 2414,  1064, 658,  1084, 662,  3712, 3290,  1058, 660,  1082, 660,  1088, 654,  1096, 646,  1098, 646,  1090, 2424,  1060, 658,  1090, 2418,  1074, 648,  1088, 656,  1084, 660,  1086, 658,  1092, 650,  1100, 2410,  1066, 658,  1084, 2426,  1072, 646,  1100, 2412,  1066, 2424,  1066, 652,  1098, 2412,  1072, 2420,  1064, 656,  1094, 650,  1100, 642,  1098, 2414,  1064, 2424,  1072, 646,  1100, 2412,  1068, 2422,  1068, 652,  1098, 644,  3708, 3296,  1066};  // PANASONIC_AC32 65836A0
uint64_t data = 0x65836A0;
```

crankyoldgit, 2020-11-29:
> Is there an software workaround?

For the record, not really. The best we can offer is the [`noise_floor`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRrecv.html#aeaa5c07a8b46f8fbb982f996cc1f9f4b) parameter.

crankyoldgit, 2020-11-29:
> I hope this is sendable though.

It should be, there are no real restrictions in generating (sending) different frequency modulations. It's the _hardware_ IR demodulator that is the problem when it comes to receiving them.

Fettkeewl, 2020-11-29:
Not having any luck with this library on my panasonic..
I'm lost at how to get this working, cant figure out if its my lacking of knowledge or a bug in the library.
I'm testing this library on the device in the image.

[image]

It's one of [theese on Aliexpress](https://www.aliexpress.com/item/4000550056776.html?spm=a2g0o.productlist.0.0.844929b4VTBduN&algo_pvid=8ad97671-7a58-4d8d-ba04-144b3fd9da0e&algo_expid=8ad97671-7a58-4d8d-ba04-144b3fd9da0e-18&btsid=2100bdf116066740085372434e321d&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)

It works very nicely when running the **IRrecvDumpV2** sketch. It pick up and translate the PANASONIC_AC protocol
after I turned the potentiometer counter-clockwise. (i.e. reducing something... i guess, frequency?)

Now trying out the **TurnOnPanasonicAC.ino** sketch.... nothing happens 

The snippet below just loops around, nothing happens at the AC, the white/neon bulb is lit faintly (observed through camera on phone)
and nothing resembling a pulse is seen.
I figured out as much to edit the code pre upload
and set the proper model according to the library
`ac.setModel(kPanasonicCkp);`

```
Sending IR command to A/C ...
Panasonic A/C remote is in the following state:
  Model: 5 (CKP), Power: On, Mode: 3 (Cool), Temp: 26C, Fan: 7 (Auto), Swing(V): 15 (Auto), Quiet: Off, Powerful: Off, Clock: 00:00, On Timer: Off, Off Timer: Off
IR Code: 0x0220E004000000060220E00400313480AF00000EE0100001000099
```

Any help is appreciated!

crankyoldgit, 2020-11-29:
The protocol you want is `PANASONIC_AC32`, the settings/example you're using is for `PANASONIC_AC`.

See #1307 for more info. The 32 bit protocol does not yet have detailed support. I.e. changing the settings programmatically.

Fettkeewl, 2020-11-30:
> The protocol you want is `PANASONIC_AC32`, the settings/example you're using is for `PANASONIC_AC`.
> 
> See #1307 for more info. The 32 bit protocol does not yet have detailed support. I.e. changing the settings programmatically.

Thank you, I've read up on the subject and am now more understanding of why it does not work. 
I think I know what must be done to produce an working example. 
I hope the two of you can make this a standard protocoll soon though, the faster I can get it into Tasmota the Happier I will be!
Bummer that I had to have one of those special ACs that wasn't supported. 

I'll be checking back from time to time to see where this progress is going. 
Thank you for your time and help! 👍

closed 2020-11-30 by Fettkeewl