crankyoldgit/IRremoteESP8266 #1350: ESP32 crashes  with Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)

Jueff, 2020-11-29:
### Version/revision of the library used
name=IRremoteESP8266
version=2.7.13
platform=ESP32

### Error behavior
In case IRReceive is initialized (interrupts enabled) the ESP32 is crashing in case a SPI flash access is active and an IRReceive interrupt occurs.
To reproduce this problem I took the IRrecvDemo sample and added some code, that access the SPI flash.
Start the program and see that it will crash after some seconds.

Suggestion: the function 
  static void USE_IRAM_ATTR gpio_intr()
is marked with the IRAM_ATTR, so the code will run as a copy in memory. But this function calls 
  timerWrite(timer, 0);  // Reset the timeout.
  timerAlarmEnable(timer);
which are both functions not marked with IRAM_ATTR.

I tried to remove this two line, then no exception occurs. But then the receive also doesn't work anymore.

### Console output
IRrecvDemo is now running and waiting for IR message on Pin 22
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x400f8fc4  PS      : 0x00060034  A0      : 0x80080f94  A1      : 0x3ffbe790  
A2      : 0x3ffc017c  A3      : 0x3ffb94f4  A4      : 0x80086959  A5      : 0x3ffb1b50  
A6      : 0x00000026  A7      : 0x3ffc16e4  A8      : 0x80080f19  A9      : 0x00000001  
A10     : 0x3ffbec20  A11     : 0x00000002  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000020  A15     : 0xffffffff  SAR     : 0x00000011  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40086f80  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400f8fc4

Backtrace: 0x400f8fc4:0x3ffbe790 0x40080f91:0x3ffbe7b0 0x40084751:0x3ffbe7d0 0x40086f7d:0x3ffb1b90 0x400f7b52:0x3ffb1c00 0x400e05c9:0x3ffb1c30 0x400e07ff:0x3ffb1c60 0x400e1d6b:0x3ffb1c90 0x400e38e6:0x3ffb1cd0 0x400e01b4:0x3ffb1d10 0x400deefc:0x3ffb1d40 0x400e7665:0x3ffb1da0 0x4000bcc5:0x3ffb1dc0 0x400e89c1:0x3ffb1de0 0x400d932f:0x3ffb1e00 0x400d967f:0x3ffb1e30 0x400d9959:0x3ffb1e70 0x400d906f:0x3ffb1ef0 0x400d12b5:0x3ffb1f20 0x400d139e:0x3ffb1f90 0x400da84d:0x3ffb1fb0 0x40088379:0x3ffb1fd0

### Stack trace
PC: 0x400f8fc4: timerWrite at C:\Users\AnonymousUser\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-timer.c line 106
EXCVADDR: 0x00000000

Decoding stack results
0x400f8fc4: timerWrite at C:\Users\AnonymousUser\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-timer.c line 106
0x40080f91: __onPinInterrupt at C:\Users\AnonymousUser\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-gpio.c line 220
0x40086f7d: spi_flash_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c line 565
0x400f7b52: esp_partition_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/partition.c line 242
0x400e05c9: spiffs_api_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs_api.c line 36
0x400e07ff: spiffs_phys_rd at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_cache.c line 161
0x400e1d6b: spiffs_obj_lu_find_entry_visitor at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c line 166
0x400e38e6: spiffs_object_find_object_index_header_by_name at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c line 1694
0x400e01b4: SPIFFS_stat at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c line 765
0x400deefc: vfs_spiffs_stat at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/esp_spiffs.c line 536
0x400e7665: esp_vfs_stat at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c line 492
0x400e89c1: stat at ../../../.././newlib/libc/syscalls/sysstat.c line 12
0x400d932f: VFSFileImpl::VFSFileImpl(VFSImpl*, char const*, char const*) at C:\Users\AnonymousUser\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS\src\vfs_api.cpp line 232
0x400d967f: std::__shared_ptr ::__shared_ptr  , VFSImpl*&, char const*, char const*&>(std::_Sp_make_shared_tag, std::allocator  const&, VFSImpl*&, char const*&&, char const*&) at c:\users\AnonymousUser\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\ext/new_allocator.h line 120
0x400d9959: VFSImpl::open(char const*, char const*) at c:\users\AnonymousUser\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr.h line 319
0x400d906f: fs::FS::open(char const*, char const*) at C:\Users\AnonymousUser\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS\src\FS.cpp line 191
0x400d12b5: ListFiles() at C:\Users\JRGEN~1\AppData\Local\Temp\arduino_modified_sketch_874785/IRrecvDemo.ino line 66
0x400d139e: loop() at C:\Users\JRGEN~1\AppData\Local\Temp\arduino_modified_sketch_874785/IRrecvDemo.ino line 61
0x400da84d: loopTask(void*) at C:\Users\AnonymousUser\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 19
0x40088379: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

#### Example code used
/*
 * IRremoteESP8266: IRrecvDemo - demonstrates receiving IR codes with IRrecv
 * This is very simple teaching code to show you how to use the library.
 * If you are trying to decode your Infra-Red remote(s) for later replay,
 * use the IRrecvDumpV2.ino (or later) example code instead of this.
 * An IR detector/demodulator must be connected to the input kRecvPin.
 * Copyright 2009 Ken Shirriff, http://arcfn.com
 * Example circuit diagram:
 *  https://github.com/crankyoldgit/IRremoteESP8266/wiki#ir-receiving
 * Changes:
 *   Version 0.2 June, 2017
 *     Changed GPIO pin to the same as other examples.
 *     Used our own method for printing a uint64_t.
 *     Changed the baud rate to 115200.
 *   Version 0.1 Sept, 2015
 *     Based on Ken Shirriff's IrsendDemo Version 0.1 July, 2009
 */

#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>

// An IR detector/demodulator is connected to GPIO pin 14(D5 on a NodeMCU
// board).
// Note: GPIO 16 won't work on the ESP8266 as it does not have interrupts.
const uint16_t kRecvPin = SCL;

IRrecv irrecv(kRecvPin);
decode_results results;

TaskHandle_t  TaskHnd ;
void SecondTask(void* parameter);

#include "SPIFFS.h"

void setup() {
  Serial.begin(115200);
  irrecv.enableIRIn();  // Start the receiver
  while (!Serial)  // Wait for the serial connection to be establised.
    delay(50);
  Serial.println();
  Serial.print("IRrecvDemo is now running and waiting for IR message on Pin ");
  Serial.println(kRecvPin);
  //xTaskCreatePinnedToCore(SecondTask,"Second",4096,NULL,1,&TaskHnd,0);
    if (!SPIFFS.begin(true)) {
    Serial.println("An Error has occurred while mounting SPIFFS");
    return;
  }

}

void loop() {
  if (irrecv.decode(&results)) {
    // print() & println() can't handle printing long longs. (uint64_t)
    serialPrintUint64(results.value, HEX);
    Serial.println("");
    irrecv.resume();  // Receive the next value
  }
  delay(100);
  ListFiles();
}

void ListFiles() {
    File root = SPIFFS.open("/");
    File file = root.openNextFile();
 
    while(file){
       Serial.print("FILE: ");
       Serial.println(file.name());
       file = root.openNextFile();
    }
}

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
yes, had a look there, didn't find this issue.

### Has this library/code previously worked as expected for you?
no

### Other useful information

crankyoldgit, 2020-11-29:
Thanks for the report. Your info is really helpful. This might be a tricky fix, as we really do need those functions in the interrupt, thus Iram.

crankyoldgit, 2020-11-29:
**Note to self:**
In theory, if I replace those functions with:

https://github.com/espressif/arduino-esp32/blob/6b0114366baf986c155e8173ab7c22bc0c5fcedc/cores/esp32/esp32-hal-timer.c#L107-L109

&

https://github.com/espressif/arduino-esp32/blob/6b0114366baf986c155e8173ab7c22bc0c5fcedc/cores/esp32/esp32-hal-timer.c#L177

It will probably work.

Jueff, 2020-11-29:
good idea, may work.
another issue which is a quick fix - the definition of 
    static ETSTimer timer;
defines the variable timer at global scope. I already had an overlap with another library, I would suggest to use a more specific name for the variable, e.g. irremote_timer.

crankyoldgit, 2020-11-30:
Good suggestion. I'll change it and see if I can namespace/scoping

crankyoldgit, 2020-11-30:
Hey @Jueff I think if got a workaround (I'm not sure I like calling it a fix) for your issue.

Can you please download and try out the code in branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1350 and let me know how it goes?

crankyoldgit, 2020-12-01:
ping.

Jueff, 2020-12-01:
pong :-) 
just starting to test

crankyoldgit, 2020-12-02:
How did your testing go?

Jueff, 2020-12-02:
Testing is complete, no more crashes.

crankyoldgit, 2020-12-02:
Huzzah!

PR #1351 (2020-12-02): [BUG] Workaround for ESP32 hw timer library calls not in IRAM.
* Manipulate the ESP32 hw timers directly rather than through the library interface.
  - Fixes a crash (only) on the ESP32 platform.
  - Required because the following lib calls are not in IRAM by default.
    * `timerWrite();`
    * `timerAlarmEnable()`
  - This a horrible kludge/hack/mess!

* Tested on a ESP8266 NodeMCU & a ESP32dev board. Working as expected.

Fixes #1350
commit 0d215e1b (2020-12-02): [BUG] Workaround for ESP32 hw timer library calls not in IRAM. (#1351)

* Manipulate the ESP32 hw timers directly rather than through the library interface.
  - Fixes a crash (only) on the ESP32 platform.
  - Required because the following lib calls are not in IRAM by default.
    * `timerWrite();`
    * `timerAlarmEnable()`
  - This a horrible kludge/hack/mess!

* Tested on a ESP8266 NodeMCU & a ESP32dev board. Working as expected.

Fixes #1350
closed 2020-12-02 by crankyoldgit