crankyoldgit/IRremoteESP8266 #1353: Example IRMQTTServer not compiling

bwint, 2020-12-03:
### Version/revision of the library used
v2.7.13

### Expected behavior
I tried to compile the **IRMQTTServer** example using PlattformIO.

### Actual behavior
Got the following **error**:
```
src\IRMQTTServer.ino: In function 'String htmlSelectAcStateProtocol(String, decode_type_t, bool)':
src\IRMQTTServer.ino:689:41: error: 'htmlOptionItem' was not declared in this scope
                                 i == def);
                                         ^
*** [.pio\build\d1_mini\src\src\IRMQTTServer.ino.o] Error 1
```

### Proposal for solution
It looks like the  `htmlOptionItem` is declared beyond the `htmlSelectAcStateProtocol` function.  
Therefore obviously the `htmlSelectAcStateProtocol` function can't use the `htmlOptionItem` function.

As soon as I moved `htmlOptionItem` function before the `htmlSelectAcStateProtocol` function everything worked like a charm.

crankyoldgit, 2020-12-03:
That's interesting. It compiles fine in our tests.
Can you please tell me what build environment (OS version, IDE(atom/visual code studio/etc) & version, and what version of Platform IO, etc, & `platformio.ini`) you're using to get that error?

Your suggested fix, is probably fine, but I want to find out why we were not catching it earlier.

crankyoldgit, 2020-12-03:
I've not been able to reproduce it with PlatformIO under Atom or Visual Code Studio running on Linux.
I'm guessing from the paths in your report, that it's on Windows of some version.

Needless to say, I agree with your PR/fix (thanks for that!) and improved it a little bit.

closed 2020-12-04 by crankyoldgit

bwint, 2020-12-04:
**My build environment:**
Visual Studio Code Version: 1.51.1 (user setup)
Commit: e5a624b788d92b8d34d1392e4c4d9789406efe8f
Date: 2020-11-10T23:35:47.374Z
Electron: 9.3.3
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT ia32 10.0.18363

**PlattformIO Version:** 
Core 5.0.3·Home 3.3.1

**plattformio.ini**
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:d1_mini_lite]
platform = espressif8266
board = d1_mini_lite
framework = arduino
lib_deps = 
	tzapu/WiFiManager@^0.15.0
	knolleary/PubSubClient@^2.8
	bblanchon/ArduinoJson@^6.17.2
	crankyoldgit/IRremoteESP8266@^2.7.13

[platformio]
src_dir = .

[env]
lib_extra_dirs = ../../
lib_ldf_mode = deep+
lib_ignore = examples
build_flags = 
framework = arduino
platform = espressif8266
monitor_speed = 115200

[common]
lib_deps_builtin = 
lib_deps_external = 
	PubSubClient@>=2.8.0
	ArduinoJson@>=6.0
    crankyoldgit/IRremoteESP8266@^2.7.13

[common_esp8266]
lib_deps_external = 
	${common.lib_deps_builtin}
	${common.lib_deps_external}
	WifiManager@>=0.14

[common_esp32]
lib_deps_external = 
	${common.lib_deps_builtin}
	${common.lib_deps_external}
	https://github.com/tzapu/WiFiManager.git#development

[env:nodemcuv2]
board = nodemcuv2
lib_deps = 
	${common_esp8266.lib_deps_external}
	crankyoldgit/IRremoteESP8266@^2.7.13

[env:d1_mini]
board = d1_mini
lib_deps = 
	${common_esp8266.lib_deps_external}
	crankyoldgit/IRremoteESP8266@^2.7.13

[env:d1_mini_no_mqtt]
board = d1_mini
build_flags = 
	${env.build_flags}
	-DMQTT_ENABLE=false
lib_deps = 
	${common_esp8266.lib_deps_external}
	crankyoldgit/IRremoteESP8266@^2.7.13

[env:esp32dev]
platform = espressif32
board = esp32dev
lib_deps = 
	${common_esp32.lib_deps_external}
	crankyoldgit/IRremoteESP8266@^2.7.13

[env:esp01_1m]
board = esp01_1m
build_flags = 
	${env.build_flags}
	-Wl,-Teagle.flash.1m64.ld
lib_deps = 
	${common_esp8266.lib_deps_external}
	crankyoldgit/IRremoteESP8266@^2.7.13
```

I tried to built it using the `[env:d1_mini]` environment.

**My build setup:**
I didn't clone the whole repo and tried to build the code in the example folder.
I generated a new PlattformIO project and used the interface to add all the necessary libraries to my project.
Than I downloaded the example code including the .h and .ini file and tried compiling the code.

Thanks for improving and accepting my PR!
Hope that information can help.
Happy to supply any other info you need.

crankyoldgit, 2020-12-04:
> **My build setup:**
> I didn't clone the whole repo and tried to build the code in the example folder.
> I generated a new PlattformIO project and used the interface to add all the necessary libraries [...]

Thanks for that. Yeah, I'd have expected that to work flawlessly too.

I can't see anything odd with your setup, though I guess something has changed in the compiler department. Dunno.
Anyway, it should be fixed now in the `master` branch, and thus will be in the next release.

commit 918f7939 (2020-12-04): [Bug] IRMQTTServer: Compiler error under PlatformIO on Windows. (#1354)

* Moved `htmlOptionItem` before `htmlSelectAcStateProtocol`
* Add a `htmlOptionItem` proto to `IRMQTTServer.h`

Fixes #1353