crankyoldgit/IRremoteESP8266 #1356: No such file or directory   #include <algorithm>

Math07170, 2020-12-05:
Using version 2.7.13 downloaded from arduino genuino library manager

I have this in console when trying to compile
C:\Users\mathi\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:17:10: fatal error: algorithm: No such file or directory

 #include <algorithm>

          ^~~~~~~~~~~

compilation terminated.

exit status 1

I've just used the example IRreceiver on the github and just expect it to compile normaly

So I've follow troubleshoting guide and I've never use this library before.

NiKiZe, 2020-12-05:
Which board are you trying to compile for?

Math07170, 2020-12-05:
Arduino MEGA2560

NiKiZe, 2020-12-05:
Not supported, read the top paragraph of the [readme](http://github.com/crankyoldgit/IRremoteESP8266/blob/master/README.md)

Math07170, 2020-12-05:
Oh yes sorry before that I just dont know what is  ESP8266, ESP32 thx

closed 2020-12-05 by Math07170

ra1n08, 2022-06-28:
i using Arduino Uno use this lib but i got error same he
C:\Users\Admin\Documents\Arduino\libraries\IRremoteESP8266-2.8.2\src\IRrecv.cpp:17:10: fatal error: algorithm: No such file or directory
 #include <algorithm>
          ^~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Uno.

NiKiZe, 2022-06-28:
As already stated

> Not supported, read the top paragraph of the [readme](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/README.md)