crankyoldgit/IRremoteESP8266 #1357: [Question] BaseOTA,h does not exist. Is it supposed to work for all ESP8266?

HubKing, 2020-12-06:
### Version/revision of the library used
2.7.13

### Describe the bug
I tried to run the sample code but BaseOTA.h did not exist, and naturally, things like "OTAwifi();" did not compile. So, I manually deleted them and ran the example. Is this because my board does not support OTA? If so, why are the code not surrounded by preprocessing directive to make it easy to exclude that part like

```
// Allow over air update
// #define OTA_ENABLE true
#if OTA_ENABLE
#include "BaseOTA.h"
#endif
...
#if OTA_ENABLE
  OTAwifi();  // start default wifi (previously saved on the ESP) for OTA
#endif
```

#### To Reproduce
Not a bug.

#### Example code used
IRrecvDumpV3

#### Expected behaviour
Not a bug

#### Output of raw data from [IRrecvDumpV2.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) or [V3](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/) (if applicable)
N/A

### What brand/model IR demodulator are you using?
HX1838

### Circuit diagram and hardware used (if applicable)
NodeMCU V3

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
No

### Has this library/code previously worked as expected for you?
Mostly Yes.

### Other useful information

crankyoldgit, 2020-12-06:
It's in the same directory as the example code.

https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/BaseOTA.h

How did you create your project from the example?

NiKiZe, 2020-12-06:
You should have BaseOTA.h in the same folder, depending on if you enabled OTA or not it defines eighter empty functions, or does includes and defines functions that does the work.

All ESPs support OTA, however not all boards have enough flash space.

HubKing, 2020-12-10:
I copied the code from the example page of an IDE, so the page did not show the file. I looked into the library directory, and found the file.

closed 2020-12-10 by HubKing