crankyoldgit/IRremoteESP8266 #1361: FYI: works with Amana A/C

MSGhero, 2020-12-21:
My [Amana AC unit](https://www.lowes.com/pd/Amana-7400-BTU-325-sq-ft-230-Volt-Through-the-Wall-Air-Conditioner-with-Heater/1001003446) seems to work with the GREE protocol. The V3 dump detected the signal from my remote without any issue, and I've played around with the `TurnOnGreeAC` demo a bit with success.

Feel free to close, this is just an FYI in case these kinds of things are tracked somewhere.

MSGhero, 2020-12-21:
Also, thanks to the contributors 😍. I thought this would be a hunker-down holiday project, but the longest part was compiling in the end.

NiKiZe, 2020-12-21:
Nice to hear that, do yo have an model number for the unit + remote?

MSGhero, 2020-12-21:
The unit is probably PBC093G00CC (not labeled, didn't purchase it). The remote is YX1FF, also Amana branded.

PR #1363 (2020-12-22): Gree: List Amana as supported.
Fixes #1361
commit 4405ee54 (2020-12-22): Gree: List Amana as supported. (#1363)

Fixes #1361
closed 2020-12-22 by crankyoldgit