crankyoldgit/IRremoteESP8266 #1364: Supporting Detailed support for Panasonic AC/Heat pump indoor unit CS-E9CKP

crankyoldgit, 2020-12-23:
> @Fettkeewl That code might work. I can't tell, as I don't have access to a CKP AC device.
> Why don't you try that software and see if it works.

Aright, having tried it out I can confirm that
1. The library covering Panasonic CKP heatpumps with remote  A75C2295 works.
2. There is a weird workaround for some reason using timers to control on/off state of the heatpump.
Why I do not know, there is a dedicated on/off button on the remote.
3. library uses 38 kHz frequency for sending and not what you wrote (far above, panasonic using 36,7 kHz)
> BTW, Panasonic uses a frequency of 36.7kHz, most IR demodulators are expecting 38kHz.

4. Cant set freq to 36.7 kHz due to lib using int as argument, tried with 37. Worked all the same.
5. I used the example code and only modified row 22 to be (IRSenderESP8266) 
6. My original remote sends signals at 4-5 meters distance without issue, my IR blaster can only send from 10 cm @ 3.3v and rougly 30-40 cm distance @ 5v ( maybe something with the frequency here, beyond me)

Now what @crankyoldgit :) How do we get this in to IRremoteESP8266?
Oh and btw  under step 5 above, I tried ToniA's implementation of your library aswell using (IRSenderIRremoteESP8266 instead)
and it worked aswell.

```
#include <Arduino.h>

#include <PanasonicCKPHeatpumpIR.h>
#include <Timer.h> // https://github.com/JChristensen/Timer

/*
    This schema demonstrates how to control the Panasonic CKP power state change. The CKP does not have discrete
    'ON' and 'OFF' commands, but only a state switch. So, if the initial power state is now known, switching the
    state does not help much.
    
    Luckily this can be implemented by using the timer:
    * The 'send' command will send all the settings AND program the timer so that the pump will turn ON in a minute
    * The 'sendPanasonicCKPCancelTimer' will cancel the timer
    
    The 'turn OFF' must be implemented the same way.
    
    Of course you can choose to not turn off the timer, but that means that the heatpump will attempt to switch ON
    (or OFF) every day at the same time, as the timer will still be active.
*/

IRSenderESP8266Alt irSender(4);     // IR led on Duemilanove digital pin 3, using Arduino PWM
//IRSenderBlaster irSender(3); // IR led on Duemilanove digital pin 3, using IR Blaster (generates the 38 kHz carrier)

PanasonicCKPHeatpumpIR *heatpumpIR;

Timer timer;

void setup()
{
  Serial.begin(115200);
  delay(500);

  heatpumpIR = new PanasonicCKPHeatpumpIR();

  Serial.println("Turning the Panasonic CKP heatpump ON by using the timer");
  heatpumpIR->send(irSender, POWER_ON, MODE_HEAT, FAN_2, 24, VDIR_UP, HDIR_AUTO);
  Serial.println("The heatpump should have beeped, and the TIMER led should be ON");

  timer.after(60000, panasonicIsOn); // Called after 1 minute
  timer.after(120000, panasonicCancelTimer); // Called after 2 minutes

}

void loop()
{
  timer.update();
}

void panasonicIsOn()
{
  Serial.println("The heatpump should should turn ON by now, the TIMER led is still ON");
}

void panasonicCancelTimer()
{
  heatpumpIR->sendPanasonicCKPCancelTimer(irSender);
  Serial.println("The TIMER led should now be OFF");
}
```

_Originally posted by @Fettkeewl in https://github.com/crankyoldgit/IRremoteESP8266/issues/1307#issuecomment-749623628_

crankyoldgit, 2020-12-23:
FYI @taxx too

Fettkeewl, 2020-12-23:
I'll get samples off what the remote sends when on/off button is pressed later today :)

crankyoldgit, 2020-12-23:
I've created a branch (https://github.com/crankyoldgit/IRremoteESP8266/tree/CKP_detailed) which has _some_ support for this A/C based on TonyA's code and https://docs.google.com/spreadsheets/d/1V7Z4edgjPXNH9vbu0QUJQzha7eIfboO-I2pNhITKtEQ/edit#gid=242843025

It is NOT in a good state for sending/creating messages yet. e.g. There is no Power control. But it should decode some of the basic parts of the messages.

Please try it out and let me know how it goes.

taxx, 2020-12-23:
Hi and sorry for the long silence.
I haven't had time to do any more analysis and I haven't had access to the remote full time.

I've recorded a lot of different data in the work in progress spreadsheet, using the code change @crankyoldgit did to support decoding the AC32 protocol. The [power on/off is here](https://docs.google.com/spreadsheets/d/1V7Z4edgjPXNH9vbu0QUJQzha7eIfboO-I2pNhITKtEQ/edit#gid=242843025&range=A72:L75). I didn't notice any difference in the power on vs the power off command.

I also tested sending the raw `unit32_t` data (both the 32 bit and the 16 bit (like `Quiet`) with the indoor unit and it worked.

Anything I can do to help? (I don't have easy access to the indoor unit at the moment, but I have the remote)

Fettkeewl, 2020-12-23:
@crankyoldgit you are a champ!
I'll try more as my wife allows me :P 

For now
The fanspeed is off by count of 1.
Is: Auto > 2  > 3 > 4 > 5 > 6
Should it not be: Auto > 1 > 2 > 3 > 4 > 5

A cycling below
```
Timestamp : 000158.217
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0xCF936A0 (32 Bits)
Mesg Desc.: Temp: 24C, Mode: 4 (Heat), Fan: 15 (Auto), Swing(H): Off, Swing(V): 2 (High)
uint16_t rawData[271] = {3404, 3570,  790, 2694,  794, 926,  818, 926,  818, 2694,  794, 2694,  794, 2694,  794, 2694,  794, 2696,  792, 2694,  794, 926,  818, 926,  816, 2694,  794, 2694,  794, 2696,  792, 2696,  792, 2696,  794, 926,  816, 926,  816, 2696,  794, 2694,  794, 926,  816, 928,  816, 926,  816, 928,  816, 926,  816, 928,  816, 2696,  792, 2696,  792, 926,  818, 926,  816, 928,  816, 926,  3434, 3572,  788, 2696,  792, 926,  816, 928,  816, 2696,  792, 2696,  792, 2696,  792, 2696,  792, 2696,  792, 2696,  792, 928,  816, 926,  816, 2696,  792, 2696,  792, 2696,  792, 2696,  794, 2696,  792, 926,  816, 928,  816, 2696,  794, 2694,  792, 928,  816, 926,  816, 926,  816, 928,  816, 928,  816, 928,  816, 2698,  790, 2696,  792, 928,  816, 928,  816, 928,  816, 928,  3432, 3572,  788, 14052,  3404, 3572,  790, 926,  816, 928,  816, 928,  816, 928,  816, 926,  816, 2696,  792, 926,  816, 2696,  792, 928,  816, 926,  818, 926,  816, 926,  816, 928,  816, 2696,  792, 926,  818, 2696,  792, 926,  816, 2696,  792, 2696,  792, 926,  816, 2696,  792, 2696,  792, 926,  818, 926,  816, 928,  816, 2696,  792, 2696,  792, 928,  816, 2696,  792, 2696,  792, 926,  816, 928,  3432, 3572,  788, 926,  818, 926,  816, 928,  816, 926,  816, 928,  816, 2696,  792, 926,  818, 2694,  794, 926,  816, 926,  818, 926,  818, 926,  816, 926,  818, 2694,  794, 926,  818, 2694,  794, 926,  818, 2696,  792, 2696,  794, 926,  818, 2694,  794, 2694,  794, 926,  816, 926,  818, 926,  818, 2694,  794, 2694,  794, 926,  818, 2694,  794, 2694,  794, 926,  818, 926,  3434, 3570,  792};  // PANASONIC_AC32 CF936A0
uint64_t data = 0xCF936A0;

Timestamp : 000164.714
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0xC2936A0 (32 Bits)
Mesg Desc.: Temp: 24C, Mode: 4 (Heat), Fan: 2 (Quiet), Swing(H): Off, Swing(V): 2 (High)
uint16_t rawData[271] = {3414, 3560,  800, 2686,  802, 918,  826, 920,  824, 2686,  802, 920,  824, 2686,  802, 920,  824, 920,  824, 2686,  802, 920,  824, 920,  824, 2686,  802, 920,  824, 2686,  802, 920,  824, 920,  822, 920,  824, 920,  822, 2688,  800, 2688,  802, 920,  822, 920,  822, 922,  824, 920,  824, 920,  824, 920,  824, 2686,  802, 2686,  802, 920,  822, 920,  824, 920,  822, 922,  3438, 3564,  798, 2688,  800, 920,  824, 920,  822, 2688,  800, 922,  822, 2688,  800, 920,  824, 922,  822, 2688,  800, 922,  822, 920,  822, 2688,  800, 922,  822, 2688,  800, 922,  822, 922,  822, 922,  822, 922,  822, 2688,  800, 2688,  800, 920,  822, 922,  822, 922,  822, 922,  822, 922,  822, 922,  822, 2690,  800, 2688,  800, 922,  822, 922,  822, 922,  822, 922,  3438, 3566,  796, 14040,  3410, 3566,  796, 922,  822, 924,  820, 924,  820, 922,  822, 922,  820, 2690,  798, 924,  820, 2690,  798, 922,  820, 924,  820, 922,  822, 922,  820, 922,  820, 2692,  798, 922,  820, 2692,  796, 924,  820, 2690,  798, 2690,  798, 922,  820, 2692,  796, 2690,  798, 924,  820, 924,  820, 924,  820, 2692,  796, 2692,  796, 924,  820, 2692,  796, 2692,  796, 924,  820, 924,  3436, 3566,  794, 924,  820, 924,  820, 924,  820, 924,  818, 924,  820, 2692,  796, 924,  820, 2692,  796, 924,  818, 928,  816, 924,  818, 924,  820, 924,  818, 2694,  794, 926,  818, 2694,  794, 926,  818, 2692,  796, 2692,  796, 924,  818, 2694,  794, 2694,  796, 926,  818, 926,  818, 924,  818, 2694,  794, 2694,  794, 926,  820, 2692,  794, 2694,  796, 924,  818, 924,  3436, 3570,  790};  // PANASONIC_AC32 C2936A0
uint64_t data = 0xC2936A0;

Timestamp : 000168.823
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0xC3936A0 (32 Bits)
Mesg Desc.: Temp: 24C, Mode: 4 (Heat), Fan: 3 (Low), Swing(H): Off, Swing(V): 2 (High)
uint16_t rawData[271] = {3400, 3576,  786, 2698,  790, 930,  814, 930,  814, 2700,  788, 2698,  790, 2700,  788, 930,  814, 930,  814, 2700,  788, 930,  814, 930,  814, 2700,  788, 2700,  788, 2700,  788, 930,  814, 930,  814, 930,  814, 930,  812, 2700,  788, 2700,  788, 930,  814, 930,  812, 932,  812, 930,  812, 932,  812, 930,  814, 2700,  788, 2700,  788, 930,  814, 930,  814, 930,  812, 932,  3428, 3578,  782, 2702,  788, 930,  812, 930,  812, 2702,  786, 2702,  786, 2700,  788, 930,  812, 932,  812, 2702,  786, 932,  812, 932,  812, 2702,  786, 2702,  786, 2702,  786, 932,  812, 932,  812, 932,  812, 932,  812, 2702,  786, 2702,  788, 930,  812, 932,  812, 932,  812, 932,  812, 932,  812, 932,  812, 2702,  786, 2702,  786, 932,  812, 932,  812, 932,  812, 932,  3428, 3580,  782, 14054,  3396, 3580,  782, 932,  812, 932,  812, 932,  812, 932,  812, 932,  812, 2702,  786, 932,  812, 2702,  786, 932,  812, 932,  812, 932,  812, 932,  810, 934,  810, 2704,  784, 932,  812, 2702,  786, 932,  810, 2704,  784, 2704,  784, 934,  810, 2704,  784, 2704,  784, 932,  810, 934,  810, 934,  810, 2704,  784, 2704,  784, 932,  810, 2706,  782, 2704,  784, 932,  810, 934,  3426, 3582,  778, 934,  810, 934,  810, 934,  808, 934,  810, 934,  810, 2706,  782, 934,  810, 2706,  782, 934,  810, 934,  808, 934,  810, 934,  808, 936,  808, 2706,  782, 938,  806, 2708,  782, 934,  808, 2706,  782, 2706,  782, 936,  808, 2706,  782, 2706,  782 [... cut ...] 828, 2682,  808, 2680,  808, 914,  828, 916,  3444, 3558,  804, 916,  828, 916,  828, 916,  828, 916,  828, 916,  828, 2682,  806, 916,  828, 2682,  806, 916,  828, 916,  828, 916,  828, 916,  828, 916,  828, 2682,  806, 916,  828, 2682,  806, 916,  828, 2682,  808, 2680,  808, 914,  828, 2682,  808, 2680,  808, 914,  828, 916,  828, 914,  830, 2680,  808, 2680,  806, 916,  828, 2682,  806, 2680,  808, 916,  828, 914,  3446, 3556,  804};  // PANASONIC_AC32 CF936A0
uint64_t data = 0xCF936A0;

```

Fettkeewl, 2020-12-23:
Temp registers as it should

```
Timestamp : 000439.589
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0xCF136A0 (32 Bits)
Mesg Desc.: Temp: 16C, Mode: 4 (Heat), Fan: 15 (Auto), Swing(H): Off, Swing(V): 2 (High)
uint16_t rawData[271] = {3522, 3456,  904, 2584,  904, 838,  906, 838,  904, 838,  906, 2582,  906, 2582,  906, 2584,  904, 2584,  904, 2584,  904, 838,  906, 838,  904, 840,  904, 2582,  906, 2584,  904, 2584,  906, 2582,  906, 838,  906, 838,  906, 2582,  906, 2582,  906, 838,  904, 838,  906, 838,  906, 838,  906, 838,  906, 838,  906, 2582,  906, 2582,  906, 838,  906, 838,  906, 838,  906, 838,  3522, 3454,  906, 2582,  906, 838,  906, 838,  906, 838,  906, 2582,  906, 2582,  906, 2582,  906, 2582,  906, 2582,  906, 838,  906, 836,  906, 838,  906, 2582,  906, 2582,  906, 2582,  906, 2580,  906, 838,  906, 838,  906, 2582,  906, 2582,  906, 838,  906, 838,  906, 836,  906, 838,  906, 838,  906, 836,  906, 2582,  906, 2582,  906, 838,  906, 836,  906, 838,  906, 836,  3524, 3454,  906, 13926,  3524, 3454,  908, 836,  906, 838,  906, 836,  906, 836,  908, 836,  906, 2582,  908, 836,  906, 2582,  906, 836,  908, 836,  908, 836,  906, 836,  906, 838,  906, 2582,  906, 836,  906, 2582,  908, 836,  906, 2582,  906, 2582,  908, 836,  906, 2582,  906, 2582,  906, 838,  906, 836,  906, 838,  906, 2582,  906, 2582,  906, 838,  906, 2582,  906, 2582,  906, 838,  906, 838,  3524, 3454,  906, 838,  906, 838,  906, 836,  906, 838,  906, 838,  906, 2582,  906, 838,  906, 2582,  906, 838,  906, 838,  906, 838,  906, 836,  906, 838,  904, 2582,  906, 838,  906, 2582,  906, 838,  906, 2584,  906, 2582,  906, 838,  904, 2582,  906, 2584,  906, 838,  906, 838,  904, 838,  906, 2582,  906, 2584,  904, 838,  904, 2584,  906, 2584,  904, 838,  904, 840,  3520, 3456,  904};  // PANASONIC_AC32 CF136A0
uint64_t data = 0xCF136A0;

Timestamp : 000442.313
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0xCF236A0 (32 Bits)
Mesg Desc.: Temp: 17C, Mode: 4 (Heat), Fan: 15 (Auto), Swing(H): Off, Swing(V): 2 (High)
uint16_t rawData[271] = {3520, 3456,  904, 838,  906, 2582,  904, 840,  904, 840,  904, 2584,  904, 2584,  904, 2584,  904, 2584,  904, 840,  904, 2584,  904, 840,  904, 840,  904, 2584,  904, 2584,  904, 2584,  904, 2584,  904, 840,  904, 840,  904, 2584,  904, 2584,  904, 840,  904, 840,  904, 840,  904, 840,  904, 840,  904, 840,  904, 2584,  904, 2584,  904, 840,  904, 840,  904, 840,  904, 840,  3520, 3458,  904, 840,  904, 2584,  904, 840,  904, 840,  904, 2584,  904, 2584,  904, 2586,  902, 2584,  904, 840,  904, 2584,  904, 840,  902, 840,  904, 2584,  904, 2584,  904, 2584,  904, 2586,  902, 840,  904, 840,  904, 2584,  904, 2584,  904, 840,  904, 840,  904, 840,  904, 840,  904, 840,  902, 842,  902, 2584,  904, 2584,  904, 840,  904, 840,  902, 840,  904, 840,  3520, 3458,  904, 13928,  3520, 3458,  902, 840,  904, 840,  902, 840,  904, 840,  902, 840,  904, 2584,  904, 840,  904, 2586,  902, 840,  902, 840,  904, 840,  902, 842,  902, 840,  904, 2584,  904, 840,  902, 2586,  902, 840,  904, 2586,  902, 2586,  902, 840,  904, 2584,  904, 2586,  902, 840,  904, 840,  902, 840,  904, 2584,  904, 2586,  902, 842,  902, 2584,  904, 2584,  904, 840,  902, 840,  3520, 3458,  902, 842,  902, 840,  902, 840,  904, 840,  902, 842,  902, 2586,  902, 840,  904, 2584,  904, 840,  902, 842,  902, 840,  904, 840,  902, 840,  904, 2586,  902, 840,  904, 2584,  904, 840,  902, 2586,  902, 2586,  902, 840,  904, 2584,  904, 2586,  902, 840,  904, 840,  902, 840,  904, 2586,  902, 2586,  902, 840,  904, 2584,  904, 2584,  904, 840,  902, 842,  3520, 3458,  902};  // PANASONIC_AC32 CF236A0
uint64_t data = 0xCF236A0;

Timestamp : 000444.631
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0xCF336A0 (32 Bits)
Mesg Desc.: Temp: 18C, Mode: 4 (Heat), Fan: 15 (Auto), Swing(H): Off, Swing(V): 2 (High)
uint16_t rawData[271] = {3520, 3458,  904, 2584,  904, 2584,  904, 840,  902, 840,  904, 2584,  904, 2586,  902, 2584,  904, 2586,  902, 2586,  902, 2586,  904, 840,  904, 840,  904, 2584,  904, 2584,  904, 2584,  904, 2584,  904, 840,  904, 840,  904, 2586,  902, 2586,  902, 842,  902, 840,  902, 840,  904, 840,  904, 840,  902, 840,  904, 2584,  904, 2584,  904, 840,  904, 840,  904, 840,  902, 842,  3520, 3458,  902, 2586,  902, 2584,  904, 840,  902, 840,  904, 2584,  904, 2584,  904, 2586,  902, 2586,  902, 2586,  902, 2586,  902, 840,  904, 840,  902, 2586,  904, 2584,  904, 2586,  902, 2584,  904, 840,  902, 842,  902, 2586,  902, 2586,  902, 840,  904, 840,  902, 842,  902, 840,  904, 840,  902, 840,  904, 2584,  904, 2584,  904, 842,  902, 842,  902, 840,  902, 842,  3520, 3458,  902, 13926,  3520, 3458,  902, 842,  902, 842,  902, 842,  902, 840,  902, 842,  902, 2586,  902, 842,  902, 2586,  902, 842,  902, 842,  902, 840,  902, 842,  902, 840,  904, 2586,  902, 840,  902, 2586,  904, 840,  902, 2586,  902, 2586,  902, 842,  902, 2586,  902, 2586,  902, 840,  904, 840,  904, 840,  902, 2586,  902, 2586,  902, 840,  902, 2586,  902, 2586,  902, 840,  902, 842,  3518, 3458,  902, 842,  902, 842,  902, 842,  902, 840,  904, 840,  902, 2586,  902, 842,  902, 2586,  902, 840,  902, 842,  902, 840,  902, 842,  902, 842,  902, 2586,  902, 840,  902, 2586,  902, 842,  902, 2586,  902, 2586,  902, 842,  902, 2586,  902, 2586,  904, 840,  902, 840,  902, 842,  902, 2586,  902, 2586,  902, 842,  902, 2586,  902, 2586,  902, 840,  902, 842,  3520, 3458,  902};  // PANASONIC_AC32 CF336A0
[... cut ...] 836, 2666,  822, 2666,  822, 908,  836, 908,  3452, 3542,  820, 908,  834, 908,  836, 908,  836, 908,  836, 908,  836, 2668,  820, 908,  836, 2668,  820, 908,  836, 908,  836, 910,  834, 908,  834, 908,  836, 2666,  822, 908,  836, 2668,  820, 908,  836, 2666,  822, 2666,  822, 908,  836, 2668,  820, 2668,  822, 908,  836, 908,  836, 908,  836, 2666,  820, 2668,  822, 908,  836, 2666,  822, 2668,  820, 910,  834, 908,  3452, 3542,  820};  // PANASONIC_AC32 CFF36A0
uint64_t data = 0xCFF36A0;

```

Fettkeewl, 2020-12-23:
This is in the following sequece

1.  ON/OFF
2. ON/OFF
3. TEMP DOWN
4. ON/OFF
5. ON/OFF

```
Timestamp : 000557.659
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0x4FF36A0 (32 Bits)
Mesg Desc.: Temp: 30C, Mode: 4 (Heat), Fan: 15 (Auto), Swing(H): Off, Swing(V): 2 (High)
uint16_t rawData[271] = {3488, 3492,  868, 2620,  868, 2622,  866, 2624,  864, 2626,  862, 2626,  864, 2626,  862, 2628,  860, 2628,  860, 2628,  860, 2626,  862, 2626,  862, 2624,  864, 2624,  866, 2624,  864, 2622,  866, 2624,  866, 874,  868, 876,  868, 2622,  866, 876,  868, 876,  868, 876,  868, 876,  868, 876,  868, 876,  868, 876,  868, 2624,  866, 876,  866, 878,  866, 876,  866, 878,  866, 876,  3484, 3496,  864, 2624,  866, 2622,  864, 2622,  866, 2622,  866, 2622,  866, 2622,  866, 2622,  866, 2620,  868, 2620,  868, 2620,  868, 2620,  868, 2620,  868, 2620,  868, 2618,  870, 2618,  870, 2618,  870, 872,  872, 872,  870, 2618,  870, 872,  870, 872,  872, 870,  872, 872,  872, 872,  872, 872,  872, 872,  872, 2618,  872, 870,  872, 872,  872, 872,  872, 872,  872, 872,  3490, 3488,  872, 13964,  3490, 3486,  874, 868,  876, 868,  876, 868,  876, 868,  876, 868,  876, 2612,  878, 866,  878, 2610,  878, 866,  878, 866,  878, 866,  878, 866,  878, 864,  880, 2608,  878, 866,  878, 2610,  878, 868,  876, 2608,  880, 2608,  880, 864,  880, 2608,  880, 2608,  880, 864,  880, 864,  880, 864,  880, 2608,  880, 2608,  880, 864,  880, 2608,  880, 2608,  880, 864,  880, 862,  3498, 3480,  882, 862,  880, 864,  880, 862,  880, 864,  880, 864,  880, 2608,  880, 864,  880, 2608,  880, 864,  880, 864,  880, 864,  880, 864,  880, 864,  880, 2608,  880, 864,  880, 2608,  880, 864,  880, 2608,  880, 2608,  880, 864,  880, 2608,  880, 2608,  880, 864,  880, 864,  880, 864,  878, 2610,  878, 2608,  880, 864,  880, 2608,  880, 2608,  880, 864,  880, 864,  3496, 3480,  880};  // PANASONIC_AC32 4FF36A0
uint64_t data = 0x4FF36A0;

Timestamp : 000562.998
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0x4FF36A0 (32 Bits)
Mesg Desc.: Temp: 30C, Mode: 4 (Heat), Fan: 15 (Auto), Swing(H): Off, Swing(V): 2 (High)
uint16_t rawData[271] = {3498, 3480,  880, 2608,  880, 2608,  880, 2608,  880, 2610,  878, 2610,  878, 2610,  878, 2610,  878, 2610,  878, 2610,  878, 2612,  876, 2612,  876, 2612,  876, 2614,  876, 2612,  876, 2612,  874, 2614,  874, 870,  874, 870,  874, 2616,  872, 870,  874, 870,  874, 870,  874, 870,  874, 870,  874, 870,  872, 870,  874, 2616,  872, 870,  872, 872,  874, 870,  872, 872,  872, 872,  3488, 3490,  872, 2616,  872, 2616,  872, 2616,  872, 2616,  872, 2618,  870, 2618,  870, 2618,  870, 2618,  870, 2616,  872, 2616,  872, 2618,  872, 2616,  870, 2618,  870, 2618,  870, 2618,  870, 2618,  870, 872,  872, 872,  872, 2618,  870, 872,  872, 872,  872, 872,  872, 872,  872, 872,  872, 872,  872, 872,  872, 2616,  872, 870,  872, 870,  874, 870,  874, 870,  874, 870,  3490, 3488,  874, 13962,  3492, 3484,  876, 866,  876, 868,  878, 866,  876, 866,  878, 866,  878, 2610,  878, 866,  878, 2610,  878, 864,  878, 866,  878, 864,  878, 866,  878, 864,  880, 2610,  878, 864,  878, 2610,  880, 864,  890, 2598,  878, 2608,  880, 864,  880, 2608,  880, 2608,  880, 864,  880, 864,  880, 862,  880, 2608,  880, 2608,  880, 864,  880, 2608,  880, 2608,  880, 862,  880, 864,  3498, 3480,  882, 862,  880, 862,  882, 862,  880, 864,  880, 862,  880, 2608,  882, 862,  880, 2608,  880, 864,  880, 864,  880, 864,  880, 862,  880, 864,  880, 2608,  880, 864,  880, 2608,  880, 864,  880, 2608,  880, 2608,  880, 864,  880, 2608,  880, 2608,  880, 864,  880, 864,  880, 864,  880, 2608,  880, 2608,  880, 864,  880, 2608,  880, 2608,  880, 864,  880, 864,  3496, 3482,  880};  // PANASONIC_AC32 4FF36A0
uint64_t data = 0x4FF36A0;

Timestamp : 000583.219
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0xCFE36A0 (32 Bits)
Mesg Desc.: Temp: 29C, Mode: 4 (Heat), Fan: 15 (Auto), Swing(H): Off, Swing(V): 2 (High)
uint16_t rawData[271] = {3524, 3452,  908, 836,  908, 2580,  906, 2580,  908, 2582,  906, 2582,  906, 2582,  908, 2580,  908, 2580,  908, 836,  906, 2582,  906, 2582,  908, 2580,  908, 2580,  906, 2582,  908, 2580,  908, 2580,  908, 836,  906, 836,  908, 2582,  906, 2582,  906, 836,  906, 838,  906, 836,  906, 838,  906, 838,  906, 838,  906, 2582,  906, 2582,  906, 838,  906, 838,  906, 836,  906, 838,  3522, 3454,  906, 838,  906, 2582,  906, 2582,  906, 2584,  904, 2582,  906, 2584,  904, 2584,  904, 2584,  904, 838,  906, 2584,  904, 2584,  904, 2584,  906, 2582,  904, 2584,  906, 2582,  906, 2582,  904, 840,  904, 838,  904, 2586,  902, 2584,  904, 838,  904, 840,  904, 838,  904, 840,  904, 838,  904, 840,  904, 2584,  904, 2584,  904, 840,  904, 840,  904, 840,  904, 840,  3520, 3456,  904, 13936,  3520, 3458,  902, 840,  904, 840,  904, 840,  904, 840,  904, 840,  904, 2586,  902, 840,  904, 2586,  902, 840,  904, 840,  904, 840,  902, 840,  904, 840,  904, 2584,  904, 840,  904, 2586,  904, 840,  904, 2584,  904, 2586,  902, 840,  904, 2584,  904, 2584,  904, 838,  904, 840,  904, 840,  904, 2584,  904, 2584,  904, 840,  904, 2584,  904, 2584,  904, 840,  904, 840,  3520, 3458,  904, 840,  904, 838,  904, 840,  904, 840,  906, 840,  904, 2584,  904, 840,  904, 2584,  904, 838,  904, 840,  904, 838,  904, 840,  904, 840,  904, 2584,  904, 838,  904, 2584,  904, 840,  904, 2584,  904, 2584,  904, 838,  906, 2584,  904, 2584,  904, 838,  904, 840,  904, 838,  904, 2584,  904, 2584,  904, 838,  904, 2584,  904, [... cut ...]  856, 2644,  844, 2644,  844, 888,  854, 888,  3472, 3518,  842, 888,  856, 888,  856, 888,  856, 888,  856, 888,  856, 2644,  844, 888,  856, 2644,  844, 888,  856, 888,  856, 888,  856, 886,  856, 888,  856, 2644,  846, 886,  856, 2644,  844, 886,  856, 2644,  844, 2644,  846, 886,  858, 2642,  846, 2642,  846, 886,  856, 886,  858, 886,  858, 2642,  846, 2642,  846, 886,  858, 2642,  846, 2642,  848, 884,  858, 884,  3476, 3516,  846};  // PANASONIC_AC32 4FE36A0
uint64_t data = 0x4FE36A0;
```

Fettkeewl, 2020-12-23:
Vertical air swing

First is Auto, then different steps and back to Auto on the last

```
Timestamp : 000827.428
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0x3080 (16 Bits)
uint16_t rawData[201] = {3512, 3466,  896, 848,  896, 848,  896, 848,  896, 848,  896, 846,  896, 848,  896, 846,  896, 2592,  896, 846,  896, 848,  896, 846,  896, 848,  896, 846,  898, 846,  896, 848,  896, 2590,  898, 846,  898, 846,  898, 848,  896, 846,  898, 2592,  898, 2590,  898, 846,  898, 846,  898, 846,  898, 846,  898, 846,  898, 846,  896, 2590,  898, 2590,  898, 846,  898, 844,  3516, 3462,  900, 844,  898, 846,  898, 844,  898, 846,  898, 844,  900, 844,  900, 844,  898, 2588,  900, 846,  898, 844,  898, 844,  900, 844,  900, 844,  900, 844,  900, 844,  900, 2588,  900, 844,  900, 844,  900, 844,  900, 844,  900, 2588,  900, 2588,  900, 842,  902, 842,  900, 844,  900, 844,  900, 844,  900, 842,  900, 2588,  902, 2586,  902, 844,  900, 842,  3518, 3460,  900, 842,  902, 842,  902, 842,  902, 842,  902, 842,  900, 842,  902, 842,  902, 2586,  902, 842,  902, 842,  902, 842,  902, 842,  902, 842,  902, 840,  902, 842,  902, 2586,  902, 840,  902, 842,  902, 840,  902, 842,  902, 2586,  902, 2586,  902, 842,  902, 842,  902, 840,  902, 842,  902, 840,  902, 842,  902, 2586,  902, 2586,  902, 840,  904, 840,  3520, 3458,  904};  // PANASONIC_AC32 3080
uint64_t data = 0x3080;

Timestamp : 000830.657
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0x32A1 (16 Bits)
uint16_t rawData[201] = {3534, 3444,  918, 2572,  916, 826,  916, 828,  916, 828,  916, 828,  916, 2572,  916, 828,  916, 2572,  916, 2572,  916, 826,  916, 828,  916, 828,  916, 826,  916, 2572,  916, 828,  916, 2572,  918, 826,  916, 2572,  916, 826,  916, 828,  916, 2572,  916, 2572,  916, 828,  916, 828,  916, 828,  916, 2572,  916, 826,  916, 828,  916, 2572,  916, 2572,  916, 828,  916, 826,  3534, 3444,  916, 2572,  916, 826,  916, 828,  916, 828,  916, 828,  916, 2572,  916, 828,  916, 2572,  916, 2572,  916, 828,  916, 828,  916, 828,  916, 826,  916, 2572,  916, 826,  916, 2572,  916, 828,  916, 2572,  918, 826,  916, 828,  916, 2572,  916, 2572,  916, 828,  916, 828,  916, 828,  916, 2572,  918, 826,  916, 826,  916, 2572,  916, 2572,  916, 826,  916, 828,  3532, 3444,  916, 2572,  916, 828,  916, 828,  916, 828,  916, 828,  916, 2572,  916, 826,  916, 2572,  916, 2572,  916, 826,  916, 828,  916, 828,  916, 828,  916, 2572,  916, 828,  916, 2572,  916, 828,  916, 2572,  916, 828,  916, 828,  916, 2572,  916, 2574,  916, 828,  916, 828,  916, 828,  916, 2574,  914, 828,  916, 828,  916, 2572,  916, 2572,  916, 828,  916, 828,  3532, 3444,  916};  // PANASONIC_AC32 32A1
uint64_t data = 0x32A1;

Timestamp : 000832.817
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0x32A2 (16 Bits)
uint16_t rawData[201] = {3530, 3446,  914, 828,  916, 2574,  914, 830,  914, 830,  914, 828,  914, 2574,  914, 828,  914, 2574,  914, 830,  914, 2574,  914, 828,  916, 828,  916, 828,  916, 2574,  914, 830,  914, 2574,  914, 830,  914, 2574,  916, 828,  914, 830,  914, 2574,  914, 2574,  916, 828,  914, 830,  914, 828,  914, 2574,  914, 830,  914, 828,  914, 2574,  916, 2574,  914, 828,  914, 830,  3530, 3448,  914, 830,  914, 2574,  914, 828,  916, 830,  914, 830,  914, 2574,  914, 828,  916, 2572,  914, 830,  914, 2574,  914, 830,  914, 830,  914, 830,  914, 2574,  914, 830,  914, 2574,  914, 830,  914, 2574,  914, 830,  914, 830,  914, 2574,  914, 2574,  914, 828,  914, 830,  914, 830,  914, 2574,  914, 828,  914, 830,  914, 2574,  914, 2574,  914, 830,  914, 830,  3532, 3446,  914, 830,  916, 2572,  914, 830,  914, 830,  914, 830,  914, 2574,  914, 830,  914, 2574,  914, 830,  914, 2574,  914, 830,  914, 830,  914, 830,  914, 2574,  914, 828,  914, 2574,  914, 830,  914, 2574,  914, 830,  914, 830,  914, 2574,  914, 2574,  914, 830,  914, 830,  914, 830,  914, 2574,  914, 830,  914, 830,  914, 2574,  914, 2574,  914, 830,  914, 830,  3532, 3446,  914};  // PANASONIC_AC32 32A2
uint64_t data = 0x32A2;

Timestamp : 000835.235
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0x32A3 (16 Bits)
uint16_t rawData[201] = {3524, 3452,  908, 2580,  908, 2580,  908, 836,  908, 834,  908, 836,  908, 2580,  908, 834,  910, 2578,  910, 2578,  910, 2580,  908, 834,  908, 836,  908, 834,  908, 2580,  908, 834,  908, 2580,  908, 836,  910, 2578,  908, 834,  910, 834,  908, 2580,  908, 2580,  908, 836,  908, 834,  908, 836,  908, 2580,  908, 836,  908, 834,  908, 2580,  910, 2580,  908, 834,  908, 836,  3524, 3452,  908, 2580,  908, 2580,  908, 836,  908, 834,  910, 834,  908, 2580,  908, 836,  908, 2580,  910, 2578,  908, 2580,  910, 834,  908, 834,  908, 836,  908, 2580,  908, 834,  910, 2580,  908, 834,  908, 2580,  910, 834,  908, 834,  910, 2580,  908, 2580,  908, 834,  910, 834,  908, 834,  910, 2580,  910, 834,  908, 836,  908, 2580,  908, 2580,  908, 834,  910, 834,  3526, 3452,  910, 2578,  908, 2580,  908, 834,  910, 834,  908, 836,  908, 2580,  908, 834,  908, 2580,  910, 2580,  908, 2580,  908, 834,  908, 836,  908, 834,  908, 2580,  910, 834,  908, 2580,  908, 836,  908, 2580,  908, 834,  908, 834,  908, 2580,  908, 2580,  910, 834,  908, 836,  908, 834,  908, 2580,  910, 834,  908, 836,  908, 2580,  908, 2580,  908, 834,  908, 836,  3526, 3452,  908};  // PANASONIC_AC32 32A3
uint64_t data = 0x32A3;

Timestamp : 000837.221
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0x32A4 (16 Bits)
[... cut ...]   900, 844,  898, 2592,  896, 2592,  896, 844,  898, 844,  3516, 3464,  896, 844,  900, 844,  900, 844,  898, 844,  898, 844,  900, 844,  900, 844,  898, 2592,  896, 844,  900, 844,  900, 844,  898, 844,  900, 844,  898, 844,  900, 844,  900, 2592,  896, 844,  898, 846,  898, 844,  898, 844,  900, 2592,  896, 2592,  896, 844,  900, 844,  898, 846,  898, 844,  898, 846,  898, 844,  898, 2594,  894, 2592,  896, 844,  898, 846,  3516, 3466,  894};  // PANASONIC_AC32 3080
uint64_t data = 0x3080;
```

Fettkeewl, 2020-12-23:
Vertical swing
Start on Auto, cycles different modes , 2 directions are displayed, D1 and D2
Theese change from Left, center, right  (L, C , R)

1.  Auto
2. D1 C - D2 C
3. D1 L - D2 L
4. D1 L - D2 C
5. D1 C - D2 R
6. D1 R - D2 R
7. Auto

```
Timestamp : 000976.030
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0x32A8 (16 Bits)
uint16_t rawData[201] = {3502, 3474,  886, 850,  894, 850,  892, 850,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 850,  892, 850,  894, 2600,  888, 850,  894, 2602,  888, 850,  894, 2602,  886, 850,  894, 2602,  886, 850,  892, 850,  894, 2600,  888, 2600,  888, 850,  894, 848,  894, 850,  894, 2600,  888, 850,  894, 850,  894, 2602,  886, 2600,  888, 850,  894, 850,  3510, 3474,  886, 850,  894, 850,  894, 850,  892, 2602,  886, 850,  894, 2602,  886, 850,  894, 2600,  888, 850,  894, 850,  894, 850,  894, 2600,  888, 850,  894, 2602,  888, 850,  894, 2602,  886, 850,  894, 2600,  888, 850,  894, 850,  894, 2600,  888, 2600,  888, 850,  894, 850,  894, 850,  894, 2600,  888, 850,  894, 850,  894, 2602,  886, 2600,  888, 850,  894, 850,  3510, 3474,  886, 850,  894, 850,  894, 850,  894, 2600,  888, 850,  894, 2600,  888, 848,  894, 2600,  888, 850,  894, 850,  894, 850,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 2602,  888, 848,  894, 2600,  888, 850,  894, 850,  894, 2600,  888, 2600,  888, 848,  894, 850,  894, 850,  894, 2600,  888, 848,  894, 850,  894, 2600,  888, 2600,  888, 848,  894, 850,  3510, 3474,  886};  // PANASONIC_AC32 32A8
uint64_t data = 0x32A8;

Timestamp : 000978.787
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0x32A9 (16 Bits)
uint16_t rawData[201] = {3504, 3474,  886, 2600,  890, 848,  896, 848,  894, 2600,  888, 850,  898, 2596,  888, 848,  894, 2600,  888, 2600,  888, 848,  894, 850,  894, 2600,  888, 848,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 850,  894, 2600,  888, 2600,  888, 850,  894, 850,  894, 850,  894, 2600,  888, 850,  894, 850,  894, 2600,  888, 2600,  888, 850,  894, 850,  3510, 3474,  888, 2600,  888, 850,  894, 850,  894, 2600,  888, 850,  892, 2600,  888, 850,  894, 2600,  888, 2600,  888, 850,  896, 848,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 850,  894, 2600,  888, 2600,  888, 850,  892, 850,  894, 850,  894, 2600,  888, 850,  894, 850,  894, 2600,  888, 2600,  888, 848,  894, 850,  3510, 3474,  888, 2600,  888, 850,  894, 850,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 2600,  888, 2600,  888, 848,  894, 850,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 850,  894, 2600,  888, 2600,  888, 850,  894, 848,  894, 850,  894, 2600,  888, 850,  894, 848,  894, 2600,  888, 2600,  888, 850,  894, 850,  3510, 3474,  886};  // PANASONIC_AC32 32A9
uint64_t data = 0x32A9;

Timestamp : 000980.524
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0x32AC (16 Bits)
uint16_t rawData[201] = {3502, 3474,  888, 848,  894, 850,  894, 2600,  888, 2600,  888, 850,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 848,  896, 2600,  888, 2600,  888, 850,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 850,  894, 2600,  888, 2600,  888, 848,  896, 848,  894, 850,  894, 2600,  888, 848,  894, 850,  894, 2600,  888, 2600,  888, 850,  894, 850,  3512, 3472,  888, 848,  894, 850,  894, 2600,  888, 2600,  888, 848,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 848,  894, 2600,  888, 2600,  888, 850,  894, 2600,  888, 848,  894, 2600,  888, 848,  894, 2600,  888, 850,  894, 850,  894, 2600,  888, 2600,  888, 848,  894, 850,  894, 850,  894, 2600,  888, 850,  894, 850,  894, 2600,  888, 2600,  888, 850,  894, 850,  3512, 3474,  886, 850,  894, 850,  894, 2600,  888, 2600,  888, 848,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 848,  894, 2600,  888, 2600,  888, 850,  894, 2600,  888, 850,  894, 2600,  890, 848,  894, 2600,  888, 848,  894, 850,  894, 2600,  888, 2600,  888, 850,  894, 850,  894, 850,  894, 2600,  888, 848,  894, 850,  894, 2600,  888, 2600,  888, 850,  894, 848,  3512, 3474,  886};  // PANASONIC_AC32 32AC
uint64_t data = 0x32AC;

Timestamp : 000982.622
Library   : v2.7.13

Protocol  : PANASONIC_AC32
Code      : 0x32AD (16 Bits)
uint16_t rawData[201] = {3504, 3472,  888, 2600,  888, 850,  894, 2600,  888, 2600,  888, 850,  894, 2600,  888, 850,  894, 2600,  888, 2598,  890, 848,  896, 2598,  888, 2600,  888, 850,  894, 2600,  888, 848,  894, 2600,  888, 850,  894, 2600,  888, 850,  894, 850,  894, 2600,  888, 2600,  888, 848,  894, 850,  894, 848,  894, 2600,  888, 850,  894, 850,  894, 2600,  888, 2600,  888, 848,  894, 850,  3510, 3474,  888, 2600,  888, 850,  894, 2600,  888, 2600,  890, 848,  894, 2600,  888, 848,  896, 2600,  888, 2600,  888, 848,  896, 2598,  888, 2600,  888, 850,  894, 2600,  888, 850,  894, 2600,  888, 848,  896, 2600,  888, 850,  894, 848,  894, 2600,  888, 2600,  888, 848,  894, 850,  894, 848,  894, 2600,  888, 850,  894, 850,  894, 2600,  888, 2600,  888, 848,  896, 848,  3510, 3474,  888, 2600,  888, 850,  894, 2600,  888, 2600,  888, 850,  894, 2600,  888, 848,  894, 2600,  888, 2600,  888, 848,  894, 2600,  888, 2600,  888, 850,  894, 2600,  888, 850,  894, 2600,  888, 848,  894, 2600,  888, 848,  894, 850,  894, 2600,  888, 2600,  888, 850,  894, 850,  894, 848,  894, 2600,  888, 848,  894, 850,  894, 2600,  890, 2598,  890, 848,  894, 850,  3510,  [... cut ...] 854,  890, 2606,  882, 2606,  882, 854,  890, 854,  3508, 3478,  882, 854,  890, 854,  890, 854,  890, 2606,  882, 854,  890, 2606,  882, 854,  890, 2606,  882, 854,  890, 854,  890, 854,  888, 2606,  882, 854,  890, 2604,  882, 854,  890, 2606,  882, 854,  890, 2606,  882, 854,  890, 854,  890, 2606,  882, 2606,  882, 854,  890, 854,  890, 854,  890, 2606,  882, 854,  888, 854,  890, 2606,  884, 2604,  884, 854,  888, 854,  3506, 3480,  882};  // PANASONIC_AC32 32A8
uint64_t data = 0x32A8;

```

crankyoldgit, 2020-12-24:
> For now
> The fanspeed is off by count of 1.
> Is: Auto > 2 > 3 > 4 > 5 > 6
> Should it not be: Auto > 1 > 2 > 3 > 4 > 5

Nope. The number you are seeing is the value it stores in the correct bits internally.
e.g.
`Fan: 2 (Quiet)` would require `IRPanasonicAc32::setFan(2);`
&
`Fan: 15 (Auto)` would require `IRPanasonicAc32::setFan(15);`
i.e. There is no `1`.

crankyoldgit, 2020-12-24:
I've updated that branch. It should support **power toggle** control via the `IRPanasonicAc32` class, and normal **power** control via the `IRac` class.

It should now be capable of sending messages. e.g. The `IRMQTTServer` example code should work for that protocol in this branch. Ditto for Tasmota.

Please test it and let me know.

FYI, I'm not going to support the 16-bit (short) messages for the finer vane control if I can avoid it. It will add a lot of complexity and code size that most people won't use. If it used the same 32-bit size, it would be a different matter.

Fettkeewl, 2020-12-24:
> Nope. The number you are seeing is the value it stores in the correct bits internally.
> e.g.
> `Fan: 2 (Quiet)` would require `IRPanasonicAc32::setFan(2);`
> &
> `Fan: 15 (Auto)` would require [...]

Alright, good to know. You did see this though? 
> Fan: 6 (UNKNOWN)

I'll try out the changes after Christmas, oh and Merry Christmas to you :)

crankyoldgit, 2020-12-24:
> Alright, good to know. You did see this though?
> 
> > Fan: 6 (UNKNOWN)

Yep. That's expected at the moment. The normal function for displaying fan speeds only caters for 5 speed settings, most only use 3-4. This protocol supports 6, which is one more than easily supported.

I'm still working out how I will deal with that or if I just ignore it.

Fettkeewl, 2020-12-25:
> I've updated that branch. It should support **power toggle** control via the `IRPanasonicAc32` class, and normal **power** control via the `IRac` class.
>
> It should now be capable of sending [...]

Alright I'm up and running with IRMQTTServer, did not do any edits to the IRMQTTServer.h file.

The IR is setup properly with the pin setup however its constantly on. No messages seem to be transmitted when trying 
the various options, the IR is not blinking atleast (as observed through camera)

I do not dare to do invert the output as per the warning unless instructed to do so.

```
// **DANGER** Optional flag to invert the output. (default = false)
//            `false`: The LED is illuminated when the GPIO is HIGH.
//            `true`: The LED is illuminated when GPIO is LOW rather than HIGH.
//            Setting this to something other than the default could
//            easily destroy your IR LED if you are overdriving it.
//            Unless you *REALLY* know what you are doing, don't change this.
const bool kInvertTxOutput = false;
```

[image]

[image]

[image]

crankyoldgit, 2020-12-26:
Can you please give us **way** more information about your settings for the program and from the debug log?

e.g.
> Simply changing from ON -> OFF wont allow me to actually send the command
I get the following in serial log:
`0000600.173: NO difference in common A/C state detected.`

I need far more information about what state you've got IRMQTTServer in, the preceding lines before that log message.
i.e. What protocol are you using? What are the MQTT messages you are sending to it? What's the "info" page look like on IRMQTTServer? etc etc.

See: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#reporting-an-issue

FYI, changing the `model` to `1` is giving you a placebo effect. The protocol `PANASONIC_AC32` does not have model support. So the code/library is just ignoring the `model` value.

Yes, a lot of those DIY IR led boards are just crap. A few 30 cent 940nm IR LEDs and a 5 cent transistor work way way better. (per the suggested circuit diagrams)

crankyoldgit, 2020-12-26:
FYI, your IR LED should not be `on` all the time. You've got something very wrong if that's the case. If you use our suggested circuits **AND** your IR LED is long solid for 10s of seconds, it will quickly burn out, as the suggested circuits deliberately over-drive the LEDs to get more "brightness" out of it, but it can only do that for fractions of a second without damage in most cases.
I've been running cheap-o IR leds in that configuration for years now in my personal setup. They are still working fine, but they only probably lit for about a second (total) per day over 3 years. e.g. 10-20 messages per day, every day.

Fettkeewl, 2020-12-26:
Thank you for your answers, I will do more thorough testing when I get the new ir leds. I just found out the board I was using was ment for proximity sensing. I bought it like 6 years ago so forgot it's use.

BTW, there is no need to close this if u don't hear from me in a while, wife's pregnant coming up on week 40 soon, so I'll be afk a bit longer when we get our touchdown :P but I will return to this because I want to solve it 🙏

Fettkeewl, 2020-12-30:
> I need far more information about what state you've got IRMQTTServer in, the preceding lines before that log message.
> i.e. What protocol are you using? What are the MQTT messages you are sending to it? What's the "info" page look like on IRMQTTServer? etc etc.

All the changes Ive done to IRMQTTServer are listed below:
* Currently MQTT is setup but not used, I'm testing the protocol using the website hosted by the device itself.
* I've setup a static IP adress 
* Serial debugging is enabled

This is my state, boot info in serial below. Will post wonky behaviour separately. 

Boot info:
```
0000000.068: IRMQTTServer v1.5.2 has booted.
0000000.079: Mounting LittleFS ...
0000000.081: mounted the file system
0000000.083: config file exists
0000000.084: Opened config file
0000000.085: Json config file parsed ok.
0000000.085: Recovered Json fields.
0000000.086: Closing the config file.
0000000.090: Unmounting LittleFS
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: http_user
*WM: Adding parameter
*WM: http_pass
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: mqtt_server
*WM: Adding parameter
*WM: mqtt_port
*WM: Adding parameter
*WM: mqtt_user
*WM: Adding parameter
*WM: mqtt_pass
*WM: Increasing _max_params to:
*WM: 20
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: mqtt_prefix
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.200
*WM: Status:
*WM: 6
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.200
0000003.029: WiFi connected. IP address:
0000003.030: 192.168.1.200
0000003.597: MDNS responder started
0000003.600: HTTP server started
0000005.002: client mqtt not connected, trying to connect
0000005.003: Attempting MQTT connection to 192.168.1.100:1883... 
0000005.003: Using mqtt username/password to connect.
0000005.450: (Re)Connected.
0000005.451: Subscription OK to:
0000005.451: IR_Blaster/send
0000005.452: Subscription OK to:
0000005.452: IR_Blaster/ac/cmnd/+
0000005.453: Subscription OK to:
0000005.456: IR_Blaster/send_0
0000005.458: Subscription OK to:
0000005.461: IR_Blaster/ac_0/cmnd/+
0000005.464: IRMQTTServer v1.5.2 just booted
0000005.468: successful client mqtt connection
0000005.472: Started listening for previous state.
0000005.477: Subscription OK to:
0000005.479: IR_Blaster/ac/stat/+
0000006.085: Receiving data by MQTT topic:
0000006.085: IR_Blaster/ac/stat/protocol
0000006.085: with payload:
0000006.085: PANASONIC_AC32
0000006.086: Checking for channel number in IR_Blaster/ac/stat/protocol
0000006.093: Channel = 0
0000006.095: It's a climate state topic. Update internal state and DON'T send
0000006.202: Receiving data by MQTT topic:
0000006.203: IR_Blaster/ac/stat/power
0000006.203: with payload:
0000006.203: on
0000006.203: Checking for channel number in IR_Blaster/ac/stat/power
0000006.208: Channel = 0
0000006.211: It's a climate state topic. Update internal state and DON'T send
0000006.318: Receiving data by MQTT topic:
0000006.318: IR_Blaster/ac/stat/mode
0000006.318: with payload:
0000006.319: heat
0000006.319: Checking for channel number in IR_Blaster/ac/stat/mode
0000006.324: Channel = 0
0000006.326: It's a climate state topic. Update internal state and DON'T send
0000006.434: Receiving data by MQTT topic:
0000006.434: IR_Blaster/ac/stat/fanspeed
0000006.434: with payload:
0000006.434: Auto
0000006.435: Checking for channel number in IR_Blaster/ac/stat/fanspeed
0000006.441: Channel = 0
0000006.443: It's a climate state topic. Update internal state and DON'T send
0000006.550: Receiving data by MQTT topic:
0000006.551: IR_Blaster/ac/stat/temp
0000006.551: with payload:
0000006.551: 25.0
0000006.551: Checking for channel number in IR_Blaster/ac/stat/temp
0000006.556: Channel = 0
0000006.559: It's a climate state topic. Update internal state and DON'T send
0000006.666: Receiving data by MQTT topic:
0000006.666: IR_Blaster/ac/stat/model
0000006.667: with payload:
0000006.667: -1
0000006.667: Checking for channel number in IR_Blaster/ac/stat/model
0000006.672: Channel = 0
0000006.674: It's a climate state topic. Update internal state and DON'T send
0000006.782: Receiving data by MQTT topic:
0000006.782: IR_Blaster/ac/stat/swingv
0000006.782: with payload:
0000006.782: Highest
0000006.783: Checking for channel number in IR_Blaster/ac/stat/swingv
0000006.789: Channel = 0
0000006.791: It's a climate state topic. Update internal state and DON'T send
0000006.899: Receiving data by MQTT topic:
0000006.899: IR_Blaster/ac/stat/swingh
0000006.899: with payload:
0000006.899: Middle
0000006.899: Checking for channel number in IR_Blaster/ac/stat/swingh
0000006.905: Channel = 0
0000006.907: It's a climate state topic. Update internal state and DON'T send
0000010.520: Unsubscribed OK from:
0000010.521: IR_Blaster/ac/stat/+
0000010.522: Difference in common A/C state detected.
0000010.523: The state was recovered from MQTT broker.
0000010.525: Finished listening for previous state.
```

Fettkeewl, 2020-12-30:
I've got the IR leds home today and started testing the AC control under http://< IP >/aircon
Following the example with a transistor, 3V3 > IR > NPN Collector > NPN Emitter > GND
with NPN Base > GPIO4. And the sending distance is greatly increased, awesome.
[image]

I've been trying all the different modes and fanspeeds and temp settings and made notes of 
the wonky behaviours found below,
I do not know how to provide additional data as the serial log only shows messages like this:
```
0003136.525: New common a/c received via HTTP
0003136.530: Difference in common A/C state detected.
0003136.531: Sending common A/C state via IR.
```

**Nevertheless results below:**

•	Power setting does nothing on its own, ON/OFF does not affect the state of the AC.
To control On/Off its basically:
Mode =/= Off  > AC will turn on
Mode = Off > AC will turn off
```java
[Turns A C On]			[Turns A C Off]
PROTOCOL: Panasonic_AC32	PROTOCOL: Panasonic_AC32	
MODEL: Default			MODEL: Default
POWER: On			POWER: On
MODE: Heat			MODE: Off
TEMP: 25 C			TEMP: 25 C
FAN: Auto			FAN: Auto
```

Works flawlessly untill the original remote is used.

•	Changing model does something in the background to affect outcome of your library.
If I turn off my AC with my original remote I can´t turn the AC on using IRremoteESP8266 without changing the 
last used Model type.
```java
[Does nothing]			[Turns A C On]
PROTOCOL: Panasonic_AC32	PROTOCOL: Panasonic_AC32	
MODEL: Default			MODEL: 1
POWER: On			POWER: On
MODE: Heat			MODE: Off
TEMP: 25 C			TEMP: 25 C
FAN: Auto			FAN: Auto
```
Does not matter what I change the model to, so long as its swapped
Then it reverts to working as mentioned in first my point.

•	If I turn on my AC with the original remote, I can’t turn it off with IRremoteESP8266 by changing Mode to OFF in IRremoteESP8266. If I change the protocol and apply a functionality like
```java
[Does nothing]			[Turns A C Off]
PROTOCOL: Panasonic_AC32	PROTOCOL: Panasonic_AC32	
MODEL: 1			MODEL: Default
POWER: On			POWER: On
MODE: Off			MODE: Heat
TEMP: 25 C			TEMP: 25 C
FAN: Auto			FAN: Auto
```

•	Swing modes do nothing except the Swing(V): Auto, which only changes my swing to be facing full down

crankyoldgit, 2020-12-31:
In a way, that's working as intended. Your ac/remote uses a power "toggle" command. There is no discrete "on" and "off" messages. So, you need to get your ESP and the A/C in sync, and keep them in sync. Using the actual remote to power on/off the a/c will get them out of sync.

The IRMQTTServer program tries to emulate a discrete/true "on" and "off" function by keeping track of what state the a/c unit should be in, and only sending a "toggle" command when needed to switch the A/C to the appropriate mode.

So for your A/C, IRMQTTServer will only "normally" send a power "toggle" message when IRMQTTServer's "power" state changes from "On" to "Off" or "Off" to "On". Any other normal setting change will not cause a "power toggle" message.
You've encountered/triggered one of the other ways that trigger a reset of IRMQTTServer's internal state. i.e. Changing the model type (which has no real effect on the operation mode of your A/C), or changing the protocol type. This happens because not all protocols and models behave the same way, hence it tries to reset the way it does things.

In short, don't power on/off the A/C with the actual remote when you want control keep controlling the A/C via the ESP.
If you get it out of sync, just cover the IR led on the ESP, and change the power state via http/mqtt. Then uncover the IR led.

or

Add an IR demodulator to the ESP, make sure it also sees the real remote messages to the A/C. You can change/compile the code to accept the real remote's message, change the state, and NOT re-produce the resulting message to the A/C unit.

or

Get the ESP in sync by using the real IR remote on the A/C to be in the same state that the ESP thinks it is.

Now, the "Mode" from Off -> Cool and Cool->Off thing. This is how [Home Assistant](https://www.home-assistant.io/) works or likes it. You probably need to change this setting in IRMQTTServer:
https://github.com/crankyoldgit/IRremoteESP8266/blob/3b57c5e736544b1f009fe8b21a73379c830c0d25/examples/IRMQTTServer/IRMQTTServer.h#L137-L151

Thanks for confirming everything else seems to be working with the new protocol control.

Fettkeewl, 2020-12-31:
Alright I understand! Thanks for explaining. 

Will dig deeper into this at another time, for now may I ask what the plan is? 
Will this protocol be out with the next release of IRremoteESP8266? Hoping to get it into Tasmota with their next version update,

crankyoldgit, 2020-12-31:
> Will this protocol be out with the next release of IRremoteESP8266?

Once it's been code reviewed, it will be merged into the `master` branch and then in the next release.

PR #1366 (2021-01-01): PANASONIC_AC32: Add limited detailed support.
* Add set/get support for:
  - Power Toggle
  - Mode
  - Temp
  - Fan Speed
  - Swing V & Swing H
* Add `IRac` class support for new changes.
* Unit test coverage for the above.
* Fix `#if` clause in `IRPanasonicAc` class.
* Extend `irutils::addFanToString()` to have an option for max.
* General code clean-up.

Note: 16 bit message settings are NOT supported.

Fixes #1364
commit f6838210 (2021-01-01): PANASONIC_AC32: Add limited detailed support. (#1366)

* Add set/get support for:
  - Power Toggle
  - Mode
  - Temp
  - Fan Speed
  - Swing V & Swing H
* Add `IRac` class support.
* Unit test coverage for the above.
* Fix `#if` clause in `IRPanasonicAc` class
* General code cleanup.
* Extend irutils::addFanToString() to have an option for max.
* Minor code cleanup

Fixed #1364
closed 2021-01-01 by crankyoldgit