crankyoldgit/IRremoteESP8266 #1376: Fujitsu AR-RAHxx setSwing freezes the AC unit

SupremeSports, 2021-01-02:
### Version/revision of the library used
2.7.2 and 2.7.13

### Describe the bug
Fujitsu AC wall unit with remote AR-RAH1U (similar to AR-RAH2E but doesn't have horizontal swing support). I've been able to get the tilt to be changed from ESP, but when trying to set the swing mode to ON (kFujitsuAcSwingVert), the AC receives the command and freezes, it will continue to operate as previously set, but not able to receive further commands either from original remote or from ESP. 

Note that all other functions (setTemp, setMode, setFan, On/Off) functions works properly until I try to setSwing(kFujitsuAcSwingVert), then nothing works anymore

#### To Reproduce
It happens each and every time I try to send the swing mode command, also, tried a ac.toggleSwingVert(true) and got the same issue.

#### Example code used

```
if (state)
  ac.setSwing(kFujitsuAcSwingVert);
else
  ac.setSwing(kFujitsuAcSwingOff);
```

Also tried both of those:
```
ac.toggleSwingVert(true);
//ac.toggleSwingVert(false);
```

#### Expected behaviour
Expecting the AC unit to switch to swing mode as it does with the original remote (it works fine with the remote)

#### Output of raw data from [IRrecvDumpV2.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) or [V3](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/) (if applicable)
_Include some serial text of the raw dumps of what the device saw._

_**Note**: Output from Tasmota is not acceptable. We can't easily use their raw format._

I don't have any IR receiver on hand, I only bought IR leds :(

### What brand/model IR demodulator are you using?
None, I simply plugged the IR LED through a transistor/resistor to GPIO4 on ESP8266-07, and pointed it toward the AC unit

### Circuit diagram and hardware used (if applicable)
I simply plugged the IR LED through a transistor/resistor to GPIO4 on ESP8266-07, and pointed it toward the AC unit

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Has this library/code previously worked as expected for you?
Yes, but never the setSwing function. Tried with 2.7.2, and just updated to 2.7.13 to test, and issue is the same.

### Other useful information
I believe that maybe the setSwing function send the wrong IR code? It completely freeze the AC unit and all I can do is to turn off the breaker and put it back on, then everything goes back to working state until I retry setSwing.

I realize that I'll need to get a IR receiver and test what the remote is sending, probably just a wrong constant, but I'm sending this issue in in order for others who could have this issue to chime in...

crankyoldgit, 2021-01-02:
> I realize that I'll need to get a IR receiver and test what the remote is sending, probably just a wrong constant, but I'm sending this issue in in order for others who could have this issue to chime in...

Yep. You got it in one. We will need the corresponding captures for the same mode(s)/setting(s) that you real-remote would produce in order to see what we need to change to fix it etc.

Also, you haven't included the full simplified code you are using to create & send the IR A/C message.
e.g.
https://github.com/crankyoldgit/IRremoteESP8266/blob/4e2cce68722915cbe99ce87204c86e667059afe9/examples/TurnOnFujitsuAC/TurnOnFujitsuAC.ino#L27-L37

Most importantly, what `model` did you use with `ac.setModel();`?

The swing settings change significantly depending which Fujitsu model you choose.

SupremeSports, 2021-01-02:
`Most importantly, what model did you use with ac.setModel();?`

```
ac.setModel(ARRAH2E); 
```

But as specified, I don't have this remote, I have the ARRAH1U that doesn't have horizontal swing. I started a modification of the library on my side and will do a pull request once working fine. Since I don't have the same remote and mine doesn't have horizontal, that's why I'm using only 

`Also, you haven't included the full simplified code you are using to create & send the IR A/C message.`
Yours pretty much sums it up, tried many configuration, I pretty much mastered the use of this library, and all other functions work fine...

```
#include <IRremoteESP8266.h>  //2.7.13
#include <IRsend.h>
#include <ir_Fujitsu.h>

IRFujitsuAC ac(4);

void setup()
{
  ac.begin();
  ac.setModel(ARRAH2E);
  delay(200);

  ac.setSwing(kFujitsuAcSwingVert); 
  ac.setMode(kFujitsuAcModeHeat); //But tried all possibilities
  ac.setFanSpeed(kFujitsuAcFanHigh); //But tried all possibilities
  ac.setTemp(24);  // 24C 
  ac.setCmd(kFujitsuAcCmdTurnOn);  //But tried all possibilities
  ac.send();
}
```

I ordered IR receiver, I'll get them in about 2 weeks, so I'll do some tests and come back to this issue

crankyoldgit, 2021-01-02:
FYI, the available model numbers are located here:
https://github.com/crankyoldgit/IRremoteESP8266/blob/4e2cce68722915cbe99ce87204c86e667059afe9/src/IRsend.h#L119-L126

These models don't have horizontal swings:
https://github.com/crankyoldgit/IRremoteESP8266/blob/4e2cce68722915cbe99ce87204c86e667059afe9/src/ir_Fujitsu.cpp#L485-L491

I'd suggest you try`ARREB1E` as the model. If I recall correctly, it's the same as `ARRAH2E` but with no horizontal. Fingers crossed.

SupremeSports, 2021-01-02:
`I'd suggest you tryARREB1E as the model. If I recall correctly, it's the same as ARRAH2E but with no horizontal. Fingers crossed.`

Perfect, I'm out of town for 2 weeks, I'll be retesting this when I'll be back. Physically, it looks pretty different, but maybe the backend is correct, I'll give it a try, anyways, using the wrong data should not mess my AC unit???

crankyoldgit, 2021-01-03:
> using the wrong data should not mess my AC unit???

I think you've already experienced the worst case.

SupremeSports, 2021-01-16:
`I'd suggest you tryARREB1E as the model`

You were right, it works... Maybe we could update the enum to make it obvious :)

closed 2021-01-16 by SupremeSports

crankyoldgit, 2021-01-16:
@SupremeSports Do you have the Model number of the A/C unit per chance?

SupremeSports, 2021-01-17:
> @SupremeSports Do you have the Model number of the A/C unit per chance?

Unit = ASU12RLF
Remote = AR-RAH1U

SupremeSports, 2021-01-17:
@crankyoldgit awesome work, very happy to contribute to such a huge project that must be incredibly hard to maintain considering the wide range of units/remote/protocols :)

crankyoldgit, 2021-01-17:
It's a challenge.

crankyoldgit, 2021-02-13:
FYI, the changes mentioned above have now been included in the new v2.7.15 release of the library.

PR #1386 (2021-01-17): Fujitsu A/C: Add warning/suggestions for AR-RAH1U devices
Correct model to use for an AR-RAH1U remote is `ARREB1E`, otherwise A/C can lock up.

Fixes #1376
commit 91846e41 (2021-01-17): Fujitsu A/C: Add warning/suggestions for AR-RAH1U devices (#1386)

* Fujitsu A/C: Add warning/suggestions for AR-RAH1U devices
Correct model to use for an AR-RAH1U remote is `ARREB1E`, otherwise A/C can lock up.
* Update SupportedProtocols.md
  - Out of release cycle update to prevent accidental device lockups.

Fixes #1376