crankyoldgit/IRremoteESP8266 #1378: Samsung and IRac.h

hoangtntx, 2021-01-06:
I have problem with IRac.h, when i use it, i can control LG and coolix ok but with samsung , i can only control on and off. I cant set temp and mode with samsung. I used sample code test with only samsung but still not ok, i set temp but it still 16 degree . 
`/* Copyright 2019 David Conran
*
* This example code demonstrates how to use the "Common" IRac class to control
* various air conditions. The IRac class does not support all the features
* for every protocol. Some have more detailed support that what the "Common"
* interface offers, and some only have a limited subset of the "Common" options.
*
* This example code will:
* o Try to turn on, then off every fully supported A/C protocol we know of.
* o It will try to put the A/C unit into Cooling mode at 25C, with a medium
*   fan speed, and no fan swinging.
* Note: Some protocols support multiple models, only the first model is tried.
*
*/
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRac.h>
#include <IRutils.h>

const uint16_t kIrLed = 4;  // The ESP GPIO pin to use that controls the IR LED.
IRac ac(kIrLed);  // Create a A/C object using GPIO to sending messages with.

void setup() {
  Serial.begin(115200);
  delay(200);

  // Set up what we want to send.
  // See state_t, opmode_t, fanspeed_t, swingv_t, & swingh_t in IRsend.h for
  // all the various options.
  ac.next.protocol = decode_type_t::SAMSUNG;  // Set a protocol to use.
  ac.next.model = 1;  // Some A/Cs have different models. Try just the first.
  ac.next.mode = stdAc::opmode_t::kCool;  // Run in cool mode initially.
  ac.next.celsius = true;  // Use Celsius for temp units. False = Fahrenheit
  ac.next.degrees = 25;  // 25 degrees.
  ac.next.fanspeed = stdAc::fanspeed_t::kMedium;  // Start the fan at medium.
  ac.next.swingv = stdAc::swingv_t::kOff;  // Don't swing the fan up or down.
  ac.next.swingh = stdAc::swingh_t::kOff;  // Don't swing the fan left or right.
  ac.next.light = false;  // Turn off any LED/Lights/Display that we can.
  ac.next.beep = false;  // Turn off any beep from the A/C if we can.
  ac.next.econo = false;  // Turn off any economy modes if we can.
  ac.next.filter = false;  // Turn off any Ion/Mold/Health filters if we can.
  ac.next.turbo = false;  // Don't use any turbo/powerful/etc modes.
  ac.next.quiet = false;  // Don't use any quiet/silent/etc modes.
  ac.next.sleep = -1;  // Don't set any sleep time or modes.
  ac.next.clean = false;  // Turn off any Cleaning options if we can.
  ac.next.clock = -1;  // Don't set any current time if we can avoid it.
  ac.next.power = false;  // Initially start with the unit off.

  Serial.println("Try to turn on & off every supported A/C type ...");
}

void loop() {
  // For every protocol the library has ...
    decode_type_t protocol = (decode_type_t)46;
    // If the protocol is supported by the IRac class ...
    if (ac.isProtocolSupported(protocol)) {
      Serial.println("Protocol " + String(protocol) + " / " +
                     typeToString(protocol) + " is supported.");
      ac.next.protocol = protocol;  // Change the protocol used.
      ac.next.celsius = true;
      ac.next.degrees = 21;
      ac.next.power = true;  // We want to turn on the A/C unit.
      Serial.println("Sending a message to turn ON the A/C unit.");
      ac.sendAc();  // Have the IRac class create and send a message.
      delay(5000);  // Wait 5 seconds.
      ac.next.celsius = true;
      ac.next.degrees = 21;
      ac.next.beep = true;  // Turn off any beep from the A/C if we can.
      ac.next.econo = true;  // Turn off any economy modes if we can.
      ac.sendAc(); 
      delay(5000);  // Wait 5 seconds.
      ac.next.power = false;  // Now we want to turn the A/C off.
      Serial.println("Send a message to turn OFF the A/C unit.");
      ac.sendAc();  // Send the message.
      delay(1000);  // Wait 1 second.
    }
  Serial.println("Starting from the begining again ...");
}`

NiKiZe, 2021-01-06:
I'm quite sure that the FAQ explains that not all units uses the same protocol.

What is the model number of your unit and remote? (Please answer this question properly)
Have you checked the supported list?

Have you, as described, taken a dump, and what does it show when you are using the original remote to change the this that you say are not working.

Also, how do you know that it isn't working?

hoangtntx, 2021-01-06:
I just check with sample CommonACcontrol and read received signal. all protocol can set temp but samsung cant set temp by Irac.h. picture below i set temp 21 and mode cool but samsung still temp 16 mode auto, cant change any mode, only turn on and off.
[image]

NiKiZe, 2021-01-06:
You are not answering the questions asked.

And pictures is not helpful, when you have the data that was asked, post it as text. Not pictures!

crankyoldgit, 2021-01-06:
`SAMSUNG_AC` is different from most protocols as it has pre-recorded/unchangable `on` & `off` commands due to the fact we can't dynamically construct the (168 bit) Extended Samsung message format. Those "saved" messages have the temp set at 16c. Once a Samsung A/C is on, and use normal commands to change modes/temp/speeds etc it will produce a different format/length (112 bit) Samsung A/C message (ones that we can dynamically construct).

See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/63e2538366f0b5050efabfc0fde84882795d830c/src/ir_Samsung.cpp#L379-L403
&
https://github.com/crankyoldgit/IRremoteESP8266/blob/63e2538366f0b5050efabfc0fde84882795d830c/src/ir_Samsung.cpp#L336-L354

Please also note that we don't accept output from other programs (like Tasmota) when debugging our library.
See: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/.github/ISSUE_TEMPLATE/problem-report.md#output-of-raw-data-from-irrecvdumpv2ino-or-v3-if-applicable
Which was in the issue template you removed/skipped over.

crankyoldgit, 2021-01-06:
You should also try: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/ControlSamsungAC/ControlSamsungAC.ino

crankyoldgit, 2021-01-06:
> What is the model number of your unit and remote? (Please answer this question properly)
> Have you checked the supported list?
> 
> Have you, as described, taken a dump, and what does it show when you are using the original remote to change the this that you say are not working.

@hoangtntx Please answer these questions etc.

crankyoldgit, 2021-01-13:
Ping @hoangtntx

crankyoldgit, 2021-01-17:
@hoangtntx Another ping. If we don't hear from you soon, we are going to assume your non-response is because you worked it out & that this issue is stale and can be closed.

crankyoldgit, 2021-01-24:
This issue appears stale. Closing due to lack of response.

closed 2021-01-24 by crankyoldgit