crankyoldgit/IRremoteESP8266 #1387: Support for SHARP AC - Model AH-A12REVP-1

GerardNL, 2021-01-17:
### Version/revision of the library used v2.7.14

### Describe the bug
I would kindly request support for SHARP AC, Model AH-A12REVP-1. It seems to be this specific model of SHARP ACs is not yet fully/correctly included. 

#### To Reproduce
I'm using Tasmoto 9.2.0(ir) with an Eachen IR module (which should include the IRremoteESP8266 v2.7.14 library) to control my Sharp Air Conditioner from Home Assistant 2021.1.x. Out of the box, several of the standard functions work just fine (temperature, mode, fan), but I'm unable to turn the AC unit ON and OFF through the build-in function of the HA. When the unit is off, it will not turn on with any other code until it received the regular "ON" command from the remote. (A totally different AC model in another room responds fine to the ON command from HA, so I don't think that's the issue)

So I have recorded the commands sent from the remote and worked out most of the digits in those command. My C++ knowledge is a bit rusty, but seems to match what is in ir_Sharp.h for the essential parts, so that seems all good.

When I press the "ON/OFF" button on the remote while the AC is on, I see this is recorded:
`{"Protocol":"SHARP_AC","Bits":104,"Data":"0xAA5ACF10CA212200088000F061","Repeat":0,"IRHVAC":{"Vendor":"SHARP_AC","Model":1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":25,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}`
So it looks like the command is well understood by the library.

When I use Home Assistant to send the "OFF" command, I see on MQQT command is sent:
`{"IRHVAC":{"Vendor":"SHARP_AC","Model":1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":25,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}`
(But I can't see the (raw?) command send to the AC, so I don't know if it sends "0xAA5ACF10CA212200088000F061"
Except for the FanSpeed, the parameters seems the same. The AC unit beeps, but doesn't turn off.

I've tried to send the commands recorded back to the AC unit, but here I seem to lack enough knowledge to get that to work. 
All I get is this --> MQT: stat/IR_Livingroom/RESULT = {"IRSend":"Done"}
and no response from the AC. So I'm not yet able to send raw data to the AC.

Below are the details I have worked out so far and the recorded commands.

Would this something that makes sense to @crankyoldgit ? Or is there any other information I can provide that would get this AC unit included? As it's hopefully just a little tweaking of existing code, it might not be too difficult?

Many thanks in advance!

Gerard

#### Output of raw data from [IRrecvDumpV2.ino]Recorded commands from the remote. 
(Sorry. I know Tasmoto output is not preferred, but not (yet) able to get others. Hopefully it's not a big problem in this case)
Turn Off (No response from AC when sent from HA): 0xAA5ACF10CA212200088000F061
Turn On (No response from AC when sent from HA): 0xAA5ACF10CA112200088000F051
Temp 30 (AC responds to command send from HA): 0xAA5ACF10CF312200088004F061
Temp 26 (AC responds to command send from HA): 0xAA5ACF10CB312200088004F021
Temp 25 (AC responds to command send from HA): 0xAA5ACF10CA312200088004F031
Temp 24 (AC responds to command send from HA): 0xAA5ACF10C9312200088004F001
Temp 23 (AC responds to command send from HA): 0xAA5ACF10C8312200088004F011
Temp 16 (AC responds to command send from HA): 0xAA5ACF10C1312200088004F081
Fan Speed - Medium = Lowest on remote (AC responds to command send from HA): 0xAA5ACF10CA315200088005F051
Fan Speed - Max = Middle on remote (AC responds to command send from HA): 0xAA5ACF10CA317200088005F071
Fan Speed - Auto (AC responds to command send from HA): 0xAA5ACF10CA312200088005F021
Vertical Swing ON/OFF Toggle: 0xAA5ACF10CA3122000F8006F061
Mode Dry (AC responds to command send from HA): 0xAA5ACF1000312300088000F001
Mode Auto (AC responds to command send from HA): 0xAA5ACF1000312000088000F031
Mode Cool (AC responds to command send from HA): 0xAA5ACF10CA312200088000F071
Turbo On (AC responds to command send from HA): 0xAA5ACF10CA612200088001F031
Turbo Off (AC responds to command send from HA): 0xAA5ACF10CA712200088001F021
I don't have the Timer functions below in Home Assistant, so I don't know if the AC would respond
Timer Cancel: 0xAA5ACF10CA8122400880C0F041
Timer On 0.5 hr: 0xAA5ACF10CA8122C00880DEF031
Timer On 1.0 hr: 0xAA5ACF10CA8122C10880C0F0D1
Timer On 1.5 hr: 0xAA5ACF10CA8122C10880DEF021
Timer On 2.0 hr: 0xAA5ACF10CA8122C20880C0F0E1
Timer On 2.5 hr: 0xAA5ACF10CA8122C20880DEF011
Timer On 3.0 hr: 0xAA5ACF10CA8122C30880C0F0F1
Timer On 3.5 hr: 0xAA5ACF10CA8122C30880DEF001
Timer On 4.0 hr: 0xAA5ACF10CA8122C40880C0F081
Timer On 4.5 hr: 0xAA5ACF10CA8122C40880DEF071
Timer On 5.0 hr: 0xAA5ACF10CA8122C50880C0F091
Timer On 5.5 hr: 0xAA5ACF10CA8122C50880DEF061
Timer On 6.0 hr: 0xAA5ACF10CA8122C60880C0F0A1
Timer On 6.5 hr: 0xAA5ACF10CA8122C60880DEF051
Timer On 7.0 hr: 0xAA5ACF10CA8122C70880C0F0B1
Timer On 7.5 hr: 0xAA5ACF10CA8122C70880DEF041
Timer On 8.0 hr: 0xAA5ACF10CA8122C80880C0F041
Timer On 8.5 hr: 0xAA5ACF10CA8122C80880DEF0B1
Timer On 9.0 hr: 0xAA5ACF10CA8122C90880C0F051
Timer On 9.5 hr: 0xAA5ACF10CA8122C90880DEF0A1
Timer On 10 hr: 0xAA5ACF10CA8122CA0880C0F061
Timer On 11 hr (double beep on AC. Remote sends command twice): 0xAA5ACF10CA8122CB0880C0F071
Timer On 12 hr (double beep on AC. Remote sends command twice): 0xAA5ACF10CA8122CC0880C0F001
Timer Off 0.5 hr: 0xAA5ACF10CA8122800880DEF071
Timer Off 1.0 hr: 0xAA5ACF10CA8122810880C0F091
Timer Off 1.5 hr: 0xAA5ACF10CA8122810880DEF061
Timer Off 2.0 hr: 0xAA5ACF10CA8122820880C0F0A1
Timer Off 2.5 hr: 0xAA5ACF10CA8122820880DEF051
Timer Off 3.0 hr: 0xAA5ACF10CA8122830880C0F0B1
Timer Off 3.5 hr: 0xAA5ACF10CA8122830880DEF041
Timer Off 4.0 hr: 0xAA5ACF10CA8122840880C0F0C1
Timer Off 4.5 hr: 0xAA5ACF10CA8122840880DEF031
Timer Off 5.0 hr: 0xAA5ACF10CA8122850880C0F0D1
Timer Off 5.5 hr: 0xAA5ACF10CA8122850880DEF021
Timer Off 6.0 hr: 0xAA5ACF10CA8122860880C0F0E1
Timer Off 6.5 hr: 0xAA5ACF10CA8122860880DEF011
Timer Off 7.0 hr: 0xAA5ACF10CA8122870880C0F0F1
Timer Off 7.5 hr: 0xAA5ACF10CA8122870880DEF001
Timer Off 8.0 hr: 0xAA5ACF10CA8122880880C0F001
Timer Off 8.5 hr: 0xAA5ACF10CA8122880880DEF0F1
Timer Off 9.0 hr: 0xAA5ACF10CA8122890880C0F011
Timer Off 9.5 hr: 0xAA5ACF10CA8122890880DEF0E1
Timer Off 10 hr: 0xAA5ACF10CA81228A0880C0F021
Timer Off 11 hr (double beep on AC. Remote sends command twice): 0xAA5ACF10CA81228B0880C0F031
Timer Off 12 hr (double beep on AC. Remote sends command twice): 0xAA5ACF10CA81228C0880C0F041
No other buttons on the remote. 

### What brand/model IR demodulator are you using?
Tasmoto 9.2.0(ir) with an Eachen IR module (which should include the IRremoteESP8266 v2.7.14 library)
Hardware: Eachen Universal IR-DC6 Remote Control running Tosmoto
I know it's a cheapy model (but popular) and there is a chance it's the cause of the problem, but doesn't seem too likely.

### I have followed the steps in the [Troubleshooting Guide]
Yes. 
I've also tried different version of Tosmoto (which had different minor versions of the IRremoteESP8266 library and tried different ACs with the same IR controller

### Has this library/code previously worked as expected for you?
No. None of the previous versions I tried have worked.
But if I understand the change logs correctly, the sharp AC code hasn't been changed since 2.6.1, and I have only tried version that were newer that that.

crankyoldgit, 2021-01-17:
Hi there. As you've correctly mentioned, we typically don't take logs/data from Tasmota  This is because there is a whole other layer of program involved which muddies the waters a lot. Tasmota doesn't give us data in a format that is always useful to us. e.g. Like your situation.

You're going to have to get more debug/info out of Tasmota _or_ get another ESP board with an IR demodulator to decode what messages are really sent (e.g. using [IRrecvDumpV3](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRrecvDumpV3)). Without more/better data it is not easy for us to help you.

I just browsed the code and the Sharp A/C support has some bizarreness with how it handles power on and off. i.e. It needs to track the previous state of the A/C to know exactly what command code to send. I don't know if/how Tasmota does this. i.e. How it calls/handles/invokes [`IRac::sendAc(etc)`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRac.html#ad60fbe1488efe2d02307d81b090b3b72) (Note: There are three very different ways it can be called/used.)

If you have the ability to flash your current ESP via serial (in case of emergency etc) you can try [IRMQTTServer](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRMQTTServer) which I do support fully and can be updated OTA (but if you've only got 1M, you probably want serial access to it). As I know it intimately, I can work out what is going on better etc.

FYI, you say:
> When I press the "ON/OFF" button on the remote while the AC is on, I see this is recorded:
{"Protocol":"SHARP_AC","Bits":104,"Data":"**0xAA5ACF10CA212200088000F061**","Repeat":0,"IRHVAC":{"Vendor":"SHARP_AC","Model":1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":25,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}

and

> Turn Off (No response from AC when sent from HA): **0xAA5ACF10CA212200088000F061**

Those are the same message codes. So, your A/C really should be responding via Tasmota/HA etc. (Assuming the actual A/C was already on before it was sent via HA etc)

Sharp A/Cs supported by this library has different codes for Off and On. So see if your actual remote produces different codes etc.

GerardNL, 2021-01-17:
Hi David,

Thanks for the quick reply!

I was afraid that my finding were not detailed enough yet. I don't have serial port yet but I do have an FT232RL FTDI Serial Adapter Module on order. Any particular hardware you'd recommend purchasing? I don't mind investing modestly to learn and contribute to the HA community. The only point would be that I'm currently in Mali, which makes shipping a bit difficult, so I usually pick new "toys" up while on a break. As my next break is scheduled in about 2 months, so there is plenty of time to order it online (e.g ali express) and have it arrive at family ahead of me. It would however take a while before I can provide the needed information. So if you want to close the topic (after your recommendations), I totally understand and will restart one hopefully in March.

crankyoldgit, 2021-01-17:
> Any particular hardware you'd recommend purchasing?

This is my off-the-top-of-my-head suggestion list for circuits like [these](https://github.com/crankyoldgit/IRremoteESP8266/wiki):
*  for the ESP: Wemos D1 Mini or similar dev board (ESP8266 or ESP32 doesn't matter)
* a circuit breadboard of some description.
* A 38kHz IR demodulator component (Avoid a [VS1838b](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#help-im-getting-very-inconsistent-results-when-capturing-an-ir-message-using-a-vs1838b-ir-demodulator) if you can)
* Some dupont cables/wires. (Optional)
* micro-usb to USB cable. (You probably have one already)
* Several 940nm IR LEDs. (Cheap as chips)  (Optional)
* An NPN transistor or two. (I bought mine for like 100 for a $1 or so) (Optional)

All of that can be found on Aliexpress/Ebay/Amazon etc etc. Avoid the pre-built "Arduino IR boards".  Often their IR range is terrible.

We can keep this open and see how we go with debugging/capturing. As the library already detects your signal, we can at least verify (albeit slowly) what the remote is sending, and I can verify that with what the library is doing etc. Alas, that doesn't account for the tasmota aspect. But if we can confirm the library works for your A/C, you can eventually point the Tasmota people to this issue to work out what might be wrong at their end, if it is at their end etc.

GerardNL, 2021-01-17:
Thanks for that. I've just ordered a small assortment of these components. Some of different types to be sure. Now the long wait for the shipments to tiny non descript envelopes from China. Looking forward dipping my toes in some hardware. I've always been on the software side :-)

crankyoldgit, 2021-01-18:
> I've always been on the software side :-)

Ditto. I have very rudimentary electronics skills. Thankfully, the circuits required to operate IR are simple and work in most trivial scenarios.

crankyoldgit, 2021-01-18:
In the meantime, can you capture a "OFF" message from the remote (when the A/C unit is on) and then an "ON" (when the unit is off).

lenkunz, 2021-01-19:
Hi.

I have Sharp AC Model: AH-PR13
I also have this problem. The remote model starts with A903. IRrecvDumpV3 detect it as A907.
Everything work fine except power command. It also refuse to respond when power is off.

The dump from IRrecvDumpV3 when I use the remote to turn the power off is
```
Protocol  : SHARP_AC
Code      : 0xAA5ACF10CC113200088000F461 (104 Bits)
Mesg Desc.: Model: 1 (A907), Power: On, Mode: 2 (Cool), Temp: 27C, Fan: 3 (Medium), Turbo: Off, Swing(V) Toggle: Off, Ion: On, Econo: -, Clean: Off
uint16_t rawData[211] = {3850, 1874,  488, 470,  484, 1370,  486, 470,  484, 1370,  488, 470,  508, 1346,  486, 470,  484, 1370,  486, 472,  482, 1370,  486, 472,  484, 1370,  488, 1370,  486, 472,  482, 1370,  488, 470,  482, 1370,  510, 1346,  486, 1370,  512, 1346,  486, 472,  482, 472,  484, 1370,  486, 1370,  488, 472,  482, 472,  484, 470,  482, 472,  484, 1368,  512, 446,  484, 470,  482, 472,  484, 472,  482, 472,  484, 1370,  488, 1368,  488, 472,  482, 472,  484, 1370,  486, 1370,  488, 1370,  486, 472,  482, 472,  500, 454,  484, 1370,  486, 472,  482, 472,  484, 470,  482, 472,  484, 1370,  486, 472,  484, 470,  484, 1368,  488, 1368,  488, 472,  484, 470,  484, 472,  484, 470,  484, 472,  482, 472,  484, 470,  484, 470,  482, 470,  484, 470,  482, 472,  482, 472,  482, 472,  482, 1370,  488, 472,  484, 470,  484, 472,  484, 470,  484, 472,  484, 470,  484, 470,  484, 470,  484, 470,  484, 470,  484, 472,  484, 1368,  488, 470,  484, 470,  484, 470,  484, 472,  482, 472,  484, 470,  484, 470,  484, 470,  484, 470,  484, 472,  484, 1368,  488, 472,  482, 1370,  488, 1370,  488, 1368,  488, 1368,  488, 1368,  488, 470,  484, 470,  482, 472,  484, 470,  484, 1368,  488, 1370,  488, 470,  430};  // SHARP_AC
uint8_t state[13] = {0xAA, 0x5A, 0xCF, 0x10, 0xCC, 0x11, 0x32, 0x00, 0x08, 0x80, 0x00, 0xF4, 0x61};
```

I hope this will be helpful.

crankyoldgit, 2021-01-19:
> I hope this will be helpful.

Actually, it does. A lot.

I think I see how the A903 differs from the A907.

I need to know a bunch more about this model.
eg. What features are reported incorrectly? (and hex codes for those that don't + description)
Does this model have a light/LED button? What about Econo mode?
What is the Full model number for the A/C unit & the remote please?

In the meantime, can you please download, test, & give feedback on this branch please: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1387

Hopefully it should work for you if/when you set the model to `3`.

lenkunz, 2021-01-20:
> Actually, it does. A lot.
>
> I think I see how the A903 differs from the A907.
>
> I need to know a bunch more about this model.
> eg. What features are reported incorrectly? (and hex codes for those that [...]

Hi I'm here to report the result and provide extra info.

# There is some data that incorrectly reported by Dump.
## The fan in auto mode (2) reported correctly. But other is largely incorrect.
**After tested with dump it is as following.**

### Auto
```
rotocol  : SHARP_AC
Code      : 0xAA5ACF10CC312200088005F401 (104 Bits)
Mesg Desc.: Model: 3 (A903), Power: On, Mode: 2 (Cool), Temp: 27C, Fan: 2 (Auto), Turbo: Off, Swing(V) Toggle: Off, Ion: On, Econo: -, Clean: Off
uint16_t rawData[211] = {3850, 1880,  482, 474,  480, 1376,  480, 474,  480, 1376,  480, 472,  482, 1376,  480, 474,  480, 1376,  482, 474,  482, 1374,  480, 474,  480, 1374,  482, 1374,  480, 474,  480, 1376,  482, 474,  482, 1376,  480, 1376,  480, 1376,  480, 1376,  480, 474,  482, 472,  480, 1376,  482, 1376,  480, 474,  480, 474,  480, 474,  480, 474,  482, 1376,  480, 474,  482, 472,  480, 474,  482, 474,  480, 472,  482, 1376,  480, 1376,  480, 474,  480, 474,  480, 1376,  482, 1374,  482, 1376,  480, 474,  480, 474,  482, 472,  480, 1376,  480, 1376,  482, 474,  480, 474,  480, 474,  482, 1376,  480, 474,  480, 474,  480, 474,  480, 1376,  482, 472,  482, 474,  480, 474,  480, 474,  482, 474,  480, 474,  480, 474,  480, 474,  482, 474,  480, 474,  480, 474,  482, 472,  480, 474,  482, 1376,  480, 474,  482, 472,  482, 474,  480, 472,  482, 474,  480, 474,  482, 472,  482, 474,  480, 474,  480, 472,  480, 474,  482, 1376,  480, 1376,  482, 472,  482, 1376,  480, 474,  506, 448,  480, 474,  482, 474,  480, 474,  480, 474,  482, 472,  482, 1376,  480, 474,  480, 1376,  482, 1376,  504, 1352,  482, 1374,  482, 1376,  482, 472,  480, 474,  482, 472,  482, 474,  480, 474,  480, 474,  482, 472,  428};  // SHARP_AC
uint8_t state[13] = {0xAA, 0x5A, 0xCF, 0x10, 0xCC, 0x31, 0x22, 0x00, 0x08, 0x80, 0x05, 0xF4, 0x01};
```
### Low
```
Protocol  : SHARP_AC
Code      : 0xAA5ACF10CC313200088005F411 (104 Bits)
Mesg Desc.: Model: 3 (A903), Power: On, Mode: 2 (Cool), Temp: 27C, Fan: 3 (Medium), Turbo: Off, Swing(V) Toggle: Off, Ion: On, Econo: -, Clean: Off
uint16_t rawData[211] = {3850, 1880,  482, 472,  482, 1374,  482, 472,  480, 1376,  482, 472,  482, 1376,  480, 474,  480, 1374,  482, 474,  480, 1376,  482, 474,  480, 1376,  480, 1376,  482, 472,  482, 1374,  480, 474,  482, 1374,  482, 1374,  482, 1374,  480, 1376,  480, 474,  480, 474,  480, 1376,  456, 1400,  482, 474,  480, 474,  456, 496,  480, 474,  480, 1376,  480, 474,  480, 474,  478, 476,  458, 496,  456, 498,  480, 1376,  480, 1376,  458, 498,  480, 474,  480, 1376,  456, 1398,  470, 1388,  480, 474,  482, 472,  482, 472,  458, 1398,  458, 1398,  480, 476,  456, 498,  480, 474,  480, 1376,  458, 496,  482, 474,  480, 1376,  478, 1378,  478, 476,  458, 496,  458, 498,  456, 498,  480, 474,  480, 474,  458, 496,  480, 474,  458, 496,  480, 474,  480, 474,  480, 474,  458, 496,  480, 1376,  456, 498,  458, 496,  480, 476,  478, 476,  456, 496,  482, 474,  480, 476,  478, 476,  480, 474,  480, 474,  480, 474,  480, 1378,  480, 1376,  480, 474,  458, 1400,  480, 474,  480, 476,  456, 498,  458, 498,  480, 474,  480, 474,  458, 496,  478, 1378,  476, 478,  480, 1376,  458, 1398,  458, 1398,  458, 1398,  480, 1378,  480, 476,  478, 474,  480, 474,  456, 1400,  480, 474,  478, 474,  480, 474,  430};  // SHARP_AC
uint8_t state[13] = {0xAA, 0x5A, 0xCF, 0x10, 0xCC, 0x31, 0x32, 0x00, 0x08, 0x80, 0x05, 0xF4, 0x11};
```
### Medium
```
Protocol  : SHARP_AC
Code      : 0xAA5ACF10CC315200088005F471 (104 Bits)
Mesg Desc.: Model: 3 (A903), Power: On, Mode: 2 (Cool), Temp: 27C, Fan: 5 (UNKNOWN), Turbo: Off, Swing(V) Toggle: Off, Ion: On, Econo: -, Clean: Off
uint16_t rawData[211] = {3824, 1902,  458, 496,  482, 1376,  458, 496,  480, 1376,  480, 474,  456, 1400,  480, 474,  480, 1376,  458, 498,  480, 1376,  458, 496,  480, 1376,  458, 1400,  458, 496,  480, 1376,  458, 496,  458, 1398,  456, 1400,  480, 1378,  456, 1400,  458, 496,  480, 476,  458, 1398,  480, 1378,  478, 474,  458, 496,  480, 474,  480, 474,  480, 1376,  458, 496,  458, 496,  480, 474,  480, 474,  480, 474,  482, 1376,  482, 1376,  456, 498,  480, 474,  480, 1376,  482, 1376,  480, 1376,  456, 498,  480, 474,  480, 474,  458, 1398,  456, 1398,  458, 498,  458, 496,  480, 474,  456, 1400,  480, 474,  480, 474,  480, 1374,  482, 474,  480, 1376,  480, 474,  478, 476,  480, 476,  480, 474,  478, 476,  480, 474,  482, 472,  480, 476,  480, 474,  462, 492,  480, 474,  480, 474,  478, 1378,  458, 496,  480, 474,  480, 474,  456, 498,  458, 496,  478, 476,  482, 474,  478, 476,  478, 476,  480, 474,  480, 474,  478, 1378,  482, 1374,  456, 498,  480, 1376,  478, 476,  456, 498,  480, 474,  456, 498,  480, 474,  480, 476,  478, 474,  480, 1376,  480, 476,  478, 1378,  456, 1400,  458, 1400,  478, 1378,  456, 1402,  476, 476,  478, 476,  456, 498,  454, 1402,  454, 1402,  480, 1378,  454, 500,  456};  // SHARP_AC
uint8_t state[13] = {0xAA, 0x5A, 0xCF, 0x10, 0xCC, 0x31, 0x52, 0x00, 0x08, 0x80, 0x05, 0xF4, 0x71};
```
### High
```
Protocol  : SHARP_AC
Code      : 0xAA5ACF10CC317200088005F451 (104 Bits)
Mesg Desc.: Model: 3 (A903), Power: On, Mode: 2 (Cool), Temp: 27C, Fan: 7 (High), Turbo: Off, Swing(V) Toggle: Off, Ion: On, Econo: -, Clean: Off
[... cut ...] 88, 484,  492, 482,  510, 464,  486, 488,  492, 478,  482, 1396,  404, 564,  466, 506,  440, 532,  408, 564,  442, 532,  436, 536,  482, 492,  410, 562,  412, 560,  408, 564,  412, 560,  408, 1468,  412, 1460,  488, 486,  480, 1390,  484, 488,  482, 490,  492, 482,  492, 480,  482, 492,  482, 490,  482, 490,  482, 1398,  430, 538,  434, 534,  496, 1378,  486, 1384,  412, 1462,  488, 1386,  488, 486,  484, 488,  484, 488,  484, 1388,  482, 490,  484, 488,  492, 482,  492};  // UNKNOWN DD5EE4E
```

crankyoldgit, 2021-01-20:
Okay, I've made some more changes to attempt to address most of those issue. Please re-download the branch and let me know how this goes.

GerardNL, 2021-01-21:
> In the meantime, can you capture a "OFF" message from the remote (when the A/C unit is on) and then an "ON" (when the unit is off).

Hi David, sorry for the slow reply. Work got in the way of more fun things. Glad to have another Sharp AC user in the thread. 

Below is what captured by the MQT console on the IR module, when using the original remote of the AC.

```
Turn AC on (from Off):
{"IrReceived":{"Protocol":"SHARP_AC","Bits":104,"Data":"0xAA5ACF10CA112200088000F051","Repeat":0,"IRHVAC":{"Vendor":"SHARP_AC","Model":1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":25,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}

Turn AC of (from Onn):
 {"IrReceived":{"Protocol":"SHARP_AC","Bits":104,"Data":"0xAA5ACF10CA212200088000F061","Repeat":0,"IRHVAC":{"Vendor":"SHARP_AC","Model":1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":25,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
```

So from that, it looks like whatever is translating the code, seems to understand "Power:On" and "Power:Off"

Anyway. Maybe it's already too late for this information. Let me read the rest of the thread. Seems interesting...

GerardNL, 2021-01-21:
You didn't ask, and maybe not too usefull as I can't (don't know how to) tell you the A9xx model number, just the model number on the box: AH-A12REVP-1. But just in case...
> I need to know a bunch more about this model.
> eg. What features are reported incorrectly? (and hex codes for those that don't + description)
Only the on/off issue. The model is so basic (no light, no ion, no econo, no kids/comfort modes)
> Does this model have a light/LED button? What about Econo mode?
None of these 2 features
> What is the Full model number for the A/C unit & the remote please?
For the remote: CRMC-A863JBEZ

crankyoldgit, 2021-01-21:
Thanks, I'll add that soon assuming the new version works for you.

You should try compiling Tasmota with the version of the library in that branch and see how it goes for you.

lenkunz, 2021-01-22:
> Okay, I've made some more changes to attempt to address most of those issue. Please re-download the branch and let me know how this goes.

Sorry for being late. My work got in my way.
Turn off and Turn on working like a charm now.

I still can't neither test if temperature setting nor mode is right. But it seems right.

There is still fan setting problem.
It seems like Medium setting works exactly like low setting. (or Auto?)

crankyoldgit, 2021-01-22:
> Turn off and Turn on working like a charm now.

Excellent. Thanks for confirming that.

> There is still fan setting problem.
It seems like Medium setting works exactly like low setting. (or Auto?)

I _think_ I know what might be causing that. I've hopefully just added a fix for that to that branch. Please download/recompile etc and let me know how it goes.

lenkunz, 2021-01-22:
> I _think_ I know what might be causing that. I've hopefully just added a fix for that to that branch. Please download/recompile etc and let me know how it goes.

I can confirm that fan command work correctly now for my remote control. ^^

GerardNL, 2021-01-22:
> Thanks, I'll add that soon assuming the new version works for you.
> 
> You should try compiling Tasmota with the version of the library in that branch and see how it goes for you.

That was an invitation for a deep dive into a new environment. A bit of trial and error, but after some reading, I managed to compile Tasmota with your new source code included. 

I didn't doubt your skills, but honestly didn't expect I got the compiling right and was prepared to see that I bricked the IR module.  But a big surpise: It worked on the first try!! So now the AC is responding to the ON and OFF commands send from HA, in the same way the AC of another brand does.

I'm very grateful for your support and pleased to see this is such an active community.

I hope the changes will make into the official branch, although there might be few users with this particular AC.

Once again, thank you so much and let me know if I can further contribute to this particular subject. The small parts are still on order, but I guess we'll not need them for this problem after all.

closed 2021-01-22 by crankyoldgit

crankyoldgit, 2021-01-22:
@GerardNL you can thank @lenkunz for providing the data needed, and making your purchase redundant. :-)

Thanks both of you for confirming it's working now.

It's now been merged into our master branch, so it will make it into Tasmota eventually.

crankyoldgit, 2021-02-13:
FYI, the changes mentioned above have now been included in the new v2.7.15 release of the library.

PR #1390 (2021-01-22): SharpAc: Add support for model A903, and improve `IRac` fan & power control.
* Add new model `A903`
  - `A903` appears to be near identical to `A705` model. Make appropriate changes
* Update unit tests accordingly.
* Update supported model info.
* Minor tweaks for this model
* Should allow power control of this model now & improve `IRac` power control of Sharp.
* Fix typo
* Adjust old suggested model for supported devices.
* Enable `IRSharpAc::convertFan()` to report the correct fan speed when passed model info.

Power control (and model detection) confirmed working by users.

Fixes #1387
commit 9499b22f (2021-01-22): SharpAc: Add support for model A903, and improve `IRac` fan & power control. (#1390)

* Add new model `A903`
  - `A903` appears to be near identical to `A705` model. Make appropriate changes
* Update unit tests accordingly.
* Update supported model info.
* Minor tweaks for this model
* Should allow power control of this model now & improve `IRac` power control of Sharp.
* Fix typo
* Adjust old suggested model for supported devices.
* Enable `IRSharpAc::convertFan()` to report the correct fan speed when passed model info.

Power control (and model detection) & Fan speed confirmed working by use [... cut]