crankyoldgit/IRremoteESP8266 #1388: sendSAMSUNG, sendNEC, and probably others too, send bits in reverse order.

egzumer, 2021-01-20:
### Version/revision of the library used
2.7.14

### Describe the bug
Bits received by IRrecv and received by IRsend are in different order

#### To Reproduce
I take a remote, and I push a power button, receiver prints:
`Protocol=SAMSUNG Address=0x707 Command=0x2 Raw-Data=0xFD020707 (32 bits)`
so I take that data and I execute with a sender device:
`irsend.sendSAMSUNG(0xFD020707);`
but the receiver device prints:
`Protocol=SAMSUNG Address=0x40BF Command=0xE0E0 Raw-Data=0xE0E040BF (32 bits)`

### Other useful information
Looks like bits are received LSB first and written that way to integer value,
but _sendSAMSUNG_ calls _sendGeneric_ with _MSBfirst_ set to true so bits are sent in reverse order.
The same is true at least for _sendNEC_, possibly for other types too.

crankyoldgit, 2021-01-21:
Are you sure you're using _our_ library for sending **&** decoding?!
`0xFD020707` is **not** a valid Samsung code for our library. `0xE0E040BF` however, is valid. I also don't recognise the text output. It's not from our IRrecvDumpV2 etc programs. Can you please supply the code if you are using our library?

This library will _not_ decode `irsend.sendSAMSUNG(0xFD020707);` as `SAMSUNG` because:
https://github.com/crankyoldgit/IRremoteESP8266/blob/91846e411de90653b4d611bf8010d491fab27cca/src/ir_Samsung.cpp#L133-L140

i.e. the first (transmitted) 8 bits is repeated/duplicated in the second 8 bits.
e.g. From `0xE0E040BF`, `0xE0` then `0xE0` meets that criteria.
Where as for `0xFD020707`, `0xFD` != `0x02`, thus it fails.

Also, your `address` and `command` are swapped compared to how we report it, also the decoder stores only 8 bit values into both the `address` & `command` values in our library. Your output is showing 16 bit values (larger than `decodeSAMSUNG()` will store in them.

FYI, every change to the library goes through an extensive list of unit tests. This sort of this (sending and decoding having different bit orders) would typically light up with a tonne of errors for **ANY** protocol.
However, just to be certain, I've drafted a specific unit test for your scenario. The library does indeed operate as expected. It decodes the sent value back to the original value. See: https://github.com/crankyoldgit/IRremoteESP8266/commit/c89a7ba22ca63f307eadbafa47d7f60848365bd5
That test passes.

I strongly think the problem is at your end, sorry.

egzumer, 2021-01-21:
I'm sorry, of course you're right. I'm stupid :), I used pro mini at the receiving end which uses IRremote. 0xFD020707 worked for me because I changed MSBfirst to false in sendSAMSUNG. When I look at the avr IRremote now, it also uses LSB for sending, so receiving has to also work in a reverse way, this is where I stumbled. I got fooled because both libraries look so similar. Thank you and sorry again.

egzumer, 2021-01-21:
Well on the other hand I don't know why this is done this way. IR codes originally are sent LSB so why the HEX representation is to be reversed and sent MSB first? How does this make sense? I guess there is no difference in the end, but still.

crankyoldgit, 2021-01-21:
The reason is legacy and backwards compatibility. The original library had it in that order, so in order for people to be able to easily port their code.

The address and command are reversed, to comply with the spec, and there is an encoderSAMSUNG function to convert between the two.

closed 2021-01-21 by crankyoldgit