crankyoldgit/IRremoteESP8266 #1389: crash after .decode()

app-vrb, 2021-01-21:
### Version/revision of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
e.g. v2.0.0, or 'master' as at 1st of June, 2017. etc._
2.7.14
### Describe the bug
A clear and concise description of what the bug is.
After I read a IR when I send a mqtt message it crash.
as you can see the memory available after the new code is detected, become negative (with ESP.getFreeHeap())
(10856) ----1>cmnd:  LEARN
(10816) ----1>received Read IR 1
(10832) ----1>Wait new IR  1
(10728) ----4>len= 59
Raw Timing[59]:
   +  8478, -  4202,    +   580, -  1522,    +   580, -   460,    +   578, -   466, 
   +   580, -   476,    +   580, -   460,    +   556, -  1538,    +   580, -  1522, 
   +   580, -   468,    +   578, -   460,    +   556, -   492,    +   578, -   476, 
   +   580, -   468,    +   580, -   468,    +   580, -   458,    +   580, -  1522, 
   +   580, -   468,    +   580, -   458,    +   580, -  1524,    +   580, -   458, 
   +   580, -  1514,    +   580, -   466,    +   580, -   468,    +   580, -  1522, 
   +   580, -   458,    +   580, -  1522,    +   580, -   466,    +   580, -   458, 
   +   580, -  1522,    +   580

(10688) ----1>Received new IR  59
(-1208) ----1>Message arrived  cmnd/14275430/STATUS
(-1208) ----1>Payload arrived  {'sid':'IR','cmnd':'STATUS'}
(-1272) ----1>mqtt_received:  cmnd/14275430/STATUS
(-1208) ----1>sid:  IR
(-1208) ----1>var:  null
(-1208) ----1>val:  null
(-1208) ----1>cmnd:  STATUS
(-1248) ----1>Received cmnd/.../STATUS IR
(-1248) ----1>Command not recognised 
(-1584) ----1>Publish Status:  [{"sid":"0","last_local_ip":"192.168.1.27","time":"2021-01-21 15:38:22"},{"var":"ERROR","val-s":"cmnd/14275430/STATUS","time":"2021-01-21 15:38:22"}]
(-1784) ----1>Mqtt call   [{"sid":"0","last_local_ip":"192.168.1.27","time":"2021-01-21 15:38:22"},{"var":"ERROR","val-s":"cmnd/14275430/STATUS","time":"2021-01-21 15:38:22"}]

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (9):
epc1=0x40251735 epc2=0x00000000 epc3=0x00000000 excvaddr=0x012200f7 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffed00 end: 3fffffb0 offset: 0190
3fffee90:  00003b58 00000000 4bc6a7f0 00000000  
3fffeea0:  0000270f 3fff48ec 3fff48ec 3fff503c  
3fffeeb0:  00000000 00000000 00000020 3fff4ac4  
3fffeec0:  3fff48ec 3fff865a 3fff863c 40256660  
3fffeed0:  00000014 3fff48ec 00000047 3fff503c  
3fffeee0:  00000000 3fff4fe4 00000020 40101027  
#### To Reproduce
_What steps did you do, and what did or didn't actually happen? How can we reproduce the issue?_

_e.g._
_1. Initialise the IRsend class._
_2. `IRsend.sendFoobar(0xdeadbeef);`_
_3. Foobar BBQ went into Cow(er)-saving mode and fried me a couple of eggs instead._

#### Example code used
_Include all relevant code snippets or links to the actual code files. Tip: [How to quote your code so it is still readable in the report](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)._

#### Expected behaviour
_What steps did you do and what should it have done? A clear and concise description of what you expected to happen._

_e.g._
_1. Initialise the IRsend class._
_2. `IRsend.sendFoobar(0xdeadbeef);`_
_3. Foobar branded BBQ turns on and cooks me some ribs._

#### Output of raw data from [IRrecvDumpV2.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) or [V3](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/) (if applicable)
_Include some serial text of the raw dumps of what the device saw._

_**Note**: Output from Tasmota is not acceptable. We can't easily use their raw format._

### What brand/model IR demodulator are you using?
_Brand/Model code_ or _None_

_If VS1838b: That is likely your problem. Please buy a better one & try again. Reporting a capture or decode issue when you use one of these is effectively wasting our & your time. See the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Help_Im_getting_very_inconsistent_results_when_capturing_an_IR_message_using_a_VS1838b_IR_demodulator)_

### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc. ESP8266 or ESP32 board type._

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes/No._

### Has this library/code previously worked as expected for you?
_Yes/No. If "Yes", which version last worked for you?_

### Other useful information
_More information is always welcome. Be verbose._

NiKiZe, 2021-01-21:
Which code is this from? What should we do to reproduce this issue? And are you sure this issue is from this library, and not some other or something else?

Could you fill in ALL parts of the template or otherwise clean it so it is readable?

app-vrb, 2021-01-21:
Hi, is a big sketch with a lot of parts, is a firmware for Home automation 
and is not easy to reproduce outside the full sketch.
I am sure is related to the infrared read function, because all the rest it 
works, and just when I launch a call to this function it crash.
The flow is like this:
From an app i make a mqtt call to start a function to learn an infrared call
The function start and wait to get a raw code
If receive the code exit and then restart the normal loop
When inside the loop receive an mqtt call it crash and as you see the 
available memory became negative and then crashI don't know what I can 
provide, if for you it make sense i can post the part of the sketch with 
the call to read the IR. Please let me know if you have some other ideas.
Thanks
Francesco

crankyoldgit, 2021-01-21:
Can you provide a link to your code? With out which there is **no way** we can help you.
It really looks like you've got a memory leak/management problem on your side of the code. Almost all the memory heap management/usage is done during the initial setup.
Your log indicates the heap is losing more than 10k. This library (unless instructed too by the caller) doesn't use anywhere near that much heap/dynamic team etc.
This points to the likelihood that memory problem is likely something your code is producing.

crankyoldgit, 2021-01-21:
> (10832) ----1>Wait new IR 1
> (10728) ----4>len= 59
> Raw Timing[59]:
>
> * 8478, -  4202,    +   580, -  1522,    +   580, -   460,    +   578, -   466,
> * 580, -   476,    +   580, -   460,    +   556, - [...]

Also, your log indicates you've already received the IR message successfully and the heap space is still intact. Your problem send to be what ever you are doing _after_ you print `Received new IR 59`.
I.e. the library had already given you you're message.

app-vrb, 2021-01-22:
Hi, I will try to isolate the problem, to make possible for you to test. If 
you want I can send also a private link or a zipped file with the complete 
project, but it will be more complicated to test.

crankyoldgit, 2021-01-22:
Do you have a link to it on github?

crankyoldgit, 2021-01-22:
FWIW, I just went over the code for the `IRrecv::decode()` function. There is **no** dynamic memory or heap allocation in that function.
https://github.com/crankyoldgit/IRremoteESP8266/blob/91846e411de90653b4d611bf8010d491fab27cca/src/IRrecv.cpp#L463-L502

The only memory allocation is done in the `IRrecv` object constructor. ie. at initialisation/instantiation time.
see:
https://github.com/crankyoldgit/IRremoteESP8266/blob/91846e411de90653b4d611bf8010d491fab27cca/src/IRrecv.cpp#L231-L298

If there isn't enough memory, it fails and will drop a debug message (if enabled) and reboot the ESP. i.e. It won't continue. You certainly won't get to do a `.decode()` call if that's the case, as the esp would have rebooted a long time ago.

I'm almost positive that the heap exhaustion is something _your_ code is doing. Not this library's. I suggest you simplify your code to track down where your memory problem is, and/or look at the example code in this library to make sure you're using it correctly.

FYI, we are not going to debug/look for the memory leak in your code. We'll provide help/assistance in using the library correctly etc. but the evidence/data you've provided is not **immediately** before & after a `IRrecv::decode()` call, so it could be anything in your code. The `decode()` function etc is used a lot by popular large tightly managed memory monitored projects (e.g. Tasmota etc). We would likely already have been warned if there was a memory leak in something so core. i.e. Another reason to think the problem is your code, not ours.

app-vrb, 2021-01-22:
Ok I will cut as much as possible my code and we will see. My feeling is a 
problem with other libraries. Anyway I will let you know.
Thanks for the moment.
Francesco

app-vrb, 2021-01-26:
For the moment I replaced with the ac library, and I don't use the 
receiver, because I was not able to change pin at runtime, or better I am 
able, but drain too much memory and after a while crash.
I need to use the receiver to replicate ac remote but I have just max 2 
kbyte available.
Please let me know if you have suggestions.
One question regarding the IRac, are you adding more type of models?

crankyoldgit, 2021-01-26:
> I don't use the receiver, because I was not able to change pin at runtime

The pin can be changed at runtime. It's set every time the `IRrecv()` object is instantiated. Just delete the object, and create/initialise a new one.

> I need to use the receiver to replicate ac remote but I have just max 2 kbyte available.

Without knowing what you are doing exactly, there is no way I can help you.
All I can do is suggest reducing the receiving buffer, perhaps don't use the [`save_buffer`](https://github.com/crankyoldgit/IRremoteESP8266/blob/0d444c909aa65222bae69335e4ff72ace7dc2e16/src/IRrecv.cpp#L239-L240) option to reduce memory, or disable protocols you don't require. See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/0d444c909aa65222bae69335e4ff72ace7dc2e16/src/IRremoteESP8266.h#L84-L87

On average every A/C protocol you disable saves you about 1-2K of flash space.

> One question regarding the IRac, are you adding more type of models?

Always. See the [Release Notes](https://github.com/crankyoldgit/IRremoteESP8266/releases).

app-vrb, 2021-01-29:
Hi David thanks for your answer.

I am able to change the pin in setup for IRsend with:

      irsend= new IRsend(PIN_IR_OUT);
      irsend->begin();

but I am not able to change the pin for IRac ac(PIN_IR_OUT);

Can you help me?

crankyoldgit, 2021-01-29:
Exactly the same way.

app-vrb, 2021-01-30:
I am trying like this:

in the definition:
IRac ac(PIN_IR_OUT);

in setup:
    ac = new IRac(PIN_IR_OUT);
    ac->begin();
but I get errors:
main_setup:230:7: error: no match for 'operator=' (operand types are 'IRac' and 'IRac*')
     ac=new IRac(PIN_IR_OUT);
       ^
/Users/francescogren/Desktop/__WorkingFolder/iDomus_0_4_fw/main_setup.ino:230:7: note: candidates are:
In file included from /Users/francescogren/Desktop/__WorkingFolder/iDomus_0_4_fw/iDomus_0_4_fw.ino:44:0:
/Users/francescogren/Documents/Arduino/libraries/IRremoteESP8266/src/IRac.h:49:7: note: IRac& IRac::operator=(const IRac&)
 class IRac {

Or if in the definition I use :
IRac *ac=NULL;

and in setup :
    ac = new IRac(PIN_IR_OUT);
    ac->begin();

I get 
main_setup:231:9: error: 'class IRac' has no member named 'begin'
     ac->begin();
         ^

probably I am doing something wrong.
Thanks

crankyoldgit, 2021-01-30:
> Or if in the definition I use :
> IRac *ac=NULL;
> 
> and in setup :
> ac = new IRac(PIN_IR_OUT);
> ac->begin();
> 
> I get
> main_setup:231:9: error: 'class IRac' has no member named 'begin'
> ac->begin();

That is because the [IRac class](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRac.html) doesn't have a `begin()` method.
See the available methods [here](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRac.html).

P.S. Remember to `delete` the `ac` object when you're done with it, otherwise you will leak memory from the heap.
But that's a [general C++ programming thing, which we don't provide help with here](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me).

app-vrb, 2021-01-30:
Hi, I am sorry to bother you, but my knowledge of C++ is limited and I 
can't find the way to set the pin.

Can you please send me the lines to put in the code in the definition 
and in the setup.

Thanks for your help.

crankyoldgit, 2021-01-30:
> Hi, I am sorry to bother you, but my knowledge of C++ is limited and I can't find the way to set the pin. Can you please send me the lines to put in the code in the definition and in the setup.

https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me

I haven't tested this code, nor will I, but it'd be something like this:

```
IRac *ac = NULL;

void loop(void) {
  // your other code
  if (do_I_need_to_change_the_output_gpio()) {
    if (ac != NULL) {
      delete ac;  // Remove the old object.
    } else {
      ac = new IRac(new_gpio_pin);
    }
  }
  // Do what ever ac stuff etc.
}
```

app-vrb, 2021-01-30:
Last easy question (easy for you) now ac is a pointer and I get error when I compile on:

ac.next.protocol .....

crankyoldgit, 2021-01-30:
You _**really**_ need to go learn the basics of C/C++.
Especially, pointers & memory management.
Try `ac->next.protocol`

Marking this issue closed, as there is no sign/evidence that there is a memory problem inherent with the library.

closed 2021-01-30 by crankyoldgit