crankyoldgit/IRremoteESP8266 #1391: Panasonic MKE

pilaGit, 2021-01-22:
### Version/revision of the library used
Tasmota v9.2.0 with IRremoteESP8266 library v2.7.13.

### Describe the bug
Not a bug but a Panasonic MKE not listed as supported.

What exactly is this Model:3 field doing?

When the MKE IR command is read after being sent by the original remote, Model is listed as 0 (zero). All other fields seem correct.

I currently use Panasonic DKE by sending Model:3 and DKEs work flawlessly. Now I need to manage Panasonic MKE. They do have many additional features (10+) but first test indicate that they respond well when I send the same command as for the DKE.

When I compare the (Tasmota) IrReceived#Data strings as read from DKE and MKE remotes, they look the same (using the same settings). I do not have problems decoding completely all the settings for MKE properly, and publishing them here, that is not an issue.

What exactly is this Model:3 field doing? I tried sending with Model:0 but it seems MKE has ignored such attempts (they were apparently sent as Model:-1).

I will now try to experiment and run MKEs for couple of days using Model:3 and try to verify if it truly works (without any additional features). Any other recommendations?

crankyoldgit, 2021-01-22:
> What exactly is this Model:3 field doing?

It's setting what model type to use with in the [`IRPanasonicAC` class object](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRPanasonicAc.html). It's setting the class object to operate/generate IR message for a DKE Panasonic mode.
See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/91846e411de90653b4d611bf8010d491fab27cca/src/IRsend.h#L144-L153
&
https://github.com/crankyoldgit/IRremoteESP8266/blob/91846e411de90653b4d611bf8010d491fab27cca/src/ir_Panasonic.h#L7-L27

You'll have to look through out the code in https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Panasonic.cpp to see what exactly differs between each supported model.

> When I compare the (Tasmota) IrReceived#Data strings as read from DKE and MKE remotes, they look the same (using the same settings). I do not have problems decoding completely all the settings for MKE properly, and publishing them here, that is not an issue.

You can follow this guide to help you decode the MKE settings.
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

> I tried sending with Model:0 but it seems MKE has ignored such attempts (they were apparently sent as Model:-1).

Attempting to change to using unsupported model numbers will be ignored. This may manifest in some other ways in Tasmota.

Btw. We do not provide Tasmota support directly.  Tasmota also doesn't use an interface to this library that offers the full functionality of the [`IRPanasonicAC` class](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRPanasonicAc.html). Hence I suggest you use our tools to see what is/isn't supported for a given model.

> Not a bug but a Panasonic MKE not listed as supported.

Looking forward to adding `MKE` as a supported device for Panasonic when you tell me what suits it best or when you've worked out the differences to other models, and what the exact model numbers are for the A/C and remote you have.

pilaGit, 2021-01-22:
OK, thanks for your answer. I will do my best to investigate ad understand. Understanding new code made by others is not a quick task :(

Some time ago, with your kind help, we improved DKE. So, I believe we can do MKE quite well :)

As for Tasmota, I use it since my SmartHome (my own program) uses 20+ a controllers per site and several RPis. IR is just a part of it. I do not ask for Tasmota support, not a problem. Tasmota authors mostly ignore users, I have found, reported and worked around over 15 clearly defined Tasmota bugs, they did not fix a single one in over a year.

I will try some more precise MKE experimenting over the weekend. Keep this issue open :)

crankyoldgit, 2021-01-23:
> Keep this issue open :)

Sure thing. Good luck.

crankyoldgit, 2021-02-02:
Friendly follow up. How did you go?

crankyoldgit, 2021-03-01:
@pilaGit 
> I will try some more precise MKE experimenting over the weekend. Keep this issue open :)

It's been over a month now. I'll close this issue if I don't hear from you soon.

crankyoldgit, 2021-03-09:
Closing due to inactivity. I'll reopen if you respond.

closed 2021-03-09 by crankyoldgit