crankyoldgit/IRremoteESP8266 #1397: Support for EcoClim

mothra78, 2021-01-31:
Hi,

Thanks a lot for your great library and advice it helped me a lot to reverse engineer the remote of my A/C.
I have one remote per room to set the desired temperature.

I have identified 2 type of frames : 
one when I press a button on the remote which send 56Bit of data in sequence of three identical bursts
an other one when I leave the remote alone it sends peridically 15 Bits of data in sequence of three identical bursts.

I have created a spreadsheet with all the information I collected to date including most of the data from the 56Bits (Ambient Temp and Programmed Temp).

Could you help in developing a decode function so I could speed up the identification of the data bits?

mothra78, 2021-01-31:
Here is the link to the Google sheet : https://docs.google.com/spreadsheets/d/1VtUXCbPTFZkH3ZEyITdRSaQ6oIVyTIGd0P-AQXJGxds/edit?usp=sharing

NiKiZe, 2021-01-31:
Could you make sure that the permissions on the sheet is so that anyone withe link can view it?

Also if you could give a raw capture in the issue on each type of frame to easily view it, that would be great.

And as we always request, the full name and model of both the remote, and the unit. Thanks!

crankyoldgit, 2021-01-31:
> one when I press a button on the remote which send 56Bit of data in sequence of three identical bursts

That's not uncommon.

> an other one when I leave the remote alone it sends peridically 15 Bits of data in sequence of three identical bursts.

Typically this is the remote telling the A/C what the current temperature is at the remote, rather than measured at the A/C.
We've got a few protocols that do this.

In order to get things started, we are going to need some raw captures using `IRrecvDumpV2` or later.

See:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol & https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

mothra78, 2021-01-31:
So access on the file is changed.

For the frames I have a 10 of each decoded in the file but basically the command frame looks like this : 

Timestamp : 000011.173
--
Library   : v2.7.14
 
Protocol  : UNKNOWN
Code      : 0x842242BF (172 Bits)
uint16_t rawData[343] = {5834, 1950,  482, 580,  506, 614,  480, 612,  456, 1738,  482, 562,  532, 582,  508, 608,  430, 1760,  456, 642,  456, 608,  484, 638,  458, 634,  456, 634,  456, 1734,  482, 1704,  402, 690,  456, 638,  404, 1786,  458, 1730,  430, 1762,  454, 638,  456, 636,  456, 1732,  426, 1764,  426, 1788,  400, 692,  402, 1764,  452, 638,  430, 1760,  424, 1786,  374, 1818,  402, 690,  374, 1786,  424, 1796,  402, 1758,  426, 1790,  376, 1782,  426, 1766,  400, 1810,  398, 1796,  400, 1788,  428, 1734,  398, 1814,  400, 1762,  470, 1742,  400, 1786,  398, 1794,  400, 1762,  398, 718,  400, 1792,  400, 1788,  400, 1788,  400, 694,  400, 694,  402, 1788,  398, 664,  5720, 1944,  426, 642,  450, 696,  442, 650,  396, 1794,  468, 602,  422, 642,  448, 696,  442, 1744,  392, 702,  392, 678,  420, 700,  394, 700,  464, 628,  466, 1720,  464, 1726,  462, 628,  464, 630,  464, 1728,  438, 1752,  462, 1722,  464, 636,  438, 626,  464, 1722,  490, 1698,  488, 1722,  464, 628,  466, 1724,  466, 626,  464, 1724,  464, 1724,  462, 1732,  462, 626,  464, 1726,  464, 1724,  464, 1722,  464, 1732,  464, 1690,  490, 1724,  464, 1726,  464, 1728,  464, 1728,  462, 1690,  492, 1728,  462, 1724,  464, 1726,  464, 1728,  464, 1720,  462, 1736,  460, 604,  490, 1718,  464, 1730,  462, 1720,  462, 630,  464, 628,  462, 1734,  462, 600,  5632, 2028,  490, 630,  464, 632,  464, 630,  462, 1724,  462, 630,  462, 636,  462, 602,  488, 1728,  464, 630,  462, 632,  464, 630,  462, 628,  462, 630,  464, 1730,  460, 1724,  464, 630,  464, 630,  462, 1728,  464, 1724,  462, 1728,  462, 630,  464, 630,  462, 1724,  464, 1728,  460, 1730,  462, 628,  460, 1732,  464, 602,  492, 1722,  464, 1726,  460, 1726,  464, 632,  464, 1696,  488, 1728,  460, 1732,  462, 1728,  462, 1694,  488, 1728,  462, 1724,  464, 1732,  460, 1700,  490, 1728,  462, 1694,  488, 1730,  462, 1720,  462, 1728,  464, 1726,  462, 1726,  460, 632,  464, 1724,  462, 1726,  460, 1730,  464, 630,  464, 632,  464, 1728,  462, 596,  7862};  // UNKNOWN 842242BF

and the status frame looks like this : 

Timestamp : 000031.942
--
Library   : v2.7.14
 
Protocol  : UNKNOWN
Code      : 0xC0071146 (49 Bits)
uint16_t rawData[97] = {5778, 1940,  458, 662,  432, 664,  428, 666,  430, 1756,  430, 664,  430, 664,  430, 1756,  430, 664,  428, 668,  428, 1758,  430, 1760,  428, 1758,  430, 668,  430, 1756,  430, 1768,  5672, 1988,  456, 610,  456, 658,  430, 670,  456, 1732,  430, 610,  506, 636,  456, 1738,  458, 636,  482, 584,  506, 1708,  430, 1758,  482, 1708,  456, 638,  430, 1754,  456, 1738,  5674, 1982,  430, 668,  458, 632,  456, 638,  456, 1736,  450, 646,  480, 588,  482, 1728,  426, 666,  428, 666,  426, 1764,  426, 1760,  454, 1744,  452, 608,  452, 1762,  424, 1766,  7770};  // UNKNOWN C0071146

mothra78, 2021-01-31:
Remote is EcoClim - HYSFR-P348 and the unit is "controleur Ecoclim ZC200DPO" it's a french model

mothra78, 2021-01-31:
Overview sheet shows you the two type of frames
RawCmd and RawStatusTemp shows the IRrecvDumpV2 Dump and my take on translation to 56/15 Bits
IR Cmd Code Table and IR Status Code Table is work in progress to find what the Bytes mean. I already identified BYTE 1 as Ambient Temp and BYTE 2 as Programmed Temp. I still need to figure out the rest.

mothra78, 2021-01-31:
RawDump Sheet contains RAW data from IRrecvDumpV2

mothra78, 2021-01-31:
Each of the three bursts starts with a 5800 | 1945 us sequence.
Binary 0 seems to be 480 | 605
Binary 1 seems to be 480 | 1700
And finally each sequence Stops with 7820 us. 
All times are averages

crankyoldgit, 2021-01-31:
Hey @mothra78 , I've whipped up some code that might aid you. It should send/decode those 56bit messages.

Please download the code in the EcoClim branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/EcoClim and let me know how it goes?
I'll take a look at your spreadsheet and the shorter message soon.

mothra78, 2021-01-31:
Great Thanks I'll have a look and provide some feedback

crankyoldgit, 2021-01-31:
I've updated that branch just now. It should also support the 15 bit messages too.

mothra78, 2021-01-31:
Thanks it will help me a lot. I have tried and basic support works. I still get a few UNKOWN frames : 

Timestamp : 000005.961
Library   : v2.7.14

Protocol  : ECOCLIM
Code      : 0x12537503FFFF0A (56 Bits)
uint16_t rawData[343] = {5856, 1920,  506, 592,  480, 614,  506, 588,  480, 1712,  478, 584,  534, 584,  506, 1684,  506, 584,  506, 590,  506, 1686,  480, 608,  480, 1710,  504, 560,  508, 618,  478, 1706,  504, 1690,  454, 638,  506, 1656,  508, 1710,  506, 1678,  480, 610,  508, 1688,  482, 610,  480, 1706,  456, 636,  456, 644,  456, 634,  482, 612,  504, 590,  454, 640,  458, 1726,  456, 1736,  456, 1732,  482, 1710,  456, 1734,  456, 1726,  480, 1710,  456, 1734,  456, 1728,  456, 1738,  456, 1706,  482, 1730,  456, 1734,  456, 1730,  480, 1706,  456, 1734,  482, 1708,  456, 1734,  458, 640,  428, 636,  480, 636,  456, 640,  480, 1712,  480, 612,  456, 1704,  508, 616,  5726, 1906,  482, 618,  502, 610,  480, 618,  480, 1708,  480, 614,  456, 632,  480, 1712,  456, 638,  456, 636,  480, 1708,  480, 612,  480, 1708,  482, 610,  482, 612,  480, 1710,  480, 1712,  480, 584,  506, 1710,  456, 1734,  506, 1652,  508, 612,  482, 1712,  480, 610,  456, 1732,  480, 620,  480, 612,  456, 632,  482, 612,  482, 612,  508, 592,  480, 1702,  482, 1708,  480, 1710,  508, 1682,  482, 1706,  482, 1704,  480, 1710,  506, 1682,  456, 1734,  480, 1708,  506, 1678,  480, 1706,  482, 1688,  504, 1712,  482, 1700,  480, 1706,  506, 1684,  506, 1682,  482, 616,  480, 612,  480, 610,  482, 618,  480, 1680,  508, 610,  482, 1710,  482, 606,  5674, 1966,  508, 612,  480, 614,  504, 590,  506, 1652,  510, 612,  480, 610,  480, 1708,  482, 612,  480, 614,  480, 1706,  482, 612,  482, 1708,  482, 612,  480, 614,  480, 1708,  506, 1682,  480, 610,  482, 1710,  482, 1706,  508, 1678,  482, 616,  480, 1708,  482, 612,  482, 1702,  482, 614,  482, 612,  482, 610,  480, 612,  482, 614,  480, 614,  482, 1706,  506, 1684,  480, 1704,  482, 1712,  480, 1710,  480, 1702,  482, 1706,  482, 1706,  482, 1708,  506, 1686,  480, 1704,  482, 1706,  506, 1684,  482, 1710,  482, 1704,  506, 1678,  508, 1680,  506, 1682,  508, 592,  506, 586,  506, 588,  506, 588,  482, 1708,  508, 582,  508, 1684,  484, 604,  7856};  // ECOCLIM 12537503FFFF0A
uint64_t data = 0x12537503FFFF0A;

Timestamp : 000016.954
Library   : v2.7.14

Protocol  : ECOCLIM
Code      : 0x12547503FFFF02 (56 Bits)
uint16_t rawData[343] = {5830, 1894,  560, 560,  508, 584,  532, 560,  510, 1682,  510, 562,  556, 562,  510, 1676,  532, 560,  534, 558,  532, 1654,  532, 562,  508, 1682,  510, 580,  510, 1682,  510, 584,  508, 586,  508, 584,  484, 1700,  510, 1678,  514, 1654,  556, 560,  534, 1658,  508, 586,  508, 1682,  508, 588,  532, 558,  508, 586,  508, 584,  510, 586,  508, 582,  532, 1658,  508, 1676,  510, 1684,  508, 1652,  560, 1654,  508, 1682,  508, 1678,  510, 1678,  508, 1682,  508, 1682,  508, 1680,  508, 1678,  532, 1660,  508, 1674,  508, 1682,  508, 1680,  508, 1676,  510, 1688,  506, 554,  560, 562,  508, 580,  508, 586,  508, 586,  508, 586,  508, 1682,  510, 584,  5672, 1968,  534, 554,  536, 584,  510, 584,  510, 1684,  508, 582,  508, 586,  508, 1686,  508, 582,  508, 586,  508, 1678,  508, 592,  508, 1674,  508, 588,  508, 1680,  508, 588,  508, 588,  508, 556,  536, 1682,  508, 1680,  506, 1678,  508, 586,  510, 1686,  508, 582,  510, 1674,  508, 586,  510, 586,  508, 584,  508, 588,  508, 586,  508, 584,  510, 1678,  510, 1682,  508, 1676,  508, 1686,  508, 1678,  508, 1684,  504, 1652,  534, 1682,  508, 1678,  508, 1686,  508, 1676,  506, 1682,  510, 1680,  508, 1682,  508, 1676,  506, 1682,  482, 1708,  506, 1680,  508, 590,  508, 558,  534, 586,  508, 586,  482, 608,  508, 594,  480, 1702,  482, 616,  5726, 1912,  534, 586,  482, 612,  508, 582,  508, 1682,  506, 592,  484, 610,  506, 1676,  482, 610,  484, 618,  482, 1680,  534, 586,  480, 1708,  480, 610,  508, 1684,  480, 614,  506, 586,  506, 586,  506, 1684,  480, 1710,  482, 1710,  478, 612,  504, 1688,  454, 642,  452, 1728,  506, 612,  482, 588,  478, 620,  480, 610,  480, 634,  482, 592,  506, 1680,  506, 1684,  504, 1684,  478, 1712,  478, 1710,  480, 1712,  478, 1706,  478, 1712,  478, 1712,  478, 1712,  480, 1708,  478, 1736,  458, 1724,  456, 1710,  476, 1714,  480, 1736,  456, 1704,  452, 1766,  454, 584,  504, 640,  480, 612,  482, 612,  456, 644,  454, 612,  482, 1728,  428, 664,  7822};  // ECOCLIM 12547503FFFF02
uint64_t data = 0x12547503FFFF02;

Timestamp : 000024.961
Library   : v2.7.14

Protocol  : ECOCLIM
Code      : 0x12557503FFFF0A (56 Bits)
uint16_t rawData[343] = {5754, 1936,  562, 582,  484, 590,  480, 638,  508, 1678,  484, 612,  508, 562,  530, 1658,  532, 580,  484, 614,  482, 1684,  508, 576,  536, 1682,  484, 608,  508, 1682,  508, 560,  532, 1684,  510, 584,  508, 1680,  482, 1702,  508, 1682,  508, 586,  508, 1682,  508, 588,  510, 1678,  508, 590,  508, 554,  534, 584,  508, 592,  506, 560,  560, 534,  536, 1674,  508, 1682,  484, 1702,  510, 1682,  508, 1676,  510, 1680,  534, 1656,  508, 1684,  510, 1674,  506, 1682,  532, 1630,  534, 1682,  532, 1658,  532, 1654,  510, 1676,  532, 1658,  532, 1658,  534, 1650,  508, 586,  534, 564,  508, 586,  508, 584,  508, 1678,  508, 586,  532, 1638,  530, 552,  5780, 1888,  534, 590,  508, 576,  510, 588,  506, 1678,  508, 584,  508, 586,  482, 1708,  508, 586,  508, 582,  508, 1686,  508, 586,  510, 1676,  508, 582,  508, 1682,  532, 560,  508, 1680,  508, 586,  512, 1676,  510, 1676,  508, 1680,  510, 586,  508, 16 [... cut ...]   424, 1742,  420, 696,  424, 1766,  394, 704,  452, 1732,  392, 680,  420, 698,  454, 642,  424, 646,  450, 666,  454, 1736,  394, 1792,  394, 1794,  396, 1794,  428, 1770,  448, 1708,  446, 1768,  392, 1798,  394, 1770,  450, 1744,  442, 1736,  418, 1796,  394, 1796,  390, 1800,  438, 1746,  390, 1776,  416, 1768,  420, 1772,  418, 1774,  416, 682,  472, 614,  420, 674,  446, 674,  450, 1718,  414, 700,  394, 1796,  392, 678,  7824};  // ECOCLIM 15597507FFFF0A
uint64_t data = 0x15597507FFFF0A;

mothra78, 2021-01-31:
For the 56Bit Frame :  
Byte1 is ambient Temp
Byte2 is Programmed Temp
First4 Bits of Byte3 are the shutter? (Volet in French) Last 4 Bits TBD
Byte4 is a Timestamp
Byte5 is 0xFF
Byte6 is 0xFF
Byte7 is TBD

mothra78, 2021-01-31:
Byte 7 first 4 Bits are depending on Slave/Master configuration : 0x7 Slave 0x0 Master

crankyoldgit, 2021-02-01:
> Thanks it will help me a lot. I have tried and basic support works. I still get a few UNKOWN frames :

I've increased the tolerance on matching by 5%, it makes it match at least one of the `UNKNOWN` examples you provided.
(Branch has been updated.)

crankyoldgit, 2021-02-01:
> For the 56Bit Frame :
> Byte1 is ambient Temp
> Byte2 is Programmed Temp
> First4 Bits of Byte3 are the shutter? (Volet in French) Last 4 Bits TBD
> Byte4 is a Timestamp
> Byte5 is 0xFF
> Byte6 is 0xFF
> Byte7 is TBD

Got the operating mode working out yet? i.e. Cool, Heat, Fan, Auto. etc

crankyoldgit, 2021-02-01:
> > For the 56Bit Frame :
> > Byte1 is ambient Temp
> > Byte2 is Programmed Temp
> > First4 Bits of Byte3 are the shutter? (Volet in French) Last 4 Bits TBD
> > Byte4 is a Timestamp
> > Byte5 is 0xFF
> > Byte6 is 0xFF
> > Byte7 is TBD
> 
> Got the operating mode working out yet? i.e. Cool, Heat, Fan, Auto. etc

Also the power on/off bit?!

mothra78, 2021-02-01:
Thanks, I have detailed the meaning of each byte/bit in the G sheet ( IR
Cmd Code Table). ON/OFF is part of byte 3.
Can you implement the decoding based on this? If not what else can I
provide?

Thanks again for your help.

crankyoldgit, 2021-02-01:
I'll tackle the "detailed" decoding after we've nailed the basic (hex) send and decode functions. I.e. It can detect 95+% of your remotes messages **and** you can reproduce the signal (via hex code) and have the A/C accept it etc.

After that, once your confident you can explain why & how **any & every** bit changes in a message, the I'll start coding it up.
eg. If you change temp, is there a checksum, and if so where and how is it calculated. etc

crankyoldgit, 2021-02-07:
Ping. @mothra78 waiting on you to confirm decoding is now better and that sending works on a real device.

mothra78, 2021-02-08:
Yes well noted, will do as soon as I can transmit.

crankyoldgit, 2021-02-13:
FYI, the changes mentioned above have now been included in the new v2.7.15 release of the library.

mothra78, 2021-02-13:
Thanks David I can confirm recieving works fine with the 5% tolerance, I
have decoded 55/56 bits from the first frame and 15/15 from the second. I
will try transmitting tests this weekend.

mothra78, 2021-02-13:
I confirm transmit works fine.
I have made a small test program : 

/* IRremoteESP8266: IRsendDemo - demonstrates sending IR codes with IRsend.
 *
 * Version 1.1 January, 2019
 * Based on Ken Shirriff's IrsendDemo Version 0.1 July, 2009,
 * Copyright 2009 Ken Shirriff, http://arcfn.com
 *
 * An IR LED circuit *MUST* be connected to the ESP8266 on a pin
 * as specified by kIrLed below.
 *
 * TL;DR: The IR LED needs to be driven by a transistor for a good result.
 *
 * Suggested circuit:
 *     https://github.com/crankyoldgit/IRremoteESP8266/wiki#ir-sending
 *
 * Common mistakes & tips:
 *   * Don't just connect the IR LED directly to the pin, it won't
 *     have enough current to drive the IR LED effectively.
 *   * Make sure you have the IR LED polarity correct.
 *     See: https://learn.sparkfun.com/tutorials/polarity/diode-and-led-polarity
 *   * Typical digital camera/phones can be used to see if the IR LED is flashed.
 *     Replace the IR LED with a normal LED if you don't have a digital camera
 *     when debugging.
 *   * Avoid using the following pins unless you really know what you are doing:
 *     * Pin 0/D3: Can interfere with the boot/program mode & support circuits.
 *     * Pin 1/TX/TXD0: Any serial transmissions from the ESP8266 will interfere.
 *     * Pin 3/RX/RXD0: Any serial transmissions to the ESP8266 will interfere.
 *   * ESP-01 modules are tricky. We suggest you use a module with more GPIOs
 *     for your first time. e.g. ESP-12 etc.
 */

#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>

const uint16_t kIrLed = 17;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).

IRsend irsend(kIrLed);  // Set the GPIO to be used to sending the message.

// Send a EcoClim A/C formatted message.
/// Status: Alpha / Completely untested.
/// @param[in] data The message to be sent.
/// @param[in] nbits The number of bits of message to be sent.
/// @param[in] repeat The number of times the command is to be repeated.
//void IRsend::sendEcoclim(const uint64_t data, const uint16_t nbits, const uint16_t repeat) {

#define EcoClimOn  0x14127589FFFF72 // Temp 14 25°c Prog 23°C On Fan Auto DIPConfig 0x7 
#define EcoClimP22 0x11117589FFFF72 // Temp 11 22°c Prog 22°C On Fan Auto DIPConfig 0x7 
#define EcoClimOff 0x1212358AFFFF7A // Temp 14 23°c Prog 23°C OFF Fan Auto DIPConfig 0x7 
#define EcoClim25 0x0A3B // 25°C DIPConfig 0x7 
#define EcoClim23 0x093B // 23°C DIPConfig 0x7 
#define EcoClim21 0x083B // 21°C DIPConfig 0x7 
#define EcoClim19 0x073B // 19°C DIPConfig 0x7 

void setup() {
  irsend.begin();
#if ESP8266
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
#else  // ESP8266
  Serial.begin(115200, SERIAL_8N1);
#endif  // ESP8266
}

void loop() {

  uint64_t data =0;
  
  Serial.println("OFF");
  irsend.sendEcoclim(EcoClimOff, 56, 3);  // 56 bits & 3 repeats
  delay(5000);
  Serial.println("ON");
  irsend.sendEcoclim(EcoClimOn, 56, 3);  // 56 bits & 3 repeats
  delay(10000);
  Serial.println("25");
  irsend.sendEcoclim(EcoClim25, 15, 3);  // 15 bits & 3 repeats
  delay(10000);
  Serial.println("21");
  irsend.sendEcoclim(EcoClim21, 15, 3);  // 15 bits & 3 repeats
  delay(20000);
  Serial.println("19");
  irsend.sendEcoclim(EcoClim19, 15, 3);  // 15 bits & 3 repeats
  delay(20000);
  Serial.println("P22");
  irsend.sendEcoclim(EcoClimP22, 56, 3);  // 56 bits & 3 repeats
  delay(10000);
  Serial.println("21");
  irsend.sendEcoclim(EcoClim21, 15, 3);  // 15 bits & 3 repeats
  delay(20000);
  Serial.println("23");
  irsend.sendEcoclim(EcoClim23, 15, 3);  // 15 bits & 3 repeats
  delay(10000);
  Serial.println("25");
  irsend.sendEcoclim(EcoClim25, 15, 3);  // 15 bits & 3 repeats
  delay(30000);
  Serial.println("OFF");
  irsend.sendEcoclim(EcoClimOff, 56, 3);  // 56 bits & 3 repeats
  delay(5000);
}

crankyoldgit, 2021-02-13:
You shouldn't need the three (3) repeats parameter.
`sendEcoclim(EcoClimOff)` for the 56 bit messages & `sendEcoclim(EcoClim25, 15)` for the.15 bit ones.

sendEcoclim() does the 3x 56 etc internally. Setting 3 repeats, will generate 12x. I.e. 9 more copies then you need.

mothra78, 2021-02-13:
OK thanks

mothra78, 2021-02-13:
Here is what I have decoded (let me know if you need an other format) : 

 union convertorEcoClimCmdFrame
 {
  struct 
  {
      uint64_t fix3 : 3;   // 010 [3 bits]
      uint64_t tbd2 : 1;   // 0.1 [1 bits]
      uint64_t DipConfig   : 4;   // 1..15 [4 bits]
      uint64_t fixed : 16;   //  [16 bits]
      uint64_t timestamp : 8;   // 00..63 [8 bits]
      uint64_t tbd1   : 4;   // 1..15 [4 bits]
      uint64_t fan : 2;   // 010 [2 bits]
      uint64_t on : 2;   // 0.1 [2 bits]
      uint64_t t_prog   : 5;   // 1..31 [5 bits]
      uint64_t mode  : 3;   // 1..12 [4 bits]
      uint64_t t_amb   : 5;   // 1..31 [5 bits]
      uint64_t pre   : 3;   // 1..7 [3 bits]
  } pieces;
  uint64_t whole; 
};

union convertorEcoClimStaFrame
 {
 struct 
 {
    uint16_t end : 3;   // 011 [3 bits]
    uint16_t DipConfig   : 4;   // 1..15 [4 bits] CONFIG
    uint16_t t_amb   : 5;   // 1..31 [5 bits] TEMP
    uint16_t fix3 : 3;   // 000 [3 bits]
} pieces;
uint16_t whole;
 };

mothra78, 2021-02-13:
#define ECOCLIM_CMD_TEMP_MIN 5 // 5°C offset on temperatures

void dump_B(decode_results *results) {
  // Dumps out the decode_results structure.
  // Call this after IRrecv::decode()
  uint16_t count = results->rawlen;
  if (results->decode_type == ECOCLIM) {
    Serial.print("ECOCLIM B: ");
   serialPrintUint64(results->value, 16);
  Serial.print(" (");
  Serial.print(results->bits, DEC);
  Serial.println(" bits)");
  uint64_t u64 = results->value ; 
 
   if (results->bits == ECOCLIM_CMD_BITS) {
    Serial.print("CMD Frame "); // 0x10127551FFFF7A
      union convertorEcoClimCmdFrame data;
      
      data.whole=u64;
      
      unsigned int t1 = data.pieces.t_amb + ECOCLIM_CMD_TEMP_MIN;
      unsigned int t2 = data.pieces.t_prog + ECOCLIM_CMD_TEMP_MIN;
      unsigned int mode = data.pieces.mode;  // 0x0 Slave Heat, 0x1 Master Heat, 0x2 Master Cool
      unsigned int tbd1 = data.pieces.tbd1;
      unsigned int fan = data.pieces.fan;
      boolean on = data.pieces.on;
      unsigned int timestamp = data.pieces.timestamp;
      uint16_t fixed = data.pieces.fixed;
      int DIPConfig = data.pieces.DipConfig ;
      unsigned int tbd2 = data.pieces.tbd2 ;
      
  if (fixed != 0xFFFF) {
    Serial.println("Err Byte 5 & 6");
  }

   Serial.print("Temp A: ");
   Serial.print(t1, DEC);
   Serial.print("°C, Temp P: ");
   Serial.print(t2, DEC);
   Serial.print("°C, fan: ");
   Serial.print(fan, DEC);
   Serial.print(", on: ");
   Serial.print(on);
   Serial.print(", mode: ");
   Serial.print(mode, HEX);
   Serial.print(", tbd1: ");
    Serial.print(tbd1, HEX);
   Serial.print(", time: ");
   Serial.print(timestamp, HEX);
   Serial.print(", DIPConfig: ");
   Serial.print(DIPConfig);
   Serial.print(", tbd2: ");
   Serial.println(tbd2, HEX);
  
  }
  else if (results->bits == ECOCLIM_STA_BITS) {
    Serial.print("STA Frame "); // 0x8BB

    union convertorEcoClimStaFrame data;
      
      data.whole=u64;
      
      unsigned int pre = data.pieces.fix3; // first 3 bits
      unsigned int t1 = data.pieces.t_amb + ECOCLIM_CMD_TEMP_MIN; // next 5 Bits
      int DIPConfig = data.pieces.DipConfig ; // next 4 Bits 
      unsigned int post = data.pieces.end; // first 3 bits

   Serial.print("Temp A: ");
   Serial.print(t1, DEC);
   Serial.print(", pre: ");
   Serial.print(pre, HEX);
   Serial.print(", master: ");
   Serial.print(DIPConfig, HEX);
   Serial.print(", post: ");
   Serial.println(post, HEX);

  }
  else {
    Serial.println("Unkown Frame");
  }
  }
}

mothra78, 2021-02-13:
// EcoClim Modes as per remote pict
const char* EcoClim_Mode[] = { 
  "SLAVE", // 0
  "COOL", // 1
  "SUN",  //2
  "RECYCLE", //3
  "FAN", //4
  "HEAT" //5
  };

// EcoClim Fan as per remote pict
const char* EcoClim_Fan[] = { 
  "Low", // 0
  "Mid", // 1
  "High",  //2
  "Auto", //3
  };

// EcoClim On/Off status
const char* EcoClim_On[] = { 
  "Off", // 0
  "On", // 1
  };

mothra78, 2021-02-13:
Seems Byte 5 & 6 are used to program the night mode. I am starting to explore this

mothra78, 2021-02-13:
And final version with time (current, On and Off)

// EcoClim Modes as per remote pict
const char* EcoClim_Mode[] = { 
  "SLAVE", // 0
  "COOL", // 1
  "SUN",  //2
  "RECYCLE", //3
  "FAN", //4
  "HEAT" //5
   "UNKOWN" //6
  "SLEEP" //7
 };

// EcoClim Fan as per remote pict
const char* EcoClim_Fan[] = { 
  "Low", // 0
  "Mid", // 1
  "High",  //2
  "Auto", //3
  };

// EcoClim On/Off status
const char* EcoClim_On[] = { 
  "Off", // 0
  "On", // 1
  };

 union convertorEcoClimCmdFrame
 {
  struct 
  {
      uint64_t fix3 : 3;   // 010 [3 bits] Fixed 0b010
      uint64_t tbd2 : 1;   // 0.1 [1 bits] Unkown
      uint64_t DipConfig   : 4;   // 0..15 [4 bits] DIP Config as per program DIP inside Remote
      uint64_t offTimeM : 3;   //  [3 bits] Minutes x10 for Off Program from current time
      uint64_t offTimeH : 5;   //  [5 bits] Hours for Off Program from current time
      uint64_t onTimeM : 3;   //  [3 bits] Minutes x10 for On Program from current time
      uint64_t onTimeH : 5;   //  [5 bits] Hours for On Program from current time
      uint64_t time : 12;   //  [12 bits] Current time in minutes
      uint64_t fan : 2;   // 0..3 [2 bits] Fan Speed Low to Auto
      uint64_t on : 1;   // 0.1 [1 bit] On Bit
      uint64_t cl : 1;   // 0.1 [1 bit] Clear Bit
      uint64_t t_prog   : 5;   // 0..31 [5 bits] Programmed Temperature in °C with a 5 degree offset
      uint64_t mode  : 3;   // 0..7 [3 bits] Current Mode from Slave to Heat
      uint64_t t_amb   : 5;   // 0..31 [5 bits] Ambiant Temperature in °C with a 5 degree offset
      uint64_t pre   : 3;   // 1..7 [3 bits] Fixed 0b000
  } pieces;
  uint64_t whole; 
};

union convertorEcoClimStaFrame
 {
 struct 
 {
    uint16_t end : 3;   // 011 [3 bits]
    uint16_t DipConfig   : 4;   // 1..15 [4 bits] DIP Config as per program DIP inside Remote
    uint16_t t_amb   : 5;   // 1..31 [5 bits] Ambiant Temperature in °C with a 5 degree offset
    uint16_t fix3 : 3;   // 000 [3 bits]
} pieces;
uint16_t whole;
 };

mothra78, 2021-02-13:
And the decoding software : 

#define ECOCLIM_CMD_TEMP_MIN 5 // 5°C offset on temperatures

void dump_B(decode_results *results) {
  // Dumps out the decode_results structure.
  // Call this after IRrecv::decode()
  uint16_t count = results->rawlen;
  if (results->decode_type == ECOCLIM) {
    Serial.print("ECOCLIM B: ");
   serialPrintUint64(results->value, 16);
  Serial.print(" (");
  Serial.print(results->bits, DEC);
  Serial.println(" bits)");
  uint64_t u64 = results->value ; 
 
   if (results->bits == ECOCLIM_CMD_BITS) {
    Serial.print("CMD Frame "); // 0x10127551FFFF7A
      union convertorEcoClimCmdFrame data;
      
      data.whole=u64;
      
      unsigned int t1 = data.pieces.t_amb + ECOCLIM_CMD_TEMP_MIN;
      unsigned int t2 = data.pieces.t_prog + ECOCLIM_CMD_TEMP_MIN;
      unsigned int mode = data.pieces.mode;  // 0x0 Slave Heat, 0x1 Master Heat, 0x2 Master Cool
      unsigned int fan = data.pieces.fan;
      boolean on = data.pieces.on;
      boolean cl = data.pieces.cl;
      uint16_t time = data.pieces.time;
      int min = time % 60;
      int hour = time / 60;
      uint8_t onTimeM = data.pieces.onTimeM;
      uint8_t onTimeH = data.pieces.onTimeH;
      uint8_t offTimeM = data.pieces.offTimeM;
      uint8_t offTimeH = data.pieces.offTimeH;
      int DIPConfig = data.pieces.DipConfig ;
      unsigned int tbd2 = data.pieces.tbd2 ;
      
   Serial.print("Temp A: ");
   Serial.print(t1, DEC);
   Serial.print("°C, Temp P: ");
   Serial.print(t2, DEC);
   Serial.print("°C, Power: ");
   Serial.print(EcoClim_On[on]);
  Serial.print(", Cl: ");
   Serial.print(cl);
   Serial.print(", fan: ");
   Serial.print(EcoClim_Fan[fan]);
   Serial.print(", mode: ");
   Serial.print(EcoClim_Mode[mode]);
    Serial.print(", time: ");
   Serial.print(hour, DEC);
    Serial.print(":");
   Serial.print(min, DEC);
   Serial.print(" , OnTime: ");
   if (onTimeH != 0x1F || onTimeM != 0x7) // OxFF in case No Prog
   {
   Serial.print(onTimeH, DEC);
   Serial.print(":");
   Serial.print(onTimeM, DEC);
   Serial.print("0");
   }
   else    Serial.print("Not Prog");
   Serial.print(", OffTime: ");
      if (offTimeH != 0x1F || offTimeM != 0x7) // OxFF in case No Prog
   {
   Serial.print(offTimeH, DEC);
   Serial.print(":");
   Serial.print(offTimeM, DEC);
   Serial.print("0");
  }
   else    Serial.print("Not Prog");   
   Serial.print(", DIPConfig: ");
   Serial.print(DIPConfig);
   Serial.print(", tbd2: ");
   Serial.println(tbd2, HEX);
  
  }
  else if (results->bits == ECOCLIM_STA_BITS) {
    Serial.print("STA Frame "); // 0x8BB

    union convertorEcoClimStaFrame data;
      
      data.whole=u64;
      
      unsigned int pre = data.pieces.fix3; // first 3 bits
      unsigned int t1 = data.pieces.t_amb + ECOCLIM_CMD_TEMP_MIN; // next 5 Bits
      int DIPConfig = data.pieces.DipConfig ; // next 4 Bits 
      unsigned int post = data.pieces.end; // first 3 bits

   Serial.print("Temp A: ");
   Serial.print(t1, DEC);
   Serial.print(", pre: ");
   Serial.print(pre, HEX);
   Serial.print(", DIPConfig: ");
   Serial.print(DIPConfig, HEX);
   Serial.print(", post: ");
   Serial.println(post, HEX);

  }
  else {
    Serial.println("Unkown Frame");
  }
  }
}

crankyoldgit, 2021-02-14:
Thanks for all of that. I've started coding it up based on what you've said so far.
Please download and try out this branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Ecoclim_detailed
And let me know how it goes. It's not all there yet, but it's a start.

I have several questions after looking at the code and data etc:
1) Are you sure the "ambient" sensor temperature is with in the ranges you specify?
2) Can you explain the modes better & in detail? I'm used to Auto, Cool, Heat, Dry, & Fan ... you've listed some unusual ones, Can you explain them to me and what they do etc. I'm assuming "Slave" is Auto, & "Sun" is Dry, is that correct?
3) Does the DIP config act as a A/C unit identifier code? So one remote doesn't accidentally control a similar device etc?
4) What does the cl/Clear bit do?
5) You don't seem to have listed any controls for the vanes/swing etc. Does this unit not have any?

mothra78, 2021-02-14:
here a couple of answers : 
1 - Yes I have tested full range from 5°C to 30°C on the screen and could get the same data via the IR.
2 and 5 - The unit I am EcoClim remote acts for each room on the amount of air blown from the Daikin central HVAC. so there are no vanes/swing etc. From the manufacturer documentation only DIP Config 0 Master and 7 Slave are documented.
DIP Config is exactly the DIP switch configuration from the back of the remote. I then tried to use other ones then from the documentation and that's how I could find other modes that I do not need in my current config. The modes I have documented are corresponding to the pictograms on the remote control (I have copied images from the remote in G Sheet (see IR Cmd Code Table sheet)).
Master DIP Config is the one a single remote in my living room next to the Daikin remote which allows me to toggle from Heat to Cool. All the other rooms are set to Slave and have no mode icon.
3 - DIP Config is exactly the DIP switch configuration from the back of the remote. From the manufacturer documentation only DIP Config 0 (OFF OFF OFF OFF) Master and 7 (OFF ON ON ON) Slave are documented.
4 - I do not know what CL Bit does it'only used in some exotic mode before the FAN Low mode.

mothra78, 2021-02-14:
Tried the new update and works fine : 

Timestamp : 000089.127
Library   : v2.7.15

Protocol  : ECOCLIM
Code      : 0x1250738FFFFF02 (56 Bits)
Mesg Desc.: Power: On, Mode: 2 (Dry), Temp: 21C, SensorTemp: 23C, Fan: 3 (Auto), Clock: 15:11
uint16_t rawData[343] = {5872, 1844,  578, 540,  546, 550,  548, 546,  550, 1662,  522, 546,  550, 542,  544, 1646,  550, 546,  550, 540,  546, 1648,  548, 548,  548, 1606,  578, 548,  550, 540,  546, 548,  548, 546,  552, 542,  544, 1674,  520, 1640,  544, 1640,  544, 550,  546, 554,  544, 1640,  544, 1650,  544, 1638,  546, 548,  548, 548,  550, 550,  548, 1636,  548, 1608,  576, 1642,  552, 1632,  552, 1638,  546, 1642,  554, 1642,  552, 1632,  552, 1638,  546, 1642,  552, 1604,  580, 1636,  548, 1646,  550, 1638,  546, 1646,  550, 1634,  550, 1636,  548, 1642,  552, 1640,  544, 1644,  552, 512,  574, 552,  544, 548,  550, 538,  548, 546,  550, 544,  554, 1642,  554, 514,  5792, 1864,  578, 548,  548, 538,  548, 548,  550, 1636,  548, 546,  550, 544,  554, 1636,  548, 546,  550, 548,  548, 1610,  574, 548,  550, 1634,  550, 548,  550, 544,  552, 546,  550, 508,  578, 552,  576, 1576,  576, 1636,  548, 1644,  552, 542,  574, 522,  576, 1614,  550, 1638,  546, 1640,  576, 516,  580, 514,  574, 526,  580, 1578,  606, 1614,  580, 1604,  582, 1606,  578, 1612,  582, 1602,  582, 1606,  578, 1614,  580, 1610,  576, 1612,  582, 1606,  578, 1608,  576, 1618,  578, 1582,  602, 1616,  580, 1606,  578, 1610,  574, 1616,  578, 1610,  574, 1614,  580, 510,  576, 520,  578, 516,  580, 512,  574, 520,  578, 518,  580, 1610,  574, 518,  5788, 1846,  606, 516,  580, 512,  574, 518,  578, 1618,  578, 490,  608, 486,  600, 1612,  574, 520,  576, 520,  576, 1610,  574, 520,  578, 1608,  576, 516,  582, 514,  572, 526,  580, 510,  578, 520,  576, 1606,  578, 1612,  572, 1618,  578, 518,  516, 546,  602, 1616,  578, 1612,  572, 1586,  608, 514,  572, 520,  576, 518,  580, 1604,  580, 1612,  572, 1616,  580, 1612,  572, 1612,  572, 1622,  572, 1610,  574, 1616,  578, 1614,  580, 1606,  578, 1606,  578, 1618,  576, 1582,  602, 1612,  582, 1608,  576, 1582,  602, 1612,  574, 1620,  574, 1612,  574, 1620,  574, 518,  580, 514,  572, 522,  576, 514,  572, 530,  578, 482,  606, 1618,  578, 510,  7920};  // ECOCLIM 1250738FFFFF02
uint64_t data = 0x1250738FFFFF02;

Timestamp : 000106.121
Library   : v2.7.15

Protocol  : ECOCLIM
Code      : 0x1250438FFFFF02 (56 Bits)
Mesg Desc.: Power: On, Mode: 2 (Dry), Temp: 21C, SensorTemp: 23C, Fan: 0 (Low), Clock: 15:11
uint16_t rawData[343] = {5866, 1890,  554, 542,  544, 550,  548, 546,  550, 1636,  550, 544,  552, 548,  548, 1606,  578, 542,  546, 550,  546, 1674,  522, 546,  552, 1608,  576, 544,  552, 540,  546, 548,  550, 544,  554, 542,  544, 1664,  520, 554,  554, 538,  550, 550,  546, 546,  550, 1608,  576, 1636,  548, 1638,  546, 548,  548, 546,  552, 544,  554, 1632,  552, 1636,  548, 1642,  554, 1634,  550, 1636,  548, 1646,  550, 1636,  548, 1638,  546, 1644,  550, 1638,  546, 1644,  550, 1634,  550, 1638,  546, 1644,  550, 1640,  544, 1642,  552, 1638,  546, 1638,  546, 1644,  552, 1636,  548, 546,  552, 546,  552, 540,  546, 548,  548, 546,  552, 544,  554, 1638,  546, 544,  5764, 1874,  578, 510,  576, 544,  552, 542,  544, 1644,  550, 540,  548, 552,  544, 1644,  552, 514,  572, 544,  552, 1636,  548, 550,  548, 1644,  550, 538,  550, 544,  552, 544,  552, 544,  542, 518,  580, 1640,  554, 538,  550, 542,  544, 548,  548, 546,  552, 1644,  552, 1604,  580, 1634,  550, 544,  554, 546,  550, 542,  546, 1640,  554, 1636,  548, 1640,  576, 1608,  576, 1614,  580, 1608,  576, 1614,  580, 1610,  574, 1612,  584, 1602,  582, 1608,  576, 1614,  582, 1606,  578, 1614,  580, 1604,  580, 1606,  578, 1612,  572, 1618,  578, 1608,  576, 1614,  582, 516,  582, 480,  606, 514,  572, 522,  574, 520,  578, 518,  580, 1606,  578, 518,  5788, 1844,  598, 522,  576, 518,  578, 516,  582, 1606,  578, 522,  576, 516,  580, 1606,  578, 516,  582, 514,  574, 1616,  578, 512,  574, 1616,  578, 516,  582, 512,  574, 520,  578, 514,  572, 522,  576, 1618,  576, 514,  572, 520,  576, 516,  582, 516,  580, 1602,  582, 1610,  574, 1614,  582, 514,  572, 494,  604, 518,  578, 1604,  580, 1608,  576, 1614,  580, 1610,  574, 1616,  580, 1608,  576, 1616,  580, 1604,  580, 1608,  576, 1614,  582, 1602,  582, 1612,  582, 1578,  606, 1608,  576, 1612,  582, 1604,  580, 1612,  572, 1588,  606, 1606,  578, 1614,  580, 512,  576, 518,  578, 516,  582, 508,  578, 524,  574, 494,  604, 1610,  576, 514,  7914};  // ECOCLIM 1250438FFFFF02
uint64_t data = 0x1250438FFFFF02;

Timestamp : 000111.127
Library   : v2.7.15

Protocol  : ECOCLIM
Code      : 0x1250538FFFFF0A (56 Bits)
Mesg Desc.: Power: On, Mode: 2 (Dry), Temp: 21C, SensorTemp: 23C, Fan: 1 (Medium), Clock: 15:11
uint16_t rawData[343] = {5872, 1886,  548, 548,  548, 514,  574, 546,  550, 1640,  554, 538,  548, 544,  554, 1638,  546, 546,  550, 544,  552, 1632,  552, 542,  546, 1650,  544, 516,  570, 550,  548, 546,  550, 542,  546, 548,  548, 1644,  552, 538,  548, 1644,  550, 544,  554, 540,  546, 1638,  548, 1646,  548, 1610,  574, 546,  550, 544,  552, 544,  552, 1630,  544, 1646,  550, 1640,  554, 1634,  552, 1634,  550, 1646,  548, 1636,  548, 1640,  554, 1636,  548, 1634,  550, 1642,  552, 1640,  546, 1612,  572, 1644,  550, 1636,  550, 1638,  546, 1644,  550, 1634,  550, 1640,  554, 1638,  546, 550,  548, 546,  550, 538,  548, 544,  552, 1642,  554, 540,  546, 1640,  544, 526,  5822, 1864,  548, 518,  578, 510,  578, 5 [... cut ...]   576, 1608,  576, 1620,  576, 488,  608, 508,  578, 1614,  582, 1608,  576, 1584,  600, 520,  578, 516,  582, 1606,  578, 516,  580, 514,  574, 524,  574, 514,  572, 526,  582, 1604,  580, 514,  574, 520,  576, 520,  576, 1610,  576, 518,  578, 1610,  574, 520,  576, 1608,  576, 524,  574, 516,  580, 1610,  574, 524,  574, 1582,  602, 518,  578, 514,  572, 526,  572, 520,  576, 516,  570, 520,  576, 518,  580, 1612,  572, 522,  7908};  // ECOCLIM 12317390454A02
uint64_t data = 0x12317390454A02;

mothra78, 2021-02-14:
Temps are correct, Clock OK and Fan mode OK

mothra78, 2021-02-14:
Power On / Off works as well

mothra78, 2021-02-14:
Readings above are done with TSOP31238 previous where using VS1838B.

crankyoldgit, 2021-02-15:
> Readings above are done with TSOP31238 previous where using VS1838B.

Okay. Thanks for the update and more info, and the confirmations.
I'll try to add more support later tonight.

crankyoldgit, 2021-02-15:
Okay, I've updated that branch with more tests and features. e.g. Timer support & Type(DIP switches)

Please try it out and let me know how it goes now.

mothra78, 2021-02-15:
Thanks a lot, i will try and provide you feedback.

mothra78, 2021-02-15:
Thanks David, it works fine, type and Timer programs are OK. Great Job!
Only thing I have found is the mode 3 which is listed a Unkown whereas it should be "Recycle".  

See below the latest traces : 

Protocol  : ECOCLIM
Code      : 0x131074E4410972 (56 Bits)
Mesg Desc.: Power: On, Mode: 0 (Auto), Temp: 21C, SensorTemp: 24C, Fan: 3 (Auto), Clock: 20:52, On Timer: 08:10, Off Timer: 01:10, Type: 7
uint16_t rawData[343] = {5888, 1854,  578, 564,  522, 548,  550, 568,  518, 1642,  554, 546,  550, 566,  522, 1644,  550, 1658,  528, 544,  552, 546,  552, 538,  550, 1642,  552, 538,  548, 546,  552, 540,  548, 572,  524, 566,  520, 1644,  552, 1664,  520, 1646,  548, 542,  544, 1642,  554, 538,  548, 544,  554, 1634,  550, 1668,  528, 1634,  550, 540,  546, 578,  518, 1638,  546, 552,  546, 578,  518, 516,  570, 1646,  550, 540,  546, 578,  518, 546,  550, 540,  546, 576,  520, 1644,  552, 544,  552, 534,  552, 540,  546, 552,  546, 1638,  546, 550,  548, 550,  548, 1640,  544, 546,  552, 1638,  546, 1640,  544, 1648,  548, 518,  580, 572,  524, 1630,  554, 544,  5762, 1896,  548, 542,  546, 574,  522, 576,  522, 1634,  550, 544,  552, 574,  522, 1640,  546, 1634,  550, 544,  520, 576,  552, 546,  552, 1610,  574, 546,  552, 512,  576, 542,  554, 540,  546, 546,  550, 1640,  546, 1644,  550, 1632,  554, 548,  548, 1634,  550, 546,  552, 540,  546, 1642,  542, 1644,  550, 1638,  546, 550,  548, 548,  550, 1634,  550, 542,  544, 550,  548, 554,  544, 1610,  574, 548,  550, 542,  544, 554,  544, 548,  550, 542,  544, 1648,  546, 522,  576, 546,  550, 536,  550, 542,  544, 1644,  550, 542,  544, 550,  548, 1642,  554, 544,  552, 1634,  552, 1634,  550, 1636,  548, 552,  546, 520,  576, 1640,  544, 550,  5788, 1866,  546, 548,  548, 542,  554, 540,  546, 1642,  552, 540,  548, 550,  546, 1640,  554, 1632,  554, 538,  548, 546,  550, 546,  552, 1640,  544, 552,  576, 490,  606, 482,  604, 520,  576, 510,  578, 1614,  580, 1608,  544, 1616,  610, 512,  574, 1612,  582, 508,  578, 516,  580, 1604,  580, 1608,  576, 1616,  580, 510,  576, 520,  576, 1608,  576, 516,  580, 514,  572, 520,  576, 1612,  582, 510,  576, 518,  578, 514,  582, 514,  572, 518,  578, 1612,  574, 522,  574, 494,  604, 512,  574, 518,  578, 1612,  572, 520,  578, 514,  572, 1618,  576, 516,  582, 1610,  574, 1608,  576, 1614,  582, 510,  576, 524,  574, 1582,  602, 522,  7918};  // ECOCLIM 131074E4410972
uint64_t data = 0x131074E4410972;

Timestamp : 000092.298
Library   : v2.7.15

Protocol  : ECOCLIM
Code      : 0x120F74E541097A (56 Bits)
Mesg Desc.: Power: On, Mode: 0 (Auto), Temp: 20C, SensorTemp: 23C, Fan: 3 (Auto), Clock: 20:53, On Timer: 08:10, Off Timer: 01:10, Type: 7
uint16_t rawData[343] = {5898, 1900,  522, 544,  552, 538,  548, 536,  552, 1620,  576, 570,  526, 566,  520, 1642,  554, 540,  548, 576,  522, 542,  544, 548,  550, 576,  522, 1634,  550, 1666,  520, 1644,  552, 1660,  524, 574,  522, 1634,  550, 1640,  544, 1640,  546, 548,  550, 1642,  554, 540,  546, 546,  552, 1638,  546, 1664,  520, 1646,  548, 544,  554, 570,  526, 1634,  550, 508,  578, 1638,  546, 546,  552, 1638,  548, 550,  546, 544,  552, 546,  552, 534,  554, 570,  526, 1630,  554, 550,  548, 536,  550, 542,  554, 542,  544, 1642,  552, 570,  518, 548,  550, 1640,  544, 546,  552, 1638,  548, 1642,  542, 1642,  554, 1634,  550, 548,  548, 1634,  550, 546,  5762, 1898,  546, 546,  552, 546,  550, 546,  550, 1608,  576, 540,  546, 544,  552, 1638,  546, 580,  518, 544,  548, 550,  554, 538,  550, 542,  544, 1644,  552, 1638,  548, 1636,  548, 1642,  552, 546,  550, 1604,  580, 1638,  548, 1636,  550, 544,  552, 1636,  548, 550,  546, 542,  546, 1642,  552, 1638,  546, 1640,  544, 550,  548, 544,  554, 1624,  550, 556,  552, 1636,  548, 544,  554, 1636,  550, 544,  552, 538,  548, 546,  550, 542,  544, 556,  550, 1632,  552, 538,  548, 550,  548, 546,  552, 540,  548, 1642,  554, 542,  544, 550,  546, 1634,  550, 548,  550, 1634,  552, 1638,  546, 1642,  554, 1638,  546, 514,  572, 1650,  544, 518,  5820, 1840,  604, 514,  582, 510,  578, 520,  576, 1606,  548, 548,  580, 514,  574, 1616,  548, 544,  574, 518,  578, 518,  578, 518,  578, 512,  574, 1614,  580, 1602,  582, 1610,  576, 1610,  574, 518,  578, 1612,  572, 1584,  602, 1614,  580, 512,  574, 1614,  580, 514,  574, 522,  574, 1612,  582, 1604,  580, 1610,  574, 520,  576, 484,  602, 1616,  580, 512,  576, 1614,  580, 512,  574, 1612,  572, 522,  574, 516,  582, 516,  582, 510,  576, 514,  572, 1618,  578, 518,  578, 514,  574, 518,  580, 514,  572, 1616,  578, 514,  574, 520,  576, 1610,  574, 526,  582, 1600,  574, 1616,  582, 1602,  578, 1612,  574, 518,  578, 1608,  576, 526,  7916};  // ECOCLIM 120F74E541097A
uint64_t data = 0x120F74E541097A;

Timestamp : 000098.292
Library   : v2.7.15

Protocol  : ECOCLIM
Code      : 0x110F44E541097A (56 Bits)
Mesg Desc.: Power: On, Mode: 0 (Auto), Temp: 20C, SensorTemp: 22C, Fan: 0 (Low), Clock: 20:53, On Timer: 08:10, Off Timer: 01:10, Type: 7
uint16_t rawData[343] = {5866, 1910,  524, 546,  554, 536,  548, 550,  548, 1636,  550, 542,  544, 552,  544, 546,  550, 1642,  552, 538,  550, 568,  518, 578,  518, 576,  520, 1634,  550, 1642,  554, 1662,  522, 1668,  528, 538,  548, 1634,  552, 546,  550, 544,  552, 542,  544, 1662,  522, 548,  548, 544,  552, 1630,  554, 1636,  548, 1668,  528, 540,  548, 546,  550, 1638,  546, 570,  526, 1634,  550, 544,  554, 1636,  550, 540,  546, 576,  520, 548,  548, 542,  546, 574,  524, 1642,  552, 542,  546, 542,  544, 554 [... cut ...] 632,  552, 1636,  548, 1642,  554, 538,  550, 546,  552, 540,  548, 546,  552, 540,  548, 550,  578, 516,  582, 508,  578, 518,  580, 514,  574, 1614,  582, 510,  578, 514,  574, 520,  578, 522,  576, 514,  574, 522,  574, 516,  582, 1608,  578, 1606,  580, 516,  580, 1606,  580, 1608,  576, 516,  582, 512,  576, 522,  576, 522,  576, 482,  604, 516,  582, 518,  580, 1606,  580, 512,  576, 518,  578, 512,  574, 1614,  582, 514,  7920};  // ECOCLIM 13317002036022
uint64_t data = 0x13317002036022;

crankyoldgit, 2021-02-16:
> 3 - DIP Config is exactly the DIP switch configuration from the back of the remote. From the manufacturer documentation only DIP Config 0 (OFF OFF OFF OFF) Master and 7 (OFF ON ON ON) Slave are documented.

Currently, The DIPs are reported via `Type:` e.g. `Type: 7` for Slave. In your latest batch of data, I noticed there are non-0 & non-7 values. eg. `Type: 1`, `Type: 2`, & `Type: 3`. What's going on with that? Seems there is more going on with those bits than we think. 

Also, the reported clock values seem all over the place. e.g. It jumps from `20:53` to `00:00` .. stays there for several minutes, then moves on to `00:01` etc. What's going on?

> Only thing I have found is the mode 3 which is listed a Unkown whereas it should be "Recycle".

I was only reporting the "standard" modes at present. I'll add something to report those other modes (Recycle/Sleep) soon.

crankyoldgit, 2021-02-16:
I've updated the branch, it should better support Sleep & Recycle modes now. Please try it out etc etc.

mothra78, 2021-02-16:
Correct for the type it's an exact copie of the DIP switch, the values seen
over the trace are when I try NOT documented and NOT usual types.

For the on off programming the values sent are using the current clock as
origin.

Let's say current time is 19:00 and I program OFF at 23:00 then 04:00 will
be transmitted in Byte 5 (strange engineering choice...).

crankyoldgit, 2021-02-16:
> Correct for the type it's an exact copie of the DIP switch, the values seen over the trace are when I try NOT documented and NOT usual types.

Phew! Cool. Currently the library will only accept _setting_ those values to the two documented values. I'll leave that as-is unless you think it realistically should allow any 4-bit pattern.

mothra78, 2021-02-16:
No fine to leave it as is. Thanks again

closed 2021-02-17 by crankyoldgit

crankyoldgit, 2021-03-24:
FYI, the aforementioned changes have been included in the new v2.7.16 release for the library.

PR #1415 (2021-02-17): Ecoclim: Add detailed A/C support
* Add support for power, mode, temp, sensor temp, fan speed, clock, Timers, & DIP switch settings.
* Unit tests updated & extended to cover new features
* Update Common AC API (`stdAc`) with support for EcoClim.

* Rename `src` files to be consistent case.
* Fix minor issue with Daikin64 and clocks.

Fixes #1397
commit a2c1516f (2021-02-17): Ecoclim: Add detailed A/C support (#1415)

* Add support for power, mode, temp, sensor temp, fan speed, clock, Timers, & DIP switch settings.
* Unit tests updated & extended to cover new features
* Update Common AC API (`stdAc`) with support for EcoClim.

* Rename `src` files to be consistent case.
* Fix minor issue with Daikin64 and clocks.

Fixes #1397