crankyoldgit/IRremoteESP8266 #1398: It is not compatible with Mitsubishi air conditioners in 0.5 degree increments.

nao-pon, 2021-02-01:
### Version/revision of the library used
2.7.14

### Describe the bug
The problem is that the temperature of 0.5 degrees is mapped above the maximum temperature of 31 degrees.

For example, 16.5 degrees is 32 degrees on the data. Similarly, 17.5 degrees is 33 degrees.

#### To Reproduce
Sends 32 degrees, which represents 16.5 degrees. Then the data will be treated as 31 degrees.

### Other useful information
I have tried the following methods to solve this problem. This requires the client to revert such mappings to their original temperature, which works well for such implementations.

- https://github.com/nao-pon/IRremoteESP8266/commit/6216cd3d1d5945c4bd854061ceab6eacf10b54b5

This requires a slight rewrite of the test, but if this is okay, I'll make a Pull Request.

crankyoldgit, 2021-02-01:
I'm not sure I fully understand what your saying with respect to how Mitsubishi handles half degrees.

Current, the library doesn't handle half degrees for Mitsubishi A/Cs.

If you can please provide some hex STATE codes captured from the actual remote + documenting what temps they are, I may be able to add support for them.
e.g. Follow [these steps](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#determining-the-bit-order-of-the-protocol) for a temp sequence, but include half degrees, and we'll see what we can do.

Also, please provide the make & model numbers of both the A/C unit _and_ the remote.

nao-pon, 2021-02-01:
@crankyoldgit Thank you for your support. The hexadecimal dump is shown below. As you know, the 15th bit represents the temperature. And when it is 0.5, 1 is set in the 14th bit.

```
0x23CB26010020080000807100000800000036	16
0x23CB26010020081000C07100000800000086	16.5
0x23CB26010020080100C07100000800000077	17
0x23CB26010020081100C07B00000800000091	17.5
0x23CB26010020080200C07B00000800000082	18
0x23CB26010020081200C07C00000800000093	18.5
0x23CB26010020080300C07C00000800000084	19
0x23CB26010020081300C07C00000800000094	19.5
0x23CB26010020080400C07C00000800000085	20
0x23CB26010020081400C07C00000800000095	20.5
0x23CB26010020080500C07C00000800000086	21
   .......
0x23CB26010020080F0080880000080000005C	31
```

The air conditioner unit is MLZ-RX5017AS, and the remote controller is SG153 (M21EDF426).

crankyoldgit, 2021-02-02:
@nao-pon Can you please try out PR #1400 / Branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1398

I think it should support your half degrees correctly now. Please let me know how it goes.

nao-pon, 2021-02-02:
@crankyoldgit Thank you for your wonderful work! I tried the Issue 1398 branch. It works perfectly with respect to temperature.

Congratulations. Thank you very much! 👍

crankyoldgit, 2021-02-02:
Thanks for confirming it works. When the PR is merged, it will close this issue. Leaving it open till then.

nao-pon, 2021-02-02:
Yes, sure! :smile:

closed 2021-02-02 by crankyoldgit

crankyoldgit, 2021-02-13:
FYI, the changes mentioned above have now been included in the new v2.7.15 release of the library.

commit 19ecbba1 (2021-02-02): MITSUBISHI_AC: Add support for half degrees.
* add `irutils::addTempFloatToString()`.
* Add half degree support to `IRMitsubishiAC::setTemp()` & `getTemp()`.
* Add unit test coverage for those changes.
* Update supported Mitsubishi AC models.
* Update TCL to use new `addTempFloatToString()`.

Misc:
  * Variable/code style cleanup in ir_Mitsubishi_test.cpp.

Fixes #1398
PR #1400 (2021-02-02): MITSUBISHI_AC: Add support for half degrees.
* Add `irutils::addTempFloatToString()`.
* Add half degree support to `IRMitsubishiAC::setTemp()` & `getTemp()`.
* Add unit test coverage for those changes.
* Update supported Mitsubishi AC models.
* Update TCL to use new `addTempFloatToString()`.

Misc:
  * Variable/code style cleanup in ir_Mitsubishi_test.cpp.

Fixes #1398
commit 98a19005 (2021-02-02): MITSUBISHI_AC: Add support for half degrees. (#1400)

* add `irutils::addTempFloatToString()`.
* Add half degree support to `IRMitsubishiAC::setTemp()` & `getTemp()`.
* Add unit test coverage for those changes.
* Update supported Mitsubishi AC models.
* Update TCL to use new `addTempFloatToString()`.

Misc:
  * Variable/code style cleanup in ir_Mitsubishi_test.cpp.

Fixes #1398