crankyoldgit/IRremoteESP8266 #1399: [Mitsubishi] False detection of SwingV mode

nao-pon, 2021-02-01:
### Version/revision of the library used
2.7.14

### Describe the bug
SwingV mode detects swing mode as Auto. In Mitsubishi air conditioner, the flap keeps moving up and down in Swing mode. With Auto, the air conditioner automatically controls the wind direction.

#### To Reproduce
There are no special steps.

#### Output of raw data from [IRrecvDumpV2.ino]
Mitsubishi MLZ-RX5017AS with remote control SG153(M21EDF426)
```
0x23CB26010020080400C88B0000080000009C "SwingV":"Highest"
0x23CB26010020080400D08B000008000000A4 "SwingV":"High"
0x23CB26010020080400D88B000008000000AC "SwingV":"Middle"
0x23CB26010020080400E08B000008000000B4 "SwingV":"Low"
0x23CB26010020080400E88B000008000000BC "SwingV":"Lowest"
0x23CB26010020080400F88B000008000000CC "SwingV":"Auto" (Is not, It's Swing mode)
0x23CB26010020080400808B00000800000054 "SwingV":"Auto"
```

Mitsubishi MSZ-GV2519 with remote control RH151 (M21ED6426)
```
0x23CB260100200804304800000000040000BD "SwingV":"Highest"
0x23CB260100200804305000000000040000C5 "SwingV":"High"
0x23CB260100200804305800000000040000CD "SwingV":"Middle"
0x23CB260100200804306000000000040000D5 "SwingV":"Low"
0x23CB260100200804306800000000040000DD "SwingV":"Lowest"
0x23CB260100200804307800000000040000ED "SwingV":"Auto" (Is not, It's Swing mode)
0x23CB260100200804308000000000040000F5 "SwingV":"Auto"
```

### What brand/model IR demodulator are you using?
HX1838: However, there seems to be no problem with the decoded data.

### Circuit diagram and hardware used (if applicable)
ESP8266

### I have followed the steps in the [Troubleshooting Guide]
Yes

### Has this library/code previously worked as expected for you?
I haven't tried previous versions.

crankyoldgit, 2021-02-01:
Can you please reproduce this with the **latest** `master` branch? e.g. after PR #1393, where I recently changed this behaviour.

crankyoldgit, 2021-02-02:
As the `IRMitsubishiAC` class doesn't produce "Highest" -> "Lowest" in it's `toString()` output. I'm assuming you're using the `IRac` class interface (or Tasmota).

The `IRac` class does not support a "Swing" state, it's limited to the following choices:
https://github.com/crankyoldgit/IRremoteESP8266/blob/1dd29a3ee62d71a363d214209ad6ca5c0b23c723/src/IRsend.h#L69-L80

Thus it (`IRac`) is doing the correct/best it can by mapping your "Swing" to "Auto" out of the available choices.

The `IRac` interface is not meant to support _every_ feature of every _manufacturer_. It's just the common set of functions.

If the Mitsubishi A/Cs you have **do not** have an Vertical Swing "Off" setting. i.e. Not moving. We could map your "Auto"  to "Off", and your "Swing" to "Auto". Looking at the code, the "Off" setting is unused so that could be the solution.

nao-pon, 2021-02-02:
@crankyoldgit Yes, I do use Tasmota.

> If the Mitsubishi A/Cs you have do not have an Vertical Swing "Off" setting. i.e. Not moving. We could map your "Auto" to "Off", and your "Swing" to "Auto". Looking at the code, the "Off" setting is unused so that could be the solution.

Yes, I really agree. I also think that such an operation is the best for Mitsubishi AC.

nao-pon, 2021-02-02:
Or is it impossible to implement the following?

```c++
 /// Common A/C settings for Vertical Swing. 
 enum class swingv_t { 
   kOff =    -1, 
   kAuto =    0, 
   kHighest = 1, 
   kHigh =    2, 
   kMiddle =  3, 
   kLow =     4, 
   kLowest =  5, 
   kSwing =   6, 
   // Add new entries before this one, and update it to point to the last entry 
   kLastSwingvEnum = kLowest, 
 }; 
```

crankyoldgit, 2021-02-02:
Technically, you'd need change `kLastSwingvEnum = kSwing,` too.

But, adding a `swing` to `swingv_t` would introduce a lot of ambiguity IMHO, as a lot of Vertical Swing's on most Brands have "Auto" perform what your A/Cs _Swing_ actually does. Plus a lot of potentially unneeded adding of extra code to retro fit that in to all other A/C classes, plus the Tasmota's of the world would need to extend their syntax/parsers etc etc.

Not impossible or impractical, but we have a suitable solution with the repurposing of "Off" to _your_ AC's Auto.
It sucks from a Mitsubishi perspective, but works for pretty most other brands.

Speaking of which. I'll get onto that now the other PR has been merged.

nao-pon, 2021-02-02:
@crankyoldgit Thank you for your detailed explanation. As for swing, it's difficult ...

Yes, I'm okay if I can identify the swinging or the auto controlled by the air conditioner. For Mitsubishi air conditioners, Off-> Air conditioner side auto, Auto-> Air conditioner side swing, I think it's good.

I look forward to working with you.

crankyoldgit, 2021-02-02:
Okay, same deal as last time, can you please download/test PR #1401 / Branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1399 and let me know how it goes for you?

nao-pon, 2021-02-03:
Yes, I tried it. The result is amazing. I can now control the auto or swing with the Mitsubishi air conditioner!

Thanks! :+1:

crankyoldgit, 2021-02-03:
Excellent. Thanks for confirming it is working as expected.

closed 2021-02-03 by crankyoldgit

crankyoldgit, 2021-02-13:
FYI, the changes mentioned above have now been included in the new v2.7.15 release of the library.

PR #1401 (2021-02-03): Mitsubishi ACs: Improve handling swing/vane settings.
* MitsubishiAC class
  - Tweak IRac/stdAc handing of swingv/vane:
    * Native Auto as stdAc's Off. 
    * Native Swing as stdAc's Auto.

Native "auto" is the A/C works out where the vanes point.
Native "swing" is the A/C just occilates the vanes up and down constant. Thus "swing" is closest to stdAc's "auto". Mapping the unused stdAc's "off" to the native "auto" so it is at least reachable by the user via `IRac` etc. 

* Various Mitsubish A/Cs
  - Improve `toString()` output for swingv & swingh positions.
  - Reduce code duplication.

Fixe [... cut]
commit ab466e4d (2021-02-03): Mitsubishi ACs: Improve handling swing/vane settings. (#1401)

* MitsubishiAC class
  - Tweak IRac/stdAc handing of swingv/vane:
    * Native Auto as stdAc's Off. 
    * Native Swing as stdAc's Auto.

Native "auto" is the A/C works out where the vanes point.
Native "swing" is the A/C just occilates the vanes up and down constant. Thus "swing" is closest to stdAc's "auto". Mapping the unused stdAc's "off" to the native "auto" so it is at least reachable by the user via `IRac` etc. 

* Various Mitsubish A/Cs
  - Improve `toString()` output for swingv & swingh positions.
  - Reduce code duplicati [... cut]