crankyoldgit/IRremoteESP8266 #1402: [Mitsubishi] Not work Power OFF command

nao-pon, 2021-02-04:
### Version/revision of the library used
Current master
https://github.com/crankyoldgit/IRremoteESP8266/commit/ab466e4dac234a433880d424bbff930315649f12

### Describe the bug
A clear and concise description of what the bug is.

#### To Reproduce
Brand: Mitsubishi Electric,  Model: MLZ-RX5017AS A/C (MITSUBISHI_AC)
Brand: Mitsubishi Electric,  Model: SG153/M21EDF426 remote (MITSUBISHI_AC)

It is not possible to turn off the air conditioner with the above devices.

#### Example code used
IRrecvDumpV3

#### Expected behaviour
Hope we can turn it off with this library

#### Output of raw data from [IRrecvDumpV2.ino]Timestamp : 000108.206
```
Library   : v2.7.14

Protocol  : MITSUBISHI_AC
Code      : 0x23CB26010020080400C05D00000803000069 (144 Bits)
Mesg Desc.: Power: On, Mode: 1 (Heat), Temp: 20C, Fan: 0 (Auto), Swing(V): 0 (Auto), Swing(H): 0 (UNKNOWN), Clock: 15:30, On Timer: 00:00, Off Timer: 00:00, Timer: -
uint16_t rawData[583] = {3396, 1632,  462, 1214,  462, 1214,  462, 398,  440, 376,  462, 398,  438, 1214,  462, 376,  462, 376,  460, 1214,  462, 1236,  438, 398,  438, 1212,  462, 376,  462, 376,  460, 1216,  460, 1214,  462, 398,  440, 1212,  462, 1236,  438, 376,  462, 378,  460, 1216,  460, 376,  462, 374,  464, 1212,  462, 398,  440, 374,  462, 376,  462, 376,  460, 398,  438, 398,  438, 374,  464, 398,  438, 378,  458, 378,  460, 376,  460, 376,  460, 376,  460, 398,  438, 400,  438, 376,  462, 378,  460, 376,  460, 378,  460, 400,  438, 1214,  462, 376,  462, 398,  438, 376,  462, 376,  462, 376,  460, 1214,  460, 400,  438, 376,  462, 378,  460, 376,  460, 376,  462, 376,  460, 1214,  462, 376,  462, 376,  462, 400,  438, 374,  462, 378,  460, 376,  460, 374,  462, 376,  462, 376,  462, 374,  462, 376,  462, 376,  462, 376,  462, 398,  438, 376,  460, 376,  462, 376,  460, 376,  462, 376,  462, 1214,  460, 1238,  438, 1212,  464, 378,  458, 1216,  460, 1236,  438, 1238,  438, 376,  460, 1214,  462, 400,  438, 376,  462, 374,  462, 376,  460, 376,  460, 378,  460, 376,  460, 376,  460, 400,  438, 376,  462, 398,  438, 376,  462, 378,  460, 376,  460, 378,  460, 378,  458, 376,  460, 376,  462, 376,  460, 378,  460, 1212,  462, 376,  460, 378,  460, 376,  462, 374,  462, 1236,  438, 1214,  462, 376,  436, 426,  438, 378,  460, 376,  460, 400,  438, 378,  460, 376,  460, 400,  438, 378,  458, 378,  460, 376,  460, 378,  460, 400,  438, 376,  460, 376,  460, 398,  438, 376,  460, 376,  460, 400,  438, 376,  462, 376,  460, 376,  460, 1238,  438, 376,  462, 400,  438, 1236,  438, 378,  460, 1238,  438, 1214,  460, 376,  460, 11278,  3372, 1680,  438, 1214,  462, 1238,  438, 376,  462, 378,  460, 400,  438, 1214,  460, 376,  462, 398,  438, 1238,  438, 1236,  438, 376,  462, 1216,  460, 376,  460, 400,  436, 1238,  438, 1216,  460, 378,  460, 1214,  462, 1214,  462, 376,  460, 400,  438, 1214,  460, 376,  460, 376,  460, 1216,  460, 376,  462, 376,  460, 374,  462, 378,  460, 376,  460, 376,  462, 400,  436, 378,  460, 378,  460, 376,  460, 398,  438, 376,  434, 426,  438, 400,  436, 402,  410, 406,  432, 404,  434, 402,  460, 376,  434, 404,  434, 1240,  436, 426,  438, 376,  436, 404,  460, 376,  462, 378,  460, 1214,  462, 400,  438, 400,  436, 376,  460, 378,  432, 404,  434, 404,  460, 1216,  458, 400,  412, 404,  460, 376,  460, 376,  460, 378,  460, 378,  432, 426,  412, 404,  458, 378,  460, 376,  460, 376,  434, 404,  460, 376,  462, 378,  460, 378,  434, 404,  458, 378,  434, 404,  434, 404,  434, 1240,  462, 1236,  438, 1238,  438, 378,  460, 1216,  458, 1216,  460, 1214,  460, 380,  458, 1214,  460, 378,  434, 404,  460, 378,  460, 378,  458, 376,  460, 376,  462, 376,  460, 376,  460, 376,  434, 404,  460, 378,  458, 378,  460, 398,  438, 378,  460, 378,  458, 378,  458, 400,  438, 376,  434, 426,  438, 378,  434, 1242,  460, 376,  460, 378,  460, 378,  460, 378,  458, 1214,  462, 1214,  460, 380,  458, 376,  460, 378,  460, 380,  458, 400,  438, 378,  458, 378,  460, 400,  438, 378,  432, 426,  412, 404,  458, 376,  434, 404,  434, 426,  438, 398,  438, 400,  438, 378,  460, 378,  432, 404,  460, 378,  458, 400,  412, 426,  412, 1240,  460, 378,  460, 376,  434, 1240,  460, 378,  434, 1240,  462, 1214,  462, 376,  460};  // MITSUBISHI_AC
uint8_t state[18] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x20, 0x08, 0x04, 0x00, 0xC0, 0x5D, 0x00, 0x00, 0x08, 0x03, 0x00, 0x00, 0x69};

Timestamp : 000127.869
Library   : v2.7.14

Protocol  : MITSUBISHI_AC
Code      : 0x23CB26010000080400C05D00000803000049 (144 Bits)
Mesg Desc.: Power: Off, Mode: 1 (Heat), Temp: 20C, Fan: 0 (Auto), Swing(V): 0 (Auto), Swing(H): 0 (UNKNOWN), Clock: 15:30, On Timer: 00:00, Off Timer: 00:00, Timer: -
uint16_t rawData[583] = {3374, 1632,  460, 1214,  460, 1238,  438, 374,  462, 376,  462, 374,  464, 1212,  462, 376,  462, 374,  464, 1214,  462, 1216,  460, 376,  462, 1236,  438, 374,  464, 374,  464, 1236,  438, 1236,  438, 374,  464, 1236,  438, 1214,  462, 374,  462, 374,  464, 1238,  438, 376,  462, 374,  462, 1238,  410, 402,  464, 374,  462, 374,  462, 374,  462, 398,  438, 374,  462, 376,  462, 374,  462, 374,  462, 374,  462, 376,  462, 374,  462, 376,  462, 374,  462, 376,  462, 398,  438, 376,  462, 376,  462, 374,  462, 376,  460, 374,  462, 376,  460, 376,  462, 374,  462, 376,  462, 374,  464, 1216,  458, 376,  462, 374,  462, 374,  462, 376,  462, 376,  460, 374,  464, 1214,  462, 374,  462, 374,  464, 374,  464, 374,  462, 376,  460, 376,  462, 374,  462, 374,  462, 374,  462, 376,  462, 376,  462, 374,  462, 398,  438, 376,  462, 376,  462, 374,  464, 376,  462, 376,  462, 376,  462, 1216,  460, 1214,  462, 1214,  462, 374,  462, 1236,  438, 1212,  464, 1214,  462, 374,  462, 1214,  462, 374,  462, 376,  462, 376,  462, 376,  462, 374,  462, 376,  462, 374,  462, 376,  462, 374,  462, 376,  462, 374,  462, 374,  464, 374,  462, 398,  440, 376,  462, 374,  462, 376,  462, 376,  462, 374,  462, 398,  438, 1214,  460, 376,  462, 378,  460, 378,  460, 374,  462, 1238,  438, 1236,  438, 374,  464, 374,  462, 376,  462, 374,  462, 374,  462, 398,  438, 374,  464, 398,  438, 400,  438, 376,  462, 374,  462, 376,  462, 376,  460, 376,  462, 374,  462, 374,  464, 398,  440, 374,  464, 374,  462, 374,  462, 374,  460, 376,  410, 1266,  462, 376,  462, 376,  462, 1212,  464, 376,  460, 376,  460, 1238,  438, 374,  462, 11310,  3372, 1656,  438, 1214,  462, 1238,  438, 376,  462, 398,  440, 374,  462, 1236,  438, 374,  462, 376,  460, 1214,  462, 1236,  438, 376,  462, 1214,  460, 376,  462, 374,  462, 1214,  462, 1214,  462, 374,  462, 1216,  460, 1214,  462, 374,  462, 376,  460, 1214,  462, 376,  462, 376,  460, 1214,  462, 376,  462, 376,  462, 374,  464, 374,  462, 374,  464, 374,  462, 376,  462, 376,  462, 376,  462, 374,  464, 376,  462, 374,  462, 374,  462, 376,  462, 374,  462, 376,  462, 376,  462, 398,  438, 376,  462, 374,  462, 376,  462, 376,  462, 374,  462, 376,  462, 376,  462, 376,  462, 1214,  460, 376,  462, 374,  462, 376,  460, 376,  462, 378,  460, 376,  460, 1216,  460, 376,  460, 376,  462, 376,  462, 376,  462, 376,  462, 378,  460, 376,  462, 374,  462, 376,  462, 376,  462, 374,  462, 376,  462, 376,  462, 374,  462, 376,  462, 374,  462, 376,  430, 404,  464, 376,  462, 1216,  460, 1214,  460, 1214,  462, 376,  462, 1214,  462, 1214,  462, 1238,  436, 376,  462, 1214,  460, 376,  462, 376,  462, 374,  462, 376,  462, 376,  460, 376,  462, 376,  462, 374,  462, 376,  462, 376,  462, 376,  462, 398,  438, 376,  462, 374,  462, 374,  462, 398,  440, 376,  462, 398,  438, 374,  464, 376,  462, 1216,  460, 374,  462, 376,  460, 374,  464, 376,  462, 1236,  438, 1238,  438, 376,  462, 376,  460, 398,  438, 376,  462, 376,  462, 374,  462, 374,  462, 376,  460, 376,  462, 374,  462, 376,  462, 376,  462, 376,  462, 376,  462, 374,  462, 376,  462, 376,  460, 376,  462, 376,  462, 374,  462, 376,  460, 376,  462, 1216,  460, 376,  462, 376,  460, 1216,  460, 376,  462, 374,  462, 1238,  438, 376,  460};  // MITSUBISHI_AC
uint8_t state[18] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x00, 0x08, 0x04, 0x00, 0xC0, 0x5D, 0x00, 0x00, 0x08, 0x03, 0x00, 0x00, 0x49};
```

### What brand/model IR demodulator are you using?
HX1838: However, there seems to be no problem with the decoded data.

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Has this library/code previously worked as expected for you?
No

nao-pon, 2021-02-04:
Brand: Mitsubishi Electric,  Model: MSZ-GV2519 A/C (MITSUBISHI_AC)
Brand: Mitsubishi Electric,  Model: RH151/M21ED6426 remote (MITSUBISHI_AC)

The above devices can be turned off with the current library without any problems.

The dump is shown below, but it seems that Mode is not OFF when turning it off with my Mitsubishi air conditioners.

```
Timestamp : 001256.125
Library   : v2.7.14

Protocol  : MITSUBISHI_AC
Code      : 0x23CB260100200804304000000000040000B5 (144 Bits)
Mesg Desc.: Power: On, Mode: 1 (Heat), Temp: 20C, Fan: 0 (Auto), Swing(V): 0 (Auto), Swing(H): 3 (Middle), Clock: 00:00, On Timer: 00:00, Off Timer: 00:00, Timer: -
uint16_t rawData[583] = {3444, 1676,  456, 1240,  458, 1242,  456, 394,  454, 394,  454, 394,  454, 1242,  456, 418,  430, 394,  454, 1240,  458, 1242,  458, 394,  454, 1240,  458, 394,  454, 394,  454, 1240,  458, 1240,  456, 394,  454, 1242,  456, 1240,  458, 394,  454, 394,  454, 1242,  456, 394,  454, 394,  454, 1240,  458, 396,  452, 394,  454, 394,  454, 394,  454, 396,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 396,  454, 392,  452, 394,  458, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 396,  454, 1240,  458, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 1240,  458, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 1242,  456, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 1242,  456, 1240,  458, 394,  454, 394,  454, 394,  456, 392,  454, 394,  454, 394,  454, 394,  454, 394,  454, 1240,  458, 394,  454, 394,  454, 396,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 396,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 1240,  458, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 396,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  456, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 1242,  456, 394,  454, 1240,  458, 394,  454, 1240,  458, 1238,  458, 394,  454, 1240,  456, 12950,  3446, 1676,  456, 1240,  456, 1240,  458, 394,  454, 394,  454, 394,  454, 1242,  456, 394,  454, 394,  454, 1242,  454, 1242,  456, 394,  454, 1242,  456, 394,  454, 394,  454, 1240,  458, 1242,  456, 394,  454, 1242,  456, 1240,  458, 394,  454, 394,  454, 1242,  456, 394,  454, 394,  456, 1240,  456, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  456, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  456, 392,  454, 394,  454, 394,  454, 1240,  458, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 1240,  458, 394,  454, 394,  454, 394,  454, 394,  454, 394,  456, 394,  454, 1242,  456, 396,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  456, 394,  454, 394,  454, 394,  454, 1242,  456, 1242,  456, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 1240,  456, 394,  454, 394,  454, 394,  534, 314,  454, 420,  430, 394,  454, 394,  454, 394,  454, 396,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  456, 392,  454, 394,  456, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  456, 392,  456, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  456, 394,  454, 394,  454, 394,  454, 1242,  456, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 396,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  456, 392,  454, 1242,  458, 394,  454, 1240,  458, 394,  454, 1240,  458, 1240,  456, 394,  454, 1240,  458};  // MITSUBISHI_AC
uint8_t state[18] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x20, 0x08, 0x04, 0x30, 0x40, 0x00, 0x00, 0x00, 0x00, 0x04, 0x00, 0x00, 0xB5};

Timestamp : 001257.598
Library   : v2.7.14

Protocol  : MITSUBISHI_AC
Code      : 0x23CB26010000080430400000000004000095 (144 Bits)
Mesg Desc.: Power: Off, Mode: 1 (Heat), Temp: 20C, Fan: 0 (Auto), Swing(V): 0 (Auto), Swing(H): 3 (Middle), Clock: 00:00, On Timer: 00:00, Off Timer: 00:00, Timer: -
uint16_t rawData[583] = {3444, 1674,  434, 1266,  458, 1242,  456, 394,  454, 394,  454, 394,  454, 1240,  458, 396,  454, 394,  454, 1240,  458, 1240,  458, 394,  454, 1242,  456, 394,  454, 394,  454, 1242,  456, 1240,  456, 394,  454, 1240,  458, 1242,  456, 396,  454, 394,  454, 1242,  456, 394,  454, 418,  430, 1244,  454, 396,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 396,  454, 394,  454, 394,  454, 394,  454, 396,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 396,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 1240,  458, 394,  454, 394,  454, 394,  454, 394,  454, 394,  456, 392,  454, 1242,  456, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 396,  454, 394,  454, 1242,  456, 1240,  456, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 1240,  456, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 396,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 396,   [... cut ...] , 394,  454, 1242,  458, 394,  454, 394,  456, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 396,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  456, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 394,  454, 1240,  456, 394,  454, 1240,  458, 394,  454, 1240,  458, 392,  456, 394,  454, 1240,  458};  // MITSUBISHI_AC
uint8_t state[18] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x00, 0x08, 0x04, 0x30, 0x40, 0x00, 0x00, 0x00, 0x00, 0x04, 0x00, 0x00, 0x95};
```

crankyoldgit, 2021-02-04:
First off, I'm confused by how you are explaining the issue(s). Please be more verbose and explain more what you are doing, and what the steps are, and what the results are.

For instance, are the captures you have provided, from the remote or from the ESP? (Please be more detail and verbose)
What do you mean by "it seems that Mode is not OFF when turning it off"? (expand and explain)

What are the sequences you are doing/code you are using to generate IR messages. (please include the smallest possible/practical example so I can reproduce your problem)

nao-pon, 2021-02-04:
Sorry for lack of explanation.

The problem is that I can't turn off the air conditioner with the commands I send from this library.
This problem only occurs with one of the two types of Mitsubishi air conditioners I have.

> it seems that Mode is not OFF when turning it off with my Mitsubishi air conditioners.

This was my misunderstanding.

Below is a dump of the commands sent by this library. It says "Power: Off, Mode: 4 (Auto)", but one air conditioner does not seem to accept "Mode: 4 (Auto)". By setting this to "Power: Off, Mode: 1 (Heat)", I was able to turn off the air conditioner.

```
Timestamp : 003803.431
Library   : v2.7.14

Protocol  : MITSUBISHI_AC
Code      : 0x23CB26010000200230C0670000000000008E (144 Bits)
Mesg Desc.: Power: Off, Mode: 4 (Auto), Temp: 18C, Fan: 0 (Auto), Swing(V): 0 (Auto), Swing(H): 3 (Middle), Clock: 17:10, On Timer: 00:00, Off Timer: 00:00, Timer: -
uint16_t rawData[583] = {3402, 1768,  476, 1286,  478, 1280,  478, 404,  476, 428,  452, 404,  474, 1282,  478, 406,  474, 410,  474, 1282,  480, 1280,  478, 406,  474, 1282,  478, 428,  452, 428,  450, 1282,  478, 1282,  478, 406,  474, 1282,  478, 1282,  478, 406,  474, 404,  474, 1284,  476, 404,  474, 430,  452, 1282,  478, 406,  474, 406,  474, 404,  474, 404,  474, 406,  474, 404,  474, 406,  474, 428,  450, 404,  476, 406,  474, 406,  474, 404,  474, 404,  474, 404,  476, 404,  476, 404,  474, 404,  476, 404,  476, 404,  474, 404,  474, 406,  474, 404,  474, 406,  474, 428,  452, 404,  474, 404,  474, 406,  474, 404,  474, 1282,  454, 428,  476, 428,  452, 404,  474, 1282,  478, 404,  474, 404,  474, 406,  474, 404,  476, 404,  474, 406,  474, 404,  474, 404,  474, 404,  474, 404,  476, 1282,  476, 1282,  478, 404,  474, 406,  474, 404,  474, 406,  474, 404,  474, 406,  474, 404,  476, 404,  474, 1282,  478, 1282,  478, 1282,  478, 1282,  478, 1282,  478, 404,  474, 404,  474, 1282,  478, 1282,  478, 406,  474, 404,  476, 406,  474, 404,  474, 404,  476, 428,  450, 404,  476, 404,  476, 404,  476, 404,  474, 404,  476, 404,  476, 404,  474, 404,  474, 406,  474, 404,  474, 430,  450, 404,  476, 404,  476, 404,  476, 404,  474, 404,  474, 404,  474, 404,  476, 404,  476, 404,  474, 404,  474, 404,  476, 404,  474, 404,  476, 404,  476, 404,  474, 406,  474, 404,  476, 404,  476, 404,  474, 406,  474, 404,  476, 404,  474, 404,  474, 406,  474, 406,  474, 404,  474, 404,  474, 404,  476, 404,  474, 404,  476, 404,  476, 404,  476, 406,  474, 1282,  476, 1282,  478, 1282,  478, 404,  476, 404,  474, 404,  474, 1282,  474, 17226,  3408, 1746,  478, 1286,  478, 1282,  478, 404,  474, 404,  474, 406,  474, 1282,  476, 406,  474, 404,  476, 1282,  478, 1282,  478, 404,  474, 1280,  478, 404,  476, 404,  474, 1282,  478, 1282,  478, 404,  474, 1282,  478, 1282,  478, 404,  476, 404,  474, 1282,  478, 404,  474, 406,  474, 1282,  478, 404,  476, 404,  476, 404,  476, 404,  474, 404,  476, 404,  474, 406,  474, 428,  452, 404,  476, 406,  474, 404,  476, 404,  474, 404,  474, 404,  476, 406,  474, 404,  474, 404,  474, 406,  474, 404,  474, 404,  476, 404,  476, 404,  474, 406,  476, 404,  474, 404,  474, 404,  476, 404,  476, 404,  474, 1282,  478, 404,  476, 406,  474, 404,  476, 1282,  478, 404,  474, 404,  476, 404,  476, 404,  474, 404,  474, 406,  476, 404,  476, 404,  474, 404,  476, 404,  476, 1282,  476, 1282,  478, 404,  474, 406,  474, 404,  476, 404,  476, 404,  474, 406,  474, 404,  476, 404,  474, 1282,  478, 1282,  478, 1282,  478, 1280,  452, 1308,  490, 390,  476, 402,  476, 1282,  478, 1282,  478, 404,  476, 402,  474, 404,  476, 404,  476, 402,  476, 404,  476, 404,  476, 404,  474, 406,  474, 406,  474, 404,  476, 404,  474, 404,  476, 404,  476, 406,  474, 404,  476, 404,  476, 404,  476, 404,  474, 406,  474, 404,  474, 404,  476, 404,  482, 396,  474, 406,  474, 404,  476, 404,  474, 404,  476, 428,  434, 420,  480, 400,  482, 394,  476, 406,  474, 404,  474, 406,  474, 404,  476, 404,  474, 404,  476, 404,  474, 404,  474, 406,  474, 404,  476, 404,  474, 404,  476, 404,  476, 404,  474, 404,  476, 402,  476, 404,  476, 404,  474, 1282,  478, 1282,  478, 1282,  478, 406,  474, 404,  476, 404,  476, 1282,  476};  // MITSUBISHI_AC
uint8_t state[18] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x00, 0x20, 0x02, 0x30, 0xC0, 0x67, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x8E};

Timestamp : 003809.025
Library   : v2.7.14

Protocol  : MITSUBISHI_AC
Code      : 0x23CB26010020080230C06700000000000096 (144 Bits)
Mesg Desc.: Power: On, Mode: 1 (Heat), Temp: 18C, Fan: 0 (Auto), Swing(V): 0 (Auto), Swing(H): 3 (Middle), Clock: 17:10, On Timer: 00:00, Off Timer: 00:00, Timer: -
uint16_t rawData[583] = {3354, 1788,  478, 1286,  478, 1282,  478, 428,  452, 428,  452, 404,  476, 1280,  478, 404,  476, 410,  474, 1282,  476, 1282,  478, 404,  474, 1282,  478, 404,  476, 404,  474, 1282,  478, 1282,  478, 404,  476, 1280,  478, 1282,  478, 428,  450, 406,  474, 1280,  478, 404,  474, 406,  476, 1280,  478, 404,  476, 402,  474, 428,  452, 428,  452, 404,  474, 404,  476, 404,  476, 404,  474, 404,  474, 406,  474, 404,  476, 404,  476, 404,  474, 404,  474, 406,  474, 406,  474, 404,  474, 404,  474, 406,  474, 404,  474, 1280,  478, 404,  476, 406,  474, 404,  474, 404,  474, 406,  474, 1282,  478, 404,  476, 406,  474, 428,  450, 406,  474, 404,  476, 1282,  478, 404,  474, 404,  476, 404,  474, 404,  476, 428,  452, 406,  474, 404,  474, 404,  476, 404,  476, 404,  474, 1280,  480, 1282,  476, 404,  474, 406,  476, 404,  474,  [... cut ...] 4, 404,  476, 404,  476, 406,  474, 404,  476, 404,  474, 404,  474, 406,  476, 404,  476, 404,  558, 318,  452, 430,  476, 404,  476, 404,  474, 404,  476, 404,  476, 404,  476, 404,  476, 404,  476, 404,  474, 406,  474, 404,  476, 404,  474, 406,  476, 404,  476, 1282,  476, 1282,  478, 404,  474, 1282,  478, 404,  476, 404,  474, 1282,  474};  // MITSUBISHI_AC
uint8_t state[18] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x20, 0x08, 0x02, 0x30, 0xC0, 0x67, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x96};
```

crankyoldgit, 2021-02-04:
> "Power: Off, Mode: 4 (Auto)" ... [SNIP] ... By setting this to "Power: Off, Mode: 1 (Heat)", I was able to turn off the air conditioner.

Okay, but _how_ are you telling it to go to these modes? Please provide the code and/or detailed instructions/steps etc. There are multiple ways you could be doing this and I can't reproduce or diagnose the path without knowing this.

nao-pon, 2021-02-04:
I'm currently controlling with Tasmota --https://github.com/hristo-atanasov/Tasmota-IRHVAC --HomeAssistant.

When I send the OFF command in this environment, it is sent with his MQTT payload as follows.
```json
{"Vendor":"MITSUBISHI_AC","Model":-1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":18,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}
```

I was able to turn off the air conditioner by sending this payload part as follows:
```json
{"Vendor":"MITSUBISHI_AC","Model":-1,"Mode":"Heat","Power":"Off","Celsius":"On","Temp":18,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}
```

nao-pon, 2021-02-04:
@crankyoldgit Is this a problem with Tasmota-IRHVAC?

crankyoldgit, 2021-02-04:
I don't know yet. This is why we prefer people using our tools/examples or providing code as it eliminates a whole layer of complexity/potential bugs/problems.

nao-pon, 2021-02-04:
Yes, I'm sorry to report the problem in an ambiguous state. Also, thank you for taking your precious time.

Specifying the mode when turning off the air conditioner is not something that this library does, and if you need instructions from the user, I think this is a problem with Tasmota-IRHVAC.

nao-pon, 2021-02-04:
I found the cause of this problem in Tasmota. When Power is Off, Mode is forcibly rewritten to Off.

https://github.com/arendst/Tasmota/blob/62c95122f2d5c9fcfe3ec9dac32b072844db7db5/tasmota/xdrv_05_irremote_full.ino#L206-L213

I rewrote this as follows, and now I can control it by issuing the appropriate command in Tasmota-IRHVAC.

```c++
  // Home Assistant wants mode to be off if power is also off & vice-versa.
  if (state.mode == stdAc::opmode_t::kOff || !state.power) {
    json.add(PSTR(D_JSON_IRHVAC_MODE), IRac::opmodeToString(state.mode));
    json.add(PSTR(D_JSON_IRHVAC_POWER),  IRac::boolToString(false));
  } else {
    json.add(PSTR(D_JSON_IRHVAC_MODE), IRac::opmodeToString(state.mode));
    json.add(PSTR(D_JSON_IRHVAC_POWER), IRac::boolToString(state.power));
  }
```

crankyoldgit, 2021-02-04:
> I found the cause of this problem in Tasmota. When Power is Off, Mode is forcibly rewritten to Off.

Yes, This is typically required because Home Assistant like it to look that way.
Setting both to Power & Mode to "Off" should result in the internal MitsubishiAC class attempting to change it's _Mode_ to Off (which doesn't exist so it will default to "Auto") and changing the _Power_ state to off.
This should produce a valid "Power off the AC" message.

So you may be barking up the wrong tree.

nao-pon, 2021-02-04:
For the time being, I made a [PR](https://github.com/arendst/Tasmota/pull/10829) for Tasmota. HomeAssistant core-2021.1.5 doesn't seem to have any problems.

Thank you very much.

closed 2021-02-04 by nao-pon