crankyoldgit/IRremoteESP8266 #1403: [Mitsubishi] support weekly timer

nao-pon, 2021-02-04:
### Version/revision of the library used
Current master
https://github.com/crankyoldgit/IRremoteESP8266/commit/ab466e4dac234a433880d424bbff930315649f12

### Describe the bug
This is not a bug.

Brand: Mitsubishi Electric,  Model: MLZ-RX5017AS A/C (MITSUBISHI_AC)
Brand: Mitsubishi Electric,  Model: SG153/M21EDF426 remote (MITSUBISHI_AC)

The above device is equipped with a weekly timer, and it seems that the on / off of the timer stored in the main unit is transmitted by a remote code.

It is included in all commands, so any control in this library will always turn off the weekly timer.

#### Example code used
IRrecvDumpV3

#### Expected behaviour
I would like to control the weekly timer with this library if possible. Or I want to be able to set irregular commands as the default.

#### Output of raw data from [IRrecvDumpV3.ino]
- Weekly timer ON
```
Timestamp : 000047.298
Library   : v2.7.14

Protocol  : MITSUBISHI_AC
Code      : 0x23CB26010020080400C05E0000080300006A (144 Bits)
Mesg Desc.: Power: On, Mode: 1 (Heat), Temp: 20C, Fan: 0 (Auto), Swing(V): 0 (Auto), Swing(H): 0 (UNKNOWN), Clock: 15:40, On Timer: 00:00, Off Timer: 00:00, Timer: -
uint16_t rawData[583] = {3372, 1658,  434, 1238,  438, 1240,  434, 426,  410, 426,  412, 406,  432, 1238,  436, 426,  412, 424,  412, 1264,  412, 1240,  436, 426,  412, 1238,  438, 426,  412, 406,  432, 1240,  436, 1240,  436, 426,  412, 1240,  436, 1240,  436, 424,  412, 406,  432, 1240,  436, 426,  412, 406,  432, 1240,  436, 404,  432, 404,  432, 426,  412, 426,  412, 406,  432, 406,  430, 426,  412, 426,  412, 406,  432, 406,  458, 372,  438, 426,  412, 406,  432, 426,  412, 426,  412, 424,  412, 426,  412, 404,  432, 426,  412, 426,  412, 1238,  436, 404,  434, 426,  412, 426,  412, 404,  434, 424,  412, 1264,  412, 426,  412, 426,  412, 426,  412, 426,  412, 406,  432, 406,  432, 1240,  436, 404,  432, 406,  432, 406,  432, 406,  432, 404,  432, 406,  432, 426,  412, 424,  412, 406,  432, 404,  432, 404,  432, 404,  432, 426,  412, 404,  432, 426,  412, 426,  410, 406,  432, 404,  432, 406,  432, 1238,  438, 1240,  436, 404,  432, 1240,  436, 1240,  436, 1240,  436, 1238,  436, 426,  412, 1238,  436, 426,  412, 404,  432, 426,  412, 404,  434, 426,  412, 426,  412, 426,  412, 426,  412, 404,  432, 404,  432, 404,  432, 426,  412, 406,  432, 404,  432, 426,  412, 426,  412, 426,  412, 406,  432, 404,  432, 406,  432, 1242,  434, 404,  432, 404,  432, 426,  412, 404,  432, 1240,  436, 1240,  436, 426,  412, 426,  412, 406,  432, 426,  412, 426,  412, 404,  432, 426,  412, 404,  434, 426,  412, 404,  432, 426,  412, 426,  412, 404,  432, 426,  412, 404,  432, 426,  412, 406,  432, 426,  412, 406,  432, 406,  432, 426,  412, 404,  432, 404,  432, 1240,  434, 426,  410, 1240,  436, 404,  434, 1238,  462, 1214,  436, 404,  432, 11304,  3370, 1658,  436, 1240,  436, 1240,  436, 404,  432, 426,  412, 426,  412, 1238,  436, 404,  434, 404,  432, 1240,  436, 1240,  436, 406,  432, 1240,  436, 404,  432, 426,  410, 1240,  436, 1240,  436, 406,  432, 1240,  436, 1240,  434, 404,  432, 426,  412, 1240,  434, 406,  432, 404,  432, 1238,  438, 424,  412, 404,  432, 426,  412, 426,  412, 404,  432, 404,  432, 426,  410, 404,  434, 404,  432, 426,  412, 406,  432, 426,  412, 404,  432, 406,  432, 404,  432, 426,  412, 426,  412, 404,  432, 404,  432, 404,  434, 1240,  436, 404,  432, 404,  432, 404,  434, 404,  432, 404,  432, 1240,  436, 426,  412, 406,  432, 404,  432, 404,  432, 426,  412, 426,  412, 1240,  436, 404,  432, 404,  432, 404,  434, 424,  412, 404,  432, 426,  412, 426,  412, 404,  432, 404,  432, 404,  434, 404,  432, 404,  434, 404,  432, 404,  432, 404,  432, 404,  434, 404,  432, 426,  412, 404,  432, 1240,  436, 1240,  436, 426,  412, 1242,  434, 1240,  436, 1240,  436, 1240,  434, 426,  412, 1240,  436, 404,  432, 404,  434, 404,  432, 426,  412, 404,  434, 426,  410, 426,  412, 426,  412, 404,  432, 426,  412, 426,  412, 404,  434, 404,  432, 404,  434, 424,  412, 426,  412, 404,  434, 404,  432, 404,  434, 404,  432, 1240,  436, 404,  432, 426,  412, 406,  432, 404,  434, 1240,  434, 1240,  436, 406,  432, 426,  410, 426,  410, 426,  412, 426,  412, 406,  432, 426,  412, 404,  432, 404,  432, 404,  434, 426,  412, 404,  432, 404,  432, 404,  432, 404,  432, 426,  410, 426,  410, 404,  434, 404,  432, 426,  412, 426,  410, 406,  432, 404,  434, 1238,  436, 426,  412, 1242,  434, 426,  412, 1240,  436, 1240,  436, 404,  434};  // MITSUBISHI_AC
uint8_t state[18] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x20, 0x08, 0x04, 0x00, 0xC0, 0x5E, 0x00, 0x00, 0x08, 0x03, 0x00, 0x00, 0x6A};
```

- Weekly timer OFF
```
Timestamp : 000053.082
Library   : v2.7.14

Protocol  : MITSUBISHI_AC
Code      : 0x23CB26010020080400C05E00000003000062 (144 Bits)
Mesg Desc.: Power: On, Mode: 1 (Heat), Temp: 20C, Fan: 0 (Auto), Swing(V): 0 (Auto), Swing(H): 0 (UNKNOWN), Clock: 15:40, On Timer: 00:00, Off Timer: 00:00, Timer: -
uint16_t rawData[583] = {3370, 1656,  436, 1242,  434, 1240,  436, 404,  434, 426,  412, 404,  432, 1240,  436, 424,  412, 426,  412, 1240,  434, 1238,  436, 426,  412, 1240,  436, 426,  412, 426,  410, 1240,  436, 1240,  436, 404,  434, 1238,  436, 1240,  436, 424,  412, 426,  412, 1238,  436, 426,  410, 404,  432, 1240,  436, 426,  412, 404,  432, 426,  412, 404,  432, 404,  432, 426,  412, 426,  412, 426,  412, 426,  412, 426,  412, 404,  432, 406,  432, 404,  432, 404,  432, 426,  412, 404,  432, 404,  432, 404,  432, 426,  412, 404,  434, 1238,  436, 404,  432, 426,  412, 426,  412, 426,  412, 424,  412, 1240,  434, 426,  412, 404,  432, 426,  410, 406,  432, 404,  432, 426,  412, 1240,  434, 426,  410, 426,  412, 426,  412, 404,  432, 426,  410, 426,  412, 406,  432, 426,  412, 404,  432, 426,  412, 426,  412, 426,  412, 426,  410, 426,  412, 426,  412, 426,  412, 426,  410, 426,  410, 426,  412, 1240,  436, 1240,  434, 426,  412, 1240,  434, 1240,  436, 1242,  434, 1238,  464, 372,  464, 1212,  436, 426,  412, 426,  412, 426,  412, 426,  414, 424,  410, 426,  412, 426,  412, 406,  432, 426,  412, 426,  412, 404,  432, 426,  412, 426,  412, 426,  412, 426,  412, 426,  412, 424,  412, 404,  434, 404,  432, 404,  432, 426,  412, 426,  412, 426,  412, 426,  412, 404,  434, 1240,  434, 1238,  438, 404,  432, 426,  412, 404,  432, 404,  432, 426,  412, 404,  432, 426,  410, 406,  432, 426,  412, 404,  432, 426,  410, 404,  434, 404,  434, 426,  412, 426,  412, 426,  412, 404,  432, 406,  432, 404,  434, 404,  432, 404,  432, 404,  432, 426,  412, 1240,  436, 404,  432, 426,  412, 426,  412, 1240,  436, 1240,  436, 426,  412, 11310,  3372, 1656,  436, 1238,  436, 1240,  436, 426,  412, 426,  412, 404,  434, 1264,  412, 426,  412, 404,  432, 1240,  434, 1238,  436, 404,  432, 1264,  412, 404,  432, 406,  432, 1240,  436, 1238,  438, 404,  432, 1240,  436, 1238,  436, 404,  432, 404,  432, 1240,  436, 404,  432, 426,  412, 1240,  460, 374,  436, 426,  412, 424,  412, 404,  432, 426,  412, 404,  432, 404,  432, 404,  434, 404,  432, 404,  434, 404,  434, 404,  432, 404,  432, 406,  432, 404,  434, 404,  432, 404,  432, 404,  434, 404,  432, 426,  412, 1240,  436, 404,  432, 404,  434, 424,  412, 426,  412, 406,  432, 1238,  436, 426,  410, 426,  412, 404,  434, 404,  432, 404,  434, 404,  432, 1240,  436, 426,  412, 404,  432, 406,  432, 426,  412, 404,  432, 406,  432, 426,  410, 406,  432, 426,  412, 404,  432, 404,  434, 404,  432, 404,  434, 404,  432, 404,  432, 406,  432, 406,  432, 424,  412, 404,  434, 1238,  438, 1264,  410, 406,  432, 1240,  436, 1240,  436, 1240,  436, 1240,  434, 404,  432, 1240,  436, 426,  410, 426,  412, 404,  432, 404,  432, 426,  412, 404,  434, 404,  432, 426,  412, 404,  432, 404,  432, 404,  432, 404,  432, 404,  432, 406,  432, 404,  432, 404,  432, 406,  432, 406,  432, 406,  432, 404,  432, 426,  412, 404,  432, 426,  412, 426,  412, 404,  432, 1240,  436, 1240,  436, 404,  432, 404,  432, 426,  412, 424,  412, 404,  432, 426,  410, 404,  434, 404,  434, 404,  432, 404,  434, 404,  434, 404,  434, 404,  434, 404,  432, 404,  432, 406,  432, 426,  412, 404,  434, 404,  434, 426,  412, 404,  432, 404,  460, 374,  438, 1238,  438, 404,  434, 404,  432, 404,  432, 1240,  436, 1240,  434, 404,  434};  // MITSUBISHI_AC
uint8_t state[18] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x20, 0x08, 0x04, 0x00, 0xC0, 0x5E, 0x00, 0x00, 0x00, 0x03, 0x00, 0x00, 0x62};
```

### What brand/model IR demodulator are you using?
HX1838: However, there seems to be no problem with the decoded data.

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Has this library/code previously worked as expected for you?
No

crankyoldgit, 2021-02-04:
_0x23CB26010020080400C05E00000**8**0300006A_ (on)
_0x23CB26010020080400C05E00000**0**03000062_ (off)

Looks like it's a single bit that needs to be flipped. So, this should be an easy addition to the `IRMitsubishiAC` class.
Note however, it could/will be controllable via `IRMitsubishiAC` but it can't be controlled by the user via the `IRac` class interface.

Which interface are you planning to use?

nao-pon, 2021-02-04:
@crankyoldgit I'm currently controlling with Tasmota <-> https://github.com/hristo-atanasov/Tasmota-IRHVAC <->HomeAssistant.

crankyoldgit, 2021-02-04:
I can add the support, but it won't be available via Tasmota, as I'm fairly confident it uses the `IRac`.
I can show you how to do a custom modification to the library to have that bit always set, but that will not be part of the standard library. i.e. You'll need to always roll/build/make your own tasmota binary.

**BUT** FYI, you are probably MUCH better off doing timer control via MQTT/tasmota/Home Assistant/etc than via the Mitsubishi A/C internal system. The former has far more functionality, scope, and ability, verses the latter which is quite limited.

nao-pon, 2021-02-04:
Yes, there is no problem with custom building and using Tasmota. I would like to keep the internal timer alive in case the Home Assistant stops.

I am very grateful for your kindness.

crankyoldgit, 2021-02-04:
Please download/test PR #1404 / Branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1403

It should add support for what you want.

For enabling it (by default) via `IRac` & thus Tasmota, I suggest you look here:
https://github.com/crankyoldgit/IRremoteESP8266/blob/05f48dc7d2ecacba311370646aa019ddc4de94e2/src/IRac.cpp#L1337-L1338

nao-pon, 2021-02-04:
@crankyoldgit Very quick Work! Thank you so much!

Yes, I now have control with the weekly timer enabled as I wish. 👍

closed 2021-02-04 by crankyoldgit

crankyoldgit, 2021-02-13:
FYI, the changes mentioned above have now been included in the new v2.7.15 release of the library.

commit 131c5540 (2021-02-04): MitsubishiAC: Add support for enabling Weekly Timer.
* Add `[s|g]etWeeklyTimerEnabled()`.
* Unit test coverage based on real data.
* Add comment on how to Weekly Timer enabling by default via `IRac` class.

Fixes #1403
PR #1404 (2021-02-04): MitsubishiAC: Add support for enabling Weekly Timer.
* Add `[s|g]etWeeklyTimerEnabled()`.
* Unit test coverage based on real data.
* Add comment on how to Weekly Timer enabling by default via `IRac` class.

Fixes #1403
commit 991bed82 (2021-02-04): MitsubishiAC: Add support for enabling Weekly Timer. (#1404)

* Add `[s|g]etWeeklyTimerEnabled()`.
* Unit test coverage based on real data.
* Add comment on how to add Weekly Timer enabling by default via `IRac` class.

Fixes #1403