crankyoldgit/IRremoteESP8266 #1405: web_ac_control example modification

Motarded603, 2021-02-04:
Hey, so I'm using the web_ac_control example for my own purposes. I'm trying to modify the code (the arduino, html/js) to instead be displaying Celsius on the webpage to Fahrenheit. I'm pretty sure I have changed everything in the ui.js file, but the webpage still shows it in Celsius.

Any help/advice is helpful. I added my source code in here to see my changes.
Thanks, Motarded.
[Fujitsu.zip](https://github.com/crankyoldgit/IRremoteESP8266/files/5928369/Fujitsu.zip)

crankyoldgit, 2021-02-04:
Ping @mariusmotea who is the author of that example code.

@Motarded603 In general we don't provide support for customising example code (see https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me) but I'll leave this Issue open for a while in case @mariusmotea or some kind soul wants to assist you.

FWIW, [IRMQTTServer](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRMQTTServer) provides a web interface, and allows control of any of our supported A/Cs via display/use of Fahrenheit or Celsius.

Motarded603, 2021-02-05:
@crankyoldgit Thanks for the response. Sorry, had to head to work after I posted this. I think I actually got it, I ended up using a second NodeMCU to test how it would handle with two of them because I'm doing a giant project. The second one shows Fahrenheit perfectly fine, but the first one I was testing with still is messed up. I'm thinking the uploaded files through SPIFFS is not updating like it should. I've tried googling how to clear the SPIFFS to upload fresh, but I'm not finding anything of that sorts? Any recommendations on this? Thanks.

crankyoldgit, 2021-02-05:
> I've tried googling how to clear the SPIFFS to upload fresh, but I'm not finding anything of that sorts? Any recommendations on this? Thanks.

"GitHub - espressif/esptool: Espressif SoC serial bootloader utility" https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region

Motarded603, 2021-02-05:
Okay, so I erased the SPIFFS... reuploaded, and it's not working like the second one that works flawlessly with the Fahrenheit instead of Celsius. Why would my changes not act like how I want it to on the second one vs the first considering I reuploaded the same code of the second to the first? This is really puzzling :/

Motarded603, 2021-02-05:
Ha! By golly I figured out the damn issue.
I remember this from my web programming class last sem... Google Chrome was saving in cache the old js, all I had to do was clear the cache of the page and it works flawlessly on both modules now! Ah... gotta love easy mistakes...

I think that solves this issue. I apparently had it fixed in the first place haha. Thanks for your help!

crankyoldgit, 2021-02-05:
Glad you have it sorted now.

closed 2021-02-05 by crankyoldgit