crankyoldgit/IRremoteESP8266 #1406: Examples missing.

Hairyloon, 2021-02-05:
I discovered this library from this project: https://www.instructables.com/WiFi-LED-Strip-Controller-Based-on-ESP-8266/ and I can't get it to work. I'm trying to migrate from the Arduino IDE to PlatformIO and not finding it entirely straightforward, but that's not your problem.

I'm getting an error saying: 'irsend' was not declared in this scope", another saying "identifier "irsend" is undefined" and a whole load saying "'irsend' is not captured"

In the process of trying to find answers to this (I've tried the "Troubleshooting" section), I found that your [examples](https://github.com/crankyoldgit/tree/master/examples) are missing. [This one](https://github.com/crankyoldgit/blob/master/examples/IRsendDemo/IRsendDemo.ino) as well.

I thought that was a proper thing to come and tell you about, so here I am doing that...

### Version/revision of the library used
2.7.14

crankyoldgit, 2021-02-05:
https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples
It's the correct URL.
Yours seem to be missing the project. i.e. "IRremoteESP8266".

crankyoldgit, 2021-02-06:
Also, looking at the Instructables source code/article you linked, it's using code from prior to v2.0 of the library.
You need to do the following: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Upgrading-to-v2.0

From the FAQ section in the wiki: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Im_getting_error_IRsend_does_not_name_a_type_messages

closed 2021-02-06 by crankyoldgit

Hairyloon, 2021-02-07:
> 
> 
> https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples
> It's the correct URL.
> Yours seem to be missing the project. i.e. "IRremoteESP8266".

I copied the links direct from your pages. https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide

> 
> Also, looking at the Instructables source code/article you linked, it's using code from prior to v2.0 of the library.
> You need to do the following: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Upgrading-to-v2.0
> 
Yeah, I tried that, and it still wouldn't have it.

crankyoldgit, 2021-02-07:
> I copied the links direct from your pages. https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide

Thanks. I've gone through every URL and I think I've corrected them all.

> Yeah, I tried that, and it still wouldn't have it.

Without seeing your code, I can't help. Have you tried the examples?

Hairyloon, 2021-02-11:
I haven't tried the examples, as I couldn't find them. I will do in due course.