crankyoldgit/IRremoteESP8266 #1408: Panasonic_AC32 power toggling on mode change

Fettkeewl, 2021-02-10:
Hi again!
I'm happy to say that theese changes made it into Tasmota :) and I've set it up and use it frequently in my home setup with home assistant. Thanks again!

I've however encountered 1 small problem with the lib in Tasmota, so I proceeded to testing out 2.7.14 on my Wemos again.
I have no super high needs for the control but:
* On/Off
* Toggle Mode
* Toggle Temp
* Toggle fanspeed

**The issue in Tasmota with your lib:**
From Tasmota when I send **Any mode other than OFF** to the HVAC it does a power toggle instead of only changing the **mode**... so I made myself an AC Send function that takes 3 parameters Mode, Temp, Fan speed and then sends the data.

AC Send with parameters below will change the AC settings accordingly and keep the AC running.
```yaml
Mode: Off
Temp: 16 - 30
FanSpeed: Auto/1-5
```
AC Send with parameters below will perform a power toggle
```yaml
Mode: Auto/Heat/Cool/Dry/Fan_Only
Temp: 16 - 30
FanSpeed: Auto/1-5
```

**The issue with 2.7.14 on my wemos:**
Mode changes work, power toggle only works when changing the **model** 

So as the behaviour is not the same I do not know how to Troubleshoot this without your help...
If you have time to help me let me know what data I can provide you with to further investigate this

Picture for reference
[image]

```yaml
# Tasmota send info
10:43:03.157 MQT: tele/TasmotaIR/RESULT = "Protocol":"PANASONIC_AC32","Bits":32,"Data":"0x0000000004F8369C","DataLSB":"0x00000000201F6C39","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC32","Model":-1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":23,"FanSpeed":"Auto","SwingV":"Highest","SwingH":"Auto","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}

# The JSON a little more readable
{ irhvac :
  {
    "Vendor": "PANASONIC_AC32",
    "Model": -1,
    "Mode": "Off",			# <-- Adjusted in HA 
    "Power": "Off",
    "Celsius": "On",
    "Temp": 23,				# <-- Adjusted in HA 
    "FanSpeed": "Auto",			# <-- Adjusted in HA 
    "SwingV": "Highest",
    "SwingH": "Auto",
    "Quiet": "Off",
    "Turbo": "Off",
    "Econo": "Off",
    "Light": "Off",
    "Filter": "Off",
    "Clean": "Off",
    "Beep": "Off",
    "Sleep": -1
  }
}
```

Heres what your library picks up from my remote when toggling the mode which
does not do a power toggle!
Somehow it does not identify the proper mode.
```yaml
11:08:19.924 MQT: tele/TasmotaIR/RESULT = "Protocol":"PANASONIC_AC32","Bits":32,"Data":"0x000000000A673690","DataLSB":"0x0000000050E66C09","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC32","Model":-1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":22,"FanSpeed":"Max","SwingV":"Highest","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
11:08:21.624 MQT: tele/TasmotaIR/RESULT = "Protocol":"PANASONIC_AC32","Bits":32,"Data":"0x000000000B643690","DataLSB":"0x00000000D0266C09","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC32","Model":-1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":19,"FanSpeed":"Max","SwingV":"Highest","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
11:08:23.475 MQT: tele/TasmotaIR/RESULT = "Protocol":"PANASONIC_AC32","Bits":32,"Data":"0x0000000009FC3690","DataLSB":"0x00000000903F6C09","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC32","Model":-1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":27,"FanSpeed":"Auto","SwingV":"Highest","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
11:08:25.075 MQT: tele/TasmotaIR/RESULT = "Protocol":"PANASONIC_AC32","Bits":32,"Data":"0x000000000D653690","DataLSB":"0x00000000B0A66C09","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC32","Model":-1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":20,"FanSpeed":"Max","SwingV":"Highest","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
11:08:26.623 MQT: tele/TasmotaIR/RESULT = "Protocol":"PANASONIC_AC32","Bits":32,"Data":"0x000000000CFB3690","DataLSB":"0x0000000030DF6C09","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC32","Model":-1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":26,"FanSpeed":"Auto","SwingV":"Highest","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
```

Heres 4 presses on the Power Toggle button on the remote 
(the power state is currently inverted so the last press which has "Power: off" actually turned on my AC)
```yaml
11:11:56.575 MQT: tele/TasmotaIR/RESULT = "Protocol":"PANASONIC_AC32","Bits":32,"Data":"0x0000000004FB3690","DataLSB":"0x0000000020DF6C09","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC32","Model":-1,"Mode":"Heat","Power":"On","Celsius":"On","Temp":26,"FanSpeed":"Auto","SwingV":"Highest","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
11:11:59.001 MQT: tele/TasmotaIR/RESULT = "Protocol":"PANASONIC_AC32","Bits":32,"Data":"0x0000000004FB3690","DataLSB":"0x0000000020DF6C09","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC32","Model":-1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":26,"FanSpeed":"Auto","SwingV":"Highest","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
11:12:01.624 MQT: tele/TasmotaIR/RESULT = "Protocol":"PANASONIC_AC32","Bits":32,"Data":"0x0000000004FB3690","DataLSB":"0x0000000020DF6C09","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC32","Model":-1,"Mode":"Heat","Power":"On","Celsius":"On","Temp":26,"FanSpeed":"Auto","SwingV":"Highest","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
11:12:03.624 MQT: tele/TasmotaIR/RESULT = "Protocol":"PANASONIC_AC32","Bits":32,"Data":"0x0000000004FB3690","DataLSB":"0x0000000020DF6C09","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC32","Model":-1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":26,"FanSpeed":"Auto","SwingV":"Highest","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
```

_Originally posted by @Fettkeewl in https://github.com/crankyoldgit/IRremoteESP8266/issues/1366#issuecomment-776601220_

crankyoldgit, 2021-02-10:
First off, let me correct an incorrect assumption and Tasmota-ism.
1) `mode`: "Off" -> "Auto" etc. does **NOT** change the power of an A/C. You **need** to use `power` for that.
  Tasmota may be doing that for you already. Don't know. Technically not this library's problem.
2) Tasmota doesn't provide the full/verbose decode of a protocol in it's output. e.g. When your capture is reported as "Power: Off" by Tasmota, the library is actually saying "Power Toggle: Off". i.e. No requested change of power detected in the IR message. Or it may be doing it correctly. I don't intimately know how Tasmota is using the library in this case.
i.e. The library doesn't report `mode` of `off` for a captured IR message, **ever**, as far as I recall.
e.g. In theory, this list of settings shouldn't be reported like this by the library. Something else in Tasmota is probably doing something to it.
```
11:08:19.924 MQT: tele/TasmotaIR/RESULT = "Protocol":"PANASONIC_AC32","Bits":32,"Data":"0x000000000A673690","DataLSB":"0x0000000050E66C09","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC32","Model":-1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":22,"FanSpeed":"Max","SwingV":"Highest","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
```

This is one of the **many** reasons we don't accept/like Tasmota _data_ in bug reports etc. Please use IRrecvDumpV2 or IRrecvDumpV3 etc. There are too many moving parts when using Tasmota which makes finding the real fault very very difficult.

> The issue with 2.7.14 on my wemos:
Mode changes work, power toggle only works when changing the model

That issue may be related to the above. (Point 1.) You don't seem to be adjusting `power` but I can't tell because you've provided no code or inadequate debug detail/logging etc.

Please note that `mode` = `off` and `power` = `off` is often tied together as a hack to deal with Home Assistant. Technically, that's done _outside_ of this library, in a project's core code.
e.g. (in one of our examples)
https://github.com/crankyoldgit/IRremoteESP8266/blob/56ee6933e7f13ea738a7d06955e15b47da289557/examples/IRMQTTServer/IRMQTTServer.h#L137-L151

Fettkeewl, 2021-02-13:
My apologies, I got my setup wrong, was missing a parameter for sending
that would store the states in tasmota :) hence making it work as intended (without trying to sync with physical remote)

closed 2021-02-13 by Fettkeewl