crankyoldgit/IRremoteESP8266 #1420: Smaller kToshibaAcMinGap

nao-pon, 2021-02-26:
### Version/revision of the library used
v2.7.15

### Describe the bug
My infrared remote control requires a smaller kToshibaAcMinGap.

#### Output of raw data from [IRrecvDumpV2.ino]
```
python auto_analyse_raw_data.py 'uint16_t rawData[295] = {4410, 4364,  558, 1606,  556, 1604,  558, 1604,  558, 1604,  558, 526,  554, 524,  556, 1606,  556, 526,  554, 526,  554, 526,  554, 526,  554, 526,  554, 1604,  558, 1602,  558, 526,  556, 1604,  558, 526,  554, 526,  556, 526,  554, 526,  554, 524,  556, 526,  554, 1602,  560, 1604,  558, 1602,  558, 1606,  558, 1604,  556, 1604,  560, 1604,  558, 1604,  556, 526,  556, 524,  556, 526,  554, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 522,  532, 1630,  558, 524,  556, 526,  556, 1604,  558, 1604,  558, 526,  554, 526,  554, 524,  554, 526,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 526,  554, 526,  556, 524,  556, 526,  554, 524,  556, 526,  554, 524,  556, 524,  556, 526,  556, 524,  554, 524,  556, 1604,  558, 1604,  558, 526,  554, 524,  556, 524,  556, 1604,  558, 4632,  4416, 4362,  556, 1604,  560, 1602,  558, 1604,  558, 1602,  558, 526,  556, 524,  556, 1604,  558, 524,  556, 524,  556, 526,  556, 524,  554, 526,  554, 1604,  558, 1606,  558, 528,  556, 1604,  560, 524,  554, 526,  554, 524,  556, 524,  556, 524,  556, 524,  556, 1604,  558, 1606,  556, 1604,  558, 1606,  558, 1604,  558, 1604,  556, 1606,  558, 1604,  558, 524,  554, 526,  554, 526,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 1604,  558, 524,  556, 524,  556, 1604,  558, 1606,  558, 524,  556, 524,  556, 524,  556, 524,  554, 526,  556, 524,  556, 528,  556, 524,  556, 524,  556, 526,  554, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 526,  556, 524,  556, 526,  556, 526,  554, 524,  556, 526,  554, 524,  556, 1604,  558, 1606,  556, 526,  554, 526,  554, 524,  556, 1604,  558};'
Found 295 timing entries.
Potential Mark Candidates:
[4416, 560]
Potential Space Candidates:
[4632, 4364, 1630, 528]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 4413
kHdrSpace  = 4363
kBitMark   = 555
kOneSpace  = 1604
kZeroSpace = 524
kSpaceGap = 4632

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+111100100000110100000011111111000000000100110000000000000000000000110001GAP(4632)
  Bits: 72
  Hex:  0xF20D03FC0130000031 (MSB first)
        0x8C00000C803FC0B04F (LSB first)
  Dec:  4465049936067165159473 (MSB first)
        2582544184064302166095 (LSB first)
  Bin:  0b111100100000110100000011111111000000000100110000000000000000000000110001 (MSB first)
        0b100011000000000000000000000011001000000000111111110000001011000001001111 (LSB first)
kHdrMark+kHdrSpace+111100100000110100000011111111000000000100110000000000000000000000110001
  Bits: 72
  Hex:  0xF20D03FC0130000031 (MSB first)
        0x8C00000C803FC0B04F (LSB first)
  Dec:  4465049936067165159473 (MSB first)
        2582544184064302166095 (LSB first)
  Bin:  0b111100100000110100000011111111000000000100110000000000000000000000110001 (MSB first)
        0b100011000000000000000000000011001000000000111111110000001011000001001111 (LSB first)

Total Nr. of suspected bits: 144
```

```
python auto_analyse_raw_data.py 'uint16_t rawData[327] = {4438, 4342,  580, 1604,  558, 1604,  558, 1580,  584, 1582,  580, 496,  556, 526,  584, 1580,  582, 522,  558, 496,  584, 498,  582, 496,  584, 498,  584, 1582,  580, 1580,  580, 498,  582, 1580,  582, 498,  584, 496,  582, 498,  582, 498,  582, 496,  584, 1578,  584, 498,  582, 498,  584, 1604,  558, 1582,  580, 1582,  582, 1578,  582, 1580,  584, 496,  582, 1582,  580, 1582,  580, 498,  582, 498,  584, 496,  584, 496,  584, 1580,  580, 498,  582, 498,  582, 1582,  582, 496,  582, 498,  582, 1582,  580, 1604,  558, 498,  582, 496,  584, 496,  584, 496,  582, 498,  582, 496,  584, 498,  582, 496,  584, 498,  582, 498,  582, 498,  582, 1582,  580, 498,  582, 498,  582, 498,  582, 496,  584, 498,  584, 498,  582, 496,  584, 498,  582, 498,  582, 496,  584, 502,  582, 498,  584, 498,  582, 498,  582, 496,  584, 1604,  558, 498,  584, 496,  582, 1582,  582, 1580,  580, 1604,  558, 498,  582, 522,  528, 1610,  582, 4632,  4414, 4366,  558, 1604,  558, 1582,  580, 1604,  558, 1582,  580, 496,  584, 498,  582, 1580,  582, 522,  558, 496,  584, 496,  584, 498,  582, 496,  584, 1582,  580, 1604,  556, 500,  582, 1604,  558, 500,  580, 498,  582, 498,  582, 498,  584, 498,  582, 1580,  582, 498,  582, 498,  582, 1578,  582, 1582,  580, 1582,  580, 1582,  580, 1582,  582, 498,  582, 1602,  560, 1580,  582, 498,  582, 500,  580, 496,  582, 498,  582, 1604,  558, 498,  582, 500,  580, 1584,  580, 498,  582, 498,  582, 1582,  582, 1580,  582, 498,  582, 498,  582, 498,  582, 498,  584, 498,  582, 500,  580, 496,  584, 498,  582, 522,  558, 498,  582, 500,  580, 1582,  580, 498,  582, 498,  556, 524,  584, 498,  582, 498,  582, 522,  558, 498,  582, 500,  582, 498,  582, 500,  580, 496,  584, 498,  582, 500,  580, 500,  582, 498,  582, 1604,  558, 500,  582, 496,  582, 1606,  558, 1580,  582, 1582,  580, 496,  582, 498,  584, 1578,  582};'
Found 327 timing entries.
Potential Mark Candidates:
[4438, 584]
Potential Space Candidates:
[4632, 4366, 1610, 526]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 4426
kHdrSpace  = 4354
kBitMark   = 578
kOneSpace  = 1587
kZeroSpace = 499
kSpaceGap = 4632

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+11110010000011010000010011111011000010010011000000000001000000000000000100111001GAP(4632)
  Bits: 80
  Hex:  0xF20D04FB093001000139 (MSB first)
        0x9C8000800C90DF20B04F (LSB first)
  Dec:  1143052855418109451829561 (MSB first)
        739050390611712853192783 (LSB first)
  Bin:  0b11110010000011010000010011111011000010010011000000000001000000000000000100111001 (MSB first)
        0b10011100100000000000000010000000000011001001000011011111001000001011000001001111 (LSB first)
kHdrMark+kHdrSpace+11110010000011010000010011111011000010010011000000000001000000000000000100111001
  Bits: 80
  Hex:  0xF20D04FB093001000139 (MSB first)
        0x9C8000800C90DF20B04F (LSB first)
  Dec:  1143052855418109451829561 (MSB first)
        739050390611712853192783 (LSB first)
  Bin:  0b11110010000011010000010011111011000010010011000000000001000000000000000100111001 (MSB first)
        0b10011100100000000000000010000000000011001001000011011111001000001011000001001111 (LSB first)

Total Nr. of suspected bits: 160
```

### What brand/model IR demodulator are you using?
Toshiba WH-UB03NJ Remote commander.
[image]

### What brand/model IR demodulator are you using?
VS1838B

### I have followed the steps in the [Troubleshooting Guide]
Yes, But as partial as possible.

### Has this library/code previously worked as expected for you?
No, I haven't tried previous versions.

### Other useful information
```diff
diff --git a/src/ir_Toshiba.cpp b/src/ir_Toshiba.cpp
index bb6d1a7..9829ed9 100644
--- a/src/ir_Toshiba.cpp
+++ b/src/ir_Toshiba.cpp
@@ -27,7 +27,7 @@ const uint16_t kToshibaAcHdrSpace = 4300;
 const uint16_t kToshibaAcBitMark = 580;
 const uint16_t kToshibaAcOneSpace = 1600;
 const uint16_t kToshibaAcZeroSpace = 490;
-const uint16_t kToshibaAcMinGap = 7400;
+const uint16_t kToshibaAcMinGap = 4600;
```
I was able to successfully decode my infrared remote with the above changes.

crankyoldgit, 2021-02-27:
Thanks for the report and data. I'll look in to it soon.
In the meantime, can you confirm if **sending** via the unmodified library works to control your AC successful?
The answer to that question will influence how I modify the library.

Thanks in advance

nao-pon, 2021-02-27:
@crankyoldgit Thank you for your reply. When I tried sending with the unmodified one, the air conditioner is generally working fine.

What is a little worrisome is that when the air conditioner receives a signal, it makes two beeping sounds. When operated with a genuine remote controller, the received sound is only once.

And, The type of air conditioner is RAS-2558V.

nao-pon, 2021-02-27:
When I tried this change, I heard two beeps in the same way. Therefore, there was no clear difference in the reception status of the air conditioner between the one that was not changed and the one that was changed.

crankyoldgit, 2021-02-27:
That's interesting. Can you please post your example code that's producing the two beeps?

Thanks for confirming the unmodified version still sends correctly. That means I can probably get away just tweaking the decoder code to match either values, as we have previous data indicating the `7400` is also correct/expected on some remotes

nao-pon, 2021-02-27:
I'm using Tasmota and HomeAssistant. I also use Mitsubishi and Panasonic, but I never get two beeps.

However, this could also be due to Tasmota or HomeAssistant, so I'll try it with examples / TurnOnToshibaAC. We will inform you of the results at a later date.

crankyoldgit, 2021-02-27:
I've looked at the code, and it seems in order to set the desired swing state of the Toshiba A/C we have to send a separate "short" command/message. So, when using the interface that Tasmota uses, yes, it will send two messages. One to set most of the desired state of the A/C, the second, to set the desired Swing state of the A/C. So two beeps is expected in that use case.

No need to chase it down etc.

crankyoldgit, 2021-02-28:
Hey @nao-pon , can you please download/test PR #1421 / branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1420 and see if it works as you expect, and let us know etc.

nao-pon, 2021-02-28:
@crankyoldgit Thank you for your quick response. I will try it and report the results.

nao-pon, 2021-03-01:
@crankyoldgit I verified it on the PR # 1421 branch. Both IR transmission and reception worked fine. thanks you.

I found another problem. This model has his swing ON / OFF toggled and the same signal is sent. At present, it seems that this signal cannot be supported.

Swing ON / OFF (Toggle) : 0xF20D01FE210425

Byte[5] becomes `04`.

NiKiZe, 2021-03-01:
> @crankyoldgit I verified it on the PR #1421 branch. Both IR transmission and reception worked fine. thanks you.
> 
> I found another problem. This model has his swing ON / OFF toggled and the same signal is sent. At present, it seems that this signal cannot be supported.
> 
> Swing ON / OFF (Toggle) : 0xF20D01FE210425
> 
> Byte[5] becomes `04`.

It is a little unclear what the issue is here, maybe a full message dump would help to explain it?

nao-pon, 2021-03-01:
@crankyoldgit Sorry... The following is a complete message dump.
```
Timestamp : 000068.313Library   : v2.7.15
Protocol  : TOSHIBA_ACCode      : 0xF20D01FE210425 (56 Bits)Mesg Desc.: Temp: 17C, Swing(V): 0 (Step)
uint16_t rawData[231] = {4412, 4364,  556, 1608,  554, 1608,  554, 1606,  558, 1604,  558, 526,  554, 530,  550, 1630,  532, 548,  532, 548,  532, 548,  532, 548,  532, 526,  554, 1608,  554, 1606,  556, 548,  532, 1608,  554, 548,  532, 548,  532, 528,  552, 528,  552, 548,  532, 548,  532, 528,  552, 1606,  556, 1630,  532, 1606,  556, 1606,  558, 1604,  558, 1606,  556, 1606,  556, 1608,  554, 548,  532, 548,  532, 550,  532, 1608,  554, 528,  552, 528,  552, 528,  554, 548,  532, 1630,  532, 548,  532, 548,  532, 548,  532, 526,  554, 548,  532, 1606,  556, 548,  532, 528,  552, 526,  554, 548,  532, 1608,  554, 526,  554, 526,  554, 1606,  556, 548,  532, 1606,  554, 4634,  4412, 4366,  556, 1608,  554, 1606,  556, 1606,  556, 1608,  554, 548,  532, 526,  554, 1608,  556, 548,  532, 526,  554, 526,  554, 526,  554, 528,  552, 1606,  556, 1608,  554, 548,  532, 1604,  556, 528,  552, 528,  552, 528,  552, 548,  532, 548,  532, 548,  532, 548,  532, 1606,  556, 1608,  556, 1608,  554, 1606,  556, 1630,  532, 1606,  556, 1606,  556, 1608,  554, 526,  554, 526,  554, 528,  552, 1608,  554, 548,  532, 526,  554, 526,  554, 548,  532, 1606,  556, 548,  532, 548,  532, 548,  532, 548,  532, 526,  552, 1608,  556, 548,  532, 526,  554, 548,  532, 548,  532, 1606,  556, 526,  554, 548,  532, 1606,  556, 548,  532, 1630,  532};  // TOSHIBA_ACuint8_t state[7] = {0xF2, 0x0D, 0x01, 0xFE, 0x21, 0x04, 0x25};
```

Also, it may be related in swing mode.

When we operate the swing with the IR remote control of the air conditioner, the temperature is treated as a fixed value of 17 ℃, which causes the temperature to be 17 ℃ on the UI side. Is it possible to use state.model to detect anything about swing mode operation?

In short, I want to detect that the received signal is in swing mode setting in the case of going through Tasmota.

crankyoldgit, 2021-03-01:
> In short, I want to detect that the received signal is in swing mode setting in the case of going through Tasmota.

Can you please try phrasing this a different way, or elaborating more on this issue? Actually, can you create a new issue for this please? This is now off-topic from the original issue. In short, I don't understand what you're trying to say or want to happen here.

> @crankyoldgit Sorry... The following is a complete message dump.
>
> ```
> Timestamp : 000068.313Library   : v2.7.15
> Protocol  : TOSHIBA_ACCode      : 0xF20D01FE210425 (56 Bits)Mesg Desc.: Temp: 17C, [...]

This signal can be reproduced via `sendToshibaAC()` and swing can be controlled by the `IRac` class interface that Tasmota uses. The latter approach can't send that message discretely though (I think).

e.g.
```
uint8_t state[7] = {0xF2, 0x0D, 0x01, 0xFE, 0x21, 0x04, 0x25};
irsend.sendToshibaAC(state, 7);
```

The underlying `IRToshibaAC` class only supports turning the swing `On` or `Off`, or `Step`ing the swing, the `IRac` class (i.e. Tasmota) does not have access to `Step`.

crankyoldgit, 2021-03-01:
> @crankyoldgit I verified it on the PR #1421 branch. Both IR transmission and reception worked fine. thanks you.

Thanks for confirming it works as expected.

closed 2021-03-01 by crankyoldgit

nao-pon, 2021-03-01:
Thanks you. Yes, I create another issues.

crankyoldgit, 2021-03-24:
FYI, the aforementioned changes have been included in the new v2.7.16 release for the library.

PR #1421 (2021-03-01): ToshibaAC: Adjust inter-message gap timing to improve matching.
* Allow for a shorted inter-message gap to handle decoding a WH-UB03NJ remote.
  - A/C does accept the longer inter-message gap. (Confirmed)
* Add unit test to confirm operation.
* Update supported models.

Kudos to @nao-pon for collecting/analysing the data.

Fixes #1420
commit 0d2dec39 (2021-03-01): ToshibaAC: Adjust inter-message gap timing to improve matching. (#1421)

* Allow for a shorted inter-message gap to handle decoding a WH-UB03NJ remote.
  - A/C does accept the longer inter-message gap. (Confirmed)
* Add unit test to confirm operation.
* Update supported models.

Kudos to @nao-pon for collecting/analysing the data.

Fixes #1420