crankyoldgit/IRremoteESP8266 #1423: [Toshiba AC] support swing toggle button

nao-pon, 2021-03-01:
### Version/revision of the library used
v2.7.15

### Describe the bug
There is a remote control with a button to turn the swing on and off. The button toggles the swing ON / OFF. There is no difference between ON and OFF.

#### To Reproduce
When submitted with that button, the library will detect it as "Swing (V): 0 (Step)".

#### Example code used
IRrecvDumpV2

#### Expected behaviour
I want the library to detect it as "Swing (V): 4 (On / Off)".

#### Output of raw data from [IRrecvDumpV2.ino]
Each time you press the swing button, the next signal will be sent.

```
Timestamp : 000068.313Library   : v2.7.15
Protocol  : TOSHIBA_ACCode      : 0xF20D01FE210425 (56 Bits)Mesg Desc.: Temp: 17C, Swing(V): 0 (Step)
uint16_t rawData[231] = {4412, 4364,  556, 1608,  554, 1608,  554, 1606,  558, 1604,  558, 526,  554, 530,  550, 1630,  532, 548,  532, 548,  532, 548,  532, 548,  532, 526,  554, 1608,  554, 1606,  556, 548,  532, 1608,  554, 548,  532, 548,  532, 528,  552, 528,  552, 548,  532, 548,  532, 528,  552, 1606,  556, 1630,  532, 1606,  556, 1606,  558, 1604,  558, 1606,  556, 1606,  556, 1608,  554, 548,  532, 548,  532, 550,  532, 1608,  554, 528,  552, 528,  552, 528,  554, 548,  532, 1630,  532, 548,  532, 548,  532, 548,  532, 526,  554, 548,  532, 1606,  556, 548,  532, 528,  552, 526,  554, 548,  532, 1608,  554, 526,  554, 526,  554, 1606,  556, 548,  532, 1606,  554, 4634,  4412, 4366,  556, 1608,  554, 1606,  556, 1606,  556, 1608,  554, 548,  532, 526,  554, 1608,  556, 548,  532, 526,  554, 526,  554, 526,  554, 528,  552, 1606,  556, 1608,  554, 548,  532, 1604,  556, 528,  552, 528,  552, 528,  552, 548,  532, 548,  532, 548,  532, 548,  532, 1606,  556, 1608,  556, 1608,  554, 1606,  556, 1630,  532, 1606,  556, 1606,  556, 1608,  554, 526,  554, 526,  554, 528,  552, 1608,  554, 548,  532, 526,  554, 526,  554, 548,  532, 1606,  556, 548,  532, 548,  532, 548,  532, 548,  532, 526,  552, 1608,  556, 548,  532, 526,  554, 548,  532, 548,  532, 1606,  556, 526,  554, 548,  532, 1606,  556, 548,  532, 1630,  532};  // TOSHIBA_ACuint8_t state[7] = {0xF2, 0x0D, 0x01, 0xFE, 0x21, 0x04, 0x25};
```

### What brand/model IR demodulator are you using?
VS1838B

### I have followed the steps in the [Troubleshooting Guide]
Yes

### Has this library/code previously worked as expected for you?
No. Maybe.

crankyoldgit, 2021-03-01:
Ah .. I get it now. I think your device differs from the previous models we support, in that they have a discrete `on` and `off` code, not a toggle one. Have you tried the on & off codes to see if your A/C will accept those codes, even though your remote doesn't?

nao-pon, 2021-03-01:
@crankyoldgit I haven't tested sending swing on and swing off commands to the air conditioner yet. Try it and report the results.

crankyoldgit, 2021-03-01:
FYI, the codes to try for discrete swing on & off should be:

```c++
uint8_t swingOnState[7] = {0xF2, 0x0D, 0x01, 0xFE, 0x21, 0x01, 0x20};
uint8_t swingOffState[7] = {0xF2, 0x0D, 0x01, 0xFE, 0x21, 0x02, 0x23};

// Turn on the swing
irsend.sendToshibaAC(swingOnState, 7);

// Turn off the swing
irsend.sendToshibaAC(swingOffState, 7);
```

crankyoldgit, 2021-03-01:
Can you please try out branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1423 and let me know how it goes.
It should handle your "swing toggle" button better now.

nao-pon, 2021-03-01:
I tried swing command acceptance. The air conditioner accepted the swing on and swing off commands and worked perfectly.

And I'm going to try https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1423.

nao-pon, 2021-03-01:
I have confirmed by https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1423 that the swing toggle button is detected correctly. Thank you!

nao-pon, 2021-03-01:
For the time being, when I build with Tasmota with this fix applied, ESP (ESP-01) will restart when the swing command is accepted.

If I undo this, it won't reboot.

crankyoldgit, 2021-03-01:
I've pushed another commit to that branch. It may fix the crash issue. Not sure, but I can then say with more confidence that it isn't the libary's fault after this fix.

Fingers crossed for you.

closed 2021-03-02 by crankyoldgit

crankyoldgit, 2021-03-24:
FYI, the aforementioned changes have been included in the new v2.7.16 release for the library.

PR #1425 (2021-03-02): ToshibaAC: Swing handling and `setRaw()` improvements.
* Add support for the Swing Toggle setting.
  - Turns out there is another setting for swing. This means it needs to grow from 2 bits to 3 bits. Adjustments as needed.
* Fix potential memory crash issue with `setRaw()`
  - Make `setRaw()` memory safer by passing a length arg.
  - Update calls to `IRToshibaAC::setRaw()`
* Unit test coverage.

Fixes #1423 

* Try to ignore changing the `IRac` class temp when we get an inbound swing IR command.
* May fix some upstream implementations & setups. e.g. Tasmota & Home Assistant.
  - Can't hurt at  [... cut]
commit e3c1208e (2021-03-02): ToshibaAC: Swing handling and `setRaw()` improvements. (#1425)

* Add support for the Swing Toggle setting.
  - Turns out there is another setting for swing. This means it needs to grow from 2 bits to 3 bits. Adjustments as needed.
* Fix potential memory crash issue with `setRaw()`
  - Make `setRaw()` memory safer by passing a length arg.
  - Update calls to `IRToshibaAC::setRaw()`
* Unit test coverage.

Fixes #1423 

* Try to ignore changing the `IRac` class temp when we get an inbound swing IR command.
* May fix some upstream implementations & setups. e.g. Tasmota & Home Assistant.
  - Can't [... cut]