crankyoldgit/IRremoteESP8266 #1424: [Toshiba AC] I want Tasmota to understand the reception of swing mode.

nao-pon, 2021-03-01:
### Version/revision of the library used
v2.7.15

### Describe the bug
This is not a bug.

The Toshiba air conditioner library treats the temperature as 17 ° C when receiving an IR signal in swing mode.

Tasmota publishes an MQTT message based on the temperature information, so the client who receives the MQTT recognizes that the temperature has been changed to 17 ° C.

Therefore, I wanted to include an MQTT message that the reception was in swing mode in some way. My idea is to use state.model. In the case of Mitsubishi air conditioner, the fixed value is "Model": -1, but in the swing mode, it is "Model": 0. I don't know if that's a good idea.

However, I would like to somehow announce in an MQTT message that I am receiving swing mode.

#### Output of raw data from [IRrecvDumpV2.ino]
Heat, 20℃, ON
```
Timestamp : 000007.345Library   : v2.7.15
Protocol  : TOSHIBA_ACCode      : 0xF20D03FC0130070036 (72 Bits)Mesg Desc.: Temp: 20C, Power: Off, Fan: 0 (Auto), Turbo: Off, Econo: Off
uint16_t rawData[295] = {4410, 4362,  558, 1604,  560, 1606,  556, 1606,  556, 1606,  556, 548,  532, 528,  552, 1608,  554, 548,  534, 524,  554, 526,  554, 548,  532, 548,  532, 1606,  556, 1608,  556, 526,  554, 1604,  558, 526,  554, 548,  532, 548,  532, 548,  532, 526,  554, 548,  534, 1608,  552, 1606,  558, 1606,  556, 1606,  556, 1606,  556, 1606,  556, 1606,  558, 1604,  556, 548,  532, 526,  554, 528,  552, 548,  532, 528,  554, 548,  532, 548,  532, 548,  532, 526,  554, 1606,  556, 526,  554, 548,  532, 1606,  556, 1606,  556, 548,  532, 548,  532, 526,  554, 526,  552, 528,  552, 548,  532, 528,  554, 548,  532, 548,  532, 1606,  556, 1606,  556, 1606,  556, 528,  554, 526,  554, 548,  532, 548,  532, 526,  554, 526,  554, 526,  554, 528,  552, 548,  532, 528,  554, 1606,  556, 1606,  556, 548,  532, 1604,  558, 1606,  556, 548,  532, 4632,  4414, 4364,  558, 1604,  556, 1606,  556, 1606,  556, 1604,  558, 548,  532, 526,  552, 1606,  556, 548,  532, 526,  552, 528,  554, 548,  532, 528,  554, 1630,  532, 1606,  556, 526,  554, 1608,  556, 548,  532, 548,  532, 526,  554, 526,  554, 526,  554, 526,  552, 1630,  532, 1608,  554, 1608,  554, 1606,  558, 1630,  534, 1606,  556, 1606,  556, 1606,  558, 546,  532, 546,  532, 548,  532, 548,  532, 548,  532, 528,  554, 548,  532, 548,  532, 526,  554, 1606,  556, 548,  532, 526,  554, 1604,  558, 1604,  558, 526,  554, 548,  532, 526,  554, 548,  532, 526,  554, 548,  532, 528,  554, 546,  532, 526,  554, 1608,  554, 1606,  558, 1606,  556, 528,  552, 528,  552, 548,  532, 548,  532, 526,  554, 548,  532, 548,  532, 548,  532, 528,  554, 548,  532, 1606,  556, 1606,  556, 528,  554, 1608,  554, 1608,  554, 528,  552};  // TOSHIBA_ACuint8_t state[9] = {0xF2, 0x0D, 0x03, 0xFC, 0x01, 0x30, 0x07, 0x00, 0x36};
```
Heat, 20℃, Off
```
Timestamp : 000009.267Library   : v2.7.15
Protocol  : TOSHIBA_ACCode      : 0xF20D03FC0130030032 (72 Bits)Mesg Desc.: Temp: 20C, Power: On, Mode: 3 (Heat), Fan: 0 (Auto), Turbo: Off, Econo: Off
uint16_t rawData[295] = {4412, 4390,  532, 1628,  534, 1608,  554, 1606,  556, 1606,  558, 548,  532, 548,  532, 1606,  556, 528,  554, 526,  552, 548,  532, 526,  554, 548,  532, 1606,  556, 1606,  556, 528,  552, 1608,  554, 548,  532, 528,  554, 526,  554, 526,  554, 548,  532, 548,  532, 1606,  558, 1606,  556, 1604,  558, 1606,  556, 1606,  558, 1604,  558, 1606,  556, 1606,  556, 548,  532, 526,  554, 526,  554, 526,  554, 548,  532, 548,  532, 526,  554, 526,  554, 548,  532, 1630,  534, 526,  552, 528,  554, 1606,  556, 1606,  556, 548,  532, 548,  532, 526,  554, 548,  532, 526,  554, 526,  554, 550,  532, 526,  554, 526,  554, 548,  532, 1606,  556, 1606,  558, 526,  554, 526,  554, 526,  554, 528,  552, 528,  554, 548,  532, 526,  554, 548,  532, 526,  552, 548,  532, 1604,  556, 1606,  558, 528,  554, 526,  554, 1606,  556, 526,  554, 4658,  4390, 4340,  580, 1604,  556, 1606,  556, 1606,  556, 1604,  558, 526,  554, 526,  554, 1606,  556, 528,  552, 528,  552, 526,  554, 548,  532, 526,  554, 1606,  556, 1608,  558, 526,  554, 1604,  556, 526,  554, 548,  532, 528,  554, 526,  554, 548,  532, 548,  532, 1606,  556, 1604,  556, 1606,  558, 1606,  556, 1604,  556, 1608,  554, 1630,  532, 1634,  532, 526,  554, 548,  532, 548,  532, 528,  552, 526,  554, 548,  532, 528,  554, 526,  554, 526,  554, 1630,  532, 530,  554, 526,  554, 1606,  556, 1604,  558, 526,  554, 526,  554, 526,  554, 526,  554, 528,  554, 526,  552, 550,  532, 526,  554, 526,  554, 526,  552, 1608,  556, 1604,  558, 526,  554, 526,  554, 548,  532, 548,  532, 550,  532, 548,  532, 548,  532, 526,  554, 526,  552, 530,  550, 1608,  556, 1606,  556, 526,  554, 526,  554, 1606,  556, 526,  554};  // TOSHIBA_ACuint8_t state[9] = {0xF2, 0x0D, 0x03, 0xFC, 0x01, 0x30, 0x03, 0x00, 0x32};
```
Swing On/Off
```
Timestamp : 000012.392Library   : v2.7.15
Protocol  : TOSHIBA_ACCode      : 0xF20D01FE210425 (56 Bits)Mesg Desc.: Temp: 17C, Swing(V): 0 (Step)
uint16_t rawData[231] = {4414, 4364,  558, 1604,  556, 1606,  556, 1606,  556, 1604,  558, 548,  532, 548,  532, 1606,  556, 548,  530, 548,  532, 526,  554, 548,  532, 548,  532, 1606,  556, 1604,  558, 548,  532, 1604,  558, 526,  554, 526,  554, 528,  554, 548,  532, 548,  532, 548,  532, 526,  554, 1604,  556, 1604,  558, 1608,  556, 1606,  558, 1604,  556, 1606,  558, 1606,  556, 1604,  558, 548,  532, 528,  552, 548,  532, 1604,  558, 548,  532, 548,  532, 526,  554, 548,  532, 1606,  556, 548,  532, 548,  532, 548,  532, 526,  554, 548,  532, 1606,  556, 548,  532, 526,  554, 548,  532, 548,  532, 1606,  558, 548,  532, 548,  532, 1608,  554, 548,  532, 1604,  558, 4636,  4410, 4368,  556, 1606,  556, 1606,  556, 1604,  558, 1606,  556, 526,  554, 526,  554, 1606,  558, 548,  532, 526,  554, 526,  554, 548,  532, 548,  532, 1606,  556, 1606,  556, 548,  532, 1606,  558, 526,  554, 548,  532, 528,  552, 548,  532, 548,  532, 528,  552, 548,  532, 1604,  558, 1606,  558, 1606,  556, 1606,  556, 1606,  558, 1606,  556, 1606,  556, 1606,  556, 528,  552, 528,  554, 548,  532, 1606,  558, 526,  554, 548,  534, 526,  554, 526,  554, 1604,  558, 526,  554, 548,  532, 548,  532, 528,  552, 548,  532, 1606,  558, 528,  552, 548,  532, 548,  532, 526,  554, 1606,  558, 526,  554, 528,  554, 1606,  556, 526,  552, 1606,  558};  // TOSHIBA_ACuint8_t state[7] = {0xF2, 0x0D, 0x01, 0xFE, 0x21, 0x04, 0x25};
```
Swing Step
```
Timestamp : 000013.760Library   : v2.7.15
Protocol  : TOSHIBA_ACCode      : 0xF20D01FE210021 (56 Bits)Mesg Desc.: Temp: 17C, Swing(V): 0 (Step)
uint16_t rawData[115] = {4414, 4366,  556, 1606,  556, 1630,  532, 1606,  556, 1606,  556, 526,  554, 548,  532, 1606,  558, 548,  532, 546,  532, 548,  532, 548,  532, 548,  532, 1604,  558, 1606,  558, 548,  532, 1606,  556, 550,  532, 526,  554, 548,  532, 548,  532, 526,  554, 548,  532, 548,  532, 1606,  556, 1604,  556, 1606,  558, 1606,  556, 1604,  558, 1606,  556, 1604,  558, 1606,  558, 548,  532, 526,  554, 548,  532, 1632,  530, 548,  532, 548,  532, 548,  532, 526,  554, 1604,  556, 528,  554, 548,  532, 526,  554, 526,  554, 548,  532, 526,  554, 548,  532, 526,  554, 548,  532, 526,  554, 1606,  556, 526,  554, 526,  554, 548,  532, 526,  554, 1606,  556};  // TOSHIBA_ACuint8_t state[7] = {0xF2, 0x0D, 0x01, 0xFE, 0x21, 0x00, 0x21};
```

### What brand/model IR demodulator are you using?
VS1838B

### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc. ESP8266 or ESP32 board type._

### I have followed the steps in the [Troubleshooting Guide]
Yes

crankyoldgit, 2021-03-01:
FYI, I've just updated the branch again. Can you please try out branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1423 (yes, same as the other issue) and let me know how it goes with this.
It should now (depending on how Tasmota does things internally, which I can't control) hopefully not over-ride the "existing" temp with the 17C value when it receives/decodes a ToshibaAC IR swing message.

Fingers crossed it works for you. If it doesn't, it may require changes in Tasmota, not this library.

nao-pon, 2021-03-01:
@crankyoldgit I tried it with the latest https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1423, but the crash still doesn't go away. I will verify the Tasmota source code.

crankyoldgit, 2021-03-01:
> @crankyoldgit I tried it with the latest https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1423, but the crash still doesn't go away. I will verify the Tasmota source code.

Well, without more info/data, I don't think I can trace it down to any part of the library.

nao-pon, 2021-03-01:
I'm so sorry. The crash was due to my mistake. IRac.cpp was not updated. I'm sorry to confuse you.

As a result, all functions worked fine. No need to change Tasmota, Home Assistant.

Thank you very much for your wonderful support!

nao-pon, 2021-03-01:
@crankyoldgit I found one problem.

1. Swing mode is Off.
2. Turn on swing mode via Tasmota. (Swing: On)
3. Press the swing toggle button on the remote control. (Swing: Off)
4. But IR library status reports On. (Actually it is Off)
5. This makes the swing state the opposite of the actual state.

crankyoldgit, 2021-03-01:
> 3\. Press the swing toggle button on the remote control. (Swing: Off)
>     But IR library status reports On. (Actually it is Off)

Can you please supply data and elaborate on this? (including how you've worked this out) I've added some unit tests to simulate what should happen in this scenario, and I can't seem to reproduce your problem. See: https://github.com/crankyoldgit/IRremoteESP8266/commit/b4596e6fe284dae5408b76fd45bbdd1a9d7afeb9 i.e. It might be a tasmota problem. Not sure yet.

nao-pon, 2021-03-02:
Yes, I took the data. I haven't solved this yet.

Ignore the "Clock" and "Weekday" in the data as they are my customizations. Please let me know if you need uncustomized data.

1. Turn on the air conditioner with the IR remote control. Swing is off.

```
Timestamp : 000824.880Library   : v2.7.15
Protocol  : TOSHIBA_ACCode      : 0xF20D03FC0100030002 (72 Bits)Mesg Desc.: Temp: 17C, Power: On, Mode: 3 (Heat), Fan: 0 (Auto), Turbo: Off, Econo: Off
uint16_t rawData[295] = {4416, 4364,  560, 1602,  560, 1604,  560, 1602,  560, 1602,  560, 524,  556, 524,  556, 1602,  560, 548,  532, 524,  556, 524,  556, 526,  556, 524,  556, 1604,  560, 1602,  560, 524,  558, 1602,  558, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 1604,  558, 1604,  558, 1606,  558, 1604,  558, 1604,  558, 1604,  558, 1604,  558, 1604,  558, 524,  556, 524,  558, 524,  556, 524,  556, 524,  556, 524,  556, 548,  532, 524,  556, 524,  556, 1604,  558, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 1604,  558, 1604,  560, 524,  556, 524,  558, 522,  556, 524,  556, 522,  558, 524,  556, 524,  558, 524,  556, 524,  556, 524,  558, 524,  556, 524,  556, 524,  558, 522,  558, 1602,  560, 526,  556, 4632,  4418, 4342,  578, 1602,  560, 1604,  560, 1604,  560, 1604,  558, 524,  556, 524,  556, 1604,  558, 524,  558, 522,  558, 524,  556, 524,  556, 524,  558, 1602,  560, 1604,  558, 524,  556, 1604,  558, 524,  556, 524,  556, 524,  558, 522,  558, 524,  556, 524,  558, 1602,  560, 1604,  558, 1604,  560, 1602,  560, 1604,  558, 1604,  560, 1604,  560, 1602,  560, 524,  556, 524,  558, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 1604,  560, 522,  558, 524,  556, 524,  556, 522,  558, 524,  558, 522,  558, 522,  556, 524,  556, 524,  558, 522,  558, 524,  558, 524,  556, 524,  556, 524,  556, 1602,  560, 1602,  560, 524,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  558, 522,  558, 524,  556, 524,  556, 524,  558, 524,  556, 524,  556, 524,  558, 522,  558, 1602,  560, 524,  556};  // TOSHIBA_ACuint8_t state[9] = {0xF2, 0x0D, 0x03, 0xFC, 0x01, 0x00, 0x03, 0x00, 0x02};
```
```
09:32:29.825 MQT: tele/tasmota_0A708A/RESULT = {"IrReceived":{"Protocol":"TOSHIBA_AC","Bits":72,"Data":"0xF20D03FC0100030002","Repeat":0,"IRHVAC":{"Vendor":"TOSHIBA_AC","Model":-1,"Mode":"Heat","Power":"On","Celsius":"On","Temp":17,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1,"Clock":-1,"Weekday":20}}}
```

2. Turn on the swing with Tasmota.

```
09:32:45.777 MQT: stat/tasmota_0A708A/RESULT = {"IRHVAC":{"Vendor":"TOSHIBA_AC","Model":-1,"Mode":"Heat","Power":"On","Celsius":"On","Temp":17,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1,"Clock":572,"Weekday":1}}
```
```
Timestamp : 000840.919Library   : v2.7.15
Protocol  : TOSHIBA_ACCode      : 0xF20D03FC0100030002 (72 Bits)Mesg Desc.: Temp: 17C, Power: On, Mode: 3 (Heat), Fan: 0 (Auto), Turbo: Off, Econo: Off
uint16_t rawData[527] = {4430, 4290,  640, 1554,  640, 1548,  642, 1550,  638, 1548,  642, 442,  638, 442,  638, 1550,  638, 452,  638, 442,  612, 466,  636, 444,  636, 442,  638, 1550,  640, 1550,  640, 442,  638, 1550,  614, 468,  638, 442,  638, 440,  638, 442,  638, 442,  612, 468,  638, 1550,  640, 1550,  640, 1550,  640, 1548,  642, 1548,  640, 1548,  642, 1550,  640, 1548,  640, 442,  636, 444,  636, 442,  636, 442,  638, 440,  638, 442,  638, 442,  610, 468,  638, 442,  612, 1576,  640, 442,  638, 442,  638, 442,  638, 440,  638, 442,  638, 442,  638, 442,  612, 470,  612, 468,  638, 442,  638, 442,  638, 440,  612, 466,  612, 468,  612, 1576,  640, 1548,  616, 468,  636, 442,  638, 442,  638, 442,  612, 468,  612, 468,  612, 468,  638, 442,  612, 468,  638, 442,  636, 442,  612, 466,  612, 468,  638, 442,  638, 1548,  640, 446,  612, 7374,  4432, 4280,  640, 1552,  638, 1550,  614, 1576,  640, 1550,  640, 442,  612, 468,  638, 1550,  640, 442,  638, 442,  636, 442,  638, 442,  638, 466,  612, 1550,  640, 1550,  612, 468,  610, 1576,  614, 468,  612, 468,  636, 442,  612, 466,  638, 442,  638, 442,  638, 1550,  640, 1550,  614, 1574,  614, 1576,  612, 1576,  614, 1576,  614, 1574,  614, 1576,  614, 466,  612, 470,  610, 468,  612, 468,  612, 468,  612, 468,  612, 468,  612, 466,  638, 442,  612, 1578,  614, 468,  638, 442,  612, 468,  612, 468,  612, 468,  612, 466,  612, 468,  636, 444,  612, 466,  612, 468,  610, 468,  612, 468,  612, 468,  638, 442,  612, 1576,  614, 1574,  614, 468,  612, 468,  612, 466,  612, 468,  612, 466,  612, 468,  612, 468,  612, 468,  636, 442,  612, 468,  612, 466,  638, 440,  612, 468,  612, 468,  612, 1576,  614, 468,  610, 7402,  4426, 4282,  614, 1574,  614, 1574,  640, 1550,  614, 1576,  614, 468,  612, 468,  612, 1574,  614, 468,  612, 468,  612, 468,  612, 468,  612, 468,  612, 1576,  614, 1574,  614, 468,  612, 1576,  614, 468,  612, 466,  612, 468,  612, 468,  610, 468,  612, 468,  612, 468,  612, 1578,  612, 1574,  614, 1576,  614, 1574,  614, 1576,  614, 1574,  614, 1576,  614, 1574,  614, 468,  612, 468,  612, 468,  612, 1576,  614, 468,  612, 468,  612, 468,  612, 468,  612, 1578,  612, 468,  612, 466,  612, 468,  612, 468,  612, 468,  612, 468,  612, 468,  612, 1576,  614, 468,  6 [... cut ...] // TOSHIBA_ACuint8_t state[7] = {0xF2, 0x0D, 0x01, 0xFE, 0x21, 0x04, 0x25};
```
```
09:32:57.633 MQT: tele/tasmota_0A708A/RESULT = {"IrReceived":{"Protocol":"TOSHIBA_AC","Bits":56,"Data":"0xF20D01FE210425","Repeat":0,"IRHVAC":{"Vendor":"TOSHIBA_AC","Model":-1,"Mode":"Heat","Power":"On","Celsius":"On","Temp":17,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1,"Clock":-1,"Weekday":20}}}
```

crankyoldgit, 2021-03-02:
Thanks for the data.

Looking at the data, and this is only an educated guess without pouring over the Tasmota code, I _think_ Tasmota is not passing the correct "previous"/existing known state of the A/C to [`IRAcUtils::decodeToState`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/namespaceIRAcUtils.html#ac5eb498bf12cb6cba023c9c1e9726949) when it makes that call.

If Tasmota doesn't. The default for `swingv` is off, so getting a _Toggle_ message will result in `swingv` being `true`/"on" etc.
i.e. The code (in that branch) basically says, _If  the swing setting is "Toggle", then if the previous state of swingv was **not** "on", i.e. "off", make it "on"/Auto, and if it is something other than "off" (i.e. On, typically auto) then make it "off"_. e.g. Invert it.
See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/b4596e6fe284dae5408b76fd45bbdd1a9d7afeb9/src/ir_Toshiba.cpp#L429-L434

TL;DR: I don't think this is the Library's problem anymore. It's now the problem of the code/project that uses the library, I think.

nao-pon, 2021-03-02:
@crankyoldgit Thanks you. I will check the source code of Tasmota.

nao-pon, 2021-03-02:
@crankyoldgit After all, Tasmota had the cause of this problem. So I created a [PR](https://github.com/arendst/Tasmota/pull/11188).

Thank you very much! 👍

crankyoldgit, 2021-03-02:
Glad you found the problem.

closed 2021-03-02 by crankyoldgit

crankyoldgit, 2021-03-24:
FYI, the aforementioned changes have been included in the new v2.7.16 release for the library.

commit ea757ac5 (2021-03-01): ToshibaAC: Handle swing messages better.
* Try to ignore changing the `IRac` class temp when we get a swing command.
* May fix some upstream implementations & setups. e.g. Tasmota & Home Assistant.
  - Can't hurt at least.

Fixes #1424
PR #1425 (2021-03-02): ToshibaAC: Swing handling and `setRaw()` improvements.
* Add support for the Swing Toggle setting.
  - Turns out there is another setting for swing. This means it needs to grow from 2 bits to 3 bits. Adjustments as needed.
* Fix potential memory crash issue with `setRaw()`
  - Make `setRaw()` memory safer by passing a length arg.
  - Update calls to `IRToshibaAC::setRaw()`
* Unit test coverage.

Fixes #1423 

* Try to ignore changing the `IRac` class temp when we get an inbound swing IR command.
* May fix some upstream implementations & setups. e.g. Tasmota & Home Assistant.
  - Can't hurt at  [... cut]
commit e3c1208e (2021-03-02): ToshibaAC: Swing handling and `setRaw()` improvements. (#1425)

* Add support for the Swing Toggle setting.
  - Turns out there is another setting for swing. This means it needs to grow from 2 bits to 3 bits. Adjustments as needed.
* Fix potential memory crash issue with `setRaw()`
  - Make `setRaw()` memory safer by passing a length arg.
  - Update calls to `IRToshibaAC::setRaw()`
* Unit test coverage.

Fixes #1423 

* Try to ignore changing the `IRac` class temp when we get an inbound swing IR command.
* May fix some upstream implementations & setups. e.g. Tasmota & Home Assistant.
  - Can't [... cut]