crankyoldgit/IRremoteESP8266 #1426: sendRaw

yaroshd81, 2021-03-05:
Some protokols have pulses more 0xFFFF (MITSUBISHI112 and other). If i want send ir-command by function sendRaw, i cant send this big pulses. Do you can todo new function, for example
```void IRsend::sendRaw(const uint16_t buf[], const uint16_t len, const uint16_t hz, uint8_t rawTick) {
  // Set IR carrier frequency
  enableIROut(hz);
  for (uint16_t i = 0; i < len; i++) {
    if (i & 1) {  // Odd bit.
      space(buf[i]*rawTick);
    } else {  // Even bit.
      mark(buf[i]*rawTick);
    }
  }
  ledOff();  // We potentially have ended with a mark(), so turn of the LED.
}
```
functions `space` and `mark` have params uint32_t.
In my code i add this function, but if i renew library from repository i will have some problems :).
Sory for my bad english.

crankyoldgit, 2021-03-05:
**TL;DR**: I think your mistake is thinking that the internal storage/capture format is the same as the "raw" format used by `sendRaw()`. It isn't. You can't just simply re-use the internal capture array for that.

First off, no protocol I've ever encountered has a `mark()` of `0xFFFF` (65535) _or larger_ , including the Mitsubishi ones. , `spaces()` on the other hand can & do. Hence why `space()` accepts a `uint32_t`, and `mark()`s don't. If you've got a `mark` pulse that long, you've _likely_ mis-captured the message.

Next, `IRrecvDumpV2+` correctly converts a capture into a valid `rawData[]` array for use with `sendRaw()` when it encounters any messages that have pulses longer than `0xFFFF`. See the following:
https://github.com/crankyoldgit/IRremoteESP8266/blob/c81809172208cff2d2e2ac3f1e1d9837baa63ae4/src/IRutils.cpp#L178-L249

>       space(buf[i]*rawTick);
>     } else {  // Even bit.
>       mark(buf[i]*rawTick);

You've added `*rawTick` here. That is not how `sendRaw()` is supposed to work.
e.g.
https://github.com/crankyoldgit/IRremoteESP8266/blob/c81809172208cff2d2e2ac3f1e1d9837baa63ae4/src/IRsend.cpp#L531-L551

I am guessing you want send the native/underlying [`rawbuf`](https://github.com/crankyoldgit/IRremoteESP8266/blob/c81809172208cff2d2e2ac3f1e1d9837baa63ae4/src/IRrecv.h#L68-L80) structure used in a capture via the `sendRaw()` command. The short answer is, you can't. `sendRaw()` is an establish format, it's not the same as how it is stored in [`rawbuf`](https://github.com/crankyoldgit/IRremoteESP8266/blob/c81809172208cff2d2e2ac3f1e1d9837baa63ae4/src/IRrecv.h#L68-L80).

You probably want to use the following code to convert to the `sendRaw()` format. i.e. [`resultToRawArray()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/IRutils_8cpp.html#a7b3bbfa1f2bf2dea2fc40a2fefe05a2a)
https://github.com/crankyoldgit/IRremoteESP8266/blob/c81809172208cff2d2e2ac3f1e1d9837baa63ae4/examples/DumbIRRepeater/DumbIRRepeater.ino#L113-L124
in the [`DumbIRRepeater.ino`](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/DumbIRRepeater/DumbIRRepeater.ino) example code.

crankyoldgit, 2021-03-15:
Another chase-up ping.

crankyoldgit, 2021-03-20:
Closing due to inactivity/no response.

closed 2021-03-20 by crankyoldgit