crankyoldgit/IRremoteESP8266 #1427: IRMQTTServer example compiling but not running

adeltc, 2021-03-06:
### Version/revision of the library used
2.7.15 master on Mars 6th

### Describe the bug
I'm trying to compile the example (without any modification) of IRMQTTServer example. It compile and upload with no issue on ESP8622 (ESP-12) but it putting random characters in the consul. 

I set the serial speed to 115200. and configured the Arduino IDE to Board "NodeMCU 1.0" with flash size : 1M FS and 1019KB for OTA, should leave 2MB for the program

#### To Reproduce
just compile and upload the IRMQTTServer.ino to a nodemcu board

crankyoldgit, 2021-03-06:
By default (i.e. unmodified), that example doesn't produce much, if any serial output at 115200.
It's all web based.
See the instructions in the code at:
https://github.com/crankyoldgit/IRremoteESP8266/blob/c81809172208cff2d2e2ac3f1e1d9837baa63ae4/examples/IRMQTTServer/IRMQTTServer.ino#L45

You may need to do a full flash erase if you've previously had a WiFiManager config on the device previously.

If you want serial output, you'll need to enable debugging:
https://github.com/crankyoldgit/IRremoteESP8266/blob/c81809172208cff2d2e2ac3f1e1d9837baa63ae4/examples/IRMQTTServer/IRMQTTServer.h#L279

See also https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Some_of_the_examples_dont_produce_output_that_I_expect_and_I_have_to_use_a_baud_rate_of_74880_Whats_up

crankyoldgit, 2021-03-09:
How are you going with this?

adeltc, 2021-03-11:
sorry for the late reply, it's all good now, Thanks! I can output when to change the value of DEBUG to true, I can also see the SSID broadcasted now. Not sure why I was not seeing before even that I flashed all the memory, not just the sketch.

closed 2021-03-12 by crankyoldgit