crankyoldgit/IRremoteESP8266 #1428: irsend.sendNEC sends the wrong code

dannutu, 2021-03-08:
### Version/revision of the library used: 
IRremoteESP8266 2.7.15

### Describe the bug
Running on ESP32-CAM with an IR LED (940nm, Vf=1.2V, I=100mA) driven from GPIO14 and +3.3V via a PN2222.
Using IRsendDemo with two hardcoded sendNEC calls:
irsend.sendNEC(0x00FF0DF2UL,32) consistently sends  0x00FFF807   (!!)
irsend.sendNEC(0x00FF1FE0UL,32) consistently sends 0x00FFB04F (!!)
The sent codes are received and decoded completely separately with a TSOP4838 connected to a Saleae logic analyzer; the NEC protocol is decoded by sigrok.
Testing with the original remote control sigrok shows the correct codes (0x00FF0DF2 and 0x00FF1FE0) every time.
I went through the examples, FAQ, troubleshooting guide and I am at a complete loss as why this happens - any idea?

#### To Reproduce
see above

#### Example code used
see above

#### Expected behaviour
_What steps did you do and what should it have done? A clear and concise description of what you expected to happen._

#### Output of raw data from [IRrecvDumpV2.ino]
I'd probably need a 2nd ESP32 for this

### What brand/model IR demodulator are you using?
Vishay TSOP4838

### I have followed the steps in the [Troubleshooting Guide]
Yes

### Has this library/code previously worked as expected for you?
No, this is the first attempt to use it

### Other useful information

crankyoldgit, 2021-03-08:
@dannutu I think the issue you are experiencing is that this library & your logic analyser/sigrok use different numeric representations of the NEC code.
i.e. If you were to decode your sent signal with this library, it would produce the same number as is sent. e.g. If you captured the first signal from the original remote via this library, it would probably produce a different code than what you got via the logic analyser.

Sadly, I can't see an easy way to convert the code from your analyser to the library's code or vise-versa. :(

The **TL;DR**: is it looks like you can't use your current method of decoding to provide the correct numeric value to use with `sendNEC()`.

P.S. You can replace `irsend.sendNEC(0x00FF0DF2UL,32)` with `irsend.sendNEC(0x00FF0DF2UL)`. You don't need to supply the Nr of Bits for a protocol if you want to send the default bit size for a protocol **and** you only are supplying a value to send. e.g. No non-standard nr. of repeats etc.

crankyoldgit, 2021-03-15:
Friendly chase up ping.

dannutu, 2021-03-15:
Hi 

Thanks for your answer last week - you were of course right :)

It's ok to close this issue if you wish (I can receive, decode and resend codes using only IRremoteESP8266 and the receiving devices actually recognize them), however please do read on as I still have a strange issue receiving IR codes and although it's not directly related to IRremoteESP8266 it would be great if you could offer any advice. 

First, the source for the difference in codes. I actually initially recorded them with ReceiveDump from Arduino-IRremote. For some reason (I didn't spend any time to sort it out) I couldn't get Arduino-IRremote on an ESP32 to send back the codes so I moved to IRremoteESP8266 assuming the two libraries use the same representation for the same NEC codes. When I found out that my devices didn't react to the codes I checked them with the logic analyzer and sigrok and I discovered that the codes sent by IRremoteESP8266 were different. That's when I created this issue. 

Then I didn't have time to play again with this due to work commitments. Then I managed to fry my logic analyzer by very briefly touching the wrong connectors and supplying it with negative voltage. Then I ordered another logic analyzer, which I received on Saturday, but I found out it wasn't working at all...

I finally spent some time over the week-end trying to sort this out without logic analyzers. I built a second IR send/receive circuit using exactly the same type of power supply, ESP32 dev board, TSOP4838, capacitor, resistors, transistor, LEDs and even wires as for the first circuit. The idea was to send the codes with one board and receive them with the other, both using the same library - IRremoteESP8266.

And this is where the strange thing happened: the second circuit would have a lot, and I really mean a lot of spurious errors while receiving. I swapped parts in and out (ESP32 boards, PSUs, TSOP4838 and the passive components) and it would still do the same. To give you an idea of how bad it was: it would be enough to slowly move my hand above the circuit to generate receive errors! This, while no IR codes (38kHz or otherwise) were sent (the first circuit was switched off at this point and I wasn't using any IR remote). If on the other hand I used the first circuit to receive the codes it would work flawlessly under exactly the same conditions. I read again the troubleshooting guide but I couldn't find anything relevant so I'm wondering what could be the cause. Any hints would be greatly appreciated even if, like I said in the beginning, I appreciate this is not directly related to IRremoteESP8266. 

Many thanks!

dannutu, 2021-03-15:
BTW, if you're curious, here's a side by side comparison for the IR codes sent by a remote and decoded using Arduino-IRremote (on the left side) and IRremoteESP8266 (on the right side). I recorded each code 3 times and each of the 3 codes for each key were consistent while using the same library. You can spot some patterns (i.e the first code 0xF609FF00 vs. 0xFF906F, which you get if you change the order of both bytes and nibbles from MSB to LSB: F6 becomes 6F and it will be the LSB, then 09 becomes 90 and so on) but not all codes follow the same rule (in fact most don't).

`IRremote:                                                                         IRremoteESP8266:         
=========                                                                         ================
Protocol=NEC Address=0x0 Command=0x9  Raw-Data=0xF609FF00 32 bits LSB first <---> Code : 0xFF906F (32 Bits)
Protocol=NEC Address=0x0 Command=0x1D Raw-Data=0xE21DFF00 32 bits LSB first <---> Code : 0xFFB847 (32 Bits)
Protocol=NEC Address=0x0 Command=0x15 Raw-Data=0xEA15FF00 32 bits LSB first <---> Code : 0xFFA857 (32 Bits)
Protocol=NEC Address=0x0 Command=0x1F Raw-Data=0xE01FFF00 32 bits LSB first <---> Code : 0xFFF807 (32 Bits)
Protocol=NEC Address=0x0 Command=0xD  Raw-Data=0xF20DFF00 32 bits LSB first <---> Code : 0xFFB04F (32 Bits)
Protocol=NEC Address=0x0 Command=0x19 Raw-Data=0xE619FF00 32 bits LSB first <---> Code : 0xFF9867 (32 Bits)
Protocol=NEC Address=0x0 Command=0x1B Raw-Data=0xE41BFF00 32 bits LSB first <---> Code : 0xFFD827 (32 Bits)
Protocol=NEC Address=0x0 Command=0x11 Raw-Data=0xEE11FF00 32 bits LSB first <---> Code : 0xFF8877 (32 Bits)
Protocol=NEC Address=0x0 Command=0x17 Raw-Data=0xE817FF00 32 bits LSB first <---> Code : 0xFFE817 (32 Bits)
Protocol=NEC Address=0x0 Command=0x12 Raw-Data=0xED12FF00 32 bits LSB first <---> Code : 0xFF48B7 (32 Bits)
Protocol=NEC Address=0x0 Command=0x16 Raw-Data=0xE916FF00 32 bits LSB first <---> Code : 0xFF6897 (32 Bits)
Protocol=NEC Address=0x0 Command=0x4D Raw-Data=0xB24DFF00 32 bits LSB first <---> Code : 0xFFB24D (32 Bits)
Protocol=NEC Address=0x0 Command=0x40 Raw-Data=0xBF40FF00 32 bits LSB first <---> Code : 0xFF02FD (32 Bits)
Protocol=NEC Address=0x0 Command=0x4C Raw-Data=0xB34CFF00 32 bits LSB first <---> Code : 0xFF32CD (32 Bits)
Protocol=NEC Address=0x0 Command=0x4  Raw-Data=0xFB04FF00 32 bits LSB first <---> Code : 0xFF20DF (32 Bits)
Protocol=NEC Address=0x0 Command=0x0  Raw-Data=0xFF00FF00 32 bits LSB first <---> Code : 0xFF00FF (32 Bits)
Protocol=NEC Address=0x0 Command=0xA  Raw-Data=0xF50AFF00 32 bits LSB first <---> Code : 0xFF50AF (32 Bits)
Protocol=NEC Address=0x0 Command=0x1E Raw-Data=0xE11EFF00 32 bits LSB first <---> Code : 0xFF7887 (32 Bits)
Protocol=NEC Address=0x0 Command=0xE  Raw-Data=0xF10EFF00 32 bits LSB first <---> Code : 0xFF708F (32 Bits)
Protocol=NEC Address=0x0 Command=0x1A Raw-Data=0xE51AFF00 32 bits LSB first <---> Code : 0xFF58A7 (32 Bits)
Protocol=NEC Address=0x0 Command=0x1C Raw-Data=0xE31CFF00 32 bits LSB first <---> Code : 0xFF38C7 (32 Bits)
Protocol=NEC Address=0x0 Command=0x14 Raw-Data=0xEB14FF00 32 bits LSB first <---> Code : 0xFF28D7 (32 Bits)
Protocol=NEC Address=0x0 Command=0xF  Raw-Data=0xF00FFF00 32 bits LSB first <---> Code : 0xFFF00F (32 Bits)
Protocol=NEC Address=0x0 Command=0xC  Raw-Data=0xF30CFF00 32 bits LSB first <---> Code : 0xFF30CF (32 Bits)
`

dannutu, 2021-03-15:
Sorry, it's not easily readable even while using the code tags.

closed 2021-03-15 by dannutu

crankyoldgit, 2021-03-15:
You poor bugger! You really are having a rough time of it.

First off, thanks for confirming there is no issue with the library per se. I'm surprised Arduino-IRremote produces a different code for NEC messages now, as this library was a fork of it. I'm guessing they must have changed something. I might peer into there code to see what's changed.

As for your hardware problems, I'd have guessed the IR receiver module was bad, but you swapped out they an have the same result.

Moving your hand over the circuit, to me, is only going to provide reflections of other IR signals, or induce some induction or capacitance in your circuit.
If there is a poor connection, that might effect it.
You seem to have swapped out every component except the breadboard (I'm assuming your using one), if suggest trying a different one or shifting all the components to different positions incase there is a bad spot/track giving you a poor connection.
 Other then that, I'm out of suggestions.

crankyoldgit, 2021-03-15:
Try using `reverseBits()` to convert between the two. See https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/IRutils_8cpp.html#a366219b6f1c46f41c6573b3e5e875e41
eg.
```c++
#include <IRutils.h>

uint32_t old_lib_code = 0xF609FF00;
irsend.sendNEC(reverseBits(old_lib_code, 32));
```

dannutu, 2021-03-15:
Thanks for the very quick answer. I actually thought about the breadboard, so I changed that as well, I just forgot to mention it.
I remember I noticed the link between Arduino-IRremote and IRremoteESP8266 (in fact that's how I got to IRremoteESP8266) and that was the reason I opened this issue initially, as I (foolishly) assumed that they would represent the same IR codes (especially from the same "manufacturer" - NEC in this case) in exactly the same way.
I wasn't prepared to spend so much time on this, I (again, foolishly) assumed it will be a quick couple of hours at most. I ended up spending an entire Sunday building and rebuilding circuits and I'm still not sure what the issue is. My gut feeling is capacitance but why would it make such a difference to fool the library into detecting ghost codes of wildly different bit lengths? 

I went through 4 different ESP32 boards, 2 breadboards, 3 PSUs, 4 TSOP4838 modules in various permutations. All had receive errors, while the original board didn't have any! That's what's so puzzling. I will rebuild it once more without a breadboard (using only Dupont wires) and if it works fine, if not I give up trying to explain it.

Thanks also for the reverseBits trick, I won't need it since I'm perfectly happy now using only your library, in the beginning I was reluctant to switch over to it because I spent time recording all my remote control units using Arduino-IRremote but since then I came up with a much faster way to do it by capturing the serial console output (/dev/ttyUSB0) and processing it using standard unix cmdline utilities (cu, grep, sort etc.). now it takes me just a couple of minutes per remote control (the time it takes me to press each key three times and go through all the keys in a consistent order - top left to bottom right, like handwriting).

crankyoldgit, 2021-03-16:
> I remember I noticed the link between Arduino-IRremote and IRremoteESP8266 (in fact that's how I got to IRremoteESP8266) and that was the reason I opened this issue initially, as I (foolishly) [...]

Well, it used to be backwards compatibility. I guess they have changed.

Re: hardware.

Hmmmm. If it's constantly giving you odd signals, it could be the data line is floating. Or as you're using an ESP32, could you have likeed the input pin with 5V? The Arduino family and esp8266 can take 3.3V & 5V input, the ESP32 can only take 3.3v.
Is the IR demodulator rated to work at 3.3v?

crankyoldgit, 2021-03-16:
See: https://github.com/Arduino-IRremote/Arduino-IRremote in the upgrading to v3.0 section. They state at the end that sendNEC has changed.

dannutu, 2021-03-21:
> See: https://github.com/Arduino-IRremote/Arduino-IRremote in the upgrading to v3.0 section. They state at the end that sendNEC has changed.

> Re: hardware.
> Hmmmm. If it's constantly giving you odd signals, it could be the data line is floating. Or as you're using an ESP32, could you have linked the input pin with 5V? (Ie. damaging it) The Arduino family and esp8266 can take 3.3V & 5V input, the ESP32 can only take 3.3v.
> Is the IR demodulator rated to work at 3.3v? A lot require 5V.

Thanks for all the additional hints, I really appreciate it.

I'd say the random "ghost" IR codes were due to the breadboards (!) I unplugged all the components from the breadboard, I soldered the pins of the capacitor and the resistor directly onto the pins of one of the TSOP4838 modules (they work on 3.3V) and I also soldered the ground, 3.3V and signal out wires onto the same TSOP4838 module pins, I used shrink tubing to keep them separate and I connected the PSU (3.3V), ESP32 and TSOP4838 using only Dupont wires and the thing works! Very few errors, almost like the good breadboard circuit. I think it has a slight tendency to begin "seeing" wrong IR codes after a while has passed without any real ones. But no amount of hand waving close to it seems to affect it which is a very clear improvement.

One last question, if you don't mind: I know the ESP32 input pins are 3.3V only but what would happen if they get 5V? Would the whole ESP32 be "fried" or just that particular input pin? And if it's just the pin, would it be completely "dead" or just "wonky"? I know I could experiment with one ESP32 module but I'd rather not :)

Cheers

crankyoldgit, 2021-03-21:
> One last question, if you don't mind: I know the ESP32 input pins are 3.3V only but what would happen if they get 5V? Would the whole ESP32 be "fried" or just that particular input pin? And if it's [...]

No idea. Why bother when a cheap logic-level shifter or some resistors/basic math can yield you 3v3 results.

Or use an ESP8266.

dannutu, 2021-03-22:
Oh, I didn't mean intentionally supplying 5V to it, I was wondering how would I be able to tell if I accidentally fed it 5V? It would be obvious if the entire ESP32 module would die or the input pin would be completely unusable, but not so obvious if it would just malfunction. I considered if this could have happened (i.e. if I accidentally fed it 5V and hence rendered it "wonky") and I realised I just couldn't tell.

crankyoldgit, 2021-03-22:
There probably is a more general/generic ESP32 forum somewhere which might have the information & experience you seek. Alas, my experience with the ESP32 is limited.