crankyoldgit/IRremoteESP8266 #1431: Added build flags in platform IO and get loads of compile errors

rowifi, 2021-03-12:
As per instructions to decode only one format I added the following into the pio.ini file
build_flags = -D_IR_ENABLE_DEFAULT_= false
	-DDECODE_NEC=true
	-DSEND_NEC=true

Then the project wont compile ( it does without the build flags)
Literally lots of similar compile errors
IRrecv.h:628:16: error: #if with no expression

crankyoldgit, 2021-03-12:
> build_flags = -D_IR_ENABLE_DEFAULT_= false

It appears you've got a space character after the `=` and before the `false`. That shouldn't be there. I think that's your problem.

I've tested building the `IRrecvDumpV2` example code with the following line in the `platformio.ini` file:
```
build_flags = -D_IR_ENABLE_DEFAULT_=false -DDECODE_NEC=true -DSEND_NEC=true -Werror
```

If it doesn't fix your problem, I'm going to need to know precisely what version or branch/commit of the library you're using.

And it compiles fine.

crankyoldgit, 2021-03-15:
@rowifi Did that fix your problem?

rowifi, 2021-03-15:
I'm sorry - I haven't had a chance to try it. I cheated by editing the library to force the settings I needed and haven't returned to check this. I will assume that this is the resolution and close the case.
Thank you for your response.

closed 2021-03-15 by rowifi