crankyoldgit/IRremoteESP8266 #1434: Calling enableIRIn() triggers ESP32 kernel error message

mottibz, 2021-03-21:
Hello,

I would first like to say a big thank you for a great library!!!

The issue I am facing is that in my code, I initially call enableIRIn() and all is fine but during execution, I need to disable that and later re-enable. After I disable it (calling disableIRIn), I later call enableIRIn, and I then I get the following message on the terminal:
[E][esp32-hal-cpu.c:93] addApbChangeCallback(): duplicate func=400814C4 arg=3FFBDFC4

The code I am using is there for a while and I did not encounter this error message previously. I suspect that one of the updates done since all worked fine is causing that.

I am developing using VSCode + PlatformIO, and have the latest updates installed.
PlatformIO: Core 5.1.1, Home 3.3.4
VSCode 1.54.3
Espressif 3.1.0

I will very highly appreciate your help in resolving this.

Thanks a lot, Motti

crankyoldgit, 2021-03-21:
What version of this library are you using?
Can you please reduce your code to the smallest possible size that reproduces the issue you're reporting? Please include that code here so we can duplicate the problem.

mottibz, 2021-03-21:
Hi, I am using the latest 2.7.15, upgraded from 2.7.13.
I tried going back to 2.7.13 (forcing using that rev in platformio.ini) but getting the same issue.
It may be related to the latest Espressif32 update?
I will try to reduce the code and see what happens.
Thanks!

mottibz, 2021-03-21:
More info:
I took the IRrecvDumpV2 example code and just added disableIR and enableIR after the initial enableIR:

 Serial.println("setup: enable IR");
  irrecv.setTolerance(kTolerancePercentage);  // Override the default tolerance.
  irrecv.enableIRIn();  // Start the receiver

  delay(500);
  Serial.println("setup: disable IR");
  irrecv.disableIRIn();

  delay(500);
  Serial.println("setup: enable IR");
  irrecv.enableIRIn();

I got the same issue:
setup: disable IR
setup: enable IR
[E][esp32-hal-cpu.c:93] addApbChangeCallback(): duplicate func=400813C0 arg=3FFBDE74

Will highly appreciate your help in resolving this! Thanks!!!

crankyoldgit, 2021-03-21:
Just confirming, you're using GPIO 14, correct?
e.g.
https://github.com/crankyoldgit/IRremoteESP8266/blob/c81809172208cff2d2e2ac3f1e1d9837baa63ae4/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L38-L41

mottibz, 2021-03-21:
I am using GPIO 16 as it is an ESP32: "const uint16_t kRecvPin = 16;"
And as I previously wrote, it worked fine until recently so I guess that it relates to the Espressif32 update, or maybe I am wrong...

crankyoldgit, 2021-03-21:
Hmm. My _existing_ setup doesn't seem to reproduce the problem.

e.g.

Here is the output:
```
IRrecvDump is now running and waiting for IR input on Pin 14
setup: enable IR
setup: disable IR
setup: enable IR
setup: Exiting.
...
```
here is the modifications to the code:
```diff
--- a/examples/IRrecvDumpV2/IRrecvDumpV2.ino
+++ b/examples/IRrecvDumpV2/IRrecvDumpV2.ino
@@ -139,8 +139,18 @@ void setup() {
   // Ignore messages with less than minimum on or off pulses.
   irrecv.setUnknownThreshold(kMinUnknownSize);
 #endif  // DECODE_HASH
+  Serial.println("setup: enable IR");
   irrecv.setTolerance(kTolerancePercentage);  // Override the default tolerance.
   irrecv.enableIRIn();  // Start the receiver
+
+  delay(500);
+  Serial.println("setup: disable IR");
+  irrecv.disableIRIn();
+
+  delay(500);
+  Serial.println("setup: enable IR");
+  irrecv.enableIRIn();
+  Serial.println("setup: Exiting.");
 }
 
 // The repeating section of the code
```

I'll try updating the ESP core and see what happens.

mottibz, 2021-03-21:
You are seeing exactly what I was previously seeing. All works fine.
This is why I suspect one of the updates that may cause this, such as the latest Espressif update.

mottibz, 2021-03-21:
Do you know how I can roll back to a previous Espressif core (on VSCode+PlatformIO)? If yes, please let me know and I will try to roll back and see if it works fine. Thanks!

crankyoldgit, 2021-03-21:
for the record, here is what I was building on/against:
Visual Studio Code:
```
Version: 1.53.2
Commit: 622cb03f7e070a9670c94bae1a45d78d7181fbd4
Date: 2021-02-11T11:48:44.518Z
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Linux x64 5.4.0-66-generic
```
and `1.12.4` of `"espressif32 ~ Espressif 32"` (via `pio platform list`)

crankyoldgit, 2021-03-21:
Yes, I haven't run `pio platform update` on my VSCode instance in a while. Upgrading to 3.1.1 esp32 now.

mottibz, 2021-03-21:
I found how to go back and tried using v3.0.0 (instead of 3.1.0) and it worked fine. Looks like the latest Espressif update breaks something...

crankyoldgit, 2021-03-21:
Okay. I can now verify/reproduce your output with 3.1(.1)
Thankfully the code works as expected ... except the the kernel message. Phew, I thought it was a crash message.

Time to look at the ESP32 library now to see if I can work out what they are complaining about.

mottibz, 2021-03-21:
No, no crash.
But, I was not sure if it affects the operation in any way so I raised a flag.
Will highly appreciate a fix so that we can sleep at night :-)

crankyoldgit, 2021-03-21:
Seems it (the check) was added in https://github.com/espressif/arduino-esp32/pull/3560 which is circa 31 Dec 2019.

crankyoldgit, 2021-03-21:
Even though the PR/commit is old, it only made it into their [1.0.5 release](https://github.com/espressif/arduino-esp32/releases/tag/1.0.5) 26 days ago. So yes, it looks like it's new. I'll double check the IR library's code to see if I can see anything obvious to fix. I didn't spot anything earlier.

mottibz, 2021-03-21:
Thanks a lot, I very highly appreciate that!
From briefly looking at the ESP code, it looks like it thinks that the function caller is trying to insert a callback that is already on the list...

crankyoldgit, 2021-03-21:
Found the issue. Turns out `timerBegin()` prefers to be called after a `timerEnd()` call. (Except initially)

PR/commit coming soon.

mottibz, 2021-03-21:
Thanks a lot, you are the one and only :-)

crankyoldgit, 2021-03-21:
Please download/try out PR #1435 / Branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1434
It should fix it. (Really tiny patch ;-)

As far as I can tell, the message is a warning of a _potential_ memory leak. i.e. Something wasn't done 100% correctly. The library was fine (i.e. no **bug** as such, but it's now cleaner, and no Kernel message. ;-)

Thanks for the report and for providing the needed code & info to reproduce the problem. That helped a lot!!

mottibz, 2021-03-21:
Always here to help.
I very highly appreciate all the effort that you are putting into the development and continued support for the library.
Small question: I always install libraries using the PlatformIO UI. How do I manually install the library from the link you send?

crankyoldgit, 2021-03-21:
> Small question: I always install libraries using the PlatformIO UI. How do I manually install the library from the link you send?

I'm no PlatformIO expert, but if you adjust your `platformio.ini` file to have something like the following in [lib_deps](https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps):
```
lib_deps =
   https://github.com/crankyoldgit/IRremoteESP8266.git#Issue1434
```
it should do the trick.

mottibz, 2021-03-21:
Thanks a lot!
I tested this using the above definition, it compiled and linked fine, and the error message is gone :-)

closed 2021-03-21 by crankyoldgit

crankyoldgit, 2021-03-24:
FYI, the aforementioned changes have been included in the new v2.7.16 release for the library.

crankyoldgit, 2023-05-07:
Have you tried the `master` branch of the library? There was a recent fix for this sort of thing. See #1984

PR #1435 (2021-03-21): Fix `addApbChangeCallback(): duplicate func` kernel msgs
When `IRrecv::enableIRn()` was called after a `IRrecv::disableIRn()` call the ESP32 would produce a kernel message/warning on the serial port. e.g.:
```
[E][esp32-hal-cpu.c:93] addApbChangeCallback(): duplicate func=400814C4 arg=3FFBDFC4
```
if the ESP32 IDF 3.3 (v1.0.5) was used.

Fix the issue by ensuring a `timerEnd()` call is made in the `disableIRIn()` call to properly clean up the timer.

Tested on an ESP32dev board. Confirmed the message no longer appears.

Fixes #1434
commit 60db0d35 (2021-03-21): Fix `addApbChangeCallback(): duplicate func` kernel msgs (#1435)

When `IRrecv::enableIRn()` was called after a `IRrecv::disableIRn()` call the ESP32 would produce a kernel message/warning on the serial port. e.g.:
```
[E][esp32-hal-cpu.c:93] addApbChangeCallback(): duplicate func=400814C4 arg=3FFBDFC4
```
if the ESP32 IDF 3.3 (v1.0.5) was used.

Fix the issue by ensuring a `timerEnd()` call is made in the `disableIRIn()` call to properly clean up the timer.

Tested on an ESP32dev board. Confirmed the message no longer appears.

Fixes #1434