crankyoldgit/IRremoteESP8266 #1436: A method to halt/pause IRrecv?

Duckle29, 2021-03-22:
Hi there. I've been looking through the doxygen and the code itself, and from what I can tell there is no method that's the inverse of `resume`

I'm working on a project that uses IRsend to send a state to an AC when it gets a message over MQTT. It however also listens for the stock remote using IRrecv to keep track of the latest state. 

Because of this I'm not always decoding a message before sending one, and can't rely on the receiving being paused from that. is there any `IRrecv::halt` or similar, and if not what would it take to make one? Is it as simple as:

```c
void IRrecv::halt(void) {
  params.recvstate = kStopState;
}
```
Or are there other things that have to be set like stopping interrupts and such?

Duckle29, 2021-03-22:
And after looking over while posting this I found `disableIRIn()` I'll use that for now. Is there any value in adding a `halt()` or would `disableIRIn()` always be preferable?

crankyoldgit, 2021-03-22:
> or would `disableIRIn()` always be preferable?

You got it in one. `disableIRIn()` & `enableIRIn()` are the way to do this. They are preferred, as they free/set up the interrupts & timers etc.

closed 2021-03-22 by crankyoldgit