crankyoldgit/IRremoteESP8266 #1440: Support for Truma Caravan A/C devices

laserir, 2021-03-24:
Here are some IRrecvDumpV3 (v2.7.13) dumps for the Truma Aventa Caravan A/C devices. Let me know if the captured data are sufficient for integration of the protocol.

The only options with this devices are:
- Temperature in °C (16-31), only available in AUTO or COOL mode
- Mode (AUTO,FAN,COOL)
- Fan (LOW,MEDIUM,HIGH), only available in FAN or COOL mode

`16 / AUTO / -
uint16_t rawData[117] = {20194, 1028,  1798, 628,  558, 662,  1190, 640,  1190, 638,  1190, 634,  1190, 638,  1188, 640,  1164, 662,  558, 658,  1192, 636,  1166, 662,  1190, 638,  1164, 658,  1190, 638,  582, 640,  556, 662,  558, 660,  1188, 638,  558, 662,  558, 662,  1190, 634,  1188, 638,  556, 664,  558, 662,  556, 660,  556, 664,  558, 662,  556, 664,  556, 662,  556, 664,  558, 662,  558, 662,  558, 660,  556, 664,  556, 664,  556, 664,  556, 660,  558, 664,  578, 640,  580, 640,  556, 660,  556, 664,  556, 664,  556, 664,  580, 636,  556, 664,  556, 664,  556, 664,  556, 660,  556, 664,  1188, 638,  1188, 640,  554, 662,  1188, 640,  1188, 638,  554, 666,  1164, 654,  582};`

`17 / AUTO / -
uint16_t rawData[117] = {20258, 964,  1802, 622,  582, 638,  1194, 634,  1194, 634,  1192, 632,  1194, 634,  1192, 636,  1194, 634,  582, 634,  1194, 634,  1192, 636,  1192, 634,  1194, 630,  1194, 634,  582, 638,  584, 638,  582, 634,  582, 638,  582, 638,  582, 638,  1194, 630,  1194, 636,  582, 638,  582, 636,  584, 634,  582, 638,  584, 638,  582, 638,  584, 632,  584, 638,  584, 638,  582, 638,  584, 634,  584, 638,  582, 638,  582, 638,  584, 634,  582, 638,  584, 638,  582, 638,  582, 634,  582, 638,  582, 638,  582, 638,  582, 634,  582, 638,  582, 638,  582, 638,  582, 634,  1192, 634,  582, 638,  1194, 634,  582, 634,  1194, 634,  1194, 634,  582, 638,  1194, 624,  586};`

`18 / AUTO / -
uint16_t rawData[117] = {20230, 990,  1802, 622,  584, 636,  1194, 634,  1192, 636,  1192, 630,  1192, 634,  1194, 634,  1192, 636,  584, 632,  1194, 634,  1194, 634,  1192, 634,  1194, 630,  1192, 636,  584, 636,  582, 638,  582, 634,  1194, 634,  1194, 634,  1192, 636,  584, 632,  1192, 636,  582, 638,  582, 638,  584, 632,  582, 638,  584, 638,  582, 638,  582, 634,  582, 638,  582, 638,  582, 638,  582, 634,  582, 638,  584, 638,  582, 638,  582, 634,  582, 638,  582, 638,  582, 638,  582, 634,  582, 638,  582, 638,  582, 638,  584, 634,  582, 638,  582, 638,  582, 638,  582, 634,  582, 638,  582, 638,  1192, 634,  582, 634,  1192, 636,  1192, 634,  582, 638,  1192, 624,  588};`

`- / FAN / LOW
uint16_t rawData[117] = {20256, 960,  1830, 594,  584, 636,  1218, 610,  1194, 632,  1194, 628,  1196, 632,  1194, 632,  1194, 634,  584, 632,  584, 636,  584, 636,  1194, 632,  1196, 628,  584, 636,  584, 636,  584, 636,  584, 632,  1194, 632,  584, 636,  584, 636,  1194, 628,  1194, 632,  584, 636,  584, 636,  584, 632,  584, 636,  584, 636,  584, 636,  584, 632,  584, 636,  584, 636,  584, 636,  584, 632,  584, 636,  584, 636,  584, 636,  584, 632,  584, 636,  584, 636,  584, 636,  584, 632,  584, 636,  584, 636,  582, 636,  584, 632,  584, 636,  584, 636,  584, 636,  584, 632,  1194, 634,  1194, 632,  584, 636,  584, 632,  584, 636,  1194, 632,  584, 636,  1194, 622,  614};`

`- / FAN / MEDIUM
uint16_t rawData[117] = {20226, 990,  1802, 624,  584, 636,  1194, 634,  1192, 634,  1192, 632,  1194, 634,  1192, 634,  1192, 636,  584, 632,  608, 612,  608, 612,  1192, 634,  584, 632,  1192, 636,  584, 636,  608, 612,  584, 632,  1192, 634,  584, 636,  610, 610,  1194, 630,  1192, 634,  608, 612,  608, 612,  608, 608,  608, 612,  584, 636,  584, 636,  584, 632,  608, 610,  584, 636,  610, 610,  584, 632,  610, 610,  608, 612,  608, 612,  608, 608,  610, 610,  608, 612,  610, 610,  610, 606,  608, 612,  608, 612,  608, 610,  584, 632,  608, 612,  610, 610,  608, 612,  584, 632,  1192, 636,  1192, 636,  608, 612,  1192, 632,  584, 636,  1192, 634,  608, 612,  1192, 624,  610};`

`- / FAN / HIGH
uint16_t rawData[117] = {20212, 1000,  1798, 628,  560, 660,  1190, 638,  1188, 640,  1166, 658,  1166, 662,  1166, 662,  1166, 662,  582, 634,  560, 660,  560, 660,  1166, 662,  1166, 658,  1190, 638,  560, 660,  558, 660,  558, 658,  1190, 638,  558, 660,  560, 660,  1190, 634,  1166, 662,  560, 660,  560, 660,  560, 656,  584, 638,  582, 638,  558, 662,  584, 632,  558, 662,  558, 662,  560, 660,  558, 658,  558, 662,  558, 662,  582, 638,  558, 658,  558, 662,  558, 662,  558, 662,  558, 658,  558, 662,  558, 662,  582, 638,  558, 658,  582, 638,  558, 662,  558, 662,  582, 634,  1166, 662,  1190, 638,  558, 662,  558, 658,  1166, 662,  1166, 660,  558, 662,  1166, 652,  582};`

`16 / COOL / LOW
uint16_t rawData[117] = {20220, 990,  1802, 622,  582, 636,  1194, 634,  1192, 634,  1194, 630,  1194, 634,  1192, 634,  1194, 634,  584, 634,  1194, 634,  582, 638,  1192, 634,  1194, 630,  584, 636,  582, 638,  582, 638,  582, 632,  1194, 634,  582, 638,  584, 636,  1192, 630,  1192, 634,  584, 636,  584, 636,  582, 634,  584, 636,  584, 636,  584, 636,  582, 632,  584, 636,  584, 636,  584, 636,  584, 632,  584, 636,  584, 636,  584, 636,  582, 634,  582, 636,  584, 636,  584, 636,  584, 632,  584, 636,  584, 636,  584, 636,  582, 634,  582, 638,  584, 636,  582, 638,  582, 632,  584, 636,  584, 636,  1192, 636,  582, 632,  584, 636,  1192, 634,  584, 636,  1192, 624,  586};`

crankyoldgit, 2021-03-24:
> Here are some IRrecvDumpV3 (v2.7.13) dumps

Can you please try the current version of the library (v2.7.16) to see if the protocol has support already added?

crankyoldgit, 2021-03-25:
I've added experimental support for this protocol in branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Truma_basic

The bit ordering/representation may be subject to change, as the data you've supplied is not enough to be sure yet.
I need a full temperature range capture, as per: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#determining-the-bit-order-of-the-protocol to work out if I've got it correct or not.

It should however be enough for you to test sending and basic decoding of codes, just the values may be subject to change once we are more sure of the format.

Please download, test, and give me feedback etc.

crankyoldgit, 2021-03-25:
Oh, and please provide detail model numbers/brands etc of the A/C unit **and** the remote.

laserir, 2021-03-25:
The manufacturer is "Truma", the model range is called "Aventa" the remote works for the "Aventa compact plus", "Aventa compact" and "Aventa comfort" models. (https://www.truma.com/de/de/produkte/truma-klimaanlagen) The remote is model/part no "40091-86700". Most likely the remote is the same as for the discontinued "Saphire" model range, but I cannot confirm for sure.

I will come back with complete captures as of your link.

v2.7.16 didn't decode the signals. For the new branch I'll have to try.

laserir, 2021-03-25:
[Truma_IR.xlsx](https://github.com/crankyoldgit/IRremoteESP8266/files/6205742/Truma_IR.xlsx)
Attached the captures, hope this will do?

crankyoldgit, 2021-03-27:
@laserir Can you please convert your spreadsheet to a publicly readable Google Spreadsheet (or similar service) for those a don't have Excel etc, and it is also updated live, rather than you having to upload a new sheet for each change, and for people needed to download the whole thing for each change etc.

As the decoder should be working in that branch ([Truma_basic](https://github.com/crankyoldgit/IRremoteESP8266/tree/Truma_basic)), it should be producing 56bit hex values (e.g. `0x45ffffffe8da81`), please provide those rather than the rawData[] ones. Only supply those if it doesn't match as `TRUMA`. I tested it with one of the rawData codes in your sheet, and it decoded fine.

laserir, 2021-03-29:
Done! Here are the captures:
https://docs.google.com/spreadsheets/d/1k-RHu0vSIB6IweiTZSa3Rxy3Z_qPUtqwcqot8uXVO6I/edit?usp=sharing

Let me know if you need anything else.

crankyoldgit, 2021-03-29:
Thanks for that. Looks like I successfully deduced the bit ordering/polarity from your earlier examples.
Your new data confirms it.

So the (hex) data format should stay the same pretty much.
Using this cell as an example. [`0x4AFFFFFFE7E081`](https://docs.google.com/spreadsheets/d/1k-RHu0vSIB6IweiTZSa3Rxy3Z_qPUtqwcqot8uXVO6I/edit#gid=0&range=E7)

The Most Significant `8` bits at Bits 48 to 55 (e.g. 0x**4A**FFFFFFE7E081) looks like it's a checksum.
Next, the temperature looks like it's stored in the `9` bits at Bits 16 to 21 (e.g. 0x4AFFFFFF**E7**E081, such that `0xE7 in binary is 0b111**00111**), i.e. `0b00111` == `0x07` == 7 (DEC). It's encoded as _that value + 10_ equals the temp in DegC. That is, 17DegC.

Another example [(`0x54FFFFFFF1E081`)](https://docs.google.com/spreadsheets/d/1k-RHu0vSIB6IweiTZSa3Rxy3Z_qPUtqwcqot8uXVO6I/edit#gid=0&range=E17) yields `0xF1`, taking the last `9` bits, gives us `0b10001` or `0x11` or `17` (Dec).
_17 + 10 = 27DegC._

The checksum bits seems to be the 8 bit sum of each 8 bit block of Bits 0 to 47, then + 5.
e.g.
`0x54FFFFFFF1E081`'s checksum is `0xFF + 0xFF + 0xFF + 0xF1 + 0xE0 + 0x81 + 5 (per above)` == 0x5**54**

So,  you're on your way to mapping out the rest of the controls/settings with that info via [these steps](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#working-out-the-details).

Let me know when you've mapped out all the controls etc.

laserir, 2021-03-29:
I added some new captures for mode and fan. Hope this is as you meant? Otherwise please give me directions which combinations to capture.

crankyoldgit, 2021-03-29:
What I'm saying is **you** need to do the analysis that's described/suggested in that wiki page. Read the whole page.

I'll convert your analysis to code, but I no longer do the breakdown/analysis anymore. I've done so many they I'm tired of doing them. You're the one with a vested interest and incentive, hence I leave that to you.

laserir, 2021-03-30:
I will try my best!

I'll try to break down the command (Example 16 °C, AUTO):
0x(49) --> **Checksum** as explained here https://github.com/crankyoldgit/IRremoteESP8266/issues/1440#issuecomment-809307120
(FF) --> Always FF
(FF) --> Always FF
(FF) --> Always FF
(E6) --> **Temperature** as explained here https://github.com/crankyoldgit/IRremoteESP8266/issues/1440#issuecomment-809307120
(E0) --> **Mode/Fan** as explained below
(81) --> Always 81

**Mode / Fan (Switch ON):**
The MODE has a base value, the FAN value is determined by an offset added to this base.

Base Values:
AUTO -> E0
COOL -> DA
FAN -> E3

Offsets for the FAN value is always +8(dec):
AUTO has no FAN value and is always E0.
COOL ranging from QUIET -> HIGH -> MEDIUM -> LOW (e.g. COOL/MEDIUM would be DA(hex) + 8(dec) + 8(dec) = EA(hex))
FAN mode ranging from HIGH -> MEDIUM -> LOW (e.g. FAN/MEDIUM would be E3(hex) + 8(dec) = EB(hex))

**Switch OFF:**
This is a bit tricky. The OFF value for the current mode has an different offset per mode:
AUTO -> +7(dec)
COOL -> +5(dec)
FAN -> +4(dec)

E.g. if mode is COOL/MEDIUM (EA) -> EA(hex) + 5(dec) = EF(hex)
or if mode is AUTO (E0) -> E0(hex) + 7(dec) = E7(hex)

Nevertheless in any mode the AUTO/OFF command (E7) will work.

crankyoldgit, 2021-03-31:
I've added experimental detailed support for this protocol in branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Truma_basic / PR #1449 

Please try it out. Hopefully it will decode things correctly, and allow you to send custom messages etc.
Let me know what works, and what doesn't. Especially the sending of IR messages, that's unverified (by you) at this point.

laserir, 2021-03-31:
All commands successfully decoded! Sending 100% correct!

crankyoldgit, 2021-03-31:
Thanks for confirming it's all working. I've updated the status to reflect that. Just waiting for the PR to be approved/merged etc.

closed 2021-04-04 by crankyoldgit

crankyoldgit, 2021-04-20:
FYI, the changes mentioned earlier have been included in the latest release of the library.

PR #1449 (2021-04-04): Experimental detailed support for Truma A/Cs.
* add `sendTruma()` & `decodeTruma()`.
* update ancillary routines as needed.
* Support settings of:
  - Power
  - Temperature
  - Operating Mode
  - Fan Speed
  - Quiet
  - Checksums.
* Update `IRac` class to add support as described.
* Add unit test coverage for new protocol & class.

Fixes #1440
commit 77ba26b4 (2021-04-04): Experimental detailed support for Truma A/Cs. (#1449)

* add `sendTruma()` & `decodeTruma()`.
* update ancillary routines as needed.
* Support settings of:
  - Power
  - Temperature
  - Operating Mode
  - Fan Speed
  - Quiet
  - Checksums.
* Update `IRac` class to add support as described.
* Add unit test coverage for new protocol & class.

Fixes #1440