crankyoldgit/IRremoteESP8266 #1444: Issue "[E][esp32-hal-cpu.c:93] addApbChangeCallback(): duplicate func=40081420 arg=3FFBE14C" also in 2.7.16

mottibz, 2021-03-25:
Hello, I tried the new version 2.7.16 and it looks like the duplicate func issue still exists as we worked on before. I am calling the IR enable function twice (no disable IR in the middle) and am getting the error message.

crankyoldgit, 2021-03-25:
That sounds like correct behaviour to me. The IRrecv enable function should only be called once to enable. There is no checking in the function to see if it is already setup. Thus the kernel warning is correct, you're causing a duplicate.

crankyoldgit, 2021-03-27:
The library will NOT keep track for you if you've called `enableIRIn()` multiple times. Doing so would require more code space and memory usage that is not needed most of the time. This is something a library user should do in their own code if they need it. It is best for the library to use the least amount of resources and leave higher-level constraints etc in optional user code.
TL;DR: I don't think it's broken. Acting as expected. Won't fix, unless you come up with a _likely_ scenario where you would need to call enableIRIn() again without calling a `disableIRIn()` prior to it.

closed 2021-03-27 by crankyoldgit