crankyoldgit/IRremoteESP8266 #1447: Wrong State for Samsung

yaroshd81, 2021-03-29:
You have 2 function for return results state AC:
1. IRAcUtils::resultAcToString
2. IRAcUtils::decodeToState
If used 1 function after recived command, we have nice result:
`resultAcToString:Power: On, Mode: 1 (Cool), Temp: 22C, Fan: 4 (Med), Swing: Off, Beep: Off, Clean: Off, Quiet: Off, Powerful: Off, Breeze: Off, Light: Off, Ion: Off`
But if used 2 function, we have bad results, because `Mode: 0 (Auto), Temp: 16C, Fan: 0 (Auto)` (All parameters have 0, because Temp: 16C = (int)0)
I fixed problem when analize 1 function with 2:
`        ac.setRaw(decode->state, decode->bits / 8);//This code in 1 function where all good working
        //ac.setRaw(decode->state);//This code in 2 function where not working
`

crankyoldgit, 2021-03-30:
Thanks for the bug report and the fix. I concur. I clearly _oopsed_ when I wrote that.

I also checked for other Protocols in `IRAcUtils::resultAcToString` that also use the bit length, and they are all present in `IRAcUtils::decodeToState`, so it looks like that's the only one I missed. Phew.

closed 2021-03-30 by crankyoldgit

crankyoldgit, 2021-04-20:
FYI, the changes mentioned earlier have been included in the latest release of the library.

commit cd8d6008 (2021-03-30): [BUG] Fix `IRAcUtils::decodeToState()` for different length Samsung msgs (#1448)

Need to pass byte length to `IRSamsungAc::setRaw()` to handle normal & **extended** messages.

Kudos to @yaroshd81 for finding the bug, and offering the solution.

Fixes #1447