crankyoldgit/IRremoteESP8266 #1451: When i try to update the library to 2.7.16 the library disappears!

EgHubs, 2021-03-30:
### Describe the bug
When I try to update the library to 2.7.16 the library disappears from the library folder on my 2017 MacBook Air with macOS Big SUR 11.2.3 (20D91) installed and Arduino 1.8.14 installed,
and when I try to downgrade to 2.7.15 and install the library is installed, if I try to update, it disappears from the library folder located on `/Users/shalaby/Documents/Arduino` 
also if I try to install it while it's not in the folder it does complete the installation but there is nothing in the folder and also I get errors on compiling, with version 2.7.15 I am good.

brianmiller1956, 2021-03-30:
I have the same problem on 2 windows10 laptops, the update appears to install but it isn't there! If I install the previous version this installs fine, if I try to update from there it just deletes the old library and doesn't install the latest.

crankyoldgit, 2021-03-31:
Thanks for reporting the issue. I've checked over our config for the 2.7.16 release, and it seems to be fine. i.e. Produced the same way as 2.7.15. So, it isn't a problem with the library per-se, nor with github where it is hosted.
My guess is there is some corruption/problem/issue at the Arduino IDE's Library Manager end.

I suggest in the meantime you try installing [it](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.7.16) from a zip file, per: https://www.arduino.cc/en/guide/libraries

Or log an issue with Arduino at https://github.com/arduino/Arduino/issues.

Perhaps try deleting the library from the IDE and try adding it again?!

I checked what & where the  Arduino IDE thinks it should be downloading: https://downloads.arduino.cc/libraries/github.com/markszabo/IRremoteESP8266-2.7.16.zip

When I download that file, it contains what I would expect.

crankyoldgit, 2021-03-31:
> and Arduino 1.8.14 installed,

Interestingly, https://www.arduino.cc/en/software indicates to me that 1.8.13 is the latest version available to download.
e.g. https://downloads.arduino.cc/arduino-1.8.13-macosx.zip

crankyoldgit, 2021-03-31:
and https://downloads.arduino.cc/arduino-1.8.14-macosx.zip doesn't seem to work/download anything.

EgHubs, 2021-03-31:
i will try to downgrade and see

EgHubs, 2021-03-31:
Alrigth, i downgraded to version 1.8.13 and the problem remains,
so i added the library by adding the zip file as you suggested and it did work.
i guess the issue isn't solved so maybe i will leave it open if you may,
thanks.

crankyoldgit, 2021-03-31:
> I went to the site and you have a point, I don't know when or where did I get this update! :)

I think you must have chosen the nightly-build stream.

> so i added the library by adding the zip file as you suggested and it did work.

Yay. Well, that proves the library is fine at least.

> i guess the issue isn't solved so maybe i will leave it open if you may,

Agreed. Something is up with the Arduino Library Manager. Not sure what I can do about it. Let's give them 24 hours to see if it is just something transient.

About the only thing I can do solely at my end is to push another release to see if it flushes their cache/whatever.

I do suggest you log a bug/issue with Arduino themselves (see issue link I included before) as I think it's there end, but again, wait and see if it clears itself up.

EgHubs, 2021-03-31:
> I think you must have chosen the nightly-build stream.

Yeah, you are right :), it was a very bad experience after all, 

> Let's give them 24 hours to see if it is just something transient.

I hope.

brianmiller1956, 2021-03-31:
Hi,
Just to add my finding, I am using 2 windows 10 machines both running arduino 1.8.13 I needed to install iRremoteESP8266 yesterday, I went through library manager and it goes through the motions of installation but at the end of the process the directory is missing and library  manager lets you install it again, I installed the earlier version without problem, library manager then says it can be updated to 2.7.16 after hitting ok it appears to install, but when I look at the library folder the directory has been deleted, gone!
I will install via the zip and see what happens next update.
Many thanks.

mosave, 2021-04-01:
Reporting same behaviour here. Latest Arduino Studio, up-to-date Win10.
Library manager successfully goes thru installation process but:
- library disappears if installing 2.7.16
- everything is Ok with 2.7.15

crankyoldgit, 2021-04-01:
I've created a pre-release (v2.7.16**a**) See: https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.7.16a
It should be exactly the same as v2.7.16. Give it a few hours (or a day) and it _should_ show up in the Arduino Library Manager ... _I think._

Let's see if that clears up the blockage/problem. Please let me know.

crankyoldgit, 2021-04-01:
Running the Arduino Library Manager Linter yields the following output:
```
$ bin/arduino-lint --library-manager update
Linting library in /home/david/git/IRremoteESP8266
Rule LS004 result: fail
WARNING: Git submodule detected. Library Manager installations and installations from GitHub's "Download ZIP" will only contain an empty folder in place of the submodule.

Finished linting project. Results:
Warning count: 1
Error count: 0
Rules passed: true
```

That seems to be the only issue it found.

crankyoldgit, 2021-04-01:
I'm attempting to remove any submodules used by the library, but realistically the _warning_ message says those submodule directories will be blank, not the actual libraries directory. i.e. This should not fix it, it's just a long shot. No _errors_ were found.

manj9501, 2021-04-15:
I have the same issue while installing v2.7.16! The library just keeps disappearing  '_'

mosave, 2021-04-16:
> I've created a pre-release (v2.7.16**a**) See: https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.7.16a
> It should be exactly the same as v2.7.16. Give it a few hours (or a day) and it [...]

Still an active issue.
No v2.7.16a available to install, Arduino Studio still cries "update required" and not able to install updated version causing library just disappears :(

brianmiller1956, 2021-04-16:
Just to confirm that the “Test” update to v2.7.16a has not yet shown itself, v2.7.16 is still the latest according to  library manager. Hope this helps.

From: Alexander
Sent: 16 April 2021 19:32
To: crankyoldgit/IRremoteESP8266
Cc: brianmiller1956; Comment
Subject: Re: [crankyoldgit/IRremoteESP8266] When i try to update the libraryto 2.7.16 the library disappears! (#1451)

I've created a pre-release (v2.7.16a) See: https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.7.16a
It should be exactly the same as v2.7.16. Give it a few hours (or a day) and it should show up in the Arduino Library Manager ... I think.
Let's see if that clears up the blockage/problem. Please let me know.
Still an active issue.
No v2.7.16a available to install, Arduino Studio still cries "update required" and not able to install updated version causing library just disappears :(

crankyoldgit, 2021-04-17:
I'll try drafting a pre-release of 2.7.17 once a couple of the current PRs get merged. Fingers crossed that might work.

AVDeveloppement, 2021-04-17:
Same problem for me too, i not help but i can say thanks for your work it's a very good library and manual install are fine for me it take less time than you write this great library!

crankyoldgit, 2021-04-18:
I've made a new (pre-)release of the library (v2.7.17), Arduino's library manager should pick it up with in 24 hours. Let me know how it goes. If it fixes this issue, I'll make it an official release.

Fingers crossed

brianmiller1956, 2021-04-18:
Hi, I have just checked this, 17 is showing as an available update in library manager, I hit update, it goes through the motions of updating but it leaves version 16 still in place with no sign of version 17, lib manager says I still have v16 installed and it can be updated, Tried this several times,  I also removed v16 and tried a new install of v17via lib manager but no directory is created and nothing is installed. Restarted IDE in between with no change, also Tried this on 2 windows 10 machines with same results. Hope this is of assistance.

From: David Conran
Sent: 18 April 2021 12:57
To: crankyoldgit/IRremoteESP8266
Cc: brianmiller1956; Comment
Subject: Re: [crankyoldgit/IRremoteESP8266] When i try to update the libraryto 2.7.16 the library disappears! (#1451)

I've made a new (pre-)release of the library (v2.7.17), Arduino's library manager should pick it up with in 24 hours. Let me know how it goes. If it fixes this issue, I'll make it an official release.
Fingers crossed

crankyoldgit, 2021-04-18:
> Hi, I have just checked this, 17 is showing as an available update in library manager, I hit update, it goes through the motions of updating but it leaves version 16 still in place with no sign of [...]

In a word, **Bugger!**

Sadly, I'm out of immediate ideas as to what could be causing the issue.

EgHubs, 2021-04-18:
> Hi, I have just checked this, 17 is showing as an available update in library manager, I hit update, it goes through the motions of updating but it leaves version 16 still in place with no sign of version 17, lib manager says I still have v16 installed and it can be updated, Tried this several times, I also removed v16 and tried a new install of v17via lib manager but no directory is created and nothing is installed. Restarted IDE in between with no change, also Tried this on 2 windows 10 machines with the same results. Hope this is of assistance.

Sadly confirming this is the same exactly here, right now only version 2.7.15 is the only one that library manager can install, version 2.7.16 still fails and also 2.7.17.
I would suggest going to all the commits that were made after publishing version 2.7.15 to check if there is anything was done by mistake.

crankyoldgit, 2021-04-19:
> I would suggest going to all the commits that were made after publishing version 2.7.15 to check if there is anything was done by mistake.

That was the first thing I did. I've just reviewed line-by-line all the changes (except the auto-generated documentation) between 2.7.15 & 2.7.16. (You can do this yourself here: https://github.com/crankyoldgit/IRremoteESP8266/compare/v2.7.15...v2.7.16

The **only** thing suspicious is we renamed a file in one of the commits in between the two versions. We've renamed files before. This has never happened previously, so it is unlikely the culprit.

Nothing, _content-wise_, has changed that should cause arduino to pick on this library.

My current _theory_ is the `.zip` file was slightly under 5Meg for v2.7.15, and slightly over for both v2.7.16 & 17. I am taking a wild guestimate that the Arduino IDE has a limit on the size of a `.zip` file it will unpack via `Library Manager` on Windows 10.
I'm currently working on trying to reduce the size of the zip file that the Arduino Library system builds for its IDEs.

Things I've looked at/tried so far:

- All commits between v2.7.15 & v2.7.16
- All actual line changes between v2.7.15 & v2.7.16
- Quadruple reviewed & improved the `library.properties` file used by Arduino's Library Manager service/backend. It's been unchanged except simple version number changes for years. So it shouldn't have exploded.
- Run and installed constant Arduino Library Manager linter checks to confirm nothing is wrong with the library setup so it should be accepted perfectly.
- Fixed any Library Manager linter issues found, even just "warnings" which were fine as-is.
- Forced the Library Manager service to build new versions of the zip file(s), in case of corruption.
- Forced/created new version numbers, , in case of corruption.
- Installed Arduino IDE 1.8.13 on Windows 10 to verify/replicate the problem (which I can).
- Verified by hand that the Arduino Library Manage json index file has the correct entries for v2.7.16+ etc.
- Verified the URLs work to download the library using a browser.
- Inspected the Arduino IDE logs to see that Library Manager is detecting, fetching, & downloading the correct file & version etc.
- Verified the IDE is downloading the file and storing it in it's `staging` directory successfully. e.g. `C:\Users\USERNAME\AppData\Local\Arduino15\staging\libraries`
- Verified said file is correct and has what I expect in it.
- Cross verified the contents of the zip files in the `staging` directory to see if they look/appear the same. (They do)

In short, all the evidence points to:
- The IRremoteESP8266 library is correctly setup for Arduino's Library Manager service.
- The Arduino's Library Manager service is picking up the library changes and zip files accordingly.
- Those zip files contain the correct contents.
- The Arduino IDE is detecting new versions of the library correctly.
- It is successfully able to download the zip file correctly.
- That zip file contains the right contents.
- **The IDE fails without logging info or warning for some unknown reason when it goes to unpack the v2.7.16+ zip files that it has downloaded.**
- **The Linux version of the Arduino IDE (1.8.13) doesn't have this issue at all. It appears it's _Windows 10 only._**

crankyoldgit, 2021-04-19:
Well, that didn't go as planed. Seems to still be caching the old zip file/version. Burning a new release number, v2.7.18. Waiting on the code review before making a release.

crankyoldgit, 2021-04-20:
v2.7.18 release seems to work with Windows 10. I'm going to assume it also works for Mac OSX.
Can you please confirm?

brianmiller1956, 2021-04-20:
Hi just tested this, shows as an upgrade in library manager from 16 to 18, installs first time, did leave v16 installed but not a problem, also installed from scratch first time, problem fixed! Well done, many thanks.

From: David Conran
Sent: 19 April 2021 17:05
To: crankyoldgit/IRremoteESP8266
Cc: brianmiller1956; Comment
Subject: Re: [crankyoldgit/IRremoteESP8266] When i try to update the libraryto 2.7.16 the library disappears! (#1451)

Well, that didn't go as planed. Seems to still be caching the old zip file/version. Burning a new release number, v2.7.18. Waiting on the code review before making a release.

crankyoldgit, 2021-04-20:
Thanks for the extra confirmation. Just waiting on @EgHubs for OSX.

manj9501, 2021-04-20:
I hereby provide auxiliary confirmation for the success of v2.7.18 on Windows 10 (if it was needed). Thanks for all the efforts @crankyoldgit !

EgHubs, 2021-04-20:
Yay! \o/ 😂 .
now it's all good and okay on macOS :)
Thanks for your great effort in solving this.

closed 2021-04-20 by EgHubs

AVDeveloppement, 2021-04-21:
Everything are ok for me on Win10, thanks for your work!  :thumbsup:
:beers: