crankyoldgit/IRremoteESP8266 #1454: Convert incoming Samsung AC signal into IRSamsungAC object for later use

Padm59, 2021-04-02:
Hello I have a question and I hope someone can help me with this.

I have an IR sender hooked up, and I am able to control my AC through MQTT and Home Assistant. I build on top of the ControleSamsungAC example and added MQTT to my needs. After sending something to the AC using the setter and ac.send I send the current status back via MQTT. To do so I use the getter functions of the IRSamsungAC object to get information about the status.

**But** I also want to see if someone is using the remote and changing the status of the AC. That's why I also hooked up a receiver to catch the remote signals.

My problem now is that I don't know how I can migrate the incoming Ir data and parse them into the IRSamsungAC object witch I use to get the status and send it back via MQTT.  

I looked in the library, but I couldn't find a function that does the job. I just don't know how to parse the incoming Samsung ac data into the IRSamsungAC object.

Showing the incoming data is no problem and I also noticed that some functions are not supported like the Long Range or Eco feature(will try to work out the decoding and will help to add them into the lib in the future). I have also an _unsupported_ remote and unit.

Hope someone knows how.

crankyoldgit, 2021-04-02:
You want to use [`IRSamsungAc::setRaw()`](
https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRSamsungAc.html#a95377e8c73b51e73e78b51a2b2fa16d4]

closed 2021-04-02 by crankyoldgit

Padm59, 2021-04-03:
How do I get the uint8_t array for the setRaw() function out of my decode_results object?

crankyoldgit, 2021-04-03:
Use something like:
https://github.com/crankyoldgit/IRremoteESP8266/blob/cd8d6008d43ffb50c7a2fa9786da3185d1c47fbf/src/IRac.cpp#L3337-L3341