crankyoldgit/IRremoteESP8266 #1467: Add HTML viewport meta tag to IRServer and IRMQTTServer examples

touchgadget, 2021-04-27:
### Version/revision of the library used
v2.7.18

### Describe the bug
The IRServer and IRMQTTServer web pages are easier to read with a viewport meta tag when using a phone or tablet in portrait mode.

Reference: https://developer.mozilla.org/en-US/docs/Web/HTML/Viewport_meta_tag

#### To Reproduce
IRServer View page in portrait mode without viewport meta tag. The user must zoom to get the page to fill the screen. With viewport meta tag, the page fills the screen.

#### Example code used
```
            "content=\"text/html;charset=utf-8\">"
            "<meta name=\"viewport\" content=\"width=device-width, initial-scale=1.0, minimum-scale=1.0, maximum-scale=5.0\">"
            "</head><body><center><h1>");
```

### Other useful information
I can generate a PR but it is only a 1 line change to two examples. The Web-AC-control example includes the viewport meta tag so not change is needed.

NiKiZe, 2021-04-27:
A PR would be great

closed 2021-04-29 by crankyoldgit

crankyoldgit, 2021-07-06:
FYI, the committed & merged changes thus far have been included in the newly released version of the library. i.e. v2.7.19

PR #1469 (2021-04-29): Add HTML viewport meta tag to IRServer and IRMQTTServer examples
The IRServer and IRMQTTServer web pages are easier to read with a viewport meta tag when using a phone or tablet in portrait mode.

Fixes #1467.
commit f152c632 (2021-04-29): Add HTML viewport meta tag to IRServer and IRMQTTServer examples (#1469)

The IRServer and IRMQTTServer web pages are easier to read with a viewport meta tag when using a phone or tablet in portrait mode.

Fixes #1467