crankyoldgit/IRremoteESP8266 #1468: Integrate DHT22 module

pasna, 2021-04-28:
@crankyoldgit I am looking forward to adding DHT22 module on board to get full function of climate control. Any suggestion ?

NiKiZe, 2021-04-28:
That is not related to this project, but there is quite a few other projects using this library, have you looked at any of them?

crankyoldgit, 2021-04-28:
@pasna I am sure someone has done that in a fork of IRMQTTServer in ages past. I was asked to add Temperature Sensor support back then, and I don't really want to add it because: a) There are so many of them, & b) It complicates the example.

If you want to add it to your own fork, I suggest having a look at Issue #842 & PR #845 where I added something similar (Vcc reporting) and what ever simple example code there is for the DHT22. It should be a fairly simple task.

But as @NiKiZe said, it's kind of out of scope for this library, but I understand it's a desirable feature.

As indicated in #842 there are several other projects that use this library **and** provide DHT22 support.
In fact, I personally use a Sonoff device with a temp sensor running Tasmota to report actual room temp to Home Assistant and IRMQTTServer to control my A/Cs.

Good luck. I can provide minimal help (I don't have a DHT22) but I won't be adding it to the library's example code.

pasna, 2021-04-29:
@crankyoldgit @NiKiZe Thanks to you all for the messages. I will look into those threads and start from there. 
Best Regards

closed 2021-04-29 by pasna