crankyoldgit/IRremoteESP8266 #1470: IRMQTT Server  transmit issue

griffisc306, 2021-05-04:
I realize this is out of scope for this project. However, My hope is that somebody has run into this. I am having an issue  that the IR will not transmit unless  my device i am using to control (phone, tablet, etc) is right next to the board. I have googled tilll im blue in the face. I have also tried all new wires, led, and even board. This issue doesnt seem to happen when i have the board plugged directly into the computer. any ideas? im grasping at straws at this point.

griffisc306, 2021-05-04:
I will also say, i can access the web UI no problem, it is specifically with sending

NiKiZe, 2021-05-04:
Sounds it is more about you being next to the device, and maybe causing reflections or similar.

Have you tried a different power supply? Also changed transistor?

griffisc306, 2021-05-04:
I am not using a transistor as it as worked without one before and on my other devices. I think you may have hit the nail on the head tho. if i put a plasic bag over it it works .

NiKiZe, 2021-05-04:
Without a transistor... read FAQ https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#basic

Reading the FAQ is something one should do before creating issues.

closed 2021-05-04 by NiKiZe