crankyoldgit/IRremoteESP8266 #1473: How to copy and determine the AC remote control?

hzhh110, 2021-05-12:
How to copy and determine the AC remote control?

hzhh110, 2021-05-12:
bool IRrecv::decodeGree(decode_results* results, uint16_t offset,
                        const uint16_t nbits, bool const strict)

How does this function work? Is there an example?

crankyoldgit, 2021-05-12:
> How to copy and determine the AC remote control?

I don't understand your question fully, but I'll answer what I think your asking.

Try using or looking at the [`IRrecvDumpV2` example code](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino).
Particularly the [`decode()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRrecv.html#aeaa5c07a8b46f8fbb982f996cc1f9f4b) which will give you a [`decode_results`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classdecode__results.html) structure, which contains a [`decode_type_t`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/IRremoteESP8266_8h.html#ad5b287a488a8c1b7b8661f029ab56fad) which is the protocol you're receiving.

> bool IRrecv::decodeGree(decode_results* results, uint16_t offset,
const uint16_t nbits, bool const strict)
> 
> How does this function work? Is there an example?

See the example files with `*recv*` in them for examples of how to use `decode()`. You can't use `decodeGree()` directly. It's used in the `decode()` routine.

crankyoldgit, 2021-05-14:
See also: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol#see-if-it-is-already-supported-first

crankyoldgit, 2021-05-26:
Ping @hzhh110 

We haven't heard from you in over 2 weeks. If we don't hear back from you soon, we are going to assume we've answered your questions and everything is now fine, and we will close this issue.

crankyoldgit, 2021-05-29:
Marking closed/stale due to no response.

closed 2021-05-29 by crankyoldgit