crankyoldgit/IRremoteESP8266 #1474: Mirage AC support

azayas97, 2021-05-13:
Hello.

I was wondering if this has support for Mirage A/C or which library should I use for this brand?

NiKiZe, 2021-05-13:
HI, Thank you for considering this Library.

1. What is the exact brand and model, of your device and remote
2. You find a list of known working and supported devices from https://github.com/crankyoldgit/IRremoteESP8266#supported-protocols
3. Did you try to search existing issues? #587 #1291
4. Why did you not use the issue template?

crankyoldgit, 2021-05-14:
Via https://github.com/crankyoldgit/IRremoteESP8266/blob/master/SupportedProtocols.md :
https://github.com/crankyoldgit/IRremoteESP8266/blob/f152c6324e6567bb799fed836545662eea04e665/src/ir_Mirage.cpp#L6-L7

Note: There is no _Detailed_ support for this protocol yet, only _Basic_ support thus far. You'd need to do analysis as described in: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

You might want to check if your A/C is actually using that protocol, it is common that brands use different protocols on some models. The best way to do that is follow the advice in: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol#see-if-it-is-already-supported-first

crankyoldgit, 2021-05-26:
Ping @azayas97 
We haven't heard back from you for almost 2 weeks. If we don't hear anything soon, we are going to assume we've answered your question or you've found your own answer, and close the issue.

crankyoldgit, 2021-05-29:
Marking this closed/stale due to lack of response.

closed 2021-05-29 by crankyoldgit

akram, 2021-08-24:
@crankyoldgit can we re-open this one? I have the same issue/question. Maybe I can help and get help.

crankyoldgit, 2021-08-24:
> @crankyoldgit can we re-open this one? I have the same issue/question. Maybe I can help and get help.

@akram please open a new issue and follow the template. Feel free to follow the advice/suggestions/recommendations in this issue though, as you will be asked the same.

akram, 2021-08-24:
@crankyoldgit please see #1573