crankyoldgit/IRremoteESP8266 #1476: Mitshubishi Air Con Vertical Swing Issue.

chaisaeng, 2021-05-13:
### Version/revision of the library used
v2.7.16
### Describe the bug
I was upgrade tasmota-ir to 9.4.0 which now upgrade IRremoteESP8266 library from v2.7.15 to v2.7.16
before the upgrade my Mitshubishi AC working as expect for the vertical swing using (part of the json "SwingV":"Lowest") after the upgrade the AC swing up and down like it is set to "Auto" not the specific position as it is desires. 
#### To Reproduce
_What steps did you do, and what did or didn't actually happen? How can we reproduce the issue?_
1.  From the old version on tasmota-ir with old version of the libray issue the ir command to set swin to lowest.
2.  Observed result working as expected
3.  Upgrade tasmota-ir to 9.4.0 with using library  v2.7.16
4.  Observed result the AC Swing up and down continuously
5.  Setting to other option from Low, Lowest, Middle, High, Highest All Swing up and down continuously.

#### Expected behaviour
The AC  Should Swing to the Position Desires i.e Middle, Lowest, Highest etc.

#### Output of raw data from [IRrecvDumpV2.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) or [V3](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/) (if applicable)
_Include some serial text of the raw dumps of what the device saw._

_**Note**: Output from Tasmota is not acceptable. We can't easily use their raw format._

### What brand/model IR demodulator are you using?
_Brand/Model code_ or _None_

_If VS1838b: That is likely your problem. Please buy a better one & try again. Reporting a capture or decode issue when you use one of these is effectively wasting our & your time. See the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Help_Im_getting_very_inconsistent_results_when_capturing_an_IR_message_using_a_VS1838b_IR_demodulator)_

### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc. ESP8266 or ESP32 board type._
esp8266
### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes/No._

### Has this library/code previously worked as expected for you?
_Yes/No. If "Yes", which version last worked for you?_
Yes, v2.7.15 
### Other useful information
_More information is always welcome. Be verbose._

crankyoldgit, 2021-05-13:
> ### Has this library/code previously worked as expected for you?
> _Yes/No. If "Yes", which version last worked for you?_
> Yes, v2.7.15

Are you sure it was working in v2.7.15? There were changes made to `MITSUBISHI_AC` back in v2.7.15, but there have been no changes since the release of v2.7.15. i.e. v2.7.16+ don't make any changes to [`ir_Mitsubishi.cpp`](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Mitsubishi.cpp) or [`ir_Mitsubishi.h`](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Mitsubishi.h). Both were last updated on the 5th of Feb. v2.7.15 was released on the 12th of Feb.

You can check through the release notes:
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/ReleaseNotes.md#v2715-20210213
&
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/ReleaseNotes.md#v2716-20210324

closed 2021-05-14 by chaisaeng

chaisaeng, 2021-05-14:
Sorry, It's my fault. I try to implement econo mode button that exist on the remote which set temperature to 25 Celsius and vertical swing auto. I got it working. sorry this cause you inconvenient and Thanks you your support response.